Barracuda09/SATPI #1: compile warnings

9000h, 2015-01-14:
I got some compile warnings like
warning: format '%d' expects argument of type 'int', but argument 4 has type 'size_t' [-Wformat]

which may can be fixed when %zd is used.

Barracuda09, 2015-01-14:
Ok..Thanks. Did not see this warning. Were is it?

9000h, 2015-01-14:
here they are

gcc -v
Es werden eingebaute Spezifikationen verwendet.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/4.6/lto-wrapper
Ziel: x86_64-linux-gnu
Konfiguriert mit: ../src/configure -v --with-pkgversion='Ubuntu/Linaro 4.6.3-1ubuntu5' --with-bugurl=file:///usr/share/doc/gcc-4.6/README.Bugs --enable-languages=c,c++,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.6 --enable-shared --enable-linker-build-id --with-system-zlib --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.6 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --enable-plugin --enable-objc-gc --disable-werror --with-arch-32=i686 --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread-Modell: posix
gcc-Version 4.6.3 (Ubuntu/Linaro 4.6.3-1ubuntu5) 
root@yavdr:/home/yavdr/satip/SATPI-master# make
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG tune.c 
tune.c: In Funktion »get_attached_frontend_count«:
tune.c:78:9: Warnung: Format »%d« erwartet Argumenttyp »int *«, aber Argument 3 hat Typ »size_t *« [-Wformat]
tune.c:80:9: Warnung: Format »%d« erwartet Argumenttyp »int *«, aber Argument 3 hat Typ »size_t *« [-Wformat]
tune.c:83:9: Warnung: Format »%d« erwartet Argumenttyp »int«, aber Argument 4 hat Typ »size_t« [-Wformat]
tune.c:83:9: Warnung: Format »%d« erwartet Argumenttyp »int«, aber Argument 5 hat Typ »size_t« [-Wformat]
tune.c:83:9: Warnung: Format »%d« erwartet Argumenttyp »int«, aber Argument 4 hat Typ »size_t« [-Wformat]
tune.c:83:9: Warnung: Format »%d« erwartet Argumenttyp »int«, aber Argument 5 hat Typ »size_t« [-Wformat]
tune.c:84:9: Warnung: Format »%d« erwartet Argumenttyp »int«, aber Argument 4 hat Typ »size_t« [-Wformat]
tune.c:84:9: Warnung: Format »%d« erwartet Argumenttyp »int«, aber Argument 5 hat Typ »size_t« [-Wformat]
tune.c:84:9: Warnung: Format »%d« erwartet Argumenttyp »int«, aber Argument 4 hat Typ »size_t« [-Wformat]
tune.c:84:9: Warnung: Format »%d« erwartet Argumenttyp »int«, aber Argument 5 hat Typ »size_t« [-Wformat]
tune.c:85:9: Warnung: Format »%d« erwartet Argumenttyp »int«, aber Argument 4 hat Typ »size_t« [-Wformat]
tune.c:85:9: Warnung: Format »%d« erwartet Argumenttyp »int«, aber Argument 5 hat Typ »size_t« [-Wformat]
tune.c:85:9: Warnung: Format »%d« erwartet Argumenttyp »int«, aber Argument 4 hat Typ »size_t« [-Wformat]
tune.c:85:9: Warnung: Format »%d« erwartet Argumenttyp »int«, aber Argument 5 hat Typ »size_t« [-Wformat]
tune.c: In Funktion »detect_attached_frontends«:
tune.c:347:3: Warnung: Format »%d« erwartet Argumenttyp »int«, aber Argument 2 hat Typ »size_t« [-Wformat]
tune.c:370:3: Warnung: Format »%d« erwartet Argumenttyp »int«, aber Argument 4 hat Typ »size_t« [-Wformat]
tune.c:370:3: Warnung: Format »%d« erwartet Argumenttyp »int«, aber Argument 4 hat Typ »size_t« [-Wformat]
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG ssdp.c 
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG rtsp.c 
rtsp.c: In Funktion »setup_rtsp«:
rtsp.c:276:34: Warnung: Vergleich zwischen vorzeichenbehafteten und vorzeichenlosen Ganzzahlausdrücken [-Wsign-compare]
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG rtp.c 
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG rtcp.c 
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG http.c 
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG utils.c 
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG connection.c 
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG applog.c 
applog.c: In Funktion »satiplog«:
applog.c:104:5: Warnung: Format »%d« erwartet Argumenttyp »int«, aber Argument 3 hat Typ »size_t« [-Wformat]
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG satpi.c 
satpi.c:40:18: Warnung: »threadID« definiert, aber nicht verwendet [-Wunused-variable]
gcc  tune.o ssdp.o rtsp.o rtp.o rtcp.o http.o utils.o connection.o applog.o satpi.o -o satpi -lpthread -lrt

Barracuda09, 2015-01-14:
Mmm.. still a bit puzzled by why I don't see these warning.
But Thanks for pointing this out.

Barracuda09, 2015-01-14:
Should be fixed now

Barracuda09, 2015-01-14:
Should still fix this too...
rtsp.c: In function 'setup_rtsp':
rtsp.c:276:34: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]

Barracuda09, 2015-01-26:
This issue is fixed

closed 2015-01-26 by Barracuda09