Barracuda09/SATPI #4: DVBviewer Lite Edition 1.0.15 no picture

9000h, 2015-01-14:
here are a debug log from a session

Jan 14 23:27:34 yavdr satpi: 0 Frontend Name: STV090x Multistandard
Jan 14 23:27:34 yavdr satpi: 1 No DTV_ENUM_DELSYS ?? (DVB_API_VERSION: 5)
Jan 14 23:27:34 yavdr satpi: 2 Frontend Type: Satellite (DVB-S)
Jan 14 23:27:34 yavdr satpi: 3 Frontend Type: Satellite (DVB-S2)
Jan 14 23:27:34 yavdr satpi: 4 Frontend Freq: 950000 Hz to 2150000 Hz
Jan 14 23:27:34 yavdr satpi: 5 Frontend srat: 1000000 symbols/s to 45000000 symbols/s
Jan 14 23:27:34 yavdr satpi: 6 Frontend Name: STV090x Multistandard
Jan 14 23:27:34 yavdr satpi: 7 No DTV_ENUM_DELSYS ?? (DVB_API_VERSION: 5)
Jan 14 23:27:34 yavdr satpi: 8 Frontend Type: Satellite (DVB-S)
Jan 14 23:27:34 yavdr satpi: 9 Frontend Type: Satellite (DVB-S2)
Jan 14 23:27:34 yavdr satpi: 10 Frontend Freq: 950000 Hz to 2150000 Hz
Jan 14 23:27:34 yavdr satpi: 11 Frontend srat: 1000000 symbols/s to 45000000 symbols/s
Jan 14 23:27:34 yavdr SatIP Server[7290]: 0 --- starting ---
Jan 14 23:27:34 yavdr SatIP Server[7290]: 1 eth0: 192.168.178.101 [00:22:4d:9d:38:0]
Jan 14 23:27:34 yavdr SatIP Server[7290]: 2 Setting up RTSP server
Jan 14 23:27:34 yavdr SatIP Server[7290]: 3 Setting up HTTP server
Jan 14 23:27:34 yavdr SatIP Server[7290]: 4 Setting up RTP server
Jan 14 23:27:34 yavdr SatIP Server[7290]: 5 Setting up RTP server
Jan 14 23:27:34 yavdr SatIP Server[7290]: 6 Setting up SSDP server
Jan 14 23:27:43 yavdr SatIP Server[7290]: 7 RTSP Connection from 192.168.178.112 Port 50310
Jan 14 23:27:43 yavdr SatIP Server[7290]: 8 OPTIONS rtsp://192.168.178.101:554/ RTSP/1.0#015#012CSeq: 1#015#012#015
Jan 14 23:27:43 yavdr SatIP Server[7290]: 9 SETUP rtsp://192.168.178.101:554/?src=1&freq=11954&msys=dvbs&plts=off&fec=34&pol=h&ro=0.35&sr=27500&mtype=qpsk&pids=0 RTSP/1.0#015#012CSeq: 2#015#012Session: 00000000#015#012Transport: RTP/AVP;unicast;client_port=52042-52043#015#012#015
Jan 14 23:27:43 yavdr SatIP Server[7290]: 10 Setup Message
Jan 14 23:27:43 yavdr SatIP Server[7290]: 11 Frontend: 0, Attaching to client 192.168.178.112
Jan 14 23:27:43 yavdr SatIP Server[7290]: 12 Start tuning
Jan 14 23:27:43 yavdr SatIP Server[7290]: 13 Sending DiSEqC
Jan 14 23:27:43 yavdr SatIP Server[7290]: 14 Frontend: 0, Tuned.
Jan 14 23:27:43 yavdr SatIP Server[7290]: 15 Frontend: 0, Opened DVR.
Jan 14 23:27:43 yavdr SatIP Server[7290]: 16 Setting filter for PID: 0
Jan 14 23:27:43 yavdr SatIP Server[7290]: 17 PLAY rtsp://192.168.178.101:554/stream=1?addpids=18 RTSP/1.0#015#012CSeq: 2#015#012Session: 00000002#015#012#015
Jan 14 23:27:43 yavdr SatIP Server[7290]: 18 Play Message
Jan 14 23:27:43 yavdr SatIP Server[7290]: 19 Setting filter for PID: 18
Jan 14 23:27:43 yavdr SatIP Server[7290]: 20 PLAY rtsp://192.168.178.101:554/stream=1?addpids=310,320,0,300,330 RTSP/1.0#015#012CSeq: 3#015#012Session: 00000002#015#012#015
Jan 14 23:27:43 yavdr SatIP Server[7290]: 21 Play Message
Jan 14 23:27:43 yavdr SatIP Server[7290]: 22 Setting filter for PID: 300 310 320 330
Jan 14 23:27:43 yavdr SatIP Server[7290]: 23 Frontend: 0, Start streaming to 192.168.178.112 (52042 - 52043)
Jan 14 23:28:36 yavdr SatIP Server[7290]: 24 GET web/desc.xml (size 1512)
Jan 14 23:28:36 yavdr SatIP Server[7290]: 25 Connection closed by Client 1
Jan 14 23:28:37 yavdr SatIP Server[7290]: 26 OPTIONS rtsp://192.168.178.101:554/ RTSP/1.0#015#012CSeq: 4#015#012Session: 00000002#015#012#015
Jan 14 23:29:31 yavdr SatIP Server[7290]: 27 OPTIONS rtsp://192.168.178.101:554/ RTSP/1.0#015#012CSeq: 5#015#012Session: 00000002#015#012#015

Barracuda09, 2015-01-14:
That is strange.. If you stop the stream and then start it again?
I use it also and it works.

9000h, 2015-01-14:
there is DESCRIBE missing in rtsp.c options line 447 and probably some code also

Barracuda09, 2015-01-14:
Yes.. I did not support that... it will not be given by an OPTION msg. So that should not be a big problem.

9000h, 2015-01-15:
ok I did try a scan with dvbviewer but still no success

Jan 15 20:13:09 yavdr satpi: 0 Frontend Name: STV090x Multistandard
Jan 15 20:13:09 yavdr satpi: 1 No DTV_ENUM_DELSYS ?? (DVB_API_VERSION: 5)
Jan 15 20:13:09 yavdr satpi: 2 Frontend Type: Satellite (DVB-S)
Jan 15 20:13:09 yavdr satpi: 3 Frontend Type: Satellite (DVB-S2)
Jan 15 20:13:09 yavdr satpi: 4 Frontend Freq: 950000 Hz to 2150000 Hz
Jan 15 20:13:09 yavdr satpi: 5 Frontend srat: 1000000 symbols/s to 45000000 symbols/s
Jan 15 20:13:09 yavdr satpi: 6 Frontend Name: STV090x Multistandard
Jan 15 20:13:09 yavdr satpi: 7 No DTV_ENUM_DELSYS ?? (DVB_API_VERSION: 5)
Jan 15 20:13:09 yavdr satpi: 8 Frontend Type: Satellite (DVB-S)
Jan 15 20:13:09 yavdr satpi: 9 Frontend Type: Satellite (DVB-S2)
Jan 15 20:13:09 yavdr satpi: 10 Frontend Freq: 950000 Hz to 2150000 Hz
Jan 15 20:13:09 yavdr satpi: 11 Frontend srat: 1000000 symbols/s to 45000000 symbols/s
Jan 15 20:13:09 yavdr SatIP Server[4899]: 0 --- starting ---
Jan 15 20:13:09 yavdr SatIP Server[4899]: 1 eth0: 192.168.178.101 [00:22:4d:9d:38:0]
Jan 15 20:13:09 yavdr SatIP Server[4899]: 2 Setting up RTSP server
Jan 15 20:13:09 yavdr SatIP Server[4899]: 3 Setting up HTTP server
Jan 15 20:13:09 yavdr SatIP Server[4899]: 4 Setting up RTP server
Jan 15 20:13:09 yavdr SatIP Server[4899]: 5 Setting up RTP server
Jan 15 20:13:09 yavdr SatIP Server[4899]: 6 Setting up SSDP server
Jan 15 20:13:10 yavdr SatIP Server[4899]: 7 GET web/desc.xml (size 1512)
Jan 15 20:13:10 yavdr SatIP Server[4899]: 8 Connection closed by Client 1
Jan 15 20:13:39 yavdr SatIP Server[4899]: 9 RTSP Connection from 192.168.178.112 Port 50290
Jan 15 20:13:39 yavdr SatIP Server[4899]: 10 OPTIONS rtsp://192.168.178.101:554/ RTSP/1.0#015#012CSeq: 1#015#012#015
Jan 15 20:13:39 yavdr SatIP Server[4899]: 11 SETUP rtsp://192.168.178.101:554/?src=1&freq=10733&msys=dvbs&plts=off&fec=89&pol=h&ro=0.35&sr=22000&mtype=&pids=0 RTSP/1.0#015#012CSeq: 2#015#012Session: 00000000#015#012Transport: RTP/AVP;unicast;client_port=52032-52033#015#012#015
Jan 15 20:13:39 yavdr SatIP Server[4899]: 12 Setup Message
Jan 15 20:13:39 yavdr SatIP Server[4899]: 13 Frontend: 0, Attaching to client 192.168.178.112
Jan 15 20:13:39 yavdr SatIP Server[4899]: 14 Start tuning
Jan 15 20:13:39 yavdr SatIP Server[4899]: 15 Sending DiSEqC
Jan 15 20:13:39 yavdr SatIP Server[4899]: 16 Frontend: 0, Tuned.
Jan 15 20:13:39 yavdr SatIP Server[4899]: 17 Frontend: 0, Opened DVR.
Jan 15 20:13:39 yavdr SatIP Server[4899]: 18 Setting filter for PID: 0
Jan 15 20:13:39 yavdr SatIP Server[4899]: 19 PLAY rtsp://192.168.178.101:554/stream=1?addpids=0 RTSP/1.0#015#012CSeq: 2#015#012Session: 00000002#015#012#015
Jan 15 20:13:39 yavdr SatIP Server[4899]: 20 Play Message
Jan 15 20:13:39 yavdr SatIP Server[4899]: 21 Frontend: 0, Start streaming to 192.168.178.112 (52032 - 52033)
Jan 15 20:13:54 yavdr SatIP Server[4899]: 22 PLAY rtsp://192.168.178.101:554/stream=1?delpids=0 RTSP/1.0#015#012CSeq: 3#015#012Session: 00000002#015#012#015
Jan 15 20:13:54 yavdr SatIP Server[4899]: 23 Play Message
Jan 15 20:13:54 yavdr SatIP Server[4899]: 24 Removing filter for PID: 0
Jan 15 20:13:54 yavdr SatIP Server[4899]: 25 PLAY rtsp://192.168.178.101:554/stream=1?src=1&freq=10733&msys=dvbs&plts=off&fec=89&pol=h&ro=0.35&sr=22000&mtype=&pids=0 RTSP/1.0#015#012CSeq: 4#015#012Session: 00000002#015#012#015
Jan 15 20:13:54 yavdr SatIP Server[4899]: 26 Play Message
Jan 15 20:13:54 yavdr SatIP Server[4899]: 27 Start tuning
Jan 15 20:13:54 yavdr SatIP Server[4899]: 28 Sending DiSEqC
Jan 15 20:13:54 yavdr SatIP Server[4899]: 29 Frontend: 0, Tuned.
Jan 15 20:13:54 yavdr SatIP Server[4899]: 30 Setting filter for PID: 0
Jan 15 20:13:54 yavdr SatIP Server[4899]: 31 PLAY rtsp://192.168.178.101:554/stream=1?addpids=0 RTSP/1.0#015#012CSeq: 5#015#012Session: 00000002#015#012#015
Jan 15 20:13:54 yavdr SatIP Server[4899]: 32 Play Message
Jan 15 20:14:09 yavdr SatIP Server[4899]: 33 PLAY rtsp://192.168.178.101:554/stream=1?delpids=0 RTSP/1.0#015#012CSeq: 6#015#012Session: 00000002#015#012#015
Jan 15 20:14:09 yavdr SatIP Server[4899]: 34 Play Message
Jan 15 20:14:09 yavdr SatIP Server[4899]: 35 Removing filter for PID: 0
Jan 15 20:14:09 yavdr SatIP Server[4899]: 36 PLAY rtsp://192.168.178.101:554/stream=1?src=1&freq=10733&msys=dvbs&plts=off&fec=89&pol=v&ro=0.35&sr=22000&mtype=&pids=0 RTSP/1.0#015#012CSeq: 7#015#012Session: 00000002#015#012#015
Jan 15 20:14:09 yavdr SatIP Server[4899]: 37 Play Message
Jan 15 20:14:09 yavdr SatIP Server[4899]: 38 Start tuning
Jan 15 20:14:10 yavdr SatIP Server[4899]: 39 Sending DiSEqC
Jan 15 20:14:10 yavdr SatIP Server[4899]: 40 Frontend: 0, Tuned.
Jan 15 20:14:10 yavdr SatIP Server[4899]: 41 Setting filter for PID: 0
Jan 15 20:14:10 yavdr SatIP Server[4899]: 42 PLAY rtsp://192.168.178.101:554/stream=1?addpids=0 RTSP/1.0#015#012CSeq: 8#015#012Session: 00000002#015#012#015
Jan 15 20:14:10 yavdr SatIP Server[4899]: 43 Play Message
Jan 15 20:14:25 yavdr SatIP Server[4899]: 44 PLAY rtsp://192.168.178.101:554/stream=1?delpids=0 RTSP/1.0#015#012CSeq: 9#015#012Session: 00000002#015#012#015
Jan 15 20:14:25 yavdr SatIP Server[4899]: 45 Play Message
Jan 15 20:14:25 yavdr SatIP Server[4899]: 46 Removing filter for PID: 0
Jan 15 20:14:25 yavdr SatIP Server[4899]: 47 PLAY rtsp://192.168.178.101:554/stream=1?src=1&freq=10733&msys=dvbs&plts=off&fec=89&pol=v&ro=0.35&sr=22000&mtype=&pids=0 RTSP/1.0#015#012CSeq: 10#015#012Session: 00000002#015#012#015
Jan 15 20:14:25 yavdr SatIP Server[4899]: 48 Play Message
[... cut ...]
Jan 15 20:15:42 yavdr SatIP Server[4899]: 99 PLAY rtsp://192.168.178.101:554/stream=1?delpids=0 RTSP/1.0#015#012CSeq: 24#015#012Session: 00000002#015#012#015
Jan 15 20:15:42 yavdr SatIP Server[4899]: 100 Play Message
Jan 15 20:15:42 yavdr SatIP Server[4899]: 101 Removing filter for PID: 0
Jan 15 20:15:42 yavdr SatIP Server[4899]: 102 TEARDOWN rtsp://192.168.178.101:554/stream=1 RTSP/1.0#015#012CSeq: 25#015#012Session: 00000002#015#012#015
Jan 15 20:15:42 yavdr SatIP Server[4899]: 103 Teardown Message, gracefull: 1
Jan 15 20:15:42 yavdr SatIP Server[4899]: 104 Frontend: 0, Stop streaming to 192.168.178.112 (52032 - 52033) (Streamed 0.000 MBytes)

Barracuda09, 2015-01-15:
Can you also add --yes-rtcp this will tell that the freq. is tuned

9000h, 2015-01-15:
still no picture in dvbviewer

root@yavdr:/home/yavdr/satip/SATPI-master# ./satpi --no-daemon --yes-ssdp --yes-rtcp
Found 2 frontend!!

Jan 15 22:11:04 yavdr satpi: 0 Frontend Name: STV090x Multistandard
Jan 15 22:11:04 yavdr satpi: 1 No DTV_ENUM_DELSYS ?? (DVB_API_VERSION: 5)
Jan 15 22:11:04 yavdr satpi: 2 Frontend Type: Satellite (DVB-S)
Jan 15 22:11:04 yavdr satpi: 3 Frontend Type: Satellite (DVB-S2)
Jan 15 22:11:04 yavdr satpi: 4 Frontend Freq: 950000 Hz to 2150000 Hz
Jan 15 22:11:04 yavdr satpi: 5 Frontend srat: 1000000 symbols/s to 45000000 symbols/s
Jan 15 22:11:04 yavdr satpi: 6 Frontend Name: STV090x Multistandard
Jan 15 22:11:04 yavdr satpi: 7 No DTV_ENUM_DELSYS ?? (DVB_API_VERSION: 5)
Jan 15 22:11:04 yavdr satpi: 8 Frontend Type: Satellite (DVB-S)
Jan 15 22:11:04 yavdr satpi: 9 Frontend Type: Satellite (DVB-S2)
Jan 15 22:11:04 yavdr satpi: 10 Frontend Freq: 950000 Hz to 2150000 Hz
Jan 15 22:11:04 yavdr satpi: 11 Frontend srat: 1000000 symbols/s to 45000000 symbols/s
Jan 15 22:11:04 yavdr SatIP Server[5670]: 0 --- starting ---
Jan 15 22:11:04 yavdr SatIP Server[5670]: 1 eth0: 192.168.178.101 [00:22:4d:9d:38:0]
Jan 15 22:11:04 yavdr SatIP Server[5670]: 2 Setting up SSDP server
Jan 15 22:11:04 yavdr SatIP Server[5670]: 3 Setting up RTCP server
Jan 15 22:11:04 yavdr SatIP Server[5670]: 4 Setting up RTCP server
Jan 15 22:11:04 yavdr SatIP Server[5670]: 5 Setting up RTP server
Jan 15 22:11:04 yavdr SatIP Server[5670]: 6 Setting up RTP server
Jan 15 22:11:04 yavdr SatIP Server[5670]: 7 Setting up HTTP server
Jan 15 22:11:04 yavdr SatIP Server[5670]: 8 Setting up RTSP server
Jan 15 22:11:40 yavdr SatIP Server[5670]: 9 RTSP Connection from 192.168.178.112 Port 50401
Jan 15 22:11:40 yavdr SatIP Server[5670]: 10 OPTIONS rtsp://192.168.178.101:554/ RTSP/1.0#015#012CSeq: 1#015#012#015
Jan 15 22:11:40 yavdr SatIP Server[5670]: 11 SETUP rtsp://192.168.178.101:554/?src=1&freq=11837&msys=dvbs&plts=off&fec=34&pol=h&ro=0.35&sr=27500&mtype=qpsk&pids=0 RTSP/1.0#015#012CSeq: 2#015#012Session: 00000000#015#012Transport: RTP/AVP;unicast;client_port=52042-52043#015#012#015
Jan 15 22:11:40 yavdr SatIP Server[5670]: 12 Setup Message
Jan 15 22:11:40 yavdr SatIP Server[5670]: 13 Frontend: 0, Attaching to client 192.168.178.112
Jan 15 22:11:40 yavdr SatIP Server[5670]: 14 Start tuning
Jan 15 22:11:40 yavdr SatIP Server[5670]: 15 Sending DiSEqC src 1 on Frontend: 12
Jan 15 22:11:40 yavdr SatIP Server[5670]: 16 Frontend: 0, Tuned.
Jan 15 22:11:40 yavdr SatIP Server[5670]: 17 Frontend: 0, Opened DVR.
Jan 15 22:11:40 yavdr SatIP Server[5670]: 18 Setting filter for PID: 0
Jan 15 22:11:40 yavdr SatIP Server[5670]: 19 PLAY rtsp://192.168.178.101:554/stream=1?addpids=18 RTSP/1.0#015#012CSeq: 2#015#012Session: 00000002#015#012#015
Jan 15 22:11:40 yavdr SatIP Server[5670]: 20 Play Message
Jan 15 22:11:40 yavdr SatIP Server[5670]: 21 Setting filter for PID: 18
Jan 15 22:11:40 yavdr SatIP Server[5670]: 22 PLAY rtsp://192.168.178.101:554/stream=1?addpids=101,102,0,100,104 RTSP/1.0#015#012CSeq: 3#015#012Session: 00000002#015#012#015
Jan 15 22:11:41 yavdr SatIP Server[5670]: 23 Frontend: 0, Start streaming to 192.168.178.112 (52042 - 52043)
Jan 15 22:11:41 yavdr SatIP Server[5670]: 24 Play Message
Jan 15 22:11:41 yavdr SatIP Server[5670]: 25 Setting filter for PID: 100 101 102 104
Jan 15 22:12:06 yavdr SatIP Server[5670]: 26 GET web/desc.xml (size 1512)
Jan 15 22:12:06 yavdr SatIP Server[5670]: 27 Connection closed by Client 1
Jan 15 22:12:35 yavdr SatIP Server[5670]: 28 OPTIONS rtsp://192.168.178.101:554/ RTSP/1.0#015#012CSeq: 4#015#012Session: 00000002#015#012#015
Jan 15 22:13:29 yavdr SatIP Server[5670]: 29 OPTIONS rtsp://192.168.178.101:554/ RTSP/1.0#015#012CSeq: 5#015#012Session: 00000002#015#012#015

Barracuda09, 2015-01-15:
Mmmm... strange.. I have the same setup as you, so it seems.
It show in the log it is tuned and start streaming..
You have a DiSEqC switch?

9000h, 2015-01-15:
yes I have
A Astra 19.2
B Hotbird 13.0
C EuroBird 9.0
D Astra2 28.2

in the VDR it works with 
1 2 3 4:
S19.2E 11700 V 9750 t v W15 [E0 10 38 F0] W15 A W15 t
S19.2E 99999 V 10600 t v W15 [E0 10 38 F1] W15 A W15 T
S19.2E 11700 H 9750 t V W15 [E0 10 38 F2] W15 A W15 t
S19.2E 99999 H 10600 t V W15 [E0 10 38 F3] W15 A W15 T

# # input 2: 13.0E
1 2 3 4:
S13.0E 11700 V 9750 t v W15 [E0 10 38 F4] W15 B W15 t
S13.0E 99999 V 10600 t v W15 [E0 10 38 F5] W15 B W15 T
S13.0E 11700 H 9750 t V W15 [E0 10 38 F6] W15 B W15 t
S13.0E 99999 H 10600 t V W15 [E0 10 38 F7] W15 B W15 T

# # input 3: 9E
1 2 3 4:
S9E 11700 V 9750 t v W15 [E0 10 38 F8] W15 A W15 t
S9E 99999 V 10600 t v W15 [E0 10 38 F9] W15 A W15 T
S9E 11700 H 9750 t V W15 [E0 10 38 FA] W15 A W15 t
S9E 99999 H 10600 t V W15 [E0 10 38 FB] W15 A W15 T

# # input 4: 28.2E
1 2 3 4:
S28.2E 11700 V 9750 t v W15 [E0 10 38 FC] W15 B W15 t
S28.2E 99999 V 10600 t v W15 [E0 10 38 FD] W15 B W15 T
S28.2E 11700 H 9750 t V W15 [E0 10 38 FE] W15 B W15 t
S28.2E 99999 H 10600 t V W15 [E0 10 38 FF] W15 B W15 T

9000h, 2015-01-15:
If I try with VLC then I get also no picture
Jan 15 22:34:59 yavdr satpi: 0 Frontend Name: STV090x Multistandard
Jan 15 22:34:59 yavdr satpi: 1 No DTV_ENUM_DELSYS ?? (DVB_API_VERSION: 5)
Jan 15 22:34:59 yavdr satpi: 2 Frontend Type: Satellite (DVB-S)
Jan 15 22:34:59 yavdr satpi: 3 Frontend Type: Satellite (DVB-S2)
Jan 15 22:34:59 yavdr satpi: 4 Frontend Freq: 950000 Hz to 2150000 Hz
Jan 15 22:34:59 yavdr satpi: 5 Frontend srat: 1000000 symbols/s to 45000000 symbols/s
Jan 15 22:34:59 yavdr satpi: 6 Frontend Name: STV090x Multistandard
Jan 15 22:34:59 yavdr satpi: 7 No DTV_ENUM_DELSYS ?? (DVB_API_VERSION: 5)
Jan 15 22:34:59 yavdr satpi: 8 Frontend Type: Satellite (DVB-S)
Jan 15 22:34:59 yavdr satpi: 9 Frontend Type: Satellite (DVB-S2)
Jan 15 22:34:59 yavdr satpi: 10 Frontend Freq: 950000 Hz to 2150000 Hz
Jan 15 22:34:59 yavdr satpi: 11 Frontend srat: 1000000 symbols/s to 45000000 symbols/s
Jan 15 22:34:59 yavdr SatIP Server[5812]: 0 --- starting ---
Jan 15 22:34:59 yavdr SatIP Server[5812]: 1 eth0: 192.168.178.101 [00:22:4d:9d:38:0]
Jan 15 22:34:59 yavdr SatIP Server[5812]: 2 Setting up SSDP server
Jan 15 22:34:59 yavdr SatIP Server[5812]: 3 Setting up RTCP server
Jan 15 22:34:59 yavdr SatIP Server[5812]: 4 Setting up RTCP server
Jan 15 22:34:59 yavdr SatIP Server[5812]: 5 Setting up RTP server
Jan 15 22:34:59 yavdr SatIP Server[5812]: 6 Setting up RTP server
Jan 15 22:34:59 yavdr SatIP Server[5812]: 7 Setting up HTTP server
Jan 15 22:34:59 yavdr SatIP Server[5812]: 8 Setting up RTSP server
Jan 15 22:35:01 yavdr CRON[5821]: (root) CMD (command -v debian-sa1 > /dev/null && debian-sa1 1 1)
Jan 15 22:35:04 yavdr SatIP Server[5812]: 9 RTSP Connection from 192.168.178.130 Port 61247
Jan 15 22:35:04 yavdr SatIP Server[5812]: 10 OPTIONS rtsp://192.168.178.101:554/?src=1&freq=12188&sr=27500&pol=h&msys=dvbs&pids=0,44,163,104,106 RTSP/1.0#015#012CSeq: 2#015#012User-Agent: LibVLC/2.2.0-rc1 (LIVE555 Streaming Media v2014.01.21)#015#012#015
Jan 15 22:35:09 yavdr SatIP Server[5812]: 11 RTSP Connection closed by Client: 192.168.178.130 ??
Jan 15 22:35:09 yavdr SatIP Server[5812]: 12 Teardown Message, gracefull: 0
Jan 15 22:35:09 yavdr SatIP Server[5812]: 13 Frontend: x, Not attached anymore to client 192.168.178.130
Jan 15 22:35:09 yavdr SatIP Server[5812]: 14 RTSP Connection from 192.168.178.130 Port 61248
Jan 15 22:35:09 yavdr SatIP Server[5812]: 15 OPTIONS rtsp://192.168.178.101:554 RTSP/1.0#015#012CSeq: 1#015#012User-Agent: RealMedia Player Version 6.0.9.1235 (linux-2.0-libc6-i386-gcc2.95)#015#012ClientChallenge: 9e26d33f2984236010ef6253fb1887f7#015#012PlayerStarttime: [28/03/2003:22:50:23 00:00]#015#012CompanyID: KnKV4M4I/B2FjJ1TToLycw==#015#012GUID: 00000000-0000-0000-0000-000000000000#015#012RegionData: 0#015#012ClientID: Linux_2.4_6.0.9.1235_play32_RN01_EN_586#015#012#015
Jan 15 22:35:09 yavdr SatIP Server[5812]: 16 RTSP Connection closed by Client: 192.168.178.130 ??
Jan 15 22:35:09 yavdr SatIP Server[5812]: 17 Teardown Message, gracefull: 0
Jan 15 22:35:09 yavdr SatIP Server[5812]: 18 Frontend: x, Not attached anymore to client 192.168.178.130
Jan 15 22:35:09 yavdr SatIP Server[5812]: 19 RTSP Connection from 192.168.178.130 Port 61249
Jan 15 22:35:09 yavdr SatIP Server[5812]: 20 OPTIONS rtsp://192.168.178.101:554/?src=1&freq=12188&sr=27500&pol=h&msys=dvbs&pids=0,44,163,104,106 RTSP/1.0#015#012CSeq: 2#015#012User-Agent: LibVLC/2.2.0-rc1 (LIVE555 Streaming Media v2014.01.21)#015#012#015
Jan 15 22:35:14 yavdr SatIP Server[5812]: 21 RTSP Connection closed by Client: 192.168.178.130 ??
Jan 15 22:35:14 yavdr SatIP Server[5812]: 22 Teardown Message, gracefull: 0
Jan 15 22:35:14 yavdr SatIP Server[5812]: 23 Frontend: x, Not attached anymore to client 192.168.178.130
Jan 15 22:35:14 yavdr SatIP Server[5812]: 24 RTSP Connection from 192.168.178.130 Port 61250
Jan 15 22:35:14 yavdr SatIP Server[5812]: 25 OPTIONS rtsp://192.168.178.101:554 RTSP/1.0#015#012CSeq: 1#015#012User-Agent: RealMedia Player Version 6.0.9.1235 (linux-2.0-libc6-i386-gcc2.95)#015#012ClientChallenge: 9e26d33f2984236010ef6253fb1887f7#015#012PlayerStarttime: [28/03/2003:22:50:23 00:00]#015#012CompanyID: KnKV4M4I/B2FjJ1TToLycw==#015#012GUID: 00000000-0000-0000-0000-000000000000#015#012RegionData: 0#015#012ClientID: Linux_2.4_6.0.9.1235_play32_RN01_EN_586#015#012#015
Jan 15 22:35:14 yavdr SatIP Server[5812]: 26 RTSP Connection closed by Client: 192.168.178.130 ??
Jan 15 22:35:14 yavdr SatIP Server[5812]: 27 Teardown Message, gracefull: 0
Jan 15 22:35:14 yavdr SatIP Server[5812]: 28 Frontend: x, Not attached anymore to client 192.168.178.130
Jan 15 22:35:14 yavdr SatIP Server[5812]: 29 RTSP Connection from 192.168.178.130 Port 61251
Jan 15 22:35:14 yavdr SatIP Server[5812]: 30 OPTIONS rtsp://192.168.178.101:554/?src=1&freq=12188&sr=27500&pol=h&msys=dvbs&pids=0,44,163,104,106 RTSP/1.0#015#012CSeq: 2#015#012User-Agent: LibVLC/2.2.0-rc1 (LIVE555 Streaming Media v2014.01.21)#015#012#015
Jan 15 22:35:19 yavdr SatIP Server[5812]: 31 RTSP Connection closed by Client: 192.168.178.130 ??
Jan 15 22:35:19 yavdr SatIP Server[5812]: 32 Teardown Message, gracefull: 0
Jan 15 22:35:19 yavdr SatIP Server[5812]: 33 Frontend: x, Not attached anymore to client 192.168.178.130
Jan 15 22:35:19 yavdr SatIP Server[5812]: 34 RTSP Connection from 192.168.178.130 Port 61252
Jan 15 22:35:19 yavdr SatIP Server[5812]: 35 OPTIONS rtsp://192.168.178.101:554 RTSP/1.0#015#012CSeq: 1#015#012User-Agent: RealMedia Player Version 6.0.9.1235 (linux-2.0-libc6-i386-gcc2.95)#015#012ClientCha [... cut ...]
Jan 15 22:35:19 yavdr SatIP Server[5812]: 37 Teardown Message, gracefull: 0
Jan 15 22:35:19 yavdr SatIP Server[5812]: 38 Frontend: x, Not attached anymore to client 192.168.178.130
Jan 15 22:35:19 yavdr SatIP Server[5812]: 39 RTSP Connection from 192.168.178.130 Port 61253
Jan 15 22:35:19 yavdr SatIP Server[5812]: 40 OPTIONS rtsp://192.168.178.101:554/?src=1&freq=12188&sr=27500&pol=h&msys=dvbs&pids=0,44,163,104,106 RTSP/1.0#015#012CSeq: 2#015#012User-Agent: LibVLC/2.2.0-rc1 (LIVE555 Streaming Media v2014.01.21)#015#012#015
^C

Barracuda09, 2015-01-15:
What is this:

Jan 15 22:35:04 yavdr SatIP Server[5812]: 10 OPTIONS rtsp://192.168.178.101:554/?src=1&freq=12188&sr=27500&pol=h&msys=dvbs&pids=0,44,163,104,106 RTSP/1.0#015#012CSeq: 2#015#012User-Agent: LibVLC/2.2.0-rc1 (LIVE555 Streaming Media v2014.01.21)#015#012#015

OPTIONS msg with tune data?

Barracuda09, 2015-01-15:
Well I did a rescan with the current git version for astra 23.5 and found 200+ programs.

Mmm... I like to know what the problem is.

9000h, 2015-01-15:
this was when I did try with VLC 2.2.0-rc1 under OSX

9000h, 2015-01-15:
latest git
root@yavdr:/home/yavdr/satip/SATPI-master# make
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG tune.c 
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG ssdp.c 
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG rtsp.c 
rtsp.c: In function 'setup_rtsp':
rtsp.c:276:34: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG rtp.c 
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG rtcp.c 
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG http.c 
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG utils.c 
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG connection.c 
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG applog.c 
applog.c: In function 'satiplog':
applog.c:104:5: warning: format '%d' expects argument of type 'int', but argument 3 has type 'size_t' [-Wformat]
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG satpi.c 
gcc  tune.o ssdp.o rtsp.o rtp.o rtcp.o http.o utils.o connection.o applog.o satpi.o -o satpi -lpthread -lrt
root@yavdr:/home/yavdr/satip/SATPI-master#

9000h, 2015-01-15:
still no picture, pid issue?
Jan 16 00:27:03 yavdr SatIP Server[6692]: 0 [satpi.c:235] --- starting ---
Jan 16 00:27:03 yavdr SatIP Server[6692]: 1 [tune.c:126] Frontend Name: STV090x Multistandard
Jan 16 00:27:03 yavdr SatIP Server[6692]: 2 [tune.c:129] No DTV_ENUM_DELSYS ?? (DVB_API_VERSION: 5)
Jan 16 00:27:03 yavdr SatIP Server[6692]: 3 [tune.c:148] Frontend Type: Satellite (DVB-S)
Jan 16 00:27:03 yavdr SatIP Server[6692]: 4 [tune.c:152] Frontend Type: Satellite (DVB-S2)
Jan 16 00:27:03 yavdr SatIP Server[6692]: 5 [tune.c:157] Frontend Freq: 950000 Hz to 2150000 Hz
Jan 16 00:27:03 yavdr SatIP Server[6692]: 6 [tune.c:158] Frontend srat: 1000000 symbols/s to 45000000 symbols/s
Jan 16 00:27:03 yavdr SatIP Server[6692]: 7 [tune.c:126] Frontend Name: STV090x Multistandard
Jan 16 00:27:03 yavdr SatIP Server[6692]: 8 [tune.c:129] No DTV_ENUM_DELSYS ?? (DVB_API_VERSION: 5)
Jan 16 00:27:03 yavdr SatIP Server[6692]: 9 [tune.c:148] Frontend Type: Satellite (DVB-S)
Jan 16 00:27:03 yavdr SatIP Server[6692]: 10 [tune.c:152] Frontend Type: Satellite (DVB-S2)
Jan 16 00:27:03 yavdr SatIP Server[6692]: 11 [tune.c:157] Frontend Freq: 950000 Hz to 2150000 Hz
Jan 16 00:27:03 yavdr SatIP Server[6692]: 12 [tune.c:158] Frontend srat: 1000000 symbols/s to 45000000 symbols/s
Jan 16 00:27:03 yavdr SatIP Server[6692]: 13 [connection.c:259] eth0: 192.168.178.101 [00:22:4d:9d:38:05]
Jan 16 00:27:03 yavdr SatIP Server[6692]: 14 [ssdp.c:92] Setting up SSDP server
Jan 16 00:27:03 yavdr SatIP Server[6692]: 15 [rtcp.c:268] Setting up RTCP server
Jan 16 00:27:03 yavdr SatIP Server[6692]: 16 [rtcp.c:268] Setting up RTCP server
Jan 16 00:27:03 yavdr SatIP Server[6692]: 17 [rtp.c:50] Setting up RTP server
Jan 16 00:27:03 yavdr SatIP Server[6692]: 18 [rtp.c:50] Setting up RTP server
Jan 16 00:27:03 yavdr SatIP Server[6692]: 19 [http.c:328] Setting up HTTP server
Jan 16 00:27:03 yavdr SatIP Server[6692]: 20 [rtsp.c:567] Setting up RTSP server
Jan 16 00:28:54 yavdr SatIP Server[6692]: 21 [connection.c:188] RTSP Connection from 192.168.178.112 Port 49171
Jan 16 00:28:54 yavdr SatIP Server[6692]: 22 [rtsp.c:617] OPTIONS rtsp://192.168.178.101:554/ RTSP/1.0#015#012CSeq: 1#015#012#015
Jan 16 00:28:54 yavdr SatIP Server[6692]: 23 [rtsp.c:617] SETUP rtsp://192.168.178.101:554/?src=1&freq=11954&msys=dvbs&plts=off&fec=34&pol=h&ro=0.35&sr=27500&mtype=qpsk&pids=0 RTSP/1.0#015#012CSeq: 2#015#012Session: 00000000#015#012Transport: RTP/AVP;unicast;client_port=52042-52043#015#012#015
Jan 16 00:28:54 yavdr SatIP Server[6692]: 24 [rtsp.c:271] Setup Message
Jan 16 00:28:54 yavdr SatIP Server[6692]: 25 [rtsp.c:306] Frontend: 0, Attaching to client 192.168.178.112
Jan 16 00:28:54 yavdr SatIP Server[6692]: 26 [tune.c:303] Start tuning
Jan 16 00:28:54 yavdr SatIP Server[6692]: 27 [tune.c:247] Sending DiSEqC
Jan 16 00:28:55 yavdr SatIP Server[6692]: 28 [tune.c:461] Frontend: 0, Tuned.
Jan 16 00:28:55 yavdr SatIP Server[6692]: 29 [tune.c:474] Frontend: 0, Opened DVR.
Jan 16 00:28:55 yavdr SatIP Server[6692]: 30 [tune.c:431] Setting filter for PID: 0
Jan 16 00:28:55 yavdr SatIP Server[6692]: 31 [rtsp.c:617] PLAY rtsp://192.168.178.101:554/stream=1?addpids=18 RTSP/1.0#015#012CSeq: 2#015#012Session: 00000002#015#012#015
Jan 16 00:28:55 yavdr SatIP Server[6692]: 32 [rtsp.c:386] Play Message
Jan 16 00:28:55 yavdr SatIP Server[6692]: 33 [tune.c:431] Setting filter for PID: 18
Jan 16 00:28:55 yavdr SatIP Server[6692]: 34 [rtp.c:104] Frontend: 0, Start streaming to 192.168.178.112 (52042 - 52043)
Jan 16 00:28:55 yavdr SatIP Server[6692]: 35 [rtsp.c:617] PLAY rtsp://192.168.178.101:554/stream=1?addpids=1110,1120,0,1100,1130 RTSP/1.0#015#012CSeq: 3#015#012Session: 00000002#015#012#015
Jan 16 00:28:55 yavdr SatIP Server[6692]: 36 [rtsp.c:386] Play Message
Jan 16 00:28:55 yavdr SatIP Server[6692]: 37 [tune.c:431] Setting filter for PID: 1100 1110 1120 1130
Jan 16 00:29:06 yavdr SatIP Server[6692]: 38 [http.c:208] GET web/desc.xml (size 1512)
Jan 16 00:29:06 yavdr SatIP Server[6692]: 39 [http.c:383] Connection closed by Client 1
Jan 16 00:29:06 yavdr SatIP Server[6692]: 40 [http.c:208] GET web/desc.xml (size 1512)
Jan 16 00:29:06 yavdr SatIP Server[6692]: 41 [http.c:383] Connection closed by Client 1
Jan 16 00:29:19 yavdr SatIP Server[6692]: 42 [rtsp.c:617] PLAY rtsp://192.168.178.101:554/stream=1?delpids=1110,1120,0,1100,1130 RTSP/1.0#015#012CSeq: 4#015#012Session: 00000002#015#012#015
Jan 16 00:29:19 yavdr SatIP Server[6692]: 43 [rtsp.c:386] Play Message
Jan 16 00:29:19 yavdr SatIP Server[6692]: 44 [tune.c:435] Removing filter for PID: 0 1100 1110 1120 1130
Jan 16 00:29:19 yavdr SatIP Server[6692]: 45 [rtsp.c:617] PLAY rtsp://192.168.178.101:554/stream=1?delpids=18 RTSP/1.0#015#012CSeq: 5#015#012Session: 00000002#015#012#015
Jan 16 00:29:19 yavdr SatIP Server[6692]: 46 [rtsp.c:386] Play Message
Jan 16 00:29:19 yavdr SatIP Server[6692]: 47 [tune.c:435] Removing filter for PID: 18
Jan 16 00:29:19 yavdr SatIP Server[6692]: 48 [rtsp.c:617] PLAY rtsp://192.168.178.101:554/stream=1?src=1&freq=11837&msys=dvbs&plts=off&fec=34&pol=h&ro=0.35&sr=27500&mtype=qpsk&pids=0 RTSP/1.0#015#012CSeq: 6#015#012Session: 00000002#015#012#015
Jan 16 00:29:19 yavdr SatIP Server[6692]: 49 [rtsp.c:386] Play Message
Jan 16 00:29:19 yavdr SatIP Server[6692]: 50 [tune.c:303] Start tuning
Jan 16 00:29:19 yavdr SatIP Server[6692]: 51 [tune.c:247] Sending DiSEqC
Jan 16 00:29:19 yavdr SatIP Server[6692]: 52 [tune.c:461] Frontend: 0, Tuned.
[... cut ...]
Jan 16 00:29:37 yavdr SatIP Server[6692]: 66 [rtsp.c:617] TEARDOWN rtsp://192.168.178.101:554/stream=1 RTSP/1.0#015#012CSeq: 11#015#012Session: 00000002#015#012#015
Jan 16 00:29:37 yavdr SatIP Server[6692]: 67 [rtsp.c:537] Teardown Message, gracefull: 1
Jan 16 00:29:37 yavdr SatIP Server[6692]: 68 [rtp.c:211] Frontend: 0, Stop streaming to 192.168.178.112 (52042 - 52043) (Streamed 0.013 MBytes)
Jan 16 00:29:37 yavdr SatIP Server[6692]: 69 [rtsp.c:494] Frontend: 0, Detached from client 192.168.178.112

9000h, 2015-01-15:
and also a scan is not working
Jan 16 00:43:46 yavdr SatIP Server[6730]: 0 [satpi.c:235] --- starting ---
Jan 16 00:43:46 yavdr SatIP Server[6730]: 1 [tune.c:126] Frontend Name: STV090x Multistandard
Jan 16 00:43:46 yavdr SatIP Server[6730]: 2 [tune.c:129] No DTV_ENUM_DELSYS ?? (DVB_API_VERSION: 5)
Jan 16 00:43:46 yavdr SatIP Server[6730]: 3 [tune.c:148] Frontend Type: Satellite (DVB-S)
Jan 16 00:43:46 yavdr SatIP Server[6730]: 4 [tune.c:152] Frontend Type: Satellite (DVB-S2)
Jan 16 00:43:46 yavdr SatIP Server[6730]: 5 [tune.c:157] Frontend Freq: 950000 Hz to 2150000 Hz
Jan 16 00:43:46 yavdr SatIP Server[6730]: 6 [tune.c:158] Frontend srat: 1000000 symbols/s to 45000000 symbols/s
Jan 16 00:43:46 yavdr SatIP Server[6730]: 7 [tune.c:126] Frontend Name: STV090x Multistandard
Jan 16 00:43:46 yavdr SatIP Server[6730]: 8 [tune.c:129] No DTV_ENUM_DELSYS ?? (DVB_API_VERSION: 5)
Jan 16 00:43:46 yavdr SatIP Server[6730]: 9 [tune.c:148] Frontend Type: Satellite (DVB-S)
Jan 16 00:43:46 yavdr SatIP Server[6730]: 10 [tune.c:152] Frontend Type: Satellite (DVB-S2)
Jan 16 00:43:46 yavdr SatIP Server[6730]: 11 [tune.c:157] Frontend Freq: 950000 Hz to 2150000 Hz
Jan 16 00:43:46 yavdr SatIP Server[6730]: 12 [tune.c:158] Frontend srat: 1000000 symbols/s to 45000000 symbols/s
Jan 16 00:43:46 yavdr SatIP Server[6730]: 13 [connection.c:259] eth0: 192.168.178.101 [00:22:4d:9d:38:05]
Jan 16 00:43:46 yavdr SatIP Server[6730]: 14 [ssdp.c:92] Setting up SSDP server
Jan 16 00:43:46 yavdr SatIP Server[6730]: 15 [rtcp.c:268] Setting up RTCP server
Jan 16 00:43:46 yavdr SatIP Server[6730]: 16 [rtcp.c:268] Setting up RTCP server
Jan 16 00:43:46 yavdr SatIP Server[6730]: 17 [rtp.c:50] Setting up RTP server
Jan 16 00:43:46 yavdr SatIP Server[6730]: 18 [rtp.c:50] Setting up RTP server
Jan 16 00:43:46 yavdr SatIP Server[6730]: 19 [http.c:328] Setting up HTTP server
Jan 16 00:43:46 yavdr SatIP Server[6730]: 20 [rtsp.c:567] Setting up RTSP server
Jan 16 00:44:00 yavdr SatIP Server[6730]: 21 [connection.c:188] RTSP Connection from 192.168.178.112 Port 49264
Jan 16 00:44:00 yavdr SatIP Server[6730]: 22 [rtsp.c:617] OPTIONS rtsp://192.168.178.101:554/ RTSP/1.0#015#012CSeq: 1#015#012#015
Jan 16 00:44:00 yavdr SatIP Server[6730]: 23 [rtsp.c:617] SETUP rtsp://192.168.178.101:554/?src=1&freq=10733&msys=dvbs&plts=off&fec=89&pol=h&ro=0.35&sr=22000&mtype=&pids=0 RTSP/1.0#015#012CSeq: 2#015#012Session: 00000000#015#012Transport: RTP/AVP;unicast;client_port=52012-52013#015#012#015
Jan 16 00:44:00 yavdr SatIP Server[6730]: 24 [rtsp.c:271] Setup Message
Jan 16 00:44:00 yavdr SatIP Server[6730]: 25 [rtsp.c:306] Frontend: 0, Attaching to client 192.168.178.112
Jan 16 00:44:00 yavdr SatIP Server[6730]: 26 [tune.c:303] Start tuning
Jan 16 00:44:00 yavdr SatIP Server[6730]: 27 [tune.c:247] Sending DiSEqC
Jan 16 00:44:00 yavdr SatIP Server[6730]: 28 [tune.c:461] Frontend: 0, Tuned.
Jan 16 00:44:00 yavdr SatIP Server[6730]: 29 [tune.c:474] Frontend: 0, Opened DVR.
Jan 16 00:44:00 yavdr SatIP Server[6730]: 30 [tune.c:431] Setting filter for PID: 0
Jan 16 00:44:00 yavdr SatIP Server[6730]: 31 [rtsp.c:617] PLAY rtsp://192.168.178.101:554/stream=1?addpids=0 RTSP/1.0#015#012CSeq: 2#015#012Session: 00000002#015#012#015
Jan 16 00:44:00 yavdr SatIP Server[6730]: 32 [rtsp.c:386] Play Message
Jan 16 00:44:00 yavdr SatIP Server[6730]: 33 [rtp.c:104] Frontend: 0, Start streaming to 192.168.178.112 (52012 - 52013)
Jan 16 00:44:15 yavdr SatIP Server[6730]: 34 [rtsp.c:617] PLAY rtsp://192.168.178.101:554/stream=1?delpids=0 RTSP/1.0#015#012CSeq: 3#015#012Session: 00000002#015#012#015
Jan 16 00:44:15 yavdr SatIP Server[6730]: 35 [rtsp.c:386] Play Message
Jan 16 00:44:15 yavdr SatIP Server[6730]: 36 [tune.c:435] Removing filter for PID: 0
Jan 16 00:44:15 yavdr SatIP Server[6730]: 37 [rtsp.c:617] PLAY rtsp://192.168.178.101:554/stream=1?src=1&freq=10733&msys=dvbs&plts=off&fec=89&pol=h&ro=0.35&sr=22000&mtype=&pids=0 RTSP/1.0#015#012CSeq: 4#015#012Session: 00000002#015#012#015
Jan 16 00:44:15 yavdr SatIP Server[6730]: 38 [rtsp.c:386] Play Message
Jan 16 00:44:15 yavdr SatIP Server[6730]: 39 [tune.c:303] Start tuning
Jan 16 00:44:15 yavdr SatIP Server[6730]: 40 [tune.c:247] Sending DiSEqC
Jan 16 00:44:15 yavdr SatIP Server[6730]: 41 [tune.c:461] Frontend: 0, Tuned.
Jan 16 00:44:15 yavdr SatIP Server[6730]: 42 [tune.c:431] Setting filter for PID: 0
Jan 16 00:44:15 yavdr SatIP Server[6730]: 43 [rtsp.c:617] PLAY rtsp://192.168.178.101:554/stream=1?addpids=0 RTSP/1.0#015#012CSeq: 5#015#012Session: 00000002#015#012#015
Jan 16 00:44:15 yavdr SatIP Server[6730]: 44 [rtsp.c:386] Play Message
Jan 16 00:44:31 yavdr SatIP Server[6730]: 45 [rtsp.c:617] PLAY rtsp://192.168.178.101:554/stream=1?delpids=0 RTSP/1.0#015#012CSeq: 6#015#012Session: 00000002#015#012#015
Jan 16 00:44:31 yavdr SatIP Server[6730]: 46 [rtsp.c:386] Play Message
Jan 16 00:44:31 yavdr SatIP Server[6730]: 47 [tune.c:435] Removing filter for PID: 0
Jan 16 00:44:31 yavdr SatIP Server[6730]: 48 [rtsp.c:617] PLAY rtsp://192.168.178.101:554/stream=1?src=1&freq=10733&msys=dvbs&plts=off&fec=89&pol=v&ro=0.35&sr=22000&mtype=&pids=0 RTSP/1.0#015#012CSeq: 7#015#012Session: 00000002#015#012#015
Jan 16 00:44:31 yavdr SatIP Server[6730]: 49 [rtsp.c:386] Play Message
Jan 16 00:44:31 yavdr SatIP Server[6730]: 50 [tune.c:303] Start tuning
Jan 16 00:44:31 yavdr SatIP Server[6730]: 51 [tune.c:247] Sending DiSEqC
Jan 16 00:44:31 yavdr SatIP Server[6730]: 52 [tune.c:461] Frontend: 0, Tuned.
[... cut ...]
Jan 16 00:46:07 yavdr SatIP Server[6730]: 114 [rtsp.c:617] TEARDOWN rtsp://192.168.178.101:554/stream=1 RTSP/1.0#015#012CSeq: 25#015#012Session: 00000002#015#012#015
Jan 16 00:46:07 yavdr SatIP Server[6730]: 115 [rtsp.c:537] Teardown Message, gracefull: 1
Jan 16 00:46:07 yavdr SatIP Server[6730]: 116 [rtp.c:211] Frontend: 0, Stop streaming to 192.168.178.112 (52012 - 52013) (Streamed 0.000 MBytes)
Jan 16 00:46:07 yavdr SatIP Server[6730]: 117 [rtsp.c:494] Frontend: 0, Detached from client 192.168.178.112

9000h, 2015-01-16:
the issue with the
Jan 16 00:44:00 yavdr SatIP Server[6730]: 22 [rtsp.c:617] OPTIONS rtsp://192.168.178.101:554/ RTSP/1.0#015#012CSeq: 1#015#012#015

could there be a issue with global and local var *msg

applog.h:	char *msg;

Barracuda09, 2015-01-16:
You mean that is comes before a SETUP message?
This is normal with DVBviewer it likes to know what it can do to the server.

But it puzzels me why it does not work for you..
I use for testing also DVBviewer and regular use tvheadend and i do not have this problem.

At some point is in the log it streamed some bytes.. 
Do you see in the web page that there is comming data from the PIDS?

Barracuda09, 2015-01-18:
Hi,

If you have some time, would you give it another try.

9000h, 2015-01-19:
next try with latest git, still no picture but could be a incompatible dvbapi

Jan 20 00:06:17 yavdr SatIP Server[5127]: 0 --- starting ---
Jan 20 00:06:17 yavdr SatIP Server[5127]: 1 Not the preferred DVB API version, for correct function it should be 5.5 or higher
Jan 20 00:06:17 yavdr SatIP Server[5127]: 2 current DVB_API_VERSION: 5.4
Jan 20 00:06:17 yavdr SatIP Server[5127]: 3 Detecting frontends in: /dev/dvb
Jan 20 00:06:17 yavdr SatIP Server[5127]: 4 Frontends found: 2
Jan 20 00:06:17 yavdr SatIP Server[5127]: 5 Open FE fd: 4 readonly 1
Jan 20 00:06:17 yavdr SatIP Server[5127]: 6 Frontend Name: STV090x Multistandard
Jan 20 00:06:17 yavdr SatIP Server[5127]: 7 Open FE fd: 4 readonly 1
Jan 20 00:06:17 yavdr SatIP Server[5127]: 8 Frontend Name: STV090x Multistandard
Jan 20 00:06:17 yavdr SatIP Server[5127]: 9 eth0: 192.168.178.101 [00:22:4d:9d:38:05]
Jan 20 00:06:17 yavdr SatIP Server[5127]: 10 Setting up SSDP server
Jan 20 00:06:17 yavdr SatIP Server[5127]: 11 Setting up RTCP server
Jan 20 00:06:17 yavdr SatIP Server[5127]: 12 Setting up RTCP server
Jan 20 00:06:17 yavdr SatIP Server[5127]: 13 Setting up RTP server
Jan 20 00:06:17 yavdr SatIP Server[5127]: 14 Setting up RTP server
Jan 20 00:06:17 yavdr SatIP Server[5127]: 15 Setting up HTTP server
Jan 20 00:06:17 yavdr SatIP Server[5127]: 16 Setting up RTSP server
Jan 20 00:06:25 yavdr SatIP Server[5127]: 17 RTSP Connection from 192.168.178.112 Port 49696
Jan 20 00:06:25 yavdr SatIP Server[5127]: 18 Setup Message
Jan 20 00:06:25 yavdr SatIP Server[5127]: 19 Frontend: 0, Attaching to client 192.168.178.112
Jan 20 00:06:25 yavdr SatIP Server[5127]: 20 Open FE fd: 12 readonly 0
Jan 20 00:06:25 yavdr SatIP Server[5127]: 21 Frontend: 0, Waiting on lock.
Jan 20 00:06:25 yavdr SatIP Server[5127]: 22 Frontend: 0, Not locked yet (status 0)...
Jan 20 00:06:26 yavdr SatIP Server[5127]: 23 Frontend: 0, Not locked yet (status 3)...
Jan 20 00:06:26 yavdr SatIP Server[5127]: 24 Frontend: 0, Not locked yet (status 3)...
Jan 20 00:06:26 yavdr SatIP Server[5127]: 25 Frontend: 0, Opened DVR fd: 13.
Jan 20 00:06:26 yavdr SatIP Server[5127]: 26 Play Message
Jan 20 00:06:26 yavdr SatIP Server[5127]: 27 Frontend: 0, Waiting on lock.
Jan 20 00:06:26 yavdr SatIP Server[5127]: 28 Frontend: 0, Not locked yet (status 0)...
Jan 20 00:06:28 yavdr SatIP Server[5127]: 29 Frontend: 0, Not locked yet (status 3)...
Jan 20 00:06:28 yavdr SatIP Server[5127]: 30 Frontend: 0, Not locked yet (status 3)...
Jan 20 00:06:28 yavdr SatIP Server[5127]: 31 Frontend: 0, Start RTCP stream to 192.168.178.112 (52010 - 52011)
Jan 20 00:06:28 yavdr SatIP Server[5127]: 32 Frontend: 0, Open Frontend Monitor for 192.168.178.112 (52010 - 52011)
Jan 20 00:06:28 yavdr SatIP Server[5127]: 33 Open FE fd: 16 readonly 1
Jan 20 00:06:28 yavdr SatIP Server[5127]: 34 Frontend: 0, Start streaming to 192.168.178.112 (52010 - 52011)
Jan 20 00:06:28 yavdr SatIP Server[5127]: 35 Play Message
Jan 20 00:06:30 yavdr SatIP Server[5127]: 36 Frontend: 0, Waiting on lock.
Jan 20 00:06:31 yavdr SatIP Server[5127]: 37 Frontend: 0, Not locked yet (status 3)...
Jan 20 00:06:31 yavdr SatIP Server[5127]: 38 Frontend: 0, Not locked yet (status 3)...
Jan 20 00:06:31 yavdr SatIP Server[5127]: 39 Frontend: 0, Not locked yet (status 3)...
Jan 20 00:07:11 yavdr SatIP Server[5127]: 40 Play Message
Jan 20 00:07:11 yavdr SatIP Server[5127]: 41 Frontend: 0, Waiting on lock.
Jan 20 00:07:12 yavdr SatIP Server[5127]: 42 Frontend: 0, Not locked yet (status 3)...
Jan 20 00:07:12 yavdr SatIP Server[5127]: 43 Frontend: 0, Not locked yet (status 3)...
Jan 20 00:07:12 yavdr SatIP Server[5127]: 44 Frontend: 0, Not locked yet (status 0)...
Jan 20 00:07:12 yavdr SatIP Server[5127]: 45 Play Message
Jan 20 00:07:13 yavdr SatIP Server[5127]: 46 Frontend: 0, Waiting on lock.
Jan 20 00:07:14 yavdr SatIP Server[5127]: 47 Frontend: 0, Not locked yet (status 0)...
Jan 20 00:07:14 yavdr SatIP Server[5127]: 48 Frontend: 0, Not locked yet (status 3)...
Jan 20 00:07:14 yavdr SatIP Server[5127]: 49 Frontend: 0, Not locked yet (status 3)...
Jan 20 00:07:14 yavdr SatIP Server[5127]: 50 Play Message
Jan 20 00:07:14 yavdr SatIP Server[5127]: 51 Frontend: 0, Waiting on lock.
Jan 20 00:07:14 yavdr SatIP Server[5127]: 52 Frontend: 0, Not locked yet (status 0)...
Jan 20 00:07:15 yavdr SatIP Server[5127]: 53 Frontend: 0, Not locked yet (status 3)...
Jan 20 00:07:15 yavdr SatIP Server[5127]: 54 Frontend: 0, Not locked yet (status 3)...
Jan 20 00:07:15 yavdr SatIP Server[5127]: 55 Play Message
Jan 20 00:07:16 yavdr SatIP Server[5127]: 56 Frontend: 0, Waiting on lock.
Jan 20 00:07:18 yavdr SatIP Server[5127]: 57 Frontend: 0, Not locked yet (status 3)...
Jan 20 00:07:18 yavdr SatIP Server[5127]: 58 Frontend: 0, Not locked yet (status 3)...
Jan 20 00:07:18 yavdr SatIP Server[5127]: 59 Frontend: 0, Not locked yet (status 3)...
Jan 20 00:07:20 yavdr SatIP Server[5127]: 60 Play Message
Jan 20 00:07:20 yavdr SatIP Server[5127]: 61 Frontend: 0, Waiting on lock.
Jan 20 00:07:20 yavdr SatIP Server[5127]: 62 Frontend: 0, Not locked yet (status 0)...
Jan 20 00:07:21 yavdr SatIP Server[5127]: 63 Frontend: 0, Not locked yet (status 3)...
Jan 20 00:07:21 yavdr SatIP Server[5127]: 64 Frontend: 0, Not locked yet (status 3)...

Barracuda09, 2015-01-20:
Hi,
Thanks for trying again. What I see is that the tune does not lock.
What is strange, do you have the correct switch output?
Can you try a direct connection to LNB.

Here it is working... scan and tune.. but I find a bit strange how the
different sat can be defined.
And run a debug log please. DVB api should work.
Thanks again.

9000h, 2015-01-20:
on the same machine vdr is working fine, hmm.

9000h, 2015-01-21:
git from today 
root@yavdr2:/usr/src/satip/SATPI-master# make
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG tune.c 
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG ssdp.c 
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG rtsp.c 
rtsp.c: In Funktion »parse_stream_string«:
rtsp.c:216:36: Fehler: »SYS_DVBC_ANNEX_A« nicht deklariert (erste Benutzung in dieser Funktion)
rtsp.c:216:36: Anmerkung: jeder nicht deklarierte Bezeichner wird nur einmal für jede Funktion, in der er vorkommt, gemeldet
make: *** [rtsp.o] Fehler 1

Barracuda09, 2015-01-21:
Oeps.... Thanks did not recompile last changes..

Now its fixed

9000h, 2015-01-21:
rtsp://192.168.178.133:554/?freq=338.000&msys=dvbc&sr=6900&mtype=256qam&pids=0,100,101,104,102,103,106

still no success with VLC you may not provide Content-Type: Content-Base:

Jan 21 22:13:05 yavdr2 SatIP Server[20660]: 0 --- starting ---
Jan 21 22:13:05 yavdr2 SatIP Server[20660]: 1 Not the preferred DVB API version, for correct function it should be 5.5 or higher
Jan 21 22:13:05 yavdr2 SatIP Server[20660]: 2 current DVB_API_VERSION: 5.4
Jan 21 22:13:05 yavdr2 SatIP Server[20660]: 3 Detecting frontends in: /dev/dvb
Jan 21 22:13:05 yavdr2 SatIP Server[20660]: 4 Frontends found: 3
Jan 21 22:13:05 yavdr2 SatIP Server[20660]: 5 Frontend Name: Philips TDA10023 DVB-C
Jan 21 22:13:05 yavdr2 SatIP Server[20660]: 6 Frontend Name: Montage DS3103/TS2022
Jan 21 22:13:05 yavdr2 SatIP Server[20660]: 7 Frontend Name: Sit2 DVB-T2/C
Jan 21 22:13:05 yavdr2 SatIP Server[20660]: 8 eth0: 192.168.178.133 [00:40:ca:a2:c7:9d]
Jan 21 22:13:05 yavdr2 SatIP Server[20660]: 9 Setting up HTTP server
Jan 21 22:13:05 yavdr2 SatIP Server[20660]: 10 Setting up RTP server
Jan 21 22:13:05 yavdr2 SatIP Server[20660]: 11 Setting up RTP server
Jan 21 22:13:05 yavdr2 SatIP Server[20660]: 12 Setting up RTCP server
Jan 21 22:13:05 yavdr2 SatIP Server[20660]: 13 Setting up RTSP server
Jan 21 22:13:05 yavdr2 SatIP Server[20660]: 14 Setting up RTCP server
Jan 21 22:13:05 yavdr2 SatIP Server[20660]: 15 Setting up SSDP server
Jan 21 22:13:17 yavdr2 SatIP Server[20660]: 16 RTSP Connection from 192.168.178.130 Port 55851
Jan 21 22:13:22 yavdr2 SatIP Server[20660]: 17 RTSP Connection closed by Client: 192.168.178.130 ??
Jan 21 22:13:22 yavdr2 SatIP Server[20660]: 18 Teardown Message, gracefull: 0
Jan 21 22:13:22 yavdr2 SatIP Server[20660]: 19 Frontend: x, Not attached anymore to client 192.168.178.130
Jan 21 22:13:22 yavdr2 SatIP Server[20660]: 20 RTSP Connection from 192.168.178.130 Port 55852
Jan 21 22:13:22 yavdr2 SatIP Server[20660]: 21 RTSP Connection closed by Client: 192.168.178.130 ??
Jan 21 22:13:22 yavdr2 SatIP Server[20660]: 22 Teardown Message, gracefull: 0
Jan 21 22:13:22 yavdr2 SatIP Server[20660]: 23 Frontend: x, Not attached anymore to client 192.168.178.130
Jan 21 22:13:22 yavdr2 SatIP Server[20660]: 24 RTSP Connection from 192.168.178.130 Port 55853
Jan 21 22:13:27 yavdr2 SatIP Server[20660]: 25 RTSP Connection closed by Client: 192.168.178.130 ??
Jan 21 22:13:27 yavdr2 SatIP Server[20660]: 26 Teardown Message, gracefull: 0
Jan 21 22:13:27 yavdr2 SatIP Server[20660]: 27 Frontend: x, Not attached anymore to client 192.168.178.130
Jan 21 22:13:27 yavdr2 SatIP Server[20660]: 28 RTSP Connection from 192.168.178.130 Port 55854
Jan 21 22:13:27 yavdr2 SatIP Server[20660]: 29 RTSP Connection closed by Client: 192.168.178.130 ??
Jan 21 22:13:27 yavdr2 SatIP Server[20660]: 30 Teardown Message, gracefull: 0
Jan 21 22:13:27 yavdr2 SatIP Server[20660]: 31 Frontend: x, Not attached anymore to client 192.168.178.130
Jan 21 22:13:27 yavdr2 SatIP Server[20660]: 32 RTSP Connection from 192.168.178.130 Port 55855
Jan 21 22:13:32 yavdr2 SatIP Server[20660]: 33 RTSP Connection closed by Client: 192.168.178.130 ??
Jan 21 22:13:32 yavdr2 SatIP Server[20660]: 34 Teardown Message, gracefull: 0
Jan 21 22:13:32 yavdr2 SatIP Server[20660]: 35 Frontend: x, Not attached anymore to client 192.168.178.130
Jan 21 22:13:32 yavdr2 SatIP Server[20660]: 36 RTSP Connection from 192.168.178.130 Port 55856
Jan 21 22:13:32 yavdr2 SatIP Server[20660]: 37 RTSP Connection closed by Client: 192.168.178.130 ??
Jan 21 22:13:32 yavdr2 SatIP Server[20660]: 38 Teardown Message, gracefull: 0
Jan 21 22:13:32 yavdr2 SatIP Server[20660]: 39 Frontend: x, Not attached anymore to client 192.168.178.130

a test with DD-TV.exe did work and I got a picture
Jan 21 22:22:02 yavdr2 SatIP Server[20674]: 0 --- starting ---
Jan 21 22:22:02 yavdr2 SatIP Server[20674]: 1 Not the preferred DVB API version, for correct function it should be 5.5 or higher
Jan 21 22:22:02 yavdr2 SatIP Server[20674]: 2 current DVB_API_VERSION: 5.4
Jan 21 22:22:02 yavdr2 SatIP Server[20674]: 3 Detecting frontends in: /dev/dvb
Jan 21 22:22:02 yavdr2 SatIP Server[20674]: 4 Frontends found: 3
Jan 21 22:22:02 yavdr2 SatIP Server[20674]: 5 Frontend Name: Philips TDA10023 DVB-C
Jan 21 22:22:02 yavdr2 SatIP Server[20674]: 6 Frontend Name: Montage DS3103/TS2022
Jan 21 22:22:02 yavdr2 SatIP Server[20674]: 7 Frontend Name: Sit2 DVB-T2/C
Jan 21 22:22:02 yavdr2 SatIP Server[20674]: 8 eth0: 192.168.178.133 [00:40:ca:a2:c7:9d]
Jan 21 22:22:02 yavdr2 SatIP Server[20674]: 9 Setting up RTSP server
Jan 21 22:22:02 yavdr2 SatIP Server[20674]: 10 Setting up RTP server
Jan 21 22:22:02 yavdr2 SatIP Server[20674]: 11 Setting up HTTP server
Jan 21 22:22:02 yavdr2 SatIP Server[20674]: 12 Setting up RTCP server
Jan 21 22:22:02 yavdr2 SatIP Server[20674]: 13 Setting up RTCP server
Jan 21 22:22:02 yavdr2 SatIP Server[20674]: 14 Setting up SSDP server
Jan 21 22:22:02 yavdr2 SatIP Server[20674]: 15 Setting up RTP server
Jan 21 22:22:12 yavdr2 SatIP Server[20674]: 16 RTSP Connection from 192.168.178.112 Port 49251
Jan 21 22:22:12 yavdr2 SatIP Server[20674]: 17 Setup Message
Jan 21 22:22:12 yavdr2 SatIP Server[20674]: 18 Frontend: 0, Attaching to client 192.168.178.112
Jan 21 22:22:12 yavdr2 SatIP Server[20674]: 19 Frontend: 0, Opened FE fd: 12.
Jan 21 22:22:12 yavdr2 SatIP Server[20674]: 20 Frontend: 0, Waiting on lock.
Jan 21 22:22:12 yavdr2 SatIP Server[20674]: 21 Frontend: 0, Not locked yet (FE status 0)...
[... cut ...]
Jan 21 22:25:19 yavdr2 SatIP Server[20674]: 64 Play Message
Jan 21 22:25:20 yavdr2 SatIP Server[20674]: 65 Play Message
Jan 21 22:25:20 yavdr2 SatIP Server[20674]: 66 Play Message
Jan 21 22:25:28 yavdr2 SatIP Server[20674]: 67 Play Message
Jan 21 22:25:28 yavdr2 SatIP Server[20674]: 68 Teardown Message, gracefull: 1
Jan 21 22:25:28 yavdr2 SatIP Server[20674]: 69 Frontend: 0, Stop streaming to 192.168.178.112 (52012 - 52013) (Streamed 0.015 MBytes)
Jan 21 22:25:28 yavdr2 SatIP Server[20674]: 70 Frontend: 0, Detached from client 192.168.178.112

Barracuda09, 2015-01-24:
I will close this defect (until somebody else gives it a try), for which i know it should work.

closed 2015-01-24 by Barracuda09

9000h, 2015-01-24:
rtsp streaming with VLC is still not working

Jan 24 10:08:34 yavdr2 SatIP Server[3597]: 0 --- starting ---
Jan 24 10:08:34 yavdr2 SatIP Server[3597]: 1 Not the preferred DVB API version, for correct function it should be 5.5 or higher
Jan 24 10:08:34 yavdr2 SatIP Server[3597]: 2 current DVB_API_VERSION: 5.4
Jan 24 10:08:34 yavdr2 SatIP Server[3597]: 3 Detecting frontends in: /dev/dvb
Jan 24 10:08:34 yavdr2 SatIP Server[3597]: 4 Frontends found: 3
Jan 24 10:08:34 yavdr2 SatIP Server[3597]: 5 Frontend Name: Philips TDA10023 DVB-C
Jan 24 10:08:34 yavdr2 SatIP Server[3597]: 6 Frontend Name: Montage DS3103/TS2022
Jan 24 10:08:34 yavdr2 SatIP Server[3597]: 7 Frontend Name: Sit2 DVB-T2/C
Jan 24 10:08:34 yavdr2 SatIP Server[3597]: 8 eth0: 192.168.178.133 [00:40:ca:a2:c7:9d]
Jan 24 10:08:34 yavdr2 SatIP Server[3597]: 9 Setting up RTSP server
Jan 24 10:08:34 yavdr2 SatIP Server[3597]: 10 Setting up HTTP server
Jan 24 10:08:34 yavdr2 SatIP Server[3597]: 11 Setting up 8 RTP servers
Jan 24 10:08:34 yavdr2 SatIP Server[3597]: 12 Setting up 8 RTCP servers
Jan 24 10:08:34 yavdr2 SatIP Server[3597]: 13 Setting up SSDP server
Jan 24 10:09:03 yavdr2 SatIP Server[3597]: 14 RTSP Connection from 192.168.178.130 Port 55466
Jan 24 10:09:08 yavdr2 SatIP Server[3597]: 15 RTSP Connection closed by Client: 192.168.178.130 ??
Jan 24 10:09:08 yavdr2 SatIP Server[3597]: 16 Frontend: x, Teardown Message, gracefull: 0
Jan 24 10:09:08 yavdr2 SatIP Server[3597]: 17 Frontend: x, Not attached anymore to client 192.168.178.130
Jan 24 10:09:08 yavdr2 SatIP Server[3597]: 18 RTSP Connection from 192.168.178.130 Port 55467
Jan 24 10:09:08 yavdr2 SatIP Server[3597]: 19 RTSP Connection closed by Client: 192.168.178.130 ??
Jan 24 10:09:08 yavdr2 SatIP Server[3597]: 20 Frontend: x, Teardown Message, gracefull: 0
Jan 24 10:09:08 yavdr2 SatIP Server[3597]: 21 Frontend: x, Not attached anymore to client 192.168.178.130
Jan 24 10:09:08 yavdr2 SatIP Server[3597]: 22 RTSP Connection from 192.168.178.130 Port 55468
Jan 24 10:09:13 yavdr2 SatIP Server[3597]: 23 RTSP Connection closed by Client: 192.168.178.130 ??
Jan 24 10:09:13 yavdr2 SatIP Server[3597]: 24 Frontend: x, Teardown Message, gracefull: 0
Jan 24 10:09:13 yavdr2 SatIP Server[3597]: 25 Frontend: x, Not attached anymore to client 192.168.178.130
Jan 24 10:09:13 yavdr2 SatIP Server[3597]: 26 RTSP Connection from 192.168.178.130 Port 55469
Jan 24 10:09:13 yavdr2 SatIP Server[3597]: 27 RTSP Connection closed by Client: 192.168.178.130 ??
Jan 24 10:09:13 yavdr2 SatIP Server[3597]: 28 Frontend: x, Teardown Message, gracefull: 0
Jan 24 10:09:13 yavdr2 SatIP Server[3597]: 29 Frontend: x, Not attached anymore to client 192.168.178.130
Jan 24 10:10:02 yavdr2 SatIP Server[3597]: 30 RTSP Connection from 192.168.178.130 Port 55470
Jan 24 10:10:07 yavdr2 SatIP Server[3597]: 31 RTSP Connection closed by Client: 192.168.178.130 ??
Jan 24 10:10:07 yavdr2 SatIP Server[3597]: 32 Frontend: x, Teardown Message, gracefull: 0
Jan 24 10:10:07 yavdr2 SatIP Server[3597]: 33 Frontend: x, Not attached anymore to client 192.168.178.130
Jan 24 10:10:07 yavdr2 SatIP Server[3597]: 34 RTSP Connection from 192.168.178.130 Port 55471
Jan 24 10:10:07 yavdr2 SatIP Server[3597]: 35 RTSP Connection closed by Client: 192.168.178.130 ??
Jan 24 10:10:07 yavdr2 SatIP Server[3597]: 36 Frontend: x, Teardown Message, gracefull: 0
Jan 24 10:10:07 yavdr2 SatIP Server[3597]: 37 Frontend: x, Not attached anymore to client 192.168.178.130
Jan 24 10:10:07 yavdr2 SatIP Server[3597]: 38 RTSP Connection from 192.168.178.130 Port 55472
Jan 24 10:10:12 yavdr2 SatIP Server[3597]: 39 RTSP Connection closed by Client: 192.168.178.130 ??
Jan 24 10:10:12 yavdr2 SatIP Server[3597]: 40 Frontend: x, Teardown Message, gracefull: 0
Jan 24 10:10:12 yavdr2 SatIP Server[3597]: 41 Frontend: x, Not attached anymore to client 192.168.178.130
Jan 24 10:10:12 yavdr2 SatIP Server[3597]: 42 RTSP Connection from 192.168.178.130 Port 55473
Jan 24 10:10:12 yavdr2 SatIP Server[3597]: 43 RTSP Connection closed by Client: 192.168.178.130 ??
Jan 24 10:10:12 yavdr2 SatIP Server[3597]: 44 Frontend: x, Teardown Message, gracefull: 0
Jan 24 10:10:12 yavdr2 SatIP Server[3597]: 45 Frontend: x, Not attached anymore to client 192.168.178.130
Jan 24 10:10:12 yavdr2 SatIP Server[3597]: 46 RTSP Connection from 192.168.178.130 Port 55474
Jan 24 10:10:17 yavdr2 SatIP Server[3597]: 47 RTSP Connection closed by Client: 192.168.178.130 ??
Jan 24 10:10:17 yavdr2 SatIP Server[3597]: 48 Frontend: x, Teardown Message, gracefull: 0
Jan 24 10:10:17 yavdr2 SatIP Server[3597]: 49 Frontend: x, Not attached anymore to client 192.168.178.130
Jan 24 10:10:17 yavdr2 SatIP Server[3597]: 50 RTSP Connection from 192.168.178.130 Port 55475
Jan 24 10:10:17 yavdr2 SatIP Server[3597]: 51 RTSP Connection closed by Client: 192.168.178.130 ??
Jan 24 10:10:17 yavdr2 SatIP Server[3597]: 52 Frontend: x, Teardown Message, gracefull: 0
Jan 24 10:10:17 yavdr2 SatIP Server[3597]: 53 Frontend: x, Not attached anymore to client 192.168.178.130
Jan 24 10:10:17 yavdr2 SatIP Server[3597]: 54 RTSP Connection from 192.168.178.130 Port 55476
Jan 24 10:10:22 yavdr2 SatIP Server[3597]: 55 RTSP Connection closed by Client: 192.168.178.130 ??
Jan 24 10:10:22 yavdr2 SatIP Server[3597]: 56 Frontend: x, Teardown Message, gracefull: 0
Jan 24 10:10:22 yavdr2 SatIP Server[3597]: 57 Frontend: x, Not attached anymore to client 192.168.178.130
Jan 24 10:10:22 yavdr2 SatIP Server[3597]: 58 RTSP Connection from 192.168.178.130 Port 55477
Jan 24 10:10:22 yavdr2 SatIP Server[3597]: 59 RTSP Connection closed by Client: 192.168.178.130 ??
Jan 24 10:10:22 yavdr2 SatIP Server[3597]: 60 Frontend: x, Teardown Message, gracefull: 0
Jan 24 10:10:22 yavdr2 SatIP Server[3597]: 61 Frontend: x, Not attached anymore to client 192.168.178.130

Barracuda09, 2015-01-24:
Hi 9000h,

Can you than please open another defect. this was about DVBViewer.
and attach a debug log?

Thanks,
Marc