Barracuda09/SATPI #6: LOG issue

9000h, 2015-01-15:
in rtcp.c line 497
-                        SI_LOG_ERROR("Frontend: x, Not attached anymore to client %s", client->ip_addr);
+                       SI_LOG_ERROR("Frontend: %d, Not attached anymore to client %s", client->fe->index, client->ip_addr);

Barracuda09, 2015-01-15:
Do you see these logs?

9000h, 2015-01-15:
yes I had this in the logs but the change creates a core dump
Core was generated by `./satpi --no-daemon --yes-ssdp'.
Program terminated with signal 11, Segmentation fault.
#0  0x0000000000405675 in teardown_session (arg=0x614610) at rtsp.c:497
497				SI_LOG_ERROR("Frontend: %d, Not attached anymore to client %s", client->fe->index, client->ip_addr);
(gdb) print client->fe
$1 = (Frontend_t *) 0x0
(gdb)

9000h, 2015-01-15:
I saw this yesterday
Jan 14 23:07:41 yavdr satpi: 0 Frontend Name: STV090x Multistandard
Jan 14 23:07:41 yavdr satpi: 1 No DTV_ENUM_DELSYS ?? (DVB_API_VERSION: 5)
Jan 14 23:07:41 yavdr satpi: 2 Frontend Type: Satellite (DVB-S)
Jan 14 23:07:41 yavdr satpi: 3 Frontend Type: Satellite (DVB-S2)
Jan 14 23:07:41 yavdr satpi: 4 Frontend Freq: 950000 Hz to 2150000 Hz
Jan 14 23:07:41 yavdr satpi: 5 Frontend srat: 1000000 symbols/s to 45000000 symbols/s
Jan 14 23:07:41 yavdr satpi: 6 Frontend Name: STV090x Multistandard
Jan 14 23:07:41 yavdr satpi: 7 No DTV_ENUM_DELSYS ?? (DVB_API_VERSION: 5)
Jan 14 23:07:41 yavdr satpi: 8 Frontend Type: Satellite (DVB-S)
Jan 14 23:07:41 yavdr satpi: 9 Frontend Type: Satellite (DVB-S2)
Jan 14 23:07:41 yavdr satpi: 10 Frontend Freq: 950000 Hz to 2150000 Hz
Jan 14 23:07:41 yavdr satpi: 11 Frontend srat: 1000000 symbols/s to 45000000 symbols/s
Jan 14 23:07:41 yavdr SatIP Server[7177]: 0 --- starting ---
Jan 14 23:07:41 yavdr SatIP Server[7177]: 1 eth0: 192.168.178.101 [00:22:4d:9d:38:0]
Jan 14 23:07:41 yavdr SatIP Server[7177]: 2 Setting up RTSP server
Jan 14 23:07:41 yavdr SatIP Server[7177]: 3 Setting up RTP server
Jan 14 23:07:41 yavdr SatIP Server[7177]: 4 Setting up HTTP server
Jan 14 23:07:41 yavdr SatIP Server[7177]: 5 Setting up RTP server
Jan 14 23:07:41 yavdr SatIP Server[7177]: 6 Setting up SSDP server
Jan 14 23:07:42 yavdr SatIP Server[7177]: 7 GET web/desc.xml (size 1512)
Jan 14 23:07:42 yavdr SatIP Server[7177]: 8 Connection closed by Client 1
Jan 14 23:07:42 yavdr SatIP Server[7177]: 9 GET web/desc.xml (size 1512)
Jan 14 23:07:42 yavdr SatIP Server[7177]: 10 Connection closed by Client 1
Jan 14 23:07:54 yavdr SatIP Server[7177]: 11 RTSP Connection from 192.168.178.130 Port 54480
Jan 14 23:07:54 yavdr SatIP Server[7177]: 12 OPTIONS rtsp://192.168.178.101:554/?src=1&freq=10744&msys=dvbs&plts=off&fec=&pol=h&ro=0.35&sr=22000&mtype=qpsk&pids=0 RTSP/1.0#015#012CSeq: 2#015#012User-Agent: LibVLC/2.2.0-rc1 (LIVE555 Streaming Media v2014.01.21)#015#012#015
Jan 14 23:07:59 yavdr SatIP Server[7177]: 13 RTSP Connection closed by Client: 192.168.178.130 ??
Jan 14 23:07:59 yavdr SatIP Server[7177]: 14 Teardown Message, gracefull: 0
Jan 14 23:07:59 yavdr SatIP Server[7177]: 15 Frontend: x, Not attached anymore to client 192.168.178.130
Jan 14 23:07:59 yavdr SatIP Server[7177]: 16 RTSP Connection from 192.168.178.130 Port 54481
Jan 14 23:07:59 yavdr SatIP Server[7177]: 17 OPTIONS rtsp://192.168.178.101:554 RTSP/1.0#015#012CSeq: 1#015#012User-Agent: RealMedia Player Version 6.0.9.1235 (linux-2.0-libc6-i386-gcc2.95)#015#012ClientChallenge: 9e26d33f2984236010ef6253fb1887f7#015#012PlayerStarttime: [28/03/2003:22:50:23 00:00]#015#012CompanyID: KnKV4M4I/B2FjJ1TToLycw==#015#012GUID: 00000000-0000-0000-0000-000000000000#015#012RegionData: 0#015#012ClientID: Linux_2.4_6.0.9.1235_play32_RN01_EN_586#015#012#015
Jan 14 23:07:59 yavdr SatIP Server[7177]: 18 RTSP Connection closed by Client: 192.168.178.130 ??
Jan 14 23:07:59 yavdr SatIP Server[7177]: 19 Teardown Message, gracefull: 0
Jan 14 23:07:59 yavdr SatIP Server[7177]: 20 Frontend: x, Not attached anymore to client 192.168.178.130
Jan 14 23:07:59 yavdr SatIP Server[7177]: 21 RTSP Connection from 192.168.178.130 Port 54482
Jan 14 23:07:59 yavdr SatIP Server[7177]: 22 OPTIONS rtsp://192.168.178.101:554/?src=1&freq=10744&msys=dvbs&plts=off&fec=&pol=h&ro=0.35&sr=22000&mtype=qpsk&pids=0 RTSP/1.0#015#012CSeq: 2#015#012User-Agent: LibVLC/2.2.0-rc1 (LIVE555 Streaming Media v2014.01.21)#015#012#015
Jan 14 23:08:04 yavdr SatIP Server[7177]: 23 RTSP Connection closed by Client: 192.168.178.130 ??

9000h, 2015-01-15:
I also get still a compile warning
make "BUILD=debug"
make[1]: Entering directory `/home/yavdr/satip/SATPI-master'
gcc -c -Wall -Wextra -Winit-self  -O0 -g3 -DDEBUG -fstack-protector-all rtsp.c 
rtsp.c: In function 'setup_rtsp':
rtsp.c:276:34: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
gcc  tune.o ssdp.o rtsp.o rtp.o rtcp.o http.o utils.o connection.o applog.o satpi.o -o satpi -lpthread -lrt

maybe it comes from comparing this
const int fe_nr
tune.h: size_t           max_fe;

Barracuda09, 2015-01-15:
Yes. if something is strange... try to first
- make clean 
- make debug

Barracuda09, 2015-01-15:
Strange:
#0 0x0000000000405675 in teardown_session (arg=0x614610) at rtsp.c:497
497 SI_LOG_ERROR("Frontend: %d, Not attached anymore to client %s", client->fe->index, client->ip_addr);

This is not the correct line it seems.

9000h, 2015-01-15:
I did change it back

closed 2015-01-15 by 9000h