Barracuda09/SATPI #7: All output in a single line on SAT>IP Log Page with Safari web browser

pbriesch, 2015-01-25:
Hi,

in Safari all output is in a single line which gets longer and longer. In Google Chrome and Firefox it works as it it supposed to.

Barracuda09, 2015-01-26:
Thanks for you input. Sorry I never tried Safari.
Will see if I can fix this.

Marc

Barracuda09, 2015-01-27:
If you like you can try it.

pbriesch, 2015-01-27:
Hi Marc,

still does not work.

Paul

Barracuda09, 2015-01-27:
Ok sorry.. I do not have Safari, I can not test this my self..

I will look into this. Thanks for trying.

Marc

pbriesch, 2015-01-27:
Marc, thank you very much for your efforts!
Paul

Barracuda09, 2015-03-02:
Hi,

Is this issue is fixed with latest commits?

Marc

pbriesch, 2015-03-02:
Yes, it is!

What was the reason?

Paul

closed 2015-03-02 by pbriesch

Barracuda09, 2015-03-03:
Hi Paul,

Well problem was with the textareas - white-space attribute