Barracuda09/SATPI #8: compile warning in function ‘describe_rtsp’

pbriesch, 2015-01-25:
I got following compile warning: 
```
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG rtsp.c
rtsp.c: In function ‘describe_rtsp’:
rtsp.c:592:2: warning: format ‘%d’ expects argument of type ‘int’, but argument 6 has type ‘size_t’ [-Wformat=]
  snprintf(rtspOk, sizeof(rtspOk), RTSP_DESCRIBE_OK, client->rtsp.cseq, server_ip_addr, strlen(rtspCont), rtspCont);
  ^
rtsp.c:592:2: warning: format ‘%d’ expects argument of type ‘int’, but argument 6 has type ‘size_t’ [-Wformat=]
```
My machine:  
   Linux version 3.16.0-29-generic (buildd@tipua) (gcc version 4.9.1 (Ubuntu 4.9.1-16ubuntu6) ) #39-Ubuntu SMP Mon Dec 15 22:27:29 UTC 2014 (Ubuntu 3.16.0-29.39-generic 3.16.7-ckt2)

9000h, 2015-01-25:
so it should be probably  %zu

Barracuda09, 2015-01-26:
Thanks for your compile report.
I will fix this.

Marc

closed 2015-01-29 by pbriesch