Barracuda09/SATPI #9: crash on pids=all

9000h, 2015-01-26:
when try to use vlc to open rtsp://192.168.178.133/?freq=338.000&msys=dvbc&sr=6900&mtype=256qam&pids=all it will crash

in this mode you can select a different program inside vlc 

Jan 27 00:29:35 yavdr2 SatIP Server[15812]: 0 --- starting ---
Jan 27 00:29:35 yavdr2 SatIP Server[15812]: 1 Not the preferred DVB API version, for correct function it should be 5.5 or higher
Jan 27 00:29:35 yavdr2 SatIP Server[15812]: 2 current DVB_API_VERSION: 5.4
Jan 27 00:29:35 yavdr2 SatIP Server[15812]: 3 Detecting frontends in: /dev/dvb
Jan 27 00:29:35 yavdr2 SatIP Server[15812]: 4 Frontends found: 4
Jan 27 00:29:35 yavdr2 SatIP Server[15812]: 5 Frontend Name: Philips TDA10023 DVB-C
Jan 27 00:29:35 yavdr2 SatIP Server[15812]: 6 Frontend Name: Montage RS6000(DVBSky)
Jan 27 00:29:35 yavdr2 SatIP Server[15812]: 7 Frontend Name: Montage RS6000(DVBSky)
Jan 27 00:29:35 yavdr2 SatIP Server[15812]: 8 Frontend Name: DRXK DVB-C DVB-T
Jan 27 00:29:35 yavdr2 SatIP Server[15812]: 9 eth0: 192.168.178.133 [00:40:ca:a2:c7:9d]
Jan 27 00:29:35 yavdr2 SatIP Server[15812]: 10 Setting up RTSP server
Jan 27 00:29:35 yavdr2 SatIP Server[15812]: 11 Setting up HTTP server
Jan 27 00:29:35 yavdr2 SatIP Server[15812]: 12 Setting up 8 RTP servers
Jan 27 00:29:35 yavdr2 SatIP Server[15812]: 13 Setting up 8 RTCP servers
Jan 27 00:29:35 yavdr2 SatIP Server[15812]: 14 Setting up SSDP server
Jan 27 00:30:11 yavdr2 SatIP Server[15812]: 15 RTSP Connection from 192.168.178.130 Port 62795
Jan 27 00:30:11 yavdr2 SatIP Server[15812]: 16 Frontend: 0, With SYS_DVBC_ANNEX_A Attaching dynamically to client 192.168.178.130
Jan 27 00:30:11 yavdr2 SatIP Server[15812]: 17 Setup Message
Jan 27 00:30:11 yavdr2 SatIP Server[15812]: 18 Modify Transport Parameters
Jan 27 00:30:11 yavdr2 SatIP Server[15812]: 19 Frontend: 0, Opened FE fd: 24.
Jan 27 00:30:12 yavdr2 SatIP Server[15812]: 20 Frontend: 0, Waiting on lock.
Jan 27 00:30:12 yavdr2 SatIP Server[15812]: 21 Frontend: 0, Not locked yet (FE status 0)...
Jan 27 00:30:12 yavdr2 SatIP Server[15812]: 22 Frontend: 0, Tuned and locked (FE status 31).
Jan 27 00:30:12 yavdr2 SatIP Server[15812]: 23 Frontend: 0, Opened DVR fd: 25.
Jan 27 00:30:12 yavdr2 SatIP Server[15812]: 24 Frontend: 0, Start streaming to 192.168.178.130 (54948 - 54949)
Jan 27 00:30:12 yavdr2 SatIP Server[15812]: 25 Frontend: 0, Start RTCP stream to 192.168.178.130 (54948 - 54949)
Jan 27 00:30:12 yavdr2 SatIP Server[15812]: 26 Frontend: 0, Open Frontend Monitor for 192.168.178.130 (54948 - 54949)
Jan 27 00:30:12 yavdr2 SatIP Server[15812]: 27 Setup Message
Jan 27 00:30:12 yavdr2 SatIP Server[15812]: 28 Modify Transport Parameters
Jan 27 00:30:12 yavdr2 SatIP Server[15812]: 29 Setup Message
Jan 27 00:30:12 yavdr2 SatIP Server[15812]: 30 Modify Transport Parameters
Jan 27 00:30:12 yavdr2 SatIP Server[15812]: 31 Setup Message
Jan 27 00:30:12 yavdr2 SatIP Server[15812]: 32 Modify Transport Parameters
Jan 27 00:30:12 yavdr2 SatIP Server[15812]: 33 Play Message
Jan 27 00:31:52 yavdr2 SatIP Server[15812]: 34 Frontend: 0, Teardown Message, gracefull: 1
Jan 27 00:31:52 yavdr2 SatIP Server[15812]: 35 RTSP Connection from 192.168.178.130 Port 62796
Jan 27 00:31:52 yavdr2 kernel: [194905.142090] thread_rtsp[15813]: segfault at 30 ip 0000000000405ae6 sp 00007f4861ee2bd0 error 6 in satpi[400000+e000]

9000h, 2015-01-26:
with make debug
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 0 [satpi.c:235] --- starting ---
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 1 [tune.c:383] Not the preferred DVB API version, for correct function it should be 5.5 or higher
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 2 [tune.c:384] current DVB_API_VERSION: 5.4
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 3 [tune.c:386] Detecting frontends in: /dev/dvb
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 4 [tune.c:396] Frontends found: 4
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 5 [tune.c:461] Open FE fd: 4 readonly 1
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 6 [tune.c:117] Frontend Name: Philips TDA10023 DVB-C
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 7 [tune.c:158] Frontend Type: Cable (Annex A)
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 8 [tune.c:172] Frontend Type Unknown: 18
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 9 [tune.c:177] Frontend Freq: 47000000 Hz to 862000000 Hz
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 10 [tune.c:178] Frontend srat: 451875 symbols/s to 7230000 symbols/s
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 11 [tune.c:461] Open FE fd: 4 readonly 1
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 12 [tune.c:117] Frontend Name: Montage RS6000(DVBSky)
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 13 [tune.c:142] Frontend Type: Satellite (DVB-S)
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 14 [tune.c:146] Frontend Type: Satellite (DVB-S2)
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 15 [tune.c:177] Frontend Freq: 950000 Hz to 2150000 Hz
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 16 [tune.c:178] Frontend srat: 1000000 symbols/s to 45000000 symbols/s
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 17 [tune.c:461] Open FE fd: 4 readonly 1
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 18 [tune.c:117] Frontend Name: Montage RS6000(DVBSky)
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 19 [tune.c:142] Frontend Type: Satellite (DVB-S)
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 20 [tune.c:146] Frontend Type: Satellite (DVB-S2)
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 21 [tune.c:177] Frontend Freq: 950000 Hz to 2150000 Hz
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 22 [tune.c:178] Frontend srat: 1000000 symbols/s to 45000000 symbols/s
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 23 [tune.c:461] Open FE fd: 4 readonly 1
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 24 [tune.c:117] Frontend Name: DRXK DVB-C DVB-T
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 25 [tune.c:158] Frontend Type: Cable (Annex A)
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 26 [tune.c:172] Frontend Type Unknown: 18
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 27 [tune.c:150] Frontend Type: Terrestrial (DVB-T)
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 28 [tune.c:177] Frontend Freq: 47000000 Hz to 865000000 Hz
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 29 [tune.c:178] Frontend srat: 870000 symbols/s to 11700000 symbols/s
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 30 [connection.c:261] eth0: 192.168.178.133 [00:40:ca:a2:c7:9d]
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 31 [rtsp.c:895] Setting up RTSP server
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 32 [http.c:445] Setting up HTTP server
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 33 [rtp.c:343] Setting up 8 RTP servers
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 34 [rtcp.c:338] Setting up 8 RTCP servers
Jan 27 00:39:38 yavdr2 SatIP Server[15954]: 35 [ssdp.c:89] Setting up SSDP server
Jan 27 00:39:54 yavdr2 SatIP Server[15954]: 36 [connection.c:190] RTSP Connection from 192.168.178.130 Port 62830
Jan 27 00:39:54 yavdr2 SatIP Server[15954]: 37 [rtsp.c:791] OPTIONS rtsp://192.168.178.133/?freq=338.000&msys=dvbc&sr=6900&mtype=256qam&pids=all RTSP/1.0
Jan 27 00:39:54 yavdr2 SatIP Server[15954]: 38 CSeq: 2
Jan 27 00:39:54 yavdr2 SatIP Server[15954]: 39 User-Agent: LibVLC/2.2.0-rc1 (LIVE555 Streaming Media v2014.01.21)
Jan 27 00:39:54 yavdr2 SatIP Server[15954]: 40 --- LINE END ---
Jan 27 00:39:54 yavdr2 SatIP Server[15954]: 41 [rtsp.c:386] Frontend: 0, With SYS_DVBC_ANNEX_A Attaching dynamically to client 192.168.178.130
Jan 27 00:39:54 yavdr2 SatIP Server[15954]: 42 [rtsp.c:791] DESCRIBE rtsp://192.168.178.133/?freq=338.000&msys=dvbc&sr=6900&mtype=256qam&pids=all RTSP/1.0
Jan 27 00:39:54 yavdr2 SatIP Server[15954]: 43 CSeq: 3
Jan 27 00:39:54 yavdr2 SatIP Server[15954]: 44 User-Agent: LibVLC/2.2.0-rc1 (LIVE555 Streaming Media v2014.01.21)
Jan 27 00:39:54 yavdr2 SatIP Server[15954]: 45 Accept: application/sdp
Jan 27 00:39:54 yavdr2 SatIP Server[15954]: 46 --- LINE END ---
Jan 27 00:39:54 yavdr2 SatIP Server[15954]: 47 [rtsp.c:595] RTSP/1.0 200 OK
Jan 27 00:39:54 yavdr2 SatIP Server[15954]: 48 CSeq: 3
Jan 27 00:39:54 yavdr2 SatIP Server[15954]: 49 Content-Type: application/sdp
Jan 27 00:39:54 yavdr2 SatIP Server[15954]: 50 Content-Base: rtsp://192.168.178.133/
Jan 27 00:39:54 yavdr2 SatIP Server[15954]: 51 Content-Length: 498
Jan 27 00:39:54 yavdr2 SatIP Server[15954]: 52 --- LINE END ---
Jan 27 00:39:54 yavdr2 SatIP Server[15954]: 53 v=0
Jan 27 00:39:54 yavdr2 SatIP Server[15954]: 54 o=- 5678901234 7890123456 IN IP4 192.168.178.133
Jan 27 00:39:54 yavdr2 SatIP Server[15954]: 55 s=SatIPServer:1 4
Jan 27 00:39:54 yavdr2 SatIP Server[15954]: 56 t=0 0
Jan 27 00:39:54 yavdr2 SatIP Server[15954]: 57 m=video 0 RTP/AVP 33
Jan 27 00:39:54 yavdr2 SatIP Server[15954]: 58 c=IN IP4 0.0.0.0
Jan 27 00:39:54 yavdr2 SatIP Server[15954]: 59 a=control:stream=0
[... cut ...]
    msg=0x7f5ed8001eb0 "OPTIONS rtsp://192.168.178.133/?freq=338.000&msys=dvbc&sr=6900&mtype=256qam&pids=all RTSP/1.0\r\nCSeq: 2\r\nUser-Agent: LibVLC/2.2.0-rc1 (LIVE555 Streaming Media v2014.01.21)\r\n\r\n", client=0x6188e0, fe=0x618160) at rtsp.c:384
#1  0x0000000000407e32 in thread_work_rtsp (arg=0x618160) at rtsp.c:801
#2  0x00007f5ef0583e9a in start_thread () from /lib/x86_64-linux-gnu/libpthread.so.0
#3  0x00007f5ef00a92ed in clone () from /lib/x86_64-linux-gnu/libc.so.6
#4  0x0000000000000000 in ?? ()
(gdb)

Barracuda09, 2015-01-29:
Hi 9000h,

The crash was not related to pids=all, but still not implemente

Thanks for testing,

Marc

closed 2015-01-30 by 9000h