Barracuda09/SATPI #11: SATPI crashes - no picture with EyeTV on Mac OS X

pbriesch, 2015-01-27:
Hi,

SATPI crashes - using EyeTV on Mac OS X as SAT>IP client. I compiled with *debug* and got following output until SATPI crashed. I hope this is of use for you.

Paul

```
0  Tue Jan 27 20:27:36.188161821 2015  7  a=control:stream=0
1  Tue Jan 27 20:27:36.188161821 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,0,0,12187000,0,SYS_DVBS,QPSK,2,0.35,27500000,FEC_3_4;pids=
2  Tue Jan 27 20:27:36.188161821 2015  7  a=inactive
3  Tue Jan 27 20:27:36.188161821 2015  7  m=video 0 RTP/AVP 33
4  Tue Jan 27 20:27:36.188161821 2015  7  c=IN IP4 0.0.0.0
5  Tue Jan 27 20:27:36.188161821 2015  7  a=control:stream=1
6  Tue Jan 27 20:27:36.188161821 2015  7  a=fmtp:33 
7  Tue Jan 27 20:27:36.188161821 2015  7  a=inactive
8  Tue Jan 27 20:27:36.188161821 2015  7  m=video 0 RTP/AVP 33
9  Tue Jan 27 20:27:36.188161821 2015  7  c=IN IP4 0.0.0.0
10  Tue Jan 27 20:27:36.188161821 2015  7  a=control:stream=2
11  Tue Jan 27 20:27:36.188161821 2015  7  a=fmtp:33 
12  Tue Jan 27 20:27:36.188161821 2015  7  a=inactive
13  Tue Jan 27 20:27:36.188161821 2015  7  m=video 0 RTP/AVP 33
14  Tue Jan 27 20:27:36.188161821 2015  7  c=IN IP4 0.0.0.0
15  Tue Jan 27 20:27:36.188161821 2015  7  a=control:stream=3
16  Tue Jan 27 20:27:36.188161821 2015  7  a=fmtp:33 
17  Tue Jan 27 20:27:36.188161821 2015  7  a=inactive
18  Tue Jan 27 20:27:37.234755474 2015  3  [rtsp.c:853] RTSP Connection closed by Client: 192.168.1.107 ??
19  Tue Jan 27 20:27:37.234838923 2015  6  [rtsp.c:706] Frontend: x, Teardown Message, gracefull: 0
20  Tue Jan 27 20:27:37.234857938 2015  3  [rtsp.c:662] Frontend: x, Not attached anymore to client 192.168.1.107
21  Tue Jan 27 20:27:37.236307436 2015  6  [connection.c:190] RTSP Connection from 192.168.1.107 Port 54936
22  Tue Jan 27 20:27:37.236622176 2015  7  [rtsp.c:791] DESCRIBE rtsp://192.168.1.161/ RTSP/1.0
23  Tue Jan 27 20:27:37.236622176 2015  7  Connection:Close
24  Tue Jan 27 20:27:37.236622176 2015  7  CSeq:1373
25  Tue Jan 27 20:27:37.236622176 2015  7  Accept:application/sdp
26  Tue Jan 27 20:27:37.236622176 2015  7  --- LINE END ---
27  Tue Jan 27 20:27:37.236858396 2015  7  [rtsp.c:595] RTSP/1.0 200 OK
28  Tue Jan 27 20:27:37.236858396 2015  7  CSeq: 0
29  Tue Jan 27 20:27:37.236858396 2015  7  Content-Type: application/sdp
30  Tue Jan 27 20:27:37.236858396 2015  7  Content-Base: rtsp://192.168.1.161/
31  Tue Jan 27 20:27:37.236858396 2015  7  Content-Length: 497
32  Tue Jan 27 20:27:37.236858396 2015  7  --- LINE END ---
33  Tue Jan 27 20:27:37.236858396 2015  7  v=0
34  Tue Jan 27 20:27:37.236858396 2015  7  o=- 5678901234 7890123456 IN IP4 192.168.1.161
35  Tue Jan 27 20:27:37.236858396 2015  7  s=SatIPServer:1 4
36  Tue Jan 27 20:27:37.236858396 2015  7  t=0 0
37  Tue Jan 27 20:27:37.236858396 2015  7  m=video 0 RTP/AVP 33
38  Tue Jan 27 20:27:37.236858396 2015  7  c=IN IP4 0.0.0.0
39  Tue Jan 27 20:27:37.236858396 2015  7  a=control:stream=0
40  Tue Jan 27 20:27:37.236858396 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,0,0,12187000,0,SYS_DVBS,QPSK,2,0.35,27500000,FEC_3_4;pids=
41  Tue Jan 27 20:27:37.236858396 2015  7  a=inactive
42  Tue Jan 27 20:27:37.236858396 2015  7  m=video 0 RTP/AVP 33
43  Tue Jan 27 20:27:37.236858396 2015  7  c=IN IP4 0.0.0.0
44  Tue Jan 27 20:27:37.236858396 2015  7  a=control:stream=1
45  Tue Jan 27 20:27:37.236858396 2015  7  a=fmtp:33 
46  Tue Jan 27 20:27:37.236858396 2015  7  a=inactive
47  Tue Jan 27 20:27:37.236858396 2015  7  m=video 0 RTP/AVP 33
48  Tue Jan 27 20:27:37.236858396 2015  7  c=IN IP4 0.0.0.0
49  Tue Jan 27 20:27:37.236858396 2015  7  a=control:stream=2
50  Tue Jan 27 20:27:37.236858396 2015  7  a=fmtp:33 
51  Tue Jan 27 20:27:37.236858396 2015  7  a=inactive
52  Tue Jan 27 20:27:37.236858396 2015  7  m=video 0 RTP/AVP 33
53  Tue Jan 27 20:27:37.236858396 2015  7  c=IN IP4 0.0.0.0
54  Tue Jan 27 20:27:37.236858396 2015  7  a=control:stream=3
55  Tue Jan 27 20:27:37.236858396 2015  7  a=fmtp:33 
56  Tue Jan 27 20:27:37.236858396 2015  7  a=inactive
57  Tue Jan 27 20:27:41.264572740 2015  3  [rtsp.c:853] RTSP Connection closed by Client: 192.168.1.107 ??
58  Tue Jan 27 20:27:41.264663642 2015  6  [rtsp.c:706] Frontend: x, Teardown Message, gracefull: 0
59  Tue Jan 27 20:27:41.264683155 2015  3  [rtsp.c:662] Frontend: x, Not attached anymore to client 192.168.1.107
60  Tue Jan 27 20:27:41.266735658 2015  6  [connection.c:190] RTSP Connection from 192.168.1.107 Port 54937
61  Tue Jan 27 20:27:41.266767391 2015  7  [rtsp.c:791] DESCRIBE rtsp://192.168.1.161/ RTSP/1.0
62  Tue Jan 27 20:27:41.266767391 2015  7  Connection:Close
[... 263 lines cut ...]
326  Tue Jan 27 20:27:56.509966891 2015  7  t=0 0
327  Tue Jan 27 20:27:56.509966891 2015  7  m=video 0 RTP/AVP 33
328  Tue Jan 27 20:27:56.509966891 2015  7  c=IN IP4 0.0.0.0
329  Tue Jan 27 20:27:56.509966891 2015  7  a=control:stream=0
330  Tue Jan 27 20:27:56.509966891 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,0,0,12187000,0,SYS_DVBS,QPSK,2,0.35,27500000,FEC_3_4;pids=
331  Tue Jan 27 20:27:56.509966891 2015  7  a=inactive
332  Tue Jan 27 20:27:56.509966891 2015  7  m=video 0 RTP/AVP 33
333  Tue Jan 27 20:27:56.509966891 2015  7  c=IN IP4 0.0.0.0
334  Tue Jan 27 20:27:56.509966891 2015  7  a=control:stream=1
335  Tue Jan 27 20:27:56.509966891 2015  7  a=fmtp:33 
336  Tue Jan 27 20:27:56.509966891 2015  7  a=inactive
337  Tue Jan 27 20:27:56.509966891 2015  7  m=video 0 RTP/AVP 33
338  Tue Jan 27 20:27:56.509966891 2015  7  c=IN IP4 0.0.0.0
339  Tue Jan 27 20:27:56.509966891 2015  7  a=control:stream=2
340  Tue Jan 27 20:27:56.509966891 2015  7  a=fmtp:33 
341  Tue Jan 27 20:27:56.509966891 2015  7  a=inactive
342  Tue Jan 27 20:27:56.509966891 2015  7  m=video 0 RTP/AVP 33
343  Tue Jan 27 20:27:56.509966891 2015  7  c=IN IP4 0.0.0.0
344  Tue Jan 27 20:27:56.509966891 2015  7  a=control:stream=3
345  Tue Jan 27 20:27:56.509966891 2015  7  a=fmtp:33 
346  Tue Jan 27 20:27:56.509966891 2015  7  a=inactive
347  Tue Jan 27 20:27:57.515379326 2015  3  [rtsp.c:853] RTSP Connection closed by Client: 192.168.1.107 ??
348  Tue Jan 27 20:27:57.515455636 2015  6  [rtsp.c:706] Frontend: x, Teardown Message, gracefull: 0
349  Tue Jan 27 20:27:57.515492573 2015  3  [rtsp.c:662] Frontend: x, Not attached anymore to client 192.168.1.107
```

Barracuda09, 2015-01-27:
Hi,

Thanks, I see also that you have no Lock on the signal (see around line 100) and no pids specified.

The crash is not so nice.. I'm trying to reproduce it here.

Marc

pbriesch, 2015-01-27:
Marc,

just found this in the output of *dmesg*

    [  519.457421] thread_rtsp[1754]: segfault at 30 ip 0000000000406962 sp 00007f1e712f9dd0 error 6 in satpi[400000+13000]

Maybe it tells you something.

Paul

Barracuda09, 2015-01-27:
Hi,
Yes I have seen this crash... but could not reproduce it (was not in gdb)

I have changed something I think is also related to this issue.
I you have time you can check it out.

Thanks
Marc

pbriesch, 2015-01-27:
Marc,
it did not crash this time. However, EyeTV did not show a picture, only the messages that somebody else is using the channel. First, there is no other client using SATPI. Second, I have two DVB-cards running.
FYI, the DVB-cards are connected to a multi switch (ASTRA and Hotbird).
Here is some logging:
````
0  Tue Jan 27 23:05:14.560903478 2015  7  --- LINE END ---
1  Tue Jan 27 23:05:14.561071506 2015  7  [rtsp.c:595] RTSP/1.0 200 OK
2  Tue Jan 27 23:05:14.561071506 2015  7  CSeq: 1047
3  Tue Jan 27 23:05:14.561071506 2015  7  Content-Type: application/sdp
4  Tue Jan 27 23:05:14.561071506 2015  7  Content-Base: rtsp://192.168.1.161/
5  Tue Jan 27 23:05:14.561071506 2015  7  Content-Length: 411
6  Tue Jan 27 23:05:14.561071506 2015  7  --- LINE END ---
7  Tue Jan 27 23:05:14.561071506 2015  7  v=0
8  Tue Jan 27 23:05:14.561071506 2015  7  o=- 5678901234 7890123456 IN IP4 192.168.1.161
9  Tue Jan 27 23:05:14.561071506 2015  7  s=SatIPServer:1 2
10  Tue Jan 27 23:05:14.561071506 2015  7  t=0 0
11  Tue Jan 27 23:05:14.561071506 2015  7  m=video 0 RTP/AVP 33
12  Tue Jan 27 23:05:14.561071506 2015  7  c=IN IP4 0.0.0.0
13  Tue Jan 27 23:05:14.561071506 2015  7  a=control:stream=0
14  Tue Jan 27 23:05:14.561071506 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,0,0,10862000,0,SYS_DVBS,QPSK,2,0.35,22000000,FEC_7_8;pids=
15  Tue Jan 27 23:05:14.561071506 2015  7  a=inactive
16  Tue Jan 27 23:05:14.561071506 2015  7  m=video 0 RTP/AVP 33
17  Tue Jan 27 23:05:14.561071506 2015  7  c=IN IP4 0.0.0.0
18  Tue Jan 27 23:05:14.561071506 2015  7  a=control:stream=1
19  Tue Jan 27 23:05:14.561071506 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=2,0,0,0,10862000,0,SYS_DVBS,QPSK,2,0.35,22000000,FEC_7_8;pids=
20  Tue Jan 27 23:05:14.561071506 2015  7  a=inactive
21  Tue Jan 27 23:05:15.759090218 2015  3  [rtsp.c:853] RTSP Connection closed by Client: 192.168.1.107 ??
22  Tue Jan 27 23:05:15.759174041 2015  6  [rtsp.c:706] Frontend: x, Teardown Message, gracefull: 0
23  Tue Jan 27 23:05:15.759193933 2015  3  [rtsp.c:662] Frontend: x, Not attached anymore to client 192.168.1.107
24  Tue Jan 27 23:05:15.761052728 2015  6  [connection.c:190] RTSP Connection from 192.168.1.107 Port 60188
25  Tue Jan 27 23:05:15.761085566 2015  7  [rtsp.c:791] SETUP rtsp://192.168.1.161:554/?src=1&freq=12187&pol=h&ro=0.35&msys=dvbs&mtype=qpsk&plts=off&sr=27500&fec=34 RTSP/1.0
26  Tue Jan 27 23:05:15.761085566 2015  7  Connection:Close
27  Tue Jan 27 23:05:15.761085566 2015  7  CSeq:1155
28  Tue Jan 27 23:05:15.761085566 2015  7  Transport:RTP/AVP;unicast;client_port=5000-5001
29  Tue Jan 27 23:05:15.761085566 2015  7  --- LINE END ---
30  Tue Jan 27 23:05:15.761151764 2015  6  [rtsp.c:386] Frontend: 0, With SYS_DVBS Attaching dynamically to client 192.168.1.107
31  Tue Jan 27 23:05:15.761192860 2015  6  [rtsp.c:425] Setup Message
32  Tue Jan 27 23:05:15.761206340 2015  7  [tune.c:349] Start tuning
33  Tue Jan 27 23:05:15.761216758 2015  7  [tune.c:273] Sending DiSEqC
34  Tue Jan 27 23:05:15.970869734 2015  6  [tune.c:528] Frontend: 0, Waiting on lock.
35  Tue Jan 27 23:05:16.261460296 2015  6  [tune.c:542] Frontend: 0, Not locked yet (FE status 0)...
36  Tue Jan 27 23:05:16.286519581 2015  6  [tune.c:542] Frontend: 0, Not locked yet (FE status 0)...
37  Tue Jan 27 23:05:16.311581037 2015  6  [tune.c:542] Frontend: 0, Not locked yet (FE status 0)...
38  Tue Jan 27 23:05:16.336653517 2015  6  [tune.c:542] Frontend: 0, Not locked yet (FE status 0)...
39  Tue Jan 27 23:05:16.376853710 2015  3  [rtsp.c:853] RTSP Connection closed by Client: 192.168.1.107 ??
40  Tue Jan 27 23:05:16.376897822 2015  6  [rtsp.c:704] Frontend: 0, Teardown Message, gracefull: 0
41  Tue Jan 27 23:05:16.380915993 2015  6  [connection.c:190] RTSP Connection from 192.168.1.107 Port 60190
42  Tue Jan 27 23:05:16.380946068 2015  7  [rtsp.c:791] TEARDOWN rtsp://192.168.1.161:554/stream=0 RTSP/1.0
43  Tue Jan 27 23:05:16.380946068 2015  7  Connection:Close
44  Tue Jan 27 23:05:16.380946068 2015  7  CSeq:1157
45  Tue Jan 27 23:05:16.380946068 2015  7  Session:00000002
46  Tue Jan 27 23:05:16.380946068 2015  7  --- LINE END ---
47  Tue Jan 27 23:05:16.381005828 2015  6  [rtsp.c:706] Frontend: x, Teardown Message, gracefull: 1
48  Tue Jan 27 23:05:16.381016849 2015  3  [rtsp.c:662] Frontend: x, Not attached anymore to client 192.168.1.107
49  Tue Jan 27 23:05:16.502136875 2015  6  [connection.c:190] RTSP Connection from 192.168.1.107 Port 60191
50  Tue Jan 27 23:05:16.502168200 2015  7  [rtsp.c:791] DESCRIBE rtsp://192.168.1.161/ RTSP/1.0
51  Tue Jan 27 23:05:16.502168200 2015  7  Connection:Close
52  Tue Jan 27 23:05:16.502168200 2015  7  CSeq:1158
53  Tue Jan 27 23:05:16.502168200 2015  7  Accept:application/sdp
54  Tue Jan 27 23:05:16.502168200 2015  7  --- LINE END ---
55  Tue Jan 27 23:05:16.502321542 2015  7  [rtsp.c:595] RTSP/1.0 200 OK
56  Tue Jan 27 23:05:16.502321542 2015  7  CSeq: 1158
57  Tue Jan 27 23:05:16.502321542 2015  7  Content-Type: application/sdp
58  Tue Jan 27 23:05:16.502321542 2015  7  Content-Base: rtsp://192.168.1.161/
59  Tue Jan 27 23:05:16.502321542 2015  7  Content-Length: 411
60  Tue Jan 27 23:05:16.502321542 2015  7  --- LINE END ---
61  Tue Jan 27 23:05:16.502321542 2015  7  v=0
62  Tue Jan 27 23:05:16.502321542 2015  7  o=- 5678901234 7890123456 IN IP4 192.168.1.161
63  Tue Jan 27 23:05:16.502321542 2015  7  s=SatIPServer:1 2
64  Tue Jan 27 23:05:16.502321542 2015  7  t=0 0
[... 260 lines cut ...]
[... cut ...]
344  Tue Jan 27 23:05:38.434458426 2015  7  [tune.c:273] Sending DiSEqC
345  Tue Jan 27 23:05:38.722870254 2015  6  [tune.c:528] Frontend: 0, Waiting on lock.
346  Tue Jan 27 23:05:39.013460080 2015  6  [tune.c:542] Frontend: 0, Not locked yet (FE status 0)...
347  Tue Jan 27 23:05:39.038525557 2015  6  [tune.c:542] Frontend: 0, Not locked yet (FE status 0)...
348  Tue Jan 27 23:05:39.063587885 2015  6  [tune.c:542] Frontend: 0, Not locked yet (FE status 0)...
349  Tue Jan 27 23:05:39.088643327 2015  6  [tune.c:542] Frontend: 0, Not locked yet (FE status 0)...

Barracuda09, 2015-01-27:
Thanks again, I see that the CSeq is now correct. But still no Lock.. I have here also a 4 way switch.
But EyeTV closes the RTSP connection (as indicated). There is I also a think the problem.

What channel is this?

pbriesch, 2015-01-27:
I have switched to several channels. If you want me to go to a special channel, please let me know. I'm going to check now if the DVB cards are working OK on my side. Just a side note: the cards respectively the driver do not support the indication of the signal strength.

Barracuda09, 2015-01-27:
Ok, but probably the status well

Barracuda09, 2015-01-27:
Hi,
This is what I use for testing it is on astra 23.5
rtsp://192.168.0.109/?src=1&&freq=11914&sr=29900&msys=dvbs&pol=h&fec=23&pids=0,1,16,17,18,40,57,88,516,2310,3002,3003,8190

pbriesch, 2015-01-27:
Sorry, I only have Astra 19.2 E and 13.0 E Hotbird.

pbriesch, 2015-01-27:
I checked my DVB adapters and they work. Below is a logging not from EyeTV but from the SAT-IP client based on vtuner that runs with VDR. The client doesn't disconnect, however, there is no picture.
I tried this client with minisat running on my pc and I got a picture. However, with minisat EyeTV does not get a picture either. 

````
0  Wed Jan 28 00:22:53.790540552 2015  6  [satpi.c:235] --- starting ---
1  Wed Jan 28 00:22:53.790626061 2015  6  [tune.c:386] Detecting frontends in: /dev/dvb
2  Wed Jan 28 00:22:53.790739274 2015  6  [tune.c:396] Frontends found: 2
3  Wed Jan 28 00:22:53.790755914 2015  7  [tune.c:461] Open FE fd: 4 readonly 1
4  Wed Jan 28 00:22:53.790784992 2015  6  [tune.c:117] Frontend Name: STB0899 Multistandard
5  Wed Jan 28 00:22:53.790808101 2015  7  [tune.c:142] Frontend Type: Satellite (DVB-S)
6  Wed Jan 28 00:22:53.790827483 2015  7  [tune.c:146] Frontend Type: Satellite (DVB-S2)
7  Wed Jan 28 00:22:53.790846880 2015  7  [tune.c:172] Frontend Type Unknown: 4
8  Wed Jan 28 00:22:53.790865468 2015  7  [tune.c:177] Frontend Freq: 950000 Hz to 2150000 Hz
9  Wed Jan 28 00:22:53.790884003 2015  7  [tune.c:178] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
10  Wed Jan 28 00:22:53.790908670 2015  7  [tune.c:461] Open FE fd: 4 readonly 1
11  Wed Jan 28 00:22:53.790919243 2015  6  [tune.c:117] Frontend Name: STB0899 Multistandard
12  Wed Jan 28 00:22:53.790926351 2015  7  [tune.c:142] Frontend Type: Satellite (DVB-S)
13  Wed Jan 28 00:22:53.790932134 2015  7  [tune.c:146] Frontend Type: Satellite (DVB-S2)
14  Wed Jan 28 00:22:53.790937655 2015  7  [tune.c:172] Frontend Type Unknown: 4
15  Wed Jan 28 00:22:53.790954362 2015  7  [tune.c:177] Frontend Freq: 950000 Hz to 2150000 Hz
16  Wed Jan 28 00:22:53.790973872 2015  7  [tune.c:178] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
17  Wed Jan 28 00:22:53.791338694 2015  6  [connection.c:261] eth0: 192.168.1.161 [00:1d:7d:a9:eb:f3]
18  Wed Jan 28 00:22:53.791348956 2015  6  [rtsp.c:895] Setting up RTSP server
19  Wed Jan 28 00:22:53.791404596 2015  6  [http.c:445] Setting up HTTP server
20  Wed Jan 28 00:22:53.791438865 2015  6  [rtp.c:343] Setting up 8 RTP servers
21  Wed Jan 28 00:22:53.791644602 2015  6  [rtcp.c:338] Setting up 8 RTCP servers
22  Wed Jan 28 00:22:53.791853302 2015  6  [ssdp.c:89] Setting up SSDP server
23  Wed Jan 28 00:22:54.201096778 2015  6  [connection.c:190] RTSP Connection from 192.168.1.150 Port 47558
24  Wed Jan 28 00:22:54.201454072 2015  7  [rtsp.c:791] OPTIONS rtsp://192.168.1.161:554/ RTSP/1.0
25  Wed Jan 28 00:22:54.201454072 2015  7  CSeq: 1
26  Wed Jan 28 00:22:54.201454072 2015  7  --- LINE END ---
27  Wed Jan 28 00:22:54.202583164 2015  7  [rtsp.c:791] SETUP rtsp://192.168.1.161/?src=1&freq=11837.0&pol=h&msys=dvbs&sr=27500&fec=34&pids=101,102,103,106,105,18,20,0,17,16,500,100 RTSP/1.0
28  Wed Jan 28 00:22:54.202583164 2015  7  CSeq: 2
29  Wed Jan 28 00:22:54.202583164 2015  7  Transport: RTP/AVP;unicast;client_port=46690-46691
30  Wed Jan 28 00:22:54.202583164 2015  7  --- LINE END ---
31  Wed Jan 28 00:22:54.202636377 2015  6  [rtsp.c:386] Frontend: 0, With SYS_DVBS Attaching dynamically to client 192.168.1.150
32  Wed Jan 28 00:22:54.202683018 2015  6  [rtsp.c:425] Setup Message
33  Wed Jan 28 00:22:54.202766354 2015  7  [tune.c:461] Open FE fd: 24 readonly 0
34  Wed Jan 28 00:22:54.202788950 2015  6  [tune.c:517] Frontend: 0, Opened FE fd: 24.
35  Wed Jan 28 00:22:54.202808773 2015  7  [tune.c:349] Start tuning
36  Wed Jan 28 00:22:54.202827998 2015  7  [tune.c:273] Sending DiSEqC
37  Wed Jan 28 00:22:54.579366165 2015  6  [tune.c:528] Frontend: 0, Waiting on lock.
38  Wed Jan 28 00:22:54.869946638 2015  6  [tune.c:542] Frontend: 0, Not locked yet (FE status 0)...
39  Wed Jan 28 00:22:54.895024529 2015  6  [tune.c:542] Frontend: 0, Not locked yet (FE status 0)...
40  Wed Jan 28 00:22:54.920098709 2015  6  [tune.c:542] Frontend: 0, Not locked yet (FE status 0)...
41  Wed Jan 28 00:22:54.945167376 2015  6  [tune.c:542] Frontend: 0, Not locked yet (FE status 0)...
42  Wed Jan 28 00:22:54.970381345 2015  6  [tune.c:560] Frontend: 0, Opened DVR fd: 26.
43  Wed Jan 28 00:22:54.970415176 2015  7  [tune.c:497] Frontend: 0, Set filter PID 0 - fd 27
44  Wed Jan 28 00:22:54.970441719 2015  7  [tune.c:497] Frontend: 0, Set filter PID 16 - fd 28
45  Wed Jan 28 00:22:54.970453892 2015  7  [tune.c:497] Frontend: 0, Set filter PID 17 - fd 29
46  Wed Jan 28 00:22:54.970466794 2015  7  [tune.c:497] Frontend: 0, Set filter PID 18 - fd 30
47  Wed Jan 28 00:22:54.970478622 2015  7  [tune.c:497] Frontend: 0, Set filter PID 20 - fd 31
48  Wed Jan 28 00:22:54.970492745 2015  7  [tune.c:497] Frontend: 0, Set filter PID 100 - fd 32
49  Wed Jan 28 00:22:54.970504363 2015  7  [tune.c:497] Frontend: 0, Set filter PID 101 - fd 33
50  Wed Jan 28 00:22:54.970516520 2015  7  [tune.c:497] Frontend: 0, Set filter PID 102 - fd 34
51  Wed Jan 28 00:22:54.970529899 2015  7  [tune.c:497] Frontend: 0, Set filter PID 103 - fd 35
52  Wed Jan 28 00:22:54.970541367 2015  7  [tune.c:497] Frontend: 0, Set filter PID 105 - fd 36
53  Wed Jan 28 00:22:54.970552966 2015  7  [tune.c:497] Frontend: 0, Set filter PID 106 - fd 37
54  Wed Jan 28 00:22:54.970566060 2015  7  [tune.c:497] Frontend: 0, Set filter PID 500 - fd 38
55  Wed Jan 28 00:22:54.970640239 2015  6  [rtp.c:98] Frontend: 0, Start streaming to 192.168.1.150 (46690 - 46691)
[... cut ...]
143  Wed Jan 28 00:24:22.741943003 2015  6  [tune.c:542] Frontend: 0, Not locked yet (FE status 0)...
144  Wed Jan 28 00:24:22.767020696 2015  6  [tune.c:542] Frontend: 0, Not locked yet (FE status 0)...
145  Wed Jan 28 00:24:22.792093168 2015  6  [tune.c:542] Frontend: 0, Not locked yet (FE status 0)...
146  Wed Jan 28 00:24:22.817165640 2015  6  [tune.c:542] Frontend: 0, Not locked yet (FE status 0)...
147  Wed Jan 28 00:24:22.842261835 2015  7  [tune.c:497] Frontend: 0, Set filter PID 101 - fd 41

pbriesch, 2015-01-27:
Since SATPI doesn't crash any more, I'm closing this issue.

closed 2015-01-27 by pbriesch