Barracuda09/SATPI #12: No picture with EyeTV on Mac OS X

pbriesch, 2015-01-27:
I separate this issue from the one I earlier opened https://github.com/Barracuda09/SATPI/issues/11#issuecomment-71744053 because the crashing has been fixed but there is still no picture with EyeTV.

Barracuda09, 2015-01-28:
Hi,
To be sure all your DVB cards have the same Satellite config. (Both Astra 19.2 E and 13.0 E Hotbird)
Because it is dynamically allocating a frontend if you need to have a specific frontend use fe=0 or fe=1

Thanks for your time and trying

Marc

Barracuda09, 2015-01-28:
Hi,

If you like you can try again.

Many Thanks,

Marc

pbriesch, 2015-01-28:
Hi Marc,

it does still not work. Here some log messages:

````
0  Wed Jan 28 21:45:51.716406772 2015  7  c=IN IP4 0.0.0.0
1  Wed Jan 28 21:45:51.716406772 2015  7  a=control:stream=1
2  Wed Jan 28 21:45:51.716406772 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=2,0,0,0,12188000,0,SYS_DVBS,QPSK,2,0.35,27500000,FEC_3_4;pids=
3  Wed Jan 28 21:45:51.716406772 2015  7  a=inactive
4  Wed Jan 28 21:45:56.816097928 2015  7  [rtsp.c:882] RTSP Connection closed as expected by Client: 192.168.1.107 ??
5  Wed Jan 28 21:45:56.817420759 2015  6  [connection.c:190] RTSP Connection from 192.168.1.107 Port 56011
6  Wed Jan 28 21:45:56.817735480 2015  7  [rtsp.c:815] DESCRIBE rtsp://192.168.1.161/ RTSP/1.0
7  Wed Jan 28 21:45:56.817735480 2015  7  Connection:Close
8  Wed Jan 28 21:45:56.817735480 2015  7  CSeq:1030
9  Wed Jan 28 21:45:56.817735480 2015  7  Accept:application/sdp
10  Wed Jan 28 21:45:56.817735480 2015  7  --- LINE END ---
11  Wed Jan 28 21:45:56.817802626 2015  6  [rtsp.c:829] RTSP Requested Connection closed by Client: 192.168.1.107
12  Wed Jan 28 21:45:56.817930393 2015  7  [rtsp.c:618] RTSP/1.0 200 OK
13  Wed Jan 28 21:45:56.817930393 2015  7  CSeq: 1030
14  Wed Jan 28 21:45:56.817930393 2015  7  Content-Type: application/sdp
15  Wed Jan 28 21:45:56.817930393 2015  7  Content-Base: rtsp://192.168.1.161/
16  Wed Jan 28 21:45:56.817930393 2015  7  Content-Length: 456
17  Wed Jan 28 21:45:56.817930393 2015  7  --- LINE END ---
18  Wed Jan 28 21:45:56.817930393 2015  7  v=0
19  Wed Jan 28 21:45:56.817930393 2015  7  o=- 5678901234 7890123456 IN IP4 192.168.1.161
20  Wed Jan 28 21:45:56.817930393 2015  7  s=SatIPServer:1 2
21  Wed Jan 28 21:45:56.817930393 2015  7  t=0 0
22  Wed Jan 28 21:45:56.817930393 2015  7  m=video 0 RTP/AVP 33
23  Wed Jan 28 21:45:56.817930393 2015  7  c=IN IP4 0.0.0.0
24  Wed Jan 28 21:45:56.817930393 2015  7  a=control:stream=0
25  Wed Jan 28 21:45:56.817930393 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,0,0,11836000,0,SYS_DVBS,QPSK,2,0.35,27500000,FEC_3_4;pids=0,16,17,18,19,20,100,101,102,104,105,106,8190
26  Wed Jan 28 21:45:56.817930393 2015  7  a=inactive
27  Wed Jan 28 21:45:56.817930393 2015  7  m=video 0 RTP/AVP 33
28  Wed Jan 28 21:45:56.817930393 2015  7  c=IN IP4 0.0.0.0
29  Wed Jan 28 21:45:56.817930393 2015  7  a=control:stream=1
30  Wed Jan 28 21:45:56.817930393 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=2,0,0,0,12188000,0,SYS_DVBS,QPSK,2,0.35,27500000,FEC_3_4;pids=
31  Wed Jan 28 21:45:56.817930393 2015  7  a=inactive
32  Wed Jan 28 21:45:58.657087200 2015  7  [rtsp.c:882] RTSP Connection closed as expected by Client: 192.168.1.107 ??
33  Wed Jan 28 21:45:58.658520944 2015  6  [connection.c:190] RTSP Connection from 192.168.1.107 Port 56012
34  Wed Jan 28 21:45:58.658556726 2015  7  [rtsp.c:815] SETUP rtsp://192.168.1.161:554/stream=0?src=1&freq=11836&pol=h&ro=0.35&msys=dvbs&mtype=qpsk&plts=off&sr=27500&fec=34 RTSP/1.0
35  Wed Jan 28 21:45:58.658556726 2015  7  Connection:Close
36  Wed Jan 28 21:45:58.658556726 2015  7  CSeq:1043
37  Wed Jan 28 21:45:58.658556726 2015  7  Transport:RTP/AVP;unicast;client_port=5000-5001
38  Wed Jan 28 21:45:58.658556726 2015  7  Session:00000002
39  Wed Jan 28 21:45:58.658556726 2015  7  --- LINE END ---
40  Wed Jan 28 21:45:58.658697721 2015  6  [rtsp.c:829] RTSP Requested Connection closed by Client: 192.168.1.107
41  Wed Jan 28 21:45:58.658725356 2015  6  [rtsp.c:448] Setup Message
42  Wed Jan 28 21:45:58.658732210 2015  6  [rtsp.c:457] Modify Transport Parameters
43  Wed Jan 28 21:45:58.658738460 2015  7  [tune.c:349] Frontend: 0, Start tuning process
44  Wed Jan 28 21:45:58.658744407 2015  7  [tune.c:273] Frontend: 0, Sending DiSEqC
45  Wed Jan 28 21:45:59.142878385 2015  6  [tune.c:528] Frontend: 0, Waiting on lock.
46  Wed Jan 28 21:45:59.433436616 2015  6  [tune.c:542] Frontend: 0, Not locked yet (FE status 0)...
47  Wed Jan 28 21:45:59.458513436 2015  6  [tune.c:542] Frontend: 0, Not locked yet (FE status 0)...
48  Wed Jan 28 21:45:59.483586093 2015  6  [tune.c:542] Frontend: 0, Not locked yet (FE status 0)...
49  Wed Jan 28 21:45:59.508661460 2015  6  [tune.c:542] Frontend: 0, Not locked yet (FE status 0)...
50  Wed Jan 28 21:45:59.549857734 2015  7  [rtsp.c:882] RTSP Connection closed as expected by Client: 192.168.1.107 ??
51  Wed Jan 28 21:45:59.551018911 2015  6  [connection.c:190] RTSP Connection from 192.168.1.107 Port 56013
52  Wed Jan 28 21:45:59.551047997 2015  7  [rtsp.c:815] PLAY rtsp://192.168.1.161:554/stream=0?pids=0,16,17,18,19,20,100,101,102,104,105,106,8190 RTSP/1.0
53  Wed Jan 28 21:45:59.551047997 2015  7  Connection:Close
54  Wed Jan 28 21:45:59.551047997 2015  7  CSeq:1044
55  Wed Jan 28 21:45:59.551047997 2015  7  Session:00000002
56  Wed Jan 28 21:45:59.551047997 2015  7  --- LINE END ---
57  Wed Jan 28 21:45:59.551105767 2015  6  [rtsp.c:829] RTSP Requested Connection closed by Client: 192.168.1.107
58  Wed Jan 28 21:45:59.551141882 2015  6  [rtsp.c:519] Play Message
59  Wed Jan 28 21:45:59.551154094 2015  7  [tune.c:349] Frontend: 0, Start tuning process
60  Wed Jan 28 21:45:59.551164594 2015  7  [tune.c:273] Frontend: 0, Sending DiSEqC
61  Wed Jan 28 21:45:59.762861162 2015  6  [tune.c:528] Frontend: 0, Waiting on lock.
62  Wed Jan 28 21:46:00.053436176 2015  6  [tune.c:542] Frontend: 0, Not locked yet (FE status 0)...
63  Wed Jan 28 21:46:00.078522731 2015  6  [tune.c:542] Frontend: 0, Not locked yet (FE status 0)...
[... cut ...]
344  Wed Jan 28 21:46:41.742805868 2015  7  --- LINE END ---
345  Wed Jan 28 21:46:41.742871024 2015  6  [rtsp.c:829] RTSP Requested Connection closed by Client: 192.168.1.107
346  Wed Jan 28 21:46:41.742909151 2015  6  [rtsp.c:519] Play Message
347  Wed Jan 28 21:46:41.742921367 2015  7  [tune.c:349] Frontend: 0, Start tuning process
348  Wed Jan 28 21:46:41.742932122 2015  7  [tune.c:273] Frontend: 0, Sending DiSEqC
349  Wed Jan 28 21:46:41.954860113 2015  6  [tune.c:528] Frontend: 0, Waiting on lock.

Barracuda09, 2015-01-28:
Mmmm.. very strange that there is no lock on the signal..just like the switch is not switched correct... what for switch do you have if I may ask.

pbriesch, 2015-01-28:
I have a KATHREIN EXR 908.
http://www.kathrein.de/en/sat/product/273439/

Barracuda09, 2015-01-28:
Hi,

Would you try something for me? just edit tune.c line 266 and change the 0xe0 to 0xe2

Thanks for trying

Marc

pbriesch, 2015-01-28:
I made this change. Furthermore, I played a bit with EyTV and switched channels rapidly. Furthermore, a second SAT>IP client, my VDR, connected as well. SATPI crashed.
dmsg:
````
thread_rtsp[1974]: segfault at 18 ip 00007fe87ca47324 sp 00007fe87c46edc0 error 4 in libpthread-2.19.so[7fe87ca3d000+19000]
````
log output:
````
0  Wed Jan 28 23:43:10.004839014 2015  7  t=0 0
1  Wed Jan 28 23:43:10.004839014 2015  7  m=video 0 RTP/AVP 33
2  Wed Jan 28 23:43:10.004839014 2015  7  c=IN IP4 0.0.0.0
3  Wed Jan 28 23:43:10.004839014 2015  7  a=control:stream=0
4  Wed Jan 28 23:43:10.004839014 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,0,0,11838700,0,SYS_DVBS,QAM_64,2,0.35,27500000,FEC_3_4;pids=0,16,17,18,20,101,102,103,105,106
5  Wed Jan 28 23:43:10.004839014 2015  7  a=inactive
6  Wed Jan 28 23:43:10.004839014 2015  7  m=video 0 RTP/AVP 33
7  Wed Jan 28 23:43:10.004839014 2015  7  c=IN IP4 0.0.0.0
8  Wed Jan 28 23:43:10.004839014 2015  7  a=control:stream=1
9  Wed Jan 28 23:43:10.004839014 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=2,0,0,0,11612000,0,SYS_DVBS,QPSK,2,0.35,22000000,FEC_5_6;pids=0,16,17,18,19,20,129,2064,2068,8190
10  Wed Jan 28 23:43:10.004839014 2015  7  a=inactive
11  Wed Jan 28 23:43:12.913851917 2015  7  [rtsp.c:882] RTSP Connection closed as expected by Client: 192.168.1.107 ??
12  Wed Jan 28 23:43:12.915167408 2015  6  [connection.c:190] RTSP Connection from 192.168.1.107 Port 62335
13  Wed Jan 28 23:43:12.915569335 2015  7  [rtsp.c:815] SETUP rtsp://192.168.1.161:554/stream=1?src=1&freq=11612&pol=h&ro=0.35&msys=dvbs&mtype=qpsk&plts=off&sr=22000&fec=56 RTSP/1.0
14  Wed Jan 28 23:43:12.915569335 2015  7  Connection:Close
15  Wed Jan 28 23:43:12.915569335 2015  7  CSeq:1030
16  Wed Jan 28 23:43:12.915569335 2015  7  Transport:RTP/AVP;unicast;client_port=5000-5001
17  Wed Jan 28 23:43:12.915569335 2015  7  Session:00000002
18  Wed Jan 28 23:43:12.915569335 2015  7  --- LINE END ---
19  Wed Jan 28 23:43:12.915646125 2015  6  [rtsp.c:829] RTSP Requested Connection closed by Client: 192.168.1.107
20  Wed Jan 28 23:43:12.915691316 2015  6  [rtsp.c:448] Setup Message
21  Wed Jan 28 23:43:12.915703390 2015  6  [rtsp.c:457] Modify Transport Parameters
22  Wed Jan 28 23:43:12.915714136 2015  7  [tune.c:349] Frontend: 1, Start tuning process
23  Wed Jan 28 23:43:12.915724746 2015  7  [tune.c:273] Frontend: 1, Sending DiSEqC
24  Wed Jan 28 23:43:13.531245460 2015  6  [tune.c:528] Frontend: 1, Waiting on lock.
25  Wed Jan 28 23:43:13.821821753 2015  6  [tune.c:542] Frontend: 1, Not locked yet (FE status 0)...
26  Wed Jan 28 23:43:13.846896997 2015  6  [tune.c:542] Frontend: 1, Not locked yet (FE status 0)...
27  Wed Jan 28 23:43:13.871966263 2015  6  [tune.c:542] Frontend: 1, Not locked yet (FE status 0)...
28  Wed Jan 28 23:43:13.897039626 2015  6  [tune.c:542] Frontend: 1, Not locked yet (FE status 0)...
29  Wed Jan 28 23:43:14.017139496 2015  7  [rtsp.c:882] RTSP Connection closed as expected by Client: 192.168.1.107 ??
30  Wed Jan 28 23:43:14.019446626 2015  6  [connection.c:190] RTSP Connection from 192.168.1.107 Port 62336
31  Wed Jan 28 23:43:14.019477218 2015  7  [rtsp.c:815] PLAY rtsp://192.168.1.161:554/stream=1?pids=0,16,17,18,19,20,129,2064,2068,8190 RTSP/1.0
32  Wed Jan 28 23:43:14.019477218 2015  7  Connection:Close
33  Wed Jan 28 23:43:14.019477218 2015  7  CSeq:1031
34  Wed Jan 28 23:43:14.019477218 2015  7  Session:00000002
35  Wed Jan 28 23:43:14.019477218 2015  7  --- LINE END ---
36  Wed Jan 28 23:43:14.019535970 2015  6  [rtsp.c:829] RTSP Requested Connection closed by Client: 192.168.1.107
37  Wed Jan 28 23:43:14.019571727 2015  6  [rtsp.c:519] Play Message
38  Wed Jan 28 23:43:14.019583543 2015  7  [tune.c:349] Frontend: 1, Start tuning process
39  Wed Jan 28 23:43:14.019593861 2015  7  [tune.c:273] Frontend: 1, Sending DiSEqC
40  Wed Jan 28 23:43:14.231265195 2015  6  [tune.c:528] Frontend: 1, Waiting on lock.
41  Wed Jan 28 23:43:14.521826915 2015  6  [tune.c:542] Frontend: 1, Not locked yet (FE status 0)...
42  Wed Jan 28 23:43:14.546906155 2015  6  [tune.c:542] Frontend: 1, Not locked yet (FE status 0)...
43  Wed Jan 28 23:43:14.571982012 2015  6  [tune.c:542] Frontend: 1, Not locked yet (FE status 0)...
44  Wed Jan 28 23:43:14.597052357 2015  6  [tune.c:542] Frontend: 1, Not locked yet (FE status 0)...
45  Wed Jan 28 23:43:14.634793918 2015  7  [rtsp.c:882] RTSP Connection closed as expected by Client: 192.168.1.107 ??
46  Wed Jan 28 23:43:14.636228278 2015  6  [connection.c:190] RTSP Connection from 192.168.1.107 Port 62337
47  Wed Jan 28 23:43:14.636635167 2015  7  [rtsp.c:815] SETUP rtsp://192.168.1.161:554/stream=1?src=1&freq=11836&pol=h&ro=0.35&msys=dvbs&mtype=qpsk&plts=off&sr=27500&fec=34 RTSP/1.0
48  Wed Jan 28 23:43:14.636635167 2015  7  Connection:Close
49  Wed Jan 28 23:43:14.636635167 2015  7  CSeq:1032
50  Wed Jan 28 23:43:14.636635167 2015  7  Transport:RTP/AVP;unicast;client_port=5000-5001
51  Wed Jan 28 23:43:14.636635167 2015  7  Session:00000002
52  Wed Jan 28 23:43:14.636635167 2015  7  --- LINE END ---
53  Wed Jan 28 23:43:14.636712410 2015  6  [rtsp.c:829] RTSP Requested Connection closed by Client: 192.168.1.107
54  Wed Jan 28 23:43:14.636757620 2015  6  [rtsp.c:448] Setup Message
55  Wed Jan 28 23:43:14.636773062 2015  6  [rtsp.c:457] Modify Transport Parameters
56  Wed Jan 28 23:43:14.636784275 2015  7  [tune.c:349] Frontend: 1, Start tuning process
57  Wed Jan 28 23:43:14.636794901 2015  7  [tune.c:273] Frontend: 1, Sending DiSEqC
[... cut ...]
344  Wed Jan 28 23:43:38.671761020 2015  7  --- LINE END ---
345  Wed Jan 28 23:43:38.671816214 2015  6  [rtsp.c:407] Frontend: 0, With SYS_DVBS Attaching dynamically to client 192.168.1.150
346  Wed Jan 28 23:43:38.671860765 2015  6  [rtsp.c:448] Setup Message
347  Wed Jan 28 23:43:38.671876697 2015  7  [tune.c:349] Frontend: 0, Start tuning process
348  Wed Jan 28 23:43:38.671888139 2015  7  [tune.c:273] Frontend: 0, Sending DiSEqC
349  Wed Jan 28 23:43:38.971263312 2015  6  [tune.c:528] Frontend: 0, Waiting on lock.

Barracuda09, 2015-01-28:
Yes.. well thanks for trying. I can not seem to reproduce the crash.. I think I know were it is.. but it does not make a lot of sense.
 Will look into this further.

Thanks,
Marc

pbriesch, 2015-01-29:
Hi Marc,
I double checked the DVB-Cards in my server and I can confirm that they work fine.
Attached find some logging from SAT>IP client with vtuner on my RaspberryPi. I did not get a picture. There is no lock.

dmesg output on my Raspberry:
````
[  931.039811] virtual DVB adapter driver, version 1.4, (c) 2010-12 Honza Petrous, SmartImp.cz
[  931.046170] DVB: registering new adapter (vTuner proxy)
[  931.049596] vtunerc: registered /dev/vtunerc0
[  932.183710] vtunerc0: setting DVB-S2 tuner vtype
[  932.183766] (NULL device *): DVB: registering adapter 0 frontend 538976288 (vTuner proxyFE DVB-S2)...
[  987.976158] vtunerc0: allocated buffer of 1316 bytes
````
logging of SATPI:
````
0  Thu Jan 29 00:58:30.408779359 2015  7  [tune.c:172] Frontend Type: Unknown 4
1  Thu Jan 29 00:58:30.408784699 2015  7  [tune.c:177] Frontend Freq: 950000 Hz to 2150000 Hz
2  Thu Jan 29 00:58:30.408793553 2015  7  [tune.c:178] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
3  Thu Jan 29 00:58:30.409140354 2015  6  [connection.c:261] eth0: 192.168.1.161 [00:1d:7d:a9:eb:f3]
4  Thu Jan 29 00:58:30.409150235 2015  6  [rtsp.c:929] Setting up RTSP server
5  Thu Jan 29 00:58:30.409204364 2015  6  [http.c:447] Setting up HTTP server
6  Thu Jan 29 00:58:30.409239085 2015  6  [rtp.c:346] Setting up 8 RTP servers
7  Thu Jan 29 00:58:30.409428529 2015  6  [rtcp.c:338] Setting up 8 RTCP servers
8  Thu Jan 29 00:58:30.409638433 2015  6  [ssdp.c:89] Setting up SSDP server
9  Thu Jan 29 00:58:30.639111271 2015  6  [connection.c:190] RTSP Connection from 192.168.1.150 Port 55701
10  Thu Jan 29 00:58:30.639437648 2015  7  [rtsp.c:815] OPTIONS rtsp://192.168.1.161:554/ RTSP/1.0
11  Thu Jan 29 00:58:30.639437648 2015  7  CSeq: 1
12  Thu Jan 29 00:58:30.639437648 2015  7  --- LINE END ---
13  Thu Jan 29 00:58:30.640692486 2015  7  [rtsp.c:815] SETUP rtsp://192.168.1.161/?src=1&freq=11840.4&pol=h&msys=dvbs&sr=27500&fec=34&pids=101,102,103,106,105,18,20,0,17,16,100,200 RTSP/1.0
14  Thu Jan 29 00:58:30.640692486 2015  7  CSeq: 2
15  Thu Jan 29 00:58:30.640692486 2015  7  Transport: RTP/AVP;unicast;client_port=46508-46509
16  Thu Jan 29 00:58:30.640692486 2015  7  --- LINE END ---
17  Thu Jan 29 00:58:30.640747963 2015  6  [rtsp.c:407] Frontend: 0, With SYS_DVBS Attaching dynamically to client 192.168.1.150
18  Thu Jan 29 00:58:30.640808787 2015  6  [rtsp.c:448] Setup Message
19  Thu Jan 29 00:58:30.640877575 2015  7  [tune.c:462] Open FE fd: 24 readonly 0
20  Thu Jan 29 00:58:30.640886373 2015  6  [tune.c:518] Frontend: 0, Opened FE fd: 24.
21  Thu Jan 29 00:58:30.640892893 2015  7  [tune.c:350] Frontend: 0, Start tuning process
22  Thu Jan 29 00:58:30.640899042 2015  7  [tune.c:279] Frontend: 0, Sending DiSEqC [e0] [10] [38] [f7]
23  Thu Jan 29 00:58:31.018867432 2015  6  [tune.c:529] Frontend: 0, Waiting on lock.
24  Thu Jan 29 00:58:31.309426528 2015  6  [tune.c:543] Frontend: 0, Not locked yet (FE status 0)...
25  Thu Jan 29 00:58:31.334512155 2015  6  [tune.c:543] Frontend: 0, Not locked yet (FE status 0)...
26  Thu Jan 29 00:58:31.359595551 2015  6  [tune.c:543] Frontend: 0, Not locked yet (FE status 0)...
27  Thu Jan 29 00:58:31.384673633 2015  6  [tune.c:543] Frontend: 0, Not locked yet (FE status 0)...
28  Thu Jan 29 00:58:31.409883304 2015  6  [tune.c:561] Frontend: 0, Opened DVR fd: 25.
29  Thu Jan 29 00:58:31.409920184 2015  7  [tune.c:498] Frontend: 0, Set filter PID 0 - fd 26
30  Thu Jan 29 00:58:31.409947310 2015  7  [tune.c:498] Frontend: 0, Set filter PID 16 - fd 27
31  Thu Jan 29 00:58:31.409970272 2015  7  [tune.c:498] Frontend: 0, Set filter PID 17 - fd 28
32  Thu Jan 29 00:58:31.409991650 2015  7  [tune.c:498] Frontend: 0, Set filter PID 18 - fd 29
33  Thu Jan 29 00:58:31.410013260 2015  7  [tune.c:498] Frontend: 0, Set filter PID 20 - fd 30
34  Thu Jan 29 00:58:31.410040071 2015  7  [tune.c:498] Frontend: 0, Set filter PID 100 - fd 31
35  Thu Jan 29 00:58:31.410052823 2015  7  [tune.c:498] Frontend: 0, Set filter PID 101 - fd 32
36  Thu Jan 29 00:58:31.410065312 2015  7  [tune.c:498] Frontend: 0, Set filter PID 102 - fd 33
37  Thu Jan 29 00:58:31.410077086 2015  7  [tune.c:498] Frontend: 0, Set filter PID 103 - fd 34
38  Thu Jan 29 00:58:31.410091216 2015  7  [tune.c:498] Frontend: 0, Set filter PID 105 - fd 35
39  Thu Jan 29 00:58:31.410103117 2015  7  [tune.c:498] Frontend: 0, Set filter PID 106 - fd 36
40  Thu Jan 29 00:58:31.410115067 2015  7  [tune.c:498] Frontend: 0, Set filter PID 200 - fd 37
41  Thu Jan 29 00:58:31.410191446 2015  6  [rtp.c:98] Frontend: 0, Start streaming to 192.168.1.150 (46508 - 46509)
42  Thu Jan 29 00:58:31.410225440 2015  6  [rtcp.c:252] Frontend: 0, Start RTCP stream to 192.168.1.150 (46508 - 46509)
43  Thu Jan 29 00:58:31.410250685 2015  6  [rtcp.c:266] Frontend: 0, Open Frontend Monitor for 192.168.1.150 (46508 - 46509)
44  Thu Jan 29 00:58:31.410261964 2015  7  [tune.c:462] Open FE fd: 38 readonly 1
45  Thu Jan 29 00:58:31.411638725 2015  7  [rtsp.c:815] PLAY rtsp://192.168.1.161/stream=0 RTSP/1.0
46  Thu Jan 29 00:58:31.411638725 2015  7  CSeq: 3
47  Thu Jan 29 00:58:31.411638725 2015  7  Session: 00000002
48  Thu Jan 29 00:58:31.411638725 2015  7  --- LINE END ---
49  Thu Jan 29 00:58:31.411709681 2015  6  [rtsp.c:519] Play Message
50  Thu Jan 29 00:58:31.411722155 2015  7  [tune.c:350] Frontend: 0, Start tuning process
51  Thu Jan 29 00:58:31.411733311 2015  7  [tune.c:279] Frontend: 0, Sending DiSEqC [e0] [10] [38] [f7]
[... cut ...]
344  Thu Jan 29 01:02:09.259566009 2015  6  [tune.c:543] Frontend: 1, Not locked yet (FE status 0)...
345  Thu Jan 29 01:02:09.284643640 2015  6  [tune.c:543] Frontend: 1, Not locked yet (FE status 0)...
346  Thu Jan 29 01:02:09.309753522 2015  7  [tune.c:498] Frontend: 1, Set filter PID 0 - fd 34
347  Thu Jan 29 01:02:09.309783620 2015  7  [tune.c:498] Frontend: 1, Set filter PID 16 - fd 35
348  Thu Jan 29 01:02:09.309830560 2015  7  [tune.c:498] Frontend: 1, Set filter PID 17 - fd 36
349  Thu Jan 29 01:02:09.309878790 2015  7  [tune.c:498] Frontend: 1, Set filter PID 20 - fd 37

Barracuda09, 2015-01-29:
Hi,

Would you be so kind to try it once more. I fixed the stupid mistake with the DiSEqC switch.

Sorry for your time and thanks,

Marc

pbriesch, 2015-01-29:
Congratulations Marc!

It works! I got a picture! Don't be sorry! It is a pleasure to help you making your program work and thank you very much for sharing it with the community.

I'm closing this issue, but we are not ready, yet. There has been a segfault for which I will open a new issue.

Regards

Paul

closed 2015-01-29 by pbriesch