Barracuda09/SATPI #13: SATPI crashes

pbriesch, 2015-01-29:
SATPI crashes with segfault. EyeTV as client. Maybe fast channel changes caused the crash.
````
[  551.076757] thread_rtsp[1614]: segfault at 18 ip 00007f6cdcdb4324 sp 00007f6cdc7dbdc0 error 4 in libpthread-2.19.so[7f6cdcdaa000+19000]
````
````
0  Thu Jan 29 11:54:28.892892164 2015  7  a=control:stream=1
1  Thu Jan 29 11:54:28.892892164 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=2,0,1,0,11954000,0,SYS_DVBS,QPSK,2,0.35,27500000,FEC_3_4;pids=,270,2171,5411,5412,5413,5415,5416
2  Thu Jan 29 11:54:28.892892164 2015  7  a=sendonly
3  Thu Jan 29 11:54:33.986434423 2015  7  [rtsp.c:882] RTSP Connection closed as expected by Client: 192.168.1.107 ??
4  Thu Jan 29 11:54:33.988389881 2015  6  [connection.c:190] RTSP Connection from 192.168.1.107 Port 49509
5  Thu Jan 29 11:54:33.988788096 2015  7  [rtsp.c:815] DESCRIBE rtsp://192.168.1.161/ RTSP/1.0
6  Thu Jan 29 11:54:33.988788096 2015  7  Connection:Close
7  Thu Jan 29 11:54:33.988788096 2015  7  CSeq:1028
8  Thu Jan 29 11:54:33.988788096 2015  7  Accept:application/sdp
9  Thu Jan 29 11:54:33.988788096 2015  7  --- LINE END ---
10  Thu Jan 29 11:54:33.988896054 2015  6  [rtsp.c:829] RTSP Requested Connection closed by Client: 192.168.1.107
11  Thu Jan 29 11:54:33.989020642 2015  7  [rtsp.c:618] RTSP/1.0 200 OK
12  Thu Jan 29 11:54:33.989020642 2015  7  CSeq: 1028
13  Thu Jan 29 11:54:33.989020642 2015  7  Content-Type: application/sdp
14  Thu Jan 29 11:54:33.989020642 2015  7  Content-Base: rtsp://192.168.1.161/
15  Thu Jan 29 11:54:33.989020642 2015  7  Content-Length: 528
16  Thu Jan 29 11:54:33.989020642 2015  7  --- LINE END ---
17  Thu Jan 29 11:54:33.989020642 2015  7  v=0
18  Thu Jan 29 11:54:33.989020642 2015  7  o=- 5678901234 7890123456 IN IP4 192.168.1.161
19  Thu Jan 29 11:54:33.989020642 2015  7  s=SatIPServer:1 2
20  Thu Jan 29 11:54:33.989020642 2015  7  t=0 0
21  Thu Jan 29 11:54:33.989020642 2015  7  m=video 0 RTP/AVP 33
22  Thu Jan 29 11:54:33.989020642 2015  7  c=IN IP4 0.0.0.0
23  Thu Jan 29 11:54:33.989020642 2015  7  a=control:stream=0
24  Thu Jan 29 11:54:33.989020642 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,1,0,11362000,0,SYS_DVBS2,PSK_8,2,0.35,22000000,FEC_2_3;pids=0,16,17,18,19,20,201,202,203,204,206,6170,6400,6410,6420,6421,6422,6423,6430,8190
25  Thu Jan 29 11:54:33.989020642 2015  7  a=inactive
26  Thu Jan 29 11:54:33.989020642 2015  7  m=video 0 RTP/AVP 33
27  Thu Jan 29 11:54:33.989020642 2015  7  c=IN IP4 0.0.0.0
28  Thu Jan 29 11:54:33.989020642 2015  7  a=control:stream=1
29  Thu Jan 29 11:54:33.989020642 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=2,0,1,0,11954000,0,SYS_DVBS,QPSK,2,0.35,27500000,FEC_3_4;pids=,270,2171,5411,5412,5413,5415,5416
30  Thu Jan 29 11:54:33.989020642 2015  7  a=sendonly
31  Thu Jan 29 11:54:38.349045276 2015  7  [rtsp.c:882] RTSP Connection closed as expected by Client: 192.168.1.107 ??
32  Thu Jan 29 11:54:38.350545257 2015  6  [connection.c:190] RTSP Connection from 192.168.1.107 Port 49510
33  Thu Jan 29 11:54:38.351479219 2015  7  [rtsp.c:815] OPTIONS rtsp://192.168.1.161/ RTSP/1.0
34  Thu Jan 29 11:54:38.351479219 2015  7  Connection:Close
35  Thu Jan 29 11:54:38.351479219 2015  7  CSeq:1074
36  Thu Jan 29 11:54:38.351479219 2015  7  Session:00000002
37  Thu Jan 29 11:54:38.351479219 2015  7  --- LINE END ---
38  Thu Jan 29 11:54:38.351587769 2015  6  [rtsp.c:829] RTSP Requested Connection closed by Client: 192.168.1.107
39  Thu Jan 29 11:54:39.073838117 2015  7  [rtsp.c:882] RTSP Connection closed as expected by Client: 192.168.1.107 ??
40  Thu Jan 29 11:54:39.075175639 2015  6  [connection.c:190] RTSP Connection from 192.168.1.107 Port 49511
41  Thu Jan 29 11:54:39.075209440 2015  7  [rtsp.c:815] DESCRIBE rtsp://192.168.1.161/ RTSP/1.0
42  Thu Jan 29 11:54:39.075209440 2015  7  Connection:Close
43  Thu Jan 29 11:54:39.075209440 2015  7  CSeq:1029
44  Thu Jan 29 11:54:39.075209440 2015  7  Accept:application/sdp
45  Thu Jan 29 11:54:39.075209440 2015  7  --- LINE END ---
46  Thu Jan 29 11:54:39.075363439 2015  6  [rtsp.c:829] RTSP Requested Connection closed by Client: 192.168.1.107
47  Thu Jan 29 11:54:39.075477031 2015  7  [rtsp.c:618] RTSP/1.0 200 OK
48  Thu Jan 29 11:54:39.075477031 2015  7  CSeq: 1029
49  Thu Jan 29 11:54:39.075477031 2015  7  Content-Type: application/sdp
50  Thu Jan 29 11:54:39.075477031 2015  7  Content-Base: rtsp://192.168.1.161/
51  Thu Jan 29 11:54:39.075477031 2015  7  Content-Length: 528
52  Thu Jan 29 11:54:39.075477031 2015  7  --- LINE END ---
53  Thu Jan 29 11:54:39.075477031 2015  7  v=0
54  Thu Jan 29 11:54:39.075477031 2015  7  o=- 5678901234 7890123456 IN IP4 192.168.1.161
55  Thu Jan 29 11:54:39.075477031 2015  7  s=SatIPServer:1 2
56  Thu Jan 29 11:54:39.075477031 2015  7  t=0 0
57  Thu Jan 29 11:54:39.075477031 2015  7  m=video 0 RTP/AVP 33
58  Thu Jan 29 11:54:39.075477031 2015  7  c=IN IP4 0.0.0.0
59  Thu Jan 29 11:54:39.075477031 2015  7  a=control:stream=0
60  Thu Jan 29 11:54:39.075477031 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,1,0,11362000,0,SYS_DVBS2,PSK_8,2,0.35,22000000,FEC_2_3;pids=0,16,17,18,19,20,201,202,203,204,206,6170,6400,6410,6420,6421,6422,6423,6430,8190
61  Thu Jan 29 11:54:39.075477031 2015  7  a=inactive
62  Thu Jan 29 11:54:39.075477031 2015  7  m=video 0 RTP/AVP 33
63  Thu Jan 29 11:54:39.075477031 2015  7  c=IN IP4 0.0.0.0
64  Thu Jan 29 11:54:39.075477031 2015  7  a=control:stream=1
[... 260 lines cut ...]
325  Thu Jan 29 11:55:04.663279859 2015  7  CSeq:1034
326  Thu Jan 29 11:55:04.663279859 2015  7  Accept:application/sdp
327  Thu Jan 29 11:55:04.663279859 2015  7  --- LINE END ---
328  Thu Jan 29 11:55:04.663364113 2015  6  [rtsp.c:829] RTSP Requested Connection closed by Client: 192.168.1.107
329  Thu Jan 29 11:55:04.663515524 2015  7  [rtsp.c:618] RTSP/1.0 200 OK
330  Thu Jan 29 11:55:04.663515524 2015  7  CSeq: 1034
331  Thu Jan 29 11:55:04.663515524 2015  7  Content-Type: application/sdp
332  Thu Jan 29 11:55:04.663515524 2015  7  Content-Base: rtsp://192.168.1.161/
333  Thu Jan 29 11:55:04.663515524 2015  7  Content-Length: 447
334  Thu Jan 29 11:55:04.663515524 2015  7  --- LINE END ---
335  Thu Jan 29 11:55:04.663515524 2015  7  v=0
336  Thu Jan 29 11:55:04.663515524 2015  7  o=- 5678901234 7890123456 IN IP4 192.168.1.161
337  Thu Jan 29 11:55:04.663515524 2015  7  s=SatIPServer:1 2
338  Thu Jan 29 11:55:04.663515524 2015  7  t=0 0
339  Thu Jan 29 11:55:04.663515524 2015  7  m=video 0 RTP/AVP 33
340  Thu Jan 29 11:55:04.663515524 2015  7  c=IN IP4 0.0.0.0
341  Thu Jan 29 11:55:04.663515524 2015  7  a=control:stream=0
342  Thu Jan 29 11:55:04.663515524 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,1,0,11362000,0,SYS_DVBS2,PSK_8,2,0.35,22000000,FEC_2_3;pids=
343  Thu Jan 29 11:55:04.663515524 2015  7  a=inactive
344  Thu Jan 29 11:55:04.663515524 2015  7  m=video 0 RTP/AVP 33
345  Thu Jan 29 11:55:04.663515524 2015  7  c=IN IP4 0.0.0.0
346  Thu Jan 29 11:55:04.663515524 2015  7  a=control:stream=1
347  Thu Jan 29 11:55:04.663515524 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=2,0,1,0,11954000,0,SYS_DVBS,QPSK,2,0.35,27500000,FEC_3_4;pids=,270,2171,5411,5412,5413,5415,5416
348  Thu Jan 29 11:55:04.663515524 2015  7  a=sendonly
349  Thu Jan 29 11:55:09.349929715 2015  7  [rtsp.c:882] RTSP Connection closed as expected by Client: 192.168.1.107 ??

Barracuda09, 2015-01-29:
Hi Paul,

A few questions:
- Can you reproduce this easy?
- Do you know how to use gdb? (if so can you run it within gdb and do a backtrace)

Thanks,

Marc

Barracuda09, 2015-01-29:
Hi Paul,

I think I fixed this issue. Can you give it another try
if you have the time.

Many thanks,

Marc

pbriesch, 2015-01-29:
Marc, 

I cannot see any crashes any more. Seems to be fixed.

Paul

closed 2015-01-29 by pbriesch