Barracuda09/SATPI #14: No picture on Elgato SAT>IP app for OS X

pbriesch, 2015-01-29:
Hi Marc,

I have tested SATPI now with the app SAT>IP for OS X of Elgato on my iPhone 4S:  https://itunes.apple.com/en/app/elgato-sat-ip/id490084846?mt=8. 

SATPI can be found by the app. However, I can't get a picture because the app tells that the channel is already being used. If I switch to an other channel it is the same thing. Attached some log output:
````
0  Thu Jan 29 15:44:02.066228980 2015  6  [satpi.c:235] --- starting ---
1  Thu Jan 29 15:44:02.066315691 2015  6  [tune.c:393] Detecting frontends in: /dev/dvb
2  Thu Jan 29 15:44:02.066426065 2015  6  [tune.c:403] Frontends found: 2
3  Thu Jan 29 15:44:02.066442818 2015  7  [tune.c:468] Open FE fd: 4 readonly 1
4  Thu Jan 29 15:44:02.066471372 2015  6  [tune.c:117] Frontend Name: STB0899 Multistandard
5  Thu Jan 29 15:44:02.066482960 2015  7  [tune.c:142] Frontend Type: Satellite (DVB-S)
6  Thu Jan 29 15:44:02.066501671 2015  7  [tune.c:146] Frontend Type: Satellite (DVB-S2)
7  Thu Jan 29 15:44:02.066524799 2015  7  [tune.c:172] Frontend Type: Unknown 4
8  Thu Jan 29 15:44:02.066543791 2015  7  [tune.c:177] Frontend Freq: 950000 Hz to 2150000 Hz
9  Thu Jan 29 15:44:02.066562510 2015  7  [tune.c:178] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
10  Thu Jan 29 15:44:02.066587076 2015  7  [tune.c:468] Open FE fd: 4 readonly 1
11  Thu Jan 29 15:44:02.066596982 2015  6  [tune.c:117] Frontend Name: STB0899 Multistandard
12  Thu Jan 29 15:44:02.066604173 2015  7  [tune.c:142] Frontend Type: Satellite (DVB-S)
13  Thu Jan 29 15:44:02.066621027 2015  7  [tune.c:146] Frontend Type: Satellite (DVB-S2)
14  Thu Jan 29 15:44:02.066639690 2015  7  [tune.c:172] Frontend Type: Unknown 4
15  Thu Jan 29 15:44:02.066658026 2015  7  [tune.c:177] Frontend Freq: 950000 Hz to 2150000 Hz
16  Thu Jan 29 15:44:02.066679079 2015  7  [tune.c:178] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
17  Thu Jan 29 15:44:02.067046699 2015  6  [connection.c:261] eth0: 192.168.1.161 [00:1d:7d:a9:eb:f3]
18  Thu Jan 29 15:44:02.067057466 2015  6  [rtsp.c:932] Setting up RTSP server
19  Thu Jan 29 15:44:02.067111211 2015  6  [http.c:447] Setting up HTTP server
20  Thu Jan 29 15:44:02.067146942 2015  6  [rtp.c:346] Setting up 8 RTP servers
21  Thu Jan 29 15:44:02.067351944 2015  6  [rtcp.c:338] Setting up 8 RTCP servers
22  Thu Jan 29 15:44:02.067562325 2015  6  [ssdp.c:89] Setting up SSDP server
23  Thu Jan 29 15:44:02.955122351 2015  6  [connection.c:190] RTSP Connection from 192.168.1.115 Port 52623
24  Thu Jan 29 15:44:02.955554420 2015  7  [rtsp.c:818] DESCRIBE rtsp://192.168.1.161/ RTSP/1.0
25  Thu Jan 29 15:44:02.955554420 2015  7  Connection:Close
26  Thu Jan 29 15:44:02.955554420 2015  7  CSeq:1016
27  Thu Jan 29 15:44:02.955554420 2015  7  Accept:application/sdp
28  Thu Jan 29 15:44:02.955554420 2015  7  --- LINE END ---
29  Thu Jan 29 15:44:02.955645401 2015  6  [rtsp.c:832] RTSP Requested Connection closed by Client: 192.168.1.115
30  Thu Jan 29 15:44:02.955752011 2015  7  [rtsp.c:621] RTSP/1.0 200 OK
31  Thu Jan 29 15:44:02.955752011 2015  7  CSeq: 1016
32  Thu Jan 29 15:44:02.955752011 2015  7  Content-Type: application/sdp
33  Thu Jan 29 15:44:02.955752011 2015  7  Content-Base: rtsp://192.168.1.161/
34  Thu Jan 29 15:44:02.955752011 2015  7  Content-Length: 247
35  Thu Jan 29 15:44:02.955752011 2015  7  --- LINE END ---
36  Thu Jan 29 15:44:02.955752011 2015  7  v=0
37  Thu Jan 29 15:44:02.955752011 2015  7  o=- 5678901234 7890123456 IN IP4 192.168.1.161
38  Thu Jan 29 15:44:02.955752011 2015  7  s=SatIPServer:1 2
39  Thu Jan 29 15:44:02.955752011 2015  7  t=0 0
40  Thu Jan 29 15:44:02.955752011 2015  7  m=video 0 RTP/AVP 33
41  Thu Jan 29 15:44:02.955752011 2015  7  c=IN IP4 0.0.0.0
42  Thu Jan 29 15:44:02.955752011 2015  7  a=control:stream=0
43  Thu Jan 29 15:44:02.955752011 2015  7  a=fmtp:33 
44  Thu Jan 29 15:44:02.955752011 2015  7  a=inactive
45  Thu Jan 29 15:44:02.955752011 2015  7  m=video 0 RTP/AVP 33
46  Thu Jan 29 15:44:02.955752011 2015  7  c=IN IP4 0.0.0.0
47  Thu Jan 29 15:44:02.955752011 2015  7  a=control:stream=1
48  Thu Jan 29 15:44:02.955752011 2015  7  a=fmtp:33 
49  Thu Jan 29 15:44:02.955752011 2015  7  a=inactive
50  Thu Jan 29 15:44:02.957552616 2015  7  [http.c:212] GET web/desc.xml (size 1594)
51  Thu Jan 29 15:44:02.957604743 2015  7  [http.c:421] Requested Connection closed by Client 1
52  Thu Jan 29 15:44:06.757531708 2015  7  [rtsp.c:885] RTSP Connection closed as expected by Client: 192.168.1.115
53  Thu Jan 29 15:44:06.760853174 2015  6  [connection.c:190] RTSP Connection from 192.168.1.115 Port 52624
54  Thu Jan 29 15:44:06.761553287 2015  7  [rtsp.c:818] DESCRIBE rtsp://192.168.1.161/ RTSP/1.0
55  Thu Jan 29 15:44:06.761553287 2015  7  Connection:Close
56  Thu Jan 29 15:44:06.761553287 2015  7  CSeq:1017
57  Thu Jan 29 15:44:06.761553287 2015  7  Accept:application/sdp
58  Thu Jan 29 15:44:06.761553287 2015  7  --- LINE END ---
59  Thu Jan 29 15:44:06.761637833 2015  6  [rtsp.c:832] RTSP Requested Connection closed by Client: 192.168.1.115
60  Thu Jan 29 15:44:06.761754578 2015  7  [rtsp.c:621] RTSP/1.0 200 OK
61  Thu Jan 29 15:44:06.761754578 2015  7  CSeq: 1017
62  Thu Jan 29 15:44:06.761754578 2015  7  Content-Type: application/sdp
63  Thu Jan 29 15:44:06.761754578 2015  7  Content-Base: rtsp://192.168.1.161/
[... 159 lines cut ...]
223  Thu Jan 29 15:44:23.216091734 2015  7  Connection:Close
224  Thu Jan 29 15:44:23.216091734 2015  7  CSeq:1023
225  Thu Jan 29 15:44:23.216091734 2015  7  Accept:application/sdp
226  Thu Jan 29 15:44:23.216091734 2015  7  --- LINE END ---
227  Thu Jan 29 15:44:23.216263786 2015  6  [rtsp.c:832] RTSP Requested Connection closed by Client: 192.168.1.115
228  Thu Jan 29 15:44:23.216365519 2015  7  [rtsp.c:621] RTSP/1.0 200 OK
229  Thu Jan 29 15:44:23.216365519 2015  7  CSeq: 1023
230  Thu Jan 29 15:44:23.216365519 2015  7  Content-Type: application/sdp
231  Thu Jan 29 15:44:23.216365519 2015  7  Content-Base: rtsp://192.168.1.161/
232  Thu Jan 29 15:44:23.216365519 2015  7  Content-Length: 247
233  Thu Jan 29 15:44:23.216365519 2015  7  --- LINE END ---
234  Thu Jan 29 15:44:23.216365519 2015  7  v=0
235  Thu Jan 29 15:44:23.216365519 2015  7  o=- 5678901234 7890123456 IN IP4 192.168.1.161
236  Thu Jan 29 15:44:23.216365519 2015  7  s=SatIPServer:1 2
237  Thu Jan 29 15:44:23.216365519 2015  7  t=0 0
238  Thu Jan 29 15:44:23.216365519 2015  7  m=video 0 RTP/AVP 33
239  Thu Jan 29 15:44:23.216365519 2015  7  c=IN IP4 0.0.0.0
240  Thu Jan 29 15:44:23.216365519 2015  7  a=control:stream=0
241  Thu Jan 29 15:44:23.216365519 2015  7  a=fmtp:33 
242  Thu Jan 29 15:44:23.216365519 2015  7  a=inactive
243  Thu Jan 29 15:44:23.216365519 2015  7  m=video 0 RTP/AVP 33
244  Thu Jan 29 15:44:23.216365519 2015  7  c=IN IP4 0.0.0.0
245  Thu Jan 29 15:44:23.216365519 2015  7  a=control:stream=1
246  Thu Jan 29 15:44:23.216365519 2015  7  a=fmtp:33 
247  Thu Jan 29 15:44:23.216365519 2015  7  a=inactive

Barracuda09, 2015-01-29:
Hi,

Thanks, Yes I will look into this.

Marc

Barracuda09, 2015-01-31:
Hi,

Do you still have no picture with Elgato SAT>IP 
I have installed it on my android tabled and I have a picture, but for some reason
it stops after some time and it send a TEARDOWN message.
Which is very strange.

pbriesch, 2015-02-01:
Yes, I now have a picture! I will close this issue. For other quirks I will open new ones.

closed 2015-02-01 by pbriesch