Barracuda09/SATPI #15: Check if other SAT>IP server is already running.

pbriesch, 2015-01-29:
Hello Marc,

it would be nice if SATPI could upon start check if there is already some other SAT>IP server running.

Thanks,

Paul

Barracuda09, 2015-01-29:
Hi,

Do you mean on the network or just a program that uses the /dev/dvb/ adapters?

Marc

pbriesch, 2015-01-29:
Well, I mean a message like this one:
````
Starting minisatip version 0.1.50, dvbapi version: 050A
[29/01 21:53:16.000]: New UDP socket 3 bound to 0.0.0.0:1900
[29/01 21:53:16.000]: setting multicast for 239.255.255.250
[29/01 21:53:16.000]: New UDP socket 4 bound to 239.255.255.250:1900
[29/01 21:53:16.000]: tcp_listen: failed: bind(): Address already in use
RTSP: Could not listen on port 554
````
In this case SATPI was already running when I started minisatip.

Barracuda09, 2015-01-29:
Ok.. I understand.

lars18th, 2019-11-20:
Hi @Barracuda09 ,

I feel this is quite easy to implement: if the RTSP port can't bind, then exit.
And the same for the HTTP port.
You agree?

Barracuda09, 2019-11-20:
Yes @lars18th this would not be to difficult to implement