Barracuda09/SATPI #16: Frequent disconnection of client

pbriesch, 2015-01-29:
Hello Marc,

I'm watching now TV on my Mac with EyeTV and SATPI. About every 3 to 10 seconds the stream interrupts for some seconds. It seems that this caused by frequent openings and closings of the connection. Here some log output and thanks again for you efforts:

Paul
````
0  Thu Jan 29 15:59:35.038768484 2015  7  --- LINE END ---
1  Thu Jan 29 15:59:35.038877753 2015  6  [rtsp.c:832] RTSP Requested Connection closed by Client: 192.168.1.107
2  Thu Jan 29 15:59:35.038966614 2015  6  [rtsp.c:522] Play Message
3  Thu Jan 29 15:59:35.039013240 2015  7  [tune.c:504] Frontend: 1, Set filter PID 200 - fd 65
4  Thu Jan 29 15:59:35.138882422 2015  7  [rtsp.c:885] RTSP Connection closed as expected by Client: 192.168.1.107
5  Thu Jan 29 15:59:35.140270538 2015  6  [connection.c:190] RTSP Connection from 192.168.1.107 Port 61695
6  Thu Jan 29 15:59:35.140799536 2015  7  [rtsp.c:818] PLAY rtsp://192.168.1.161:554/stream=1?delpids=200&addpids=300 RTSP/1.0
7  Thu Jan 29 15:59:35.140799536 2015  7  Connection:Close
8  Thu Jan 29 15:59:35.140799536 2015  7  CSeq:1210
9  Thu Jan 29 15:59:35.140799536 2015  7  Session:00000002
10  Thu Jan 29 15:59:35.140799536 2015  7  --- LINE END ---
11  Thu Jan 29 15:59:35.140906280 2015  6  [rtsp.c:832] RTSP Requested Connection closed by Client: 192.168.1.107
12  Thu Jan 29 15:59:35.140964138 2015  6  [rtsp.c:522] Play Message
13  Thu Jan 29 15:59:35.140986227 2015  7  [tune.c:508] Frontend: 1, Remove filter PID 200 - Packet Count: 1
14  Thu Jan 29 15:59:35.141034343 2015  7  [tune.c:504] Frontend: 1, Set filter PID 300 - fd 65
15  Thu Jan 29 15:59:35.237352941 2015  7  [rtsp.c:885] RTSP Connection closed as expected by Client: 192.168.1.107
16  Thu Jan 29 15:59:35.238498905 2015  6  [connection.c:190] RTSP Connection from 192.168.1.107 Port 61696
17  Thu Jan 29 15:59:35.238529264 2015  7  [rtsp.c:818] PLAY rtsp://192.168.1.161:554/stream=1?delpids=300&addpids=600 RTSP/1.0
18  Thu Jan 29 15:59:35.238529264 2015  7  Connection:Close
19  Thu Jan 29 15:59:35.238529264 2015  7  CSeq:1211
20  Thu Jan 29 15:59:35.238529264 2015  7  Session:00000002
21  Thu Jan 29 15:59:35.238529264 2015  7  --- LINE END ---
22  Thu Jan 29 15:59:35.238685964 2015  6  [rtsp.c:832] RTSP Requested Connection closed by Client: 192.168.1.107
23  Thu Jan 29 15:59:35.238718799 2015  6  [rtsp.c:522] Play Message
24  Thu Jan 29 15:59:35.238729469 2015  7  [tune.c:508] Frontend: 1, Remove filter PID 300 - Packet Count: 1
25  Thu Jan 29 15:59:35.238760274 2015  7  [tune.c:504] Frontend: 1, Set filter PID 600 - fd 65
26  Thu Jan 29 15:59:35.338159451 2015  7  [rtsp.c:885] RTSP Connection closed as expected by Client: 192.168.1.107
27  Thu Jan 29 15:59:35.340557877 2015  6  [connection.c:190] RTSP Connection from 192.168.1.107 Port 61697
28  Thu Jan 29 15:59:35.340755386 2015  7  [rtsp.c:818] PLAY rtsp://192.168.1.161:554/stream=1?delpids=600&addpids=700 RTSP/1.0
29  Thu Jan 29 15:59:35.340755386 2015  7  Connection:Close
30  Thu Jan 29 15:59:35.340755386 2015  7  CSeq:1212
31  Thu Jan 29 15:59:35.340755386 2015  7  Session:00000002
32  Thu Jan 29 15:59:35.340755386 2015  7  --- LINE END ---
33  Thu Jan 29 15:59:35.340814801 2015  6  [rtsp.c:832] RTSP Requested Connection closed by Client: 192.168.1.107
34  Thu Jan 29 15:59:35.340848662 2015  6  [rtsp.c:522] Play Message
35  Thu Jan 29 15:59:35.340860303 2015  7  [tune.c:508] Frontend: 1, Remove filter PID 600 - Packet Count: 1
36  Thu Jan 29 15:59:35.340899610 2015  7  [tune.c:504] Frontend: 1, Set filter PID 700 - fd 65
37  Thu Jan 29 15:59:35.438092790 2015  7  [rtsp.c:885] RTSP Connection closed as expected by Client: 192.168.1.107
38  Thu Jan 29 15:59:35.439399323 2015  6  [connection.c:190] RTSP Connection from 192.168.1.107 Port 61698
39  Thu Jan 29 15:59:35.439847796 2015  7  [rtsp.c:818] PLAY rtsp://192.168.1.161:554/stream=1?delpids=700&addpids=800 RTSP/1.0
40  Thu Jan 29 15:59:35.439847796 2015  7  Connection:Close
41  Thu Jan 29 15:59:35.439847796 2015  7  CSeq:1213
42  Thu Jan 29 15:59:35.439847796 2015  7  Session:00000002
43  Thu Jan 29 15:59:35.439847796 2015  7  --- LINE END ---
44  Thu Jan 29 15:59:35.439953575 2015  6  [rtsp.c:832] RTSP Requested Connection closed by Client: 192.168.1.107
45  Thu Jan 29 15:59:35.439992324 2015  6  [rtsp.c:522] Play Message
46  Thu Jan 29 15:59:35.440027814 2015  7  [tune.c:508] Frontend: 1, Remove filter PID 700 - Packet Count: 1
47  Thu Jan 29 15:59:35.440065215 2015  7  [tune.c:504] Frontend: 1, Set filter PID 800 - fd 65
48  Thu Jan 29 15:59:35.538118803 2015  7  [rtsp.c:885] RTSP Connection closed as expected by Client: 192.168.1.107
49  Thu Jan 29 15:59:35.539424710 2015  6  [connection.c:190] RTSP Connection from 192.168.1.107 Port 61699
50  Thu Jan 29 15:59:35.539872596 2015  7  [rtsp.c:818] PLAY rtsp://192.168.1.161:554/stream=1?delpids=800&addpids=650 RTSP/1.0
51  Thu Jan 29 15:59:35.539872596 2015  7  Connection:Close
52  Thu Jan 29 15:59:35.539872596 2015  7  CSeq:1214
53  Thu Jan 29 15:59:35.539872596 2015  7  Session:00000002
54  Thu Jan 29 15:59:35.539872596 2015  7  --- LINE END ---
55  Thu Jan 29 15:59:35.539980232 2015  6  [rtsp.c:832] RTSP Requested Connection closed by Client: 192.168.1.107
56  Thu Jan 29 15:59:35.540042127 2015  6  [rtsp.c:522] Play Message
57  Thu Jan 29 15:59:35.540070070 2015  7  [tune.c:504] Frontend: 1, Set filter PID 650 - fd 67
58  Thu Jan 29 15:59:35.540079722 2015  7  [tune.c:508] Frontend: 1, Remove filter PID 800 - Packet Count: 1
59  Thu Jan 29 15:59:35.638018431 2015  7  [rtsp.c:885] RTSP Connection closed as expected by Client: 192.168.1.107
60  Thu Jan 29 15:59:35.639207545 2015  6  [connection.c:190] RTSP Connection from 192.168.1.107 Port 61700
61  Thu Jan 29 15:59:35.639237267 2015  7  [rtsp.c:818] PLAY rtsp://192.168.1.161:554/stream=1?delpids=650&addpids=500 RTSP/1.0
62  Thu Jan 29 15:59:35.639237267 2015  7  Connection:Close
63  Thu Jan 29 15:59:35.639237267 2015  7  CSeq:1215
[... 261 lines cut ...]
325  Thu Jan 29 15:59:53.676015188 2015  7  Connection:Close
326  Thu Jan 29 15:59:53.676015188 2015  7  CSeq:1046
327  Thu Jan 29 15:59:53.676015188 2015  7  Accept:application/sdp
328  Thu Jan 29 15:59:53.676015188 2015  7  --- LINE END ---
329  Thu Jan 29 15:59:53.676124371 2015  6  [rtsp.c:832] RTSP Requested Connection closed by Client: 192.168.1.107
330  Thu Jan 29 15:59:53.676252891 2015  7  [rtsp.c:621] RTSP/1.0 200 OK
331  Thu Jan 29 15:59:53.676252891 2015  7  CSeq: 1046
332  Thu Jan 29 15:59:53.676252891 2015  7  Content-Type: application/sdp
333  Thu Jan 29 15:59:53.676252891 2015  7  Content-Base: rtsp://192.168.1.161/
334  Thu Jan 29 15:59:53.676252891 2015  7  Content-Length: 533
335  Thu Jan 29 15:59:53.676252891 2015  7  --- LINE END ---
336  Thu Jan 29 15:59:53.676252891 2015  7  v=0
337  Thu Jan 29 15:59:53.676252891 2015  7  o=- 5678901234 7890123456 IN IP4 192.168.1.161
338  Thu Jan 29 15:59:53.676252891 2015  7  s=SatIPServer:1 2
339  Thu Jan 29 15:59:53.676252891 2015  7  t=0 0
340  Thu Jan 29 15:59:53.676252891 2015  7  m=video 0 RTP/AVP 33
341  Thu Jan 29 15:59:53.676252891 2015  7  c=IN IP4 0.0.0.0
342  Thu Jan 29 15:59:53.676252891 2015  7  a=control:stream=0
343  Thu Jan 29 15:59:53.676252891 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,1,0,11836000,0,SYS_DVBS,QPSK,2,0.35,27500000,FEC_3_4;pids=0,16,17,18,19,20,84,100,101,102,103,104,105,106,2070,2171,8190
344  Thu Jan 29 15:59:53.676252891 2015  7  a=sendonly
345  Thu Jan 29 15:59:53.676252891 2015  7  m=video 0 RTP/AVP 33
346  Thu Jan 29 15:59:53.676252891 2015  7  c=IN IP4 0.0.0.0
347  Thu Jan 29 15:59:53.676252891 2015  7  a=control:stream=1
348  Thu Jan 29 15:59:53.676252891 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=2,0,1,0,11954000,0,SYS_DVBS,QPSK,2,0.35,27500000,FEC_3_4;pids=0,16,17,18,19,20,84,100,110,120,121,122,125,130,131,950,8190
349  Thu Jan 29 15:59:53.676252891 2015  7  a=sendonly

Barracuda09, 2015-01-29:
Hi,

Thanks for testing.
I'm not sure if that is the problem. I will look into his.
How is VLC working?

Marc

pbriesch, 2015-01-29:
I have not yet tested vlc since I don't know exactly how to define the URI. Do you have any hints or examples for me?

Barracuda09, 2015-01-29:
Ok here is an example for Astra 23.5 of some FTA channel:
rtsp://192.168.0.109/?src=1&freq=11914&sr=29900&msys=dvbs&pol=h&fec=23&pids=0,1,16,17,18,40,57,88,516,2310,3002,3003,8190

pbriesch, 2015-01-29:
Here is a good source for channel lists: http://www.inverto.tv/products/product.php?id=242
Click the tab for **Product support** and download the lists by clicking on **IDL400S_DLNA_and_M3U_Channel_Lists**.
I have now VLC for Mac, Version 3.0.0-git Vetinari (Intel 64bit), running on one channel and there seem to be no problems, no interruptions in the stream.
````
0  Thu Jan 29 23:20:40.049121456 2015  7  v=0
1  Thu Jan 29 23:20:40.049121456 2015  7  o=- 5678901234 7890123456 IN IP4 192.168.1.161
2  Thu Jan 29 23:20:40.049121456 2015  7  s=SatIPServer:1 2
3  Thu Jan 29 23:20:40.049121456 2015  7  t=0 0
4  Thu Jan 29 23:20:40.049121456 2015  7  m=video 0 RTP/AVP 33
5  Thu Jan 29 23:20:40.049121456 2015  7  c=IN IP4 0.0.0.0
6  Thu Jan 29 23:20:40.049121456 2015  7  a=control:stream=0
7  Thu Jan 29 23:20:40.049121456 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,1,0,10744000,0,SYS_DVBS,QPSK,2,0.35,22000000,FEC_5_6;pids=
8  Thu Jan 29 23:20:40.049121456 2015  7  a=inactive
9  Thu Jan 29 23:20:40.049121456 2015  7  m=video 0 RTP/AVP 33
10  Thu Jan 29 23:20:40.049121456 2015  7  c=IN IP4 0.0.0.0
11  Thu Jan 29 23:20:40.049121456 2015  7  a=control:stream=1
12  Thu Jan 29 23:20:40.049121456 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=2,0,1,0,11954000,0,SYS_DVBS,QPSK,2,0.35,27500000,FEC_5_6;pids=0,100,110,120,121,125,130
13  Thu Jan 29 23:20:40.049121456 2015  7  a=sendonly
14  Thu Jan 29 23:20:44.441430043 2015  7  [rtsp.c:881] RTSP Connection closed as expected by Client: 192.168.1.2
15  Thu Jan 29 23:20:44.442094013 2015  6  [connection.c:190] RTSP Connection from 192.168.1.2 Port 53120
16  Thu Jan 29 23:20:44.442127758 2015  7  [rtsp.c:814] DESCRIBE rtsp://192.168.1.161/ RTSP/1.0
17  Thu Jan 29 23:20:44.442127758 2015  7  Connection:Close
18  Thu Jan 29 23:20:44.442127758 2015  7  CSeq:1114
19  Thu Jan 29 23:20:44.442127758 2015  7  Accept:application/sdp
20  Thu Jan 29 23:20:44.442127758 2015  7  --- LINE END ---
21  Thu Jan 29 23:20:44.442220679 2015  6  [rtsp.c:828] RTSP Requested Connection closed by Client: 192.168.1.2
22  Thu Jan 29 23:20:44.442305964 2015  7  [rtsp.c:617] RTSP/1.0 200 OK
23  Thu Jan 29 23:20:44.442305964 2015  7  CSeq: 1114
24  Thu Jan 29 23:20:44.442305964 2015  7  Content-Type: application/sdp
25  Thu Jan 29 23:20:44.442305964 2015  7  Content-Base: rtsp://192.168.1.161/
26  Thu Jan 29 23:20:44.442305964 2015  7  Content-Length: 436
27  Thu Jan 29 23:20:44.442305964 2015  7  --- LINE END ---
28  Thu Jan 29 23:20:44.442305964 2015  7  v=0
29  Thu Jan 29 23:20:44.442305964 2015  7  o=- 5678901234 7890123456 IN IP4 192.168.1.161
30  Thu Jan 29 23:20:44.442305964 2015  7  s=SatIPServer:1 2
31  Thu Jan 29 23:20:44.442305964 2015  7  t=0 0
32  Thu Jan 29 23:20:44.442305964 2015  7  m=video 0 RTP/AVP 33
33  Thu Jan 29 23:20:44.442305964 2015  7  c=IN IP4 0.0.0.0
34  Thu Jan 29 23:20:44.442305964 2015  7  a=control:stream=0
35  Thu Jan 29 23:20:44.442305964 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,1,0,10744000,0,SYS_DVBS,QPSK,2,0.35,22000000,FEC_5_6;pids=
36  Thu Jan 29 23:20:44.442305964 2015  7  a=inactive
37  Thu Jan 29 23:20:44.442305964 2015  7  m=video 0 RTP/AVP 33
38  Thu Jan 29 23:20:44.442305964 2015  7  c=IN IP4 0.0.0.0
39  Thu Jan 29 23:20:44.442305964 2015  7  a=control:stream=1
40  Thu Jan 29 23:20:44.442305964 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=2,0,1,0,11954000,0,SYS_DVBS,QPSK,2,0.35,27500000,FEC_5_6;pids=0,100,110,120,121,125,130
41  Thu Jan 29 23:20:44.442305964 2015  7  a=sendonly
42  Thu Jan 29 23:20:45.087956948 2015  7  [rtsp.c:881] RTSP Connection closed as expected by Client: 192.168.1.2
43  Thu Jan 29 23:20:45.088289954 2015  6  [connection.c:190] RTSP Connection from 192.168.1.2 Port 53121
44  Thu Jan 29 23:20:45.088322040 2015  7  [rtsp.c:814] DESCRIBE rtsp://192.168.1.161/ RTSP/1.0
45  Thu Jan 29 23:20:45.088322040 2015  7  Connection:Close
46  Thu Jan 29 23:20:45.088322040 2015  7  CSeq:1113
47  Thu Jan 29 23:20:45.088322040 2015  7  Accept:application/sdp
48  Thu Jan 29 23:20:45.088322040 2015  7  --- LINE END ---
49  Thu Jan 29 23:20:45.088502466 2015  6  [rtsp.c:828] RTSP Requested Connection closed by Client: 192.168.1.2
50  Thu Jan 29 23:20:45.088595334 2015  7  [rtsp.c:617] RTSP/1.0 200 OK
51  Thu Jan 29 23:20:45.088595334 2015  7  CSeq: 1113
52  Thu Jan 29 23:20:45.088595334 2015  7  Content-Type: application/sdp
53  Thu Jan 29 23:20:45.088595334 2015  7  Content-Base: rtsp://192.168.1.161/
54  Thu Jan 29 23:20:45.088595334 2015  7  Content-Length: 436
55  Thu Jan 29 23:20:45.088595334 2015  7  --- LINE END ---
56  Thu Jan 29 23:20:45.088595334 2015  7  v=0
57  Thu Jan 29 23:20:45.088595334 2015  7  o=- 5678901234 7890123456 IN IP4 192.168.1.161
58  Thu Jan 29 23:20:45.088595334 2015  7  s=SatIPServer:1 2
59  Thu Jan 29 23:20:45.088595334 2015  7  t=0 0
60  Thu Jan 29 23:20:45.088595334 2015  7  m=video 0 RTP/AVP 33
61  Thu Jan 29 23:20:45.088595334 2015  7  c=IN IP4 0.0.0.0
62  Thu Jan 29 23:20:45.088595334 2015  7  a=control:stream=0
63  Thu Jan 29 23:20:45.088595334 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,1,0,10744000,0,SYS_DVBS,QPSK,2,0.35,22000000,FEC_5_6;pids=
64  Thu Jan 29 23:20:45.088595334 2015  7  a=inactive
65  Thu Jan 29 23:20:45.088595334 2015  7  m=video 0 RTP/AVP 33
[... 259 lines cut ...]
325  Thu Jan 29 23:21:10.297128204 2015  7  Connection:Close
326  Thu Jan 29 23:21:10.297128204 2015  7  CSeq:1118
327  Thu Jan 29 23:21:10.297128204 2015  7  Accept:application/sdp
328  Thu Jan 29 23:21:10.297128204 2015  7  --- LINE END ---
[... cut ...]
343  Thu Jan 29 23:21:10.297352442 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,1,0,10744000,0,SYS_DVBS,QPSK,2,0.35,22000000,FEC_5_6;pids=
344  Thu Jan 29 23:21:10.297352442 2015  7  a=inactive
345  Thu Jan 29 23:21:10.297352442 2015  7  m=video 0 RTP/AVP 33
346  Thu Jan 29 23:21:10.297352442 2015  7  c=IN IP4 0.0.0.0
347  Thu Jan 29 23:21:10.297352442 2015  7  a=control:stream=1
348  Thu Jan 29 23:21:10.297352442 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=2,0,1,0,11954000,0,SYS_DVBS,QPSK,2,0.35,27500000,FEC_5_6;pids=0,100,110,120,121,125,130
349  Thu Jan 29 23:21:10.297352442 2015  7  a=sendonly

pbriesch, 2015-02-02:
Hello Marc,

just want to let you know that I the iOS app HD+ https://itunes.apple.com/en/app/hd-plus-guide/id674999909?mt=8 works so far well with SATPI. There is also a version available for Android in the Google Play store https://play.google.com/store/apps/details?id=air.de.hdplus.tvguide.
With EyeTV for iOS i'm still not able to get a picture.

Paul

Barracuda09, 2015-02-02:
Hi Paul,

Did EyeTV worked before? Or just not all the time.
Can you provide an other debug log?

Thanks for testing,

Marc

pbriesch, 2015-02-02:
EyeTV on iOS never worked before.
SATPI was already running when I started EyeTV. It took roughly about 20 s until the app recognized SATPI. Then the app put out the message that the channel is unavailable: http://pastebin.com/6ct0gcMV

Now I started the app again and suddenly it works! However, disconnects after about 20 to 30 s. Strange. Log for this is here: http://pastebin.com/pbFjjVWU

pbriesch, 2015-02-02:
Here is a log that covers some disconnects; http://pastebin.com/AMT3aXbJ

Barracuda09, 2015-02-03:
Hi Paul,

If you have time, will you give it another try.

Thanks,

Marc

pbriesch, 2015-02-03:
Hi Marc,

EyeTV on iOS seems to work now with SATPI without disconnects. Great! Good job! 
Here is the log: http://pastebin.com/C0hZk1N0

However, it takes about 10 s to 20 s until EyeTV finds SATPI: http://pastebin.com/6uYW8qyk
Maybe we should open a new issue for the initial connection problem?

I'wondering about following lines, but only you can judge this:
````
23  Tue Feb 3 20:10:26.360651283 2015  6  [ssdp.c:256] SAT>IP Client 192.168.1.115 : contacted us sending reply
24  Tue Feb 3 20:10:26.411388239 2015  6  [ssdp.c:256] SAT>IP Client 192.168.1.115 : contacted us sending reply
25  Tue Feb 3 20:10:26.457216463 2015  6  [ssdp.c:256] SAT>IP Client 192.168.1.115 : contacted us sending reply

35  Tue Feb 3 20:11:19.417140166 2015  7  [rtsp.c:648] RTSP/1.0 404 Not Found

pbriesch, 2015-02-03:
Here a log running EyeTV for OS X. I could see as well that EyeTV needed some time to eventually connect to SATIP. It connects much faster to the DIGIBIT R1. No disconnects, Steady movie stream. 

http://pastebin.com/J4JmiKL0

Barracuda09, 2015-02-03:
Hi,

Line 35 is because there were no streams setup (according to specs see
 http://www.satip.info/sites/satip/files/resource/satip_specification_version_1_2_2.pdf)
page 37:
- If no session has been previously setup on the SAT>IP server
(and thus no stream has been created), the response to the DESCRIBE request will be error message 404 (stream not found).

Line 23..25 is someone from this IP contacted the SATPI server and its sending a reply back.

pbriesch, 2015-02-03:
Thanks Marc, great Job. I close this issue now.

closed 2015-02-03 by pbriesch