Barracuda09/SATPI #18: Only one client can get stream

pbriesch, 2015-02-03:
Hi Marc,

when EyeTV on OS X was running (IP 192.168.1.107) , I could connect with EyeTV on iOS to SATPI (IP 192.168.1.115), however, the message "already in use" was shown.  http://pastebin.com/dAHNDGtn
The server (IP 192.168.1.161) has however two DVB-S2 cards.

Thanks for looking into this.

Paul

pbriesch, 2015-02-03:
Now I quit EyeTV on my Mac but the iPhone still doesn't show a picture. The message "already in use"  is still being displayed. http://pastebin.com/V8MJrxNK

After starting EyeTV on the Mac again it did connect to SATPI, showed the message "already in use" but after some channel switching a picture came. http://pastebin.com/V6YaG3Z8

Barracuda09, 2015-02-03:
Do you see that the frontends are attached (see in web page)

pbriesch, 2015-02-03:
After a restart of SATPI I can now watch with both iPhone and Mac at the same time. I first started EyeTV on the iPhone, then on the Mac. However, when I was checking the web page, the second delivery system was shown as *undefined*, shortly later as *dvbs*. 

````
0  Tue Feb 3 21:48:01.807250423 2015  7  [rtsp.c:848] DESCRIBE rtsp://192.168.1.161/ RTSP/1.0
1  Tue Feb 3 21:48:01.807250423 2015  7  Connection:Close
2  Tue Feb 3 21:48:01.807250423 2015  7  CSeq:1018
3  Tue Feb 3 21:48:01.807250423 2015  7  Accept:application/sdp
4  Tue Feb 3 21:48:01.807250423 2015  7  --- LINE END ---
5  Tue Feb 3 21:48:01.807358954 2015  6  [rtsp.c:862] RTSP Client 192.168.1.107: Requested Connection closed
6  Tue Feb 3 21:48:01.807515834 2015  7  [rtsp.c:648] RTSP/1.0 200 OK
7  Tue Feb 3 21:48:01.807515834 2015  7  CSeq: 1018
8  Tue Feb 3 21:48:01.807515834 2015  7  Content-Type: application/sdp
9  Tue Feb 3 21:48:01.807515834 2015  7  Content-Base: rtsp://192.168.1.161/
10  Tue Feb 3 21:48:01.807515834 2015  7  Content-Length: 478
11  Tue Feb 3 21:48:01.807515834 2015  7  --- LINE END ---
12  Tue Feb 3 21:48:01.807515834 2015  7  v=0
13  Tue Feb 3 21:48:01.807515834 2015  7  o=- 5678901234 7890123456 IN IP4 192.168.1.161
14  Tue Feb 3 21:48:01.807515834 2015  7  s=SatIPServer:1 2
15  Tue Feb 3 21:48:01.807515834 2015  7  t=0 0
16  Tue Feb 3 21:48:01.807515834 2015  7  m=video 0 RTP/AVP 33
17  Tue Feb 3 21:48:01.807515834 2015  7  c=IN IP4 0.0.0.0
18  Tue Feb 3 21:48:01.807515834 2015  7  a=control:stream=0
19  Tue Feb 3 21:48:01.807515834 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,1,0,11954,0,dvbs,qpsk,2,0.35,27500,34;pids=0,16,17,18,20,100,110,120,121,122,130,131
20  Tue Feb 3 21:48:01.807515834 2015  7  a=sendonly
21  Tue Feb 3 21:48:01.807515834 2015  7  m=video 0 RTP/AVP 33
22  Tue Feb 3 21:48:01.807515834 2015  7  c=IN IP4 0.0.0.0
23  Tue Feb 3 21:48:01.807515834 2015  7  a=control:stream=1
24  Tue Feb 3 21:48:01.807515834 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=2,0,1,0,12110,0,dvbs,qpsk,2,0.35,27500,34;pids=0,16,17,18,19,20,2171,2600,2601,2602,2603,2604,2670,8190
25  Tue Feb 3 21:48:01.807515834 2015  7  a=sendonly
26  Tue Feb 3 21:48:06.880059136 2015  7  [rtsp.c:913] RTSP Client 192.168.1.107: Connection closed as expected
27  Tue Feb 3 21:48:06.882179061 2015  6  [connection.c:263] RTSP Connection from 192.168.1.107 Port 55931
28  Tue Feb 3 21:48:06.882214005 2015  7  [rtsp.c:848] DESCRIBE rtsp://192.168.1.161/ RTSP/1.0
29  Tue Feb 3 21:48:06.882214005 2015  7  Connection:Close
30  Tue Feb 3 21:48:06.882214005 2015  7  CSeq:1019
31  Tue Feb 3 21:48:06.882214005 2015  7  Accept:application/sdp
32  Tue Feb 3 21:48:06.882214005 2015  7  --- LINE END ---
33  Tue Feb 3 21:48:06.882367918 2015  6  [rtsp.c:862] RTSP Client 192.168.1.107: Requested Connection closed
34  Tue Feb 3 21:48:06.882447142 2015  7  [rtsp.c:648] RTSP/1.0 200 OK
35  Tue Feb 3 21:48:06.882447142 2015  7  CSeq: 1019
36  Tue Feb 3 21:48:06.882447142 2015  7  Content-Type: application/sdp
37  Tue Feb 3 21:48:06.882447142 2015  7  Content-Base: rtsp://192.168.1.161/
38  Tue Feb 3 21:48:06.882447142 2015  7  Content-Length: 478
39  Tue Feb 3 21:48:06.882447142 2015  7  --- LINE END ---
40  Tue Feb 3 21:48:06.882447142 2015  7  v=0
41  Tue Feb 3 21:48:06.882447142 2015  7  o=- 5678901234 7890123456 IN IP4 192.168.1.161
42  Tue Feb 3 21:48:06.882447142 2015  7  s=SatIPServer:1 2
43  Tue Feb 3 21:48:06.882447142 2015  7  t=0 0
44  Tue Feb 3 21:48:06.882447142 2015  7  m=video 0 RTP/AVP 33
45  Tue Feb 3 21:48:06.882447142 2015  7  c=IN IP4 0.0.0.0
46  Tue Feb 3 21:48:06.882447142 2015  7  a=control:stream=0
47  Tue Feb 3 21:48:06.882447142 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,1,0,11954,0,dvbs,qpsk,2,0.35,27500,34;pids=0,16,17,18,20,100,110,120,121,122,130,131
48  Tue Feb 3 21:48:06.882447142 2015  7  a=sendonly
49  Tue Feb 3 21:48:06.882447142 2015  7  m=video 0 RTP/AVP 33
50  Tue Feb 3 21:48:06.882447142 2015  7  c=IN IP4 0.0.0.0
51  Tue Feb 3 21:48:06.882447142 2015  7  a=control:stream=1
52  Tue Feb 3 21:48:06.882447142 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=2,0,1,0,12110,0,dvbs,qpsk,2,0.35,27500,34;pids=0,16,17,18,19,20,2171,2600,2601,2602,2603,2604,2670,8190
53  Tue Feb 3 21:48:06.882447142 2015  7  a=sendonly
54  Tue Feb 3 21:48:11.981618043 2015  7  [rtsp.c:913] RTSP Client 192.168.1.107: Connection closed as expected
55  Tue Feb 3 21:48:11.982849812 2015  6  [connection.c:263] RTSP Connection from 192.168.1.107 Port 55933
56  Tue Feb 3 21:48:11.982883190 2015  7  [rtsp.c:848] DESCRIBE rtsp://192.168.1.161/ RTSP/1.0
57  Tue Feb 3 21:48:11.982883190 2015  7  Connection:Close
58  Tue Feb 3 21:48:11.982883190 2015  7  CSeq:1020
59  Tue Feb 3 21:48:11.982883190 2015  7  Accept:application/sdp
60  Tue Feb 3 21:48:11.982883190 2015  7  --- LINE END ---
61  Tue Feb 3 21:48:11.983036051 2015  6  [rtsp.c:862] RTSP Client 192.168.1.107: Requested Connection closed
62  Tue Feb 3 21:48:11.983118590 2015  7  [rtsp.c:648] RTSP/1.0 200 OK
63  Tue Feb 3 21:48:11.983118590 2015  7  CSeq: 1020
64  Tue Feb 3 21:48:11.983118590 2015  7  Content-Type: application/sdp
65  Tue Feb 3 21:48:11.983118590 2015  7  Content-Base: rtsp://192.168.1.161/
66  Tue Feb 3 21:48:11.983118590 2015  7  Content-Length: 478
[... 258 lines cut ...]
325  Tue Feb 3 21:48:57.775914403 2015  7  Connection:Close
326  Tue Feb 3 21:48:57.775914403 2015  7  CSeq:1029
327  Tue Feb 3 21:48:57.775914403 2015  7  Accept:application/sdp
[... cut ...]
343  Tue Feb 3 21:48:57.776147106 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,1,0,11954,0,dvbs,qpsk,2,0.35,27500,34;pids=0,16,17,18,20,100,110,120,121,122,130,131
344  Tue Feb 3 21:48:57.776147106 2015  7  a=sendonly
345  Tue Feb 3 21:48:57.776147106 2015  7  m=video 0 RTP/AVP 33
346  Tue Feb 3 21:48:57.776147106 2015  7  c=IN IP4 0.0.0.0
347  Tue Feb 3 21:48:57.776147106 2015  7  a=control:stream=1
348  Tue Feb 3 21:48:57.776147106 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=2,0,1,0,12110,0,dvbs,qpsk,2,0.35,27500,34;pids=0,16,17,18,19,20,2171,2600,2601,2602,2603,2604,2670,8190
349  Tue Feb 3 21:48:57.776147106 2015  7  a=sendonly

Barracuda09, 2015-02-03:
Hi Paul,

Yes there is still some issues (I have seen here that one frontend is not released)
I'm looking into this problem.

Thanks for testing this,

Marc

pbriesch, 2015-02-03:
I did run an other test. First I made sure both frontends are attached. EyeTV on Mac was running fine. EyeTV on iPhone did connect, but "channel unavailable" message. After restarting SATIP both clients could connect and both showed TV :-)

pbriesch, 2015-02-03:
Hi Marc,

it seems that one stream still continues even after I have quit all clients. Please see attached screen shot. Connection 1 is alive but there is no client and the Log has stopped.

[image]

Barracuda09, 2015-02-03:
Yes I have seen this also... you do not have by any chance a log?
You can also recall it from syslog

pbriesch, 2015-02-03:
Here is my syslog: https://www.dropbox.com/s/qs4545o9ppnrh7f/syslog?dl=0

Barracuda09, 2015-02-06:
Hi,

I have rewritten the RTSP Engine. So if you like you can give it a retry.
Thanks for testing and regards,

Marc

pbriesch, 2015-02-06:
Marc,

I got some compiler warnings:
````
pbriesch@ubuntu:~/satpi$ make
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG tune.c
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG ssdp.c
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG rtsp.c
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG rtp.c
rtp.c: In function ‘start_rtp’:
rtp.c:355:3: warning: format ‘%d’ expects argument of type ‘int’, but argument 4 has type ‘size_t’ [-Wformat=]
   snprintf(np, sizeof(np), "thread_rtp%d", i);
   ^
rtp.c:355:3: warning: format ‘%d’ expects argument of type ‘int’, but argument 4 has type ‘size_t’ [-Wformat=]
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG rtcp.c
rtcp.c: In function ‘start_rtcp’:
rtcp.c:346:3: warning: format ‘%d’ expects argument of type ‘int’, but argument 4 has type ‘size_t’ [-Wformat=]
   snprintf(np, sizeof(np), "thread_rtcp%d", i);
   ^
rtcp.c:346:3: warning: format ‘%d’ expects argument of type ‘int’, but argument 4 has type ‘size_t’ [-Wformat=]
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG http.c
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG utils.c
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG connection.c
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG applog.c
gcc -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG satpi.c
gcc  tune.o ssdp.o rtsp.o rtp.o rtcp.o http.o utils.o connection.o applog.o satpi.o -o satpi -lpthread -lrt
pbriesch@ubuntu:~/satpi$

Barracuda09, 2015-02-06:
Hi,

Thanks.. strange that my gcc is not spiting this warning....but fixed this.
Sorry but I still have some resource leaks I have to find. 

Regards,

Marc

pbriesch, 2015-02-06:
Marc,
I have now VDR with vdr-plugin-satip running. Unfortunately, the ver doesn't show a picture. Here the log and the screenshot of the Webpage:
[image]
[image]

````
0  Fri Feb 6 22:53:53.364186765 2015  7  --- LINE END ---
1  Fri Feb 6 22:53:53.364266088 2015  6  [rtsp.c:760] Frontend: 0, Teardown Message, gracefull: 1
2  Fri Feb 6 22:53:53.459140073 2015  6  [rtp.c:208] Frontend: 0, Stop streaming to 192.168.1.21 (53069 - 53070) (Streamed 0.000 MBytes)
3  Fri Feb 6 22:53:53.560821476 2015  6  [rtsp.c:715] Frontend: 0, Detached from client 192.168.1.21 with session ID: 0288582906
4  Fri Feb 6 22:53:53.562335424 2015  6  [connection.c:263] RTSP Connection from 192.168.1.21 Port 55589 (fd: 24)
5  Fri Feb 6 22:53:53.567419533 2015  6  [rtsp.c:878] Found empty client slot giving Session ID: 0671693627
6  Fri Feb 6 22:53:53.567437891 2015  7  [rtsp.c:897] OPTIONS rtsp://192.168.1.161/ RTSP/1.0
7  Fri Feb 6 22:53:53.567437891 2015  7  CSeq: 1
8  Fri Feb 6 22:53:53.567437891 2015  7  User-Agent: vdr-satip/1.0.2-GIT-0-156-g9e89621 (device 0)
9  Fri Feb 6 22:53:53.567437891 2015  7  --- LINE END ---
10  Fri Feb 6 22:53:53.572604029 2015  6  [rtsp.c:849] Client connection with sessionID 0671693627
11  Fri Feb 6 22:53:53.572621440 2015  6  [rtsp.c:854] Found Client with sessionID 0671693627
12  Fri Feb 6 22:53:53.572634081 2015  7  [rtsp.c:897] SETUP rtsp://192.168.1.161/?freq=11493&src=1&sr=22000&pol=h&fec=23&mtype=8psk&ro=0.35&msys=dvbs2 RTSP/1.0
13  Fri Feb 6 22:53:53.572634081 2015  7  CSeq: 2
14  Fri Feb 6 22:53:53.572634081 2015  7  Session: 0671693627
15  Fri Feb 6 22:53:53.572634081 2015  7  Transport: RTP/AVP;unicast;client_port=53069-53070
16  Fri Feb 6 22:53:53.572634081 2015  7  User-Agent: vdr-satip/1.0.2-GIT-0-156-g9e89621 (device 0)
17  Fri Feb 6 22:53:53.572634081 2015  7  If-Modified-Since: Thu, 01 Jan 1970 00:00:00 GMT
18  Fri Feb 6 22:53:53.572634081 2015  7  --- LINE END ---
19  Fri Feb 6 22:53:53.572741503 2015  6  [rtsp.c:409] Frontend: 0, With dvbs2 Attaching dynamically to client 192.168.1.21 with session ID: 0671693627
20  Fri Feb 6 22:53:53.572787318 2015  6  [rtsp.c:455] Setup Message
21  Fri Feb 6 22:53:53.572833744 2015  7  [tune.c:476] Open FE fd: 25 readonly 0
22  Fri Feb 6 22:53:53.572842575 2015  6  [tune.c:532] Frontend: 0, Opened FE fd: 25.
23  Fri Feb 6 22:53:53.572863082 2015  7  [tune.c:360] Frontend: 0, Start tuning process
24  Fri Feb 6 22:53:53.572871751 2015  6  [tune.c:289] Frontend: 0, Sending DiSEqC [e0] [10] [38] [f2]
25  Fri Feb 6 22:53:53.951554438 2015  6  [tune.c:543] Frontend: 0, Waiting on lock.
26  Fri Feb 6 22:53:54.186666609 2015  6  [tune.c:557] Frontend: 0, Not locked yet (FE status 26)...
27  Fri Feb 6 22:53:54.213111329 2015  6  [tune.c:557] Frontend: 0, Not locked yet (FE status 26)...
28  Fri Feb 6 22:53:54.239554528 2015  6  [tune.c:557] Frontend: 0, Not locked yet (FE status 26)...
29  Fri Feb 6 22:53:54.266011971 2015  6  [tune.c:557] Frontend: 0, Not locked yet (FE status 26)...
30  Fri Feb 6 22:53:54.291205368 2015  6  [tune.c:575] Frontend: 0, Opened DVR fd: 26.
31  Fri Feb 6 22:53:54.291275261 2015  6  [rtcp.c:252] Frontend: 0, Start RTCP stream to 192.168.1.21 (53069 - 53070)
32  Fri Feb 6 22:53:54.291286993 2015  6  [rtcp.c:266] Frontend: 0, Open Frontend Monitor for 192.168.1.21 (53069 - 53070)
33  Fri Feb 6 22:53:54.291297700 2015  7  [tune.c:476] Open FE fd: 27 readonly 1
34  Fri Feb 6 22:53:54.296322366 2015  6  [rtsp.c:849] Client connection with sessionID 0671693627
35  Fri Feb 6 22:53:54.296340836 2015  6  [rtsp.c:854] Found Client with sessionID 0671693627
36  Fri Feb 6 22:53:54.296353661 2015  7  [rtsp.c:897] PLAY rtsp://192.168.1.161/stream=0?freq=11361&src=1&sr=22000&pol=h&fec=23&mtype=8psk&ro=0.35&msys=dvbs2 RTSP/1.0
37  Fri Feb 6 22:53:54.296353661 2015  7  CSeq: 3
38  Fri Feb 6 22:53:54.296353661 2015  7  Session: 0671693627
39  Fri Feb 6 22:53:54.296353661 2015  7  User-Agent: vdr-satip/1.0.2-GIT-0-156-g9e89621 (device 0)
40  Fri Feb 6 22:53:54.296353661 2015  7  --- LINE END ---
41  Fri Feb 6 22:53:54.296466870 2015  6  [rtsp.c:520] Play Message
42  Fri Feb 6 22:53:54.296481686 2015  7  [tune.c:360] Frontend: 0, Start tuning process
43  Fri Feb 6 22:53:54.296502219 2015  6  [tune.c:289] Frontend: 0, Sending DiSEqC [e0] [10] [38] [f2]
44  Fri Feb 6 22:53:54.507566106 2015  6  [tune.c:543] Frontend: 0, Waiting on lock.
45  Fri Feb 6 22:53:54.742679453 2015  6  [tune.c:557] Frontend: 0, Not locked yet (FE status 26)...
46  Fri Feb 6 22:53:54.769143791 2015  6  [tune.c:557] Frontend: 0, Not locked yet (FE status 26)...
47  Fri Feb 6 22:53:54.795586178 2015  6  [tune.c:557] Frontend: 0, Not locked yet (FE status 26)...
48  Fri Feb 6 22:53:54.822047110 2015  6  [tune.c:557] Frontend: 0, Not locked yet (FE status 26)...
49  Fri Feb 6 22:53:54.847129369 2015  6  [rtp.c:98] Frontend: 0, Start streaming to 192.168.1.21 (53069 - 53070)
50  Fri Feb 6 22:53:54.852193227 2015  6  [rtsp.c:849] Client connection with sessionID 0671693627
51  Fri Feb 6 22:53:54.852212019 2015  6  [rtsp.c:854] Found Client with sessionID 0671693627
52  Fri Feb 6 22:53:54.852224518 2015  7  [rtsp.c:897] TEARDOWN rtsp://192.168.1.161/stream=0 RTSP/1.0
53  Fri Feb 6 22:53:54.852224518 2015  7  CSeq: 4
54  Fri Feb 6 22:53:54.852224518 2015  7  Session: 0671693627
55  Fri Feb 6 22:53:54.852224518 2015  7  User-Agent: vdr-satip/1.0.2-GIT-0-156-g9e89621 (device 0)
56  Fri Feb 6 22:53:54.852224518 2015  7  --- LINE END ---
[... cut ...]
344  Fri Feb 6 22:56:23.237298319 2015  7  [http.c:463] Requested Connection closed by Client 4
345  Fri Feb 6 22:56:27.435585168 2015  7  [http.c:244] GET web/log.html (size 1784)
346  Fri Feb 6 22:56:27.516149059 2015  7  [http.c:244] GET web/stylesheet.css (size 1463)
347  Fri Feb 6 22:56:27.516508476 2015  7  [http.c:244] GET web/loadxmldoc.js (size 521)
348  Fri Feb 6 22:56:27.569519646 2015  7  [http.c:244] GET web/bg-grain.png (size 5636)
349  Fri Feb 6 22:56:27.569590394 2015  7  [http.c:244] GET web/grid_noise.png (size 4638)

pbriesch, 2015-02-06:
In addition  to VDR I have started EyeTV on my Mac. EyeTV did connect, however, no picture. I quit EyeTV and started it again. The I got a picture. Please not the two entries of the Mac's IP address in the screenshot.
[image]
````
0  Fri Feb 6 23:03:50.042909518 2015  7  --- LINE END ---
1  Fri Feb 6 23:03:50.042969200 2015  6  [rtsp.c:911] RTSP Client 192.168.1.107: Requested Connection closed with session ID: 0926347328
2  Fri Feb 6 23:03:50.043085428 2015  7  [rtsp.c:646] RTSP/1.0 200 OK
3  Fri Feb 6 23:03:50.043085428 2015  7  CSeq: 1010
4  Fri Feb 6 23:03:50.043085428 2015  7  Content-Type: application/sdp
5  Fri Feb 6 23:03:50.043085428 2015  7  Content-Base: rtsp://192.168.1.161/
6  Fri Feb 6 23:03:50.043085428 2015  7  Content-Length: 439
7  Fri Feb 6 23:03:50.043085428 2015  7  --- LINE END ---
8  Fri Feb 6 23:03:50.043085428 2015  7  v=0
9  Fri Feb 6 23:03:50.043085428 2015  7  o=- 5678901234 7890123456 IN IP4 192.168.1.161
10  Fri Feb 6 23:03:50.043085428 2015  7  s=SatIPServer:1 2
11  Fri Feb 6 23:03:50.043085428 2015  7  t=0 0
12  Fri Feb 6 23:03:50.043085428 2015  7  m=video 0 RTP/AVP 33
13  Fri Feb 6 23:03:50.043085428 2015  7  c=IN IP4 0.0.0.0
14  Fri Feb 6 23:03:50.043085428 2015  7  a=control:stream=0
15  Fri Feb 6 23:03:50.043085428 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,1,0,10892,0,dvbs,8psk,2,0.35,27500,34;pids=
16  Fri Feb 6 23:03:50.043085428 2015  7  a=inactive
17  Fri Feb 6 23:03:50.043085428 2015  7  m=video 0 RTP/AVP 33
18  Fri Feb 6 23:03:50.043085428 2015  7  c=IN IP4 0.0.0.0
19  Fri Feb 6 23:03:50.043085428 2015  7  a=control:stream=1
20  Fri Feb 6 23:03:50.043085428 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=2,0,1,0,10744,0,dvbs,qpsk,2,0.35,22000,56;pids=0,16,17,18,19,20,400,401,402,403,404,407,408,470,2171,8190
21  Fri Feb 6 23:03:50.043085428 2015  7  a=sendonly
22  Fri Feb 6 23:03:55.058474188 2015  7  [rtsp.c:969] RTSP Client 192.168.1.107: Connection closed with session ID: 0926347328 (fd: 53)
23  Fri Feb 6 23:03:55.063600110 2015  6  [connection.c:263] RTSP Connection from 192.168.1.107 Port 51810 (fd: 53)
24  Fri Feb 6 23:03:55.068689724 2015  6  [rtsp.c:878] Found empty client slot giving Session ID: 0409872828
25  Fri Feb 6 23:03:55.068707874 2015  7  [rtsp.c:897] DESCRIBE rtsp://192.168.1.161/ RTSP/1.0
26  Fri Feb 6 23:03:55.068707874 2015  7  Connection:Close
27  Fri Feb 6 23:03:55.068707874 2015  7  CSeq:1011
28  Fri Feb 6 23:03:55.068707874 2015  7  Accept:application/sdp
29  Fri Feb 6 23:03:55.068707874 2015  7  --- LINE END ---
30  Fri Feb 6 23:03:55.068811023 2015  6  [rtsp.c:911] RTSP Client 192.168.1.107: Requested Connection closed with session ID: 0409872828
31  Fri Feb 6 23:03:55.068923161 2015  7  [rtsp.c:646] RTSP/1.0 200 OK
32  Fri Feb 6 23:03:55.068923161 2015  7  CSeq: 1011
33  Fri Feb 6 23:03:55.068923161 2015  7  Content-Type: application/sdp
34  Fri Feb 6 23:03:55.068923161 2015  7  Content-Base: rtsp://192.168.1.161/
35  Fri Feb 6 23:03:55.068923161 2015  7  Content-Length: 439
36  Fri Feb 6 23:03:55.068923161 2015  7  --- LINE END ---
37  Fri Feb 6 23:03:55.068923161 2015  7  v=0
38  Fri Feb 6 23:03:55.068923161 2015  7  o=- 5678901234 7890123456 IN IP4 192.168.1.161
39  Fri Feb 6 23:03:55.068923161 2015  7  s=SatIPServer:1 2
40  Fri Feb 6 23:03:55.068923161 2015  7  t=0 0
41  Fri Feb 6 23:03:55.068923161 2015  7  m=video 0 RTP/AVP 33
42  Fri Feb 6 23:03:55.068923161 2015  7  c=IN IP4 0.0.0.0
43  Fri Feb 6 23:03:55.068923161 2015  7  a=control:stream=0
44  Fri Feb 6 23:03:55.068923161 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,1,0,10892,0,dvbs,8psk,2,0.35,27500,34;pids=
45  Fri Feb 6 23:03:55.068923161 2015  7  a=inactive
46  Fri Feb 6 23:03:55.068923161 2015  7  m=video 0 RTP/AVP 33
47  Fri Feb 6 23:03:55.068923161 2015  7  c=IN IP4 0.0.0.0
48  Fri Feb 6 23:03:55.068923161 2015  7  a=control:stream=1
49  Fri Feb 6 23:03:55.068923161 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=2,0,1,0,10744,0,dvbs,qpsk,2,0.35,22000,56;pids=0,16,17,18,19,20,400,401,402,403,404,407,408,470,2171,8190
50  Fri Feb 6 23:03:55.068923161 2015  7  a=sendonly
51  Fri Feb 6 23:04:00.078531008 2015  7  [rtsp.c:969] RTSP Client 192.168.1.107: Connection closed with session ID: 0409872828 (fd: 53)
52  Fri Feb 6 23:04:00.083873235 2015  6  [connection.c:263] RTSP Connection from 192.168.1.107 Port 51813 (fd: 53)
53  Fri Feb 6 23:04:00.088963561 2015  6  [rtsp.c:878] Found empty client slot giving Session ID: 1810187268
54  Fri Feb 6 23:04:00.088982206 2015  7  [rtsp.c:897] DESCRIBE rtsp://192.168.1.161/ RTSP/1.0
55  Fri Feb 6 23:04:00.088982206 2015  7  Connection:Close
56  Fri Feb 6 23:04:00.088982206 2015  7  CSeq:1012
57  Fri Feb 6 23:04:00.088982206 2015  7  Accept:application/sdp
58  Fri Feb 6 23:04:00.088982206 2015  7  --- LINE END ---
59  Fri Feb 6 23:04:00.089110778 2015  6  [rtsp.c:911] RTSP Client 192.168.1.107: Requested Connection closed with session ID: 1810187268
60  Fri Feb 6 23:04:00.089215291 2015  7  [rtsp.c:646] RTSP/1.0 200 OK
61  Fri Feb 6 23:04:00.089215291 2015  7  CSeq: 1012
62  Fri Feb 6 23:04:00.089215291 2015  7  Content-Type: application/sdp
63  Fri Feb 6 23:04:00.089215291 2015  7  Content-Base: rtsp://192.168.1.161/
64  Fri Feb 6 23:04:00.089215291 2015  7  Content-Length: 439
65  Fri Feb 6 23:04:00.089215291 2015  7  --- LINE END ---
66  Fri Feb 6 23:04:00.089215291 2015  7  v=0
[... 258 lines cut ...]
325  Fri Feb 6 23:04:40.516829122 2015  7  Connection:Close
326  Fri Feb 6 23:04:40.516829122 2015  7  CSeq:1020
[... cut ...]
343  Fri Feb 6 23:04:40.517117192 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,1,0,10892,0,dvbs,8psk,2,0.35,27500,34;pids=
344  Fri Feb 6 23:04:40.517117192 2015  7  a=inactive
345  Fri Feb 6 23:04:40.517117192 2015  7  m=video 0 RTP/AVP 33
346  Fri Feb 6 23:04:40.517117192 2015  7  c=IN IP4 0.0.0.0
347  Fri Feb 6 23:04:40.517117192 2015  7  a=control:stream=1
348  Fri Feb 6 23:04:40.517117192 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=2,0,1,0,10744,0,dvbs,qpsk,2,0.35,22000,56;pids=0,16,17,18,19,20,400,401,402,403,404,407,408,470,2171,8190
349  Fri Feb 6 23:04:40.517117192 2015  7  a=sendonly

pbriesch, 2015-02-06:
I was now able to connect with EyeTV on my Mac mini and MacBookPro at the same time. One channel on ASTRA and one on Hotbird. Somehow it takes some starts and stops of EyeTV to be able to show a picture. But I think this is an other issue.
[image]
````
0  Fri Feb 6 23:35:03.709904716 2015  6  [rtsp.c:849] Client connection with sessionID 1123908057
1  Fri Feb 6 23:35:03.709925786 2015  6  [rtsp.c:854] Found Client with sessionID 1123908057
2  Fri Feb 6 23:35:03.709946493 2015  7  [rtsp.c:897] PLAY rtsp://192.168.1.161:554/stream=1?delpids=102&addpids=3658 RTSP/1.0
3  Fri Feb 6 23:35:03.709946493 2015  7  Connection:Close
4  Fri Feb 6 23:35:03.709946493 2015  7  CSeq:1099
5  Fri Feb 6 23:35:03.709946493 2015  7  Session:1123908057
6  Fri Feb 6 23:35:03.709946493 2015  7  --- LINE END ---
7  Fri Feb 6 23:35:03.710036339 2015  6  [rtsp.c:911] RTSP Client 192.168.1.111: Requested Connection closed with session ID: 1123908057
8  Fri Feb 6 23:35:03.710071645 2015  6  [rtsp.c:520] Play Message
9  Fri Feb 6 23:35:03.710081852 2015  7  [tune.c:516] Frontend: 1, Remove filter PID 102 - Packet Count: 1
10  Fri Feb 6 23:35:03.710132165 2015  7  [tune.c:512] Frontend: 1, Set filter PID 3658 - fd 77
11  Fri Feb 6 23:35:03.715265502 2015  6  [rtsp.c:878] Found empty client slot giving Session ID: 0178484062
12  Fri Feb 6 23:35:03.715283878 2015  7  [rtsp.c:897] DESCRIBE rtsp://192.168.1.161/ RTSP/1.0
13  Fri Feb 6 23:35:03.715283878 2015  7  Connection:Close
14  Fri Feb 6 23:35:03.715283878 2015  7  CSeq:1030
15  Fri Feb 6 23:35:03.715283878 2015  7  Accept:application/sdp
16  Fri Feb 6 23:35:03.715283878 2015  7  --- LINE END ---
17  Fri Feb 6 23:35:03.715381144 2015  6  [rtsp.c:911] RTSP Client 192.168.1.111: Requested Connection closed with session ID: 0178484062
18  Fri Feb 6 23:35:03.715505164 2015  7  [rtsp.c:646] RTSP/1.0 200 OK
19  Fri Feb 6 23:35:03.715505164 2015  7  CSeq: 1030
20  Fri Feb 6 23:35:03.715505164 2015  7  Content-Type: application/sdp
21  Fri Feb 6 23:35:03.715505164 2015  7  Content-Base: rtsp://192.168.1.161/
22  Fri Feb 6 23:35:03.715505164 2015  7  Content-Length: 482
23  Fri Feb 6 23:35:03.715505164 2015  7  --- LINE END ---
24  Fri Feb 6 23:35:03.715505164 2015  7  v=0
25  Fri Feb 6 23:35:03.715505164 2015  7  o=- 5678901234 7890123456 IN IP4 192.168.1.161
26  Fri Feb 6 23:35:03.715505164 2015  7  s=SatIPServer:1 2
27  Fri Feb 6 23:35:03.715505164 2015  7  t=0 0
28  Fri Feb 6 23:35:03.715505164 2015  7  m=video 0 RTP/AVP 33
29  Fri Feb 6 23:35:03.715505164 2015  7  c=IN IP4 0.0.0.0
30  Fri Feb 6 23:35:03.715505164 2015  7  a=control:stream=0
31  Fri Feb 6 23:35:03.715505164 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,1,0,10744,0,dvbs,qpsk,2,0.35,22000,56;pids=0,16,17,18,19,20,400,401,402,403,404,407,408,470,2171,8190
32  Fri Feb 6 23:35:03.715505164 2015  7  a=sendonly
33  Fri Feb 6 23:35:03.715505164 2015  7  m=video 0 RTP/AVP 33
34  Fri Feb 6 23:35:03.715505164 2015  7  c=IN IP4 0.0.0.0
35  Fri Feb 6 23:35:03.715505164 2015  7  a=control:stream=1
36  Fri Feb 6 23:35:03.715505164 2015  7  a=fmtp:33 ver=1.0;src=2;tuner=2,0,1,0,11541,1,dvbs,qpsk,2,0.35,22000,56;pids=0,16,17,18,19,20,300,301,302,3622,3658,8190
37  Fri Feb 6 23:35:03.715505164 2015  7  a=sendonly
38  Fri Feb 6 23:35:03.900285552 2015  7  [rtsp.c:969] RTSP Client 192.168.1.111: Connection closed with session ID: 1123908057 (fd: 76)
39  Fri Feb 6 23:35:03.905400959 2015  6  [connection.c:263] RTSP Connection from 192.168.1.111 Port 50862 (fd: 76)
40  Fri Feb 6 23:35:03.910484136 2015  6  [rtsp.c:849] Client connection with sessionID 1123908057
41  Fri Feb 6 23:35:03.910503423 2015  6  [rtsp.c:854] Found Client with sessionID 1123908057
42  Fri Feb 6 23:35:03.910516665 2015  7  [rtsp.c:897] PLAY rtsp://192.168.1.161:554/stream=1?delpids=3658 RTSP/1.0
43  Fri Feb 6 23:35:03.910516665 2015  7  Connection:Close
44  Fri Feb 6 23:35:03.910516665 2015  7  CSeq:1100
45  Fri Feb 6 23:35:03.910516665 2015  7  Session:1123908057
46  Fri Feb 6 23:35:03.910516665 2015  7  --- LINE END ---
47  Fri Feb 6 23:35:03.910666376 2015  6  [rtsp.c:911] RTSP Client 192.168.1.111: Requested Connection closed with session ID: 1123908057
48  Fri Feb 6 23:35:03.910698943 2015  6  [rtsp.c:520] Play Message
49  Fri Feb 6 23:35:03.910722385 2015  7  [tune.c:516] Frontend: 1, Remove filter PID 3658 - Packet Count: 1
50  Fri Feb 6 23:35:04.751192078 2015  7  [rtsp.c:969] RTSP Client 192.168.1.107: Connection closed with session ID: 2094709092 (fd: 79)
51  Fri Feb 6 23:35:04.756328720 2015  6  [connection.c:263] RTSP Connection from 192.168.1.107 Port 53042 (fd: 77)
52  Fri Feb 6 23:35:04.761410833 2015  6  [rtsp.c:878] Found empty client slot giving Session ID: 0222596815
53  Fri Feb 6 23:35:04.761429674 2015  7  [rtsp.c:897] DESCRIBE rtsp://192.168.1.161/ RTSP/1.0
54  Fri Feb 6 23:35:04.761429674 2015  7  Connection:Close
55  Fri Feb 6 23:35:04.761429674 2015  7  CSeq:1203
56  Fri Feb 6 23:35:04.761429674 2015  7  Accept:application/sdp
57  Fri Feb 6 23:35:04.761429674 2015  7  --- LINE END ---
58  Fri Feb 6 23:35:04.761533212 2015  6  [rtsp.c:911] RTSP Client 192.168.1.107: Requested Connection closed with session ID: 0222596815
59  Fri Feb 6 23:35:04.761646212 2015  7  [rtsp.c:646] RTSP/1.0 200 OK
60  Fri Feb 6 23:35:04.761646212 2015  7  CSeq: 1203
61  Fri Feb 6 23:35:04.761646212 2015  7  Content-Type: application/sdp
62  Fri Feb 6 23:35:04.761646212 2015  7  Content-Base: rtsp://192.168.1.161/
[... cut ...]
343  Fri Feb 6 23:35:15.908068786 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,1,0,10744,0,dvbs,qpsk,2,0.35,22000,56;pids=0,16,17,18,19,20,400,401,402,403,404,407,408,470,2171,8190
344  Fri Feb 6 23:35:15.908068786 2015  7  a=sendonly
345  Fri Feb 6 23:35:15.908068786 2015  7  m=video 0 RTP/AVP 33
346  Fri Feb 6 23:35:15.908068786 2015  7  c=IN IP4 0.0.0.0
347  Fri Feb 6 23:35:15.908068786 2015  7  a=control:stream=1
348  Fri Feb 6 23:35:15.908068786 2015  7  a=fmtp:33 ver=1.0;src=2;tuner=2,0,1,0,11541,1,dvbs,qpsk,2,0.35,22000,56;pids=0,16,17,18,19,20,300,301,302,3622,8190
349  Fri Feb 6 23:35:15.908068786 2015  7  a=sendonly

Barracuda09, 2015-02-06:
Hi,

Thanks for testing.. Yes there is an resource leak... So there is still some work to do  :-)

Marc

Barracuda09, 2015-02-08:
Hi,

I have fixed some resource leaks. So if you like and have some
time can you please give it another try?
Many thanks,

Marc

pbriesch, 2015-02-08:
Hello Marc,

briefly tested a little bit. When I have more time today I can do more. Client 1 EyteTV on Mac 1 and client 2 on Mac 2.

[image]

[image]

````
0  Sun Feb 8 12:52:04.305470447 2015  7  --- LINE END ---
1  Sun Feb 8 12:52:04.305631521 2015  6  [rtsp.c:917] RTSP Client 192.168.1.2: Requested Connection closed with fd: 28 and Session ID: 1111505545
2  Sun Feb 8 12:52:04.305664008 2015  6  [rtsp.c:520] Play Message
3  Sun Feb 8 12:52:04.305674794 2015  7  [tune.c:520] Frontend: 1, Remove filter PID: 336 - fd: 69 - Packet Count: 1
4  Sun Feb 8 12:52:04.305737947 2015  7  [tune.c:516] Frontend: 1, Set filter PID: 7931 - fd: 68
5  Sun Feb 8 12:52:04.495406883 2015  7  [rtsp.c:975] RTSP Client 192.168.1.2: Connection closed with fd: 28 and Session ID: 1111505545
6  Sun Feb 8 12:52:04.500530146 2015  6  [connection.c:263] RTSP Connection from 192.168.1.2 Port 62446 (fd: 28)
7  Sun Feb 8 12:52:04.505622112 2015  6  [rtsp.c:855] Client connection with sessionID 1111505545
8  Sun Feb 8 12:52:04.505642075 2015  6  [rtsp.c:860] RTSP Client 192.168.1.2: Found by sessionID 1111505545 with fd: 28
9  Sun Feb 8 12:52:04.505655659 2015  7  [rtsp.c:903] PLAY rtsp://192.168.1.161:554/stream=1?delpids=7931&addpids=3400 RTSP/1.0
10  Sun Feb 8 12:52:04.505655659 2015  7  Connection:Close
11  Sun Feb 8 12:52:04.505655659 2015  7  CSeq:1157
12  Sun Feb 8 12:52:04.505655659 2015  7  Session:1111505545
13  Sun Feb 8 12:52:04.505655659 2015  7  --- LINE END ---
14  Sun Feb 8 12:52:04.505819085 2015  6  [rtsp.c:917] RTSP Client 192.168.1.2: Requested Connection closed with fd: 28 and Session ID: 1111505545
15  Sun Feb 8 12:52:04.505852025 2015  6  [rtsp.c:520] Play Message
16  Sun Feb 8 12:52:04.505894182 2015  7  [tune.c:516] Frontend: 1, Set filter PID: 3400 - fd: 69
17  Sun Feb 8 12:52:04.505920014 2015  7  [tune.c:520] Frontend: 1, Remove filter PID: 7931 - fd: 68 - Packet Count: 2
18  Sun Feb 8 12:52:04.596447427 2015  7  [rtsp.c:975] RTSP Client 192.168.1.2: Connection closed with fd: 28 and Session ID: 1111505545
19  Sun Feb 8 12:52:04.601557348 2015  6  [connection.c:263] RTSP Connection from 192.168.1.2 Port 62447 (fd: 28)
20  Sun Feb 8 12:52:04.606650832 2015  6  [rtsp.c:855] Client connection with sessionID 1111505545
21  Sun Feb 8 12:52:04.606671375 2015  6  [rtsp.c:860] RTSP Client 192.168.1.2: Found by sessionID 1111505545 with fd: 28
22  Sun Feb 8 12:52:04.606685157 2015  7  [rtsp.c:903] PLAY rtsp://192.168.1.161:554/stream=1?delpids=3400&addpids=3500 RTSP/1.0
23  Sun Feb 8 12:52:04.606685157 2015  7  Connection:Close
24  Sun Feb 8 12:52:04.606685157 2015  7  CSeq:1158
25  Sun Feb 8 12:52:04.606685157 2015  7  Session:1111505545
26  Sun Feb 8 12:52:04.606685157 2015  7  --- LINE END ---
27  Sun Feb 8 12:52:04.606845898 2015  6  [rtsp.c:917] RTSP Client 192.168.1.2: Requested Connection closed with fd: 28 and Session ID: 1111505545
28  Sun Feb 8 12:52:04.606878778 2015  6  [rtsp.c:520] Play Message
29  Sun Feb 8 12:52:04.606901213 2015  7  [tune.c:520] Frontend: 1, Remove filter PID: 3400 - fd: 69 - Packet Count: 1
30  Sun Feb 8 12:52:04.606947388 2015  7  [tune.c:516] Frontend: 1, Set filter PID: 3500 - fd: 68
31  Sun Feb 8 12:52:04.695342752 2015  7  [rtsp.c:975] RTSP Client 192.168.1.2: Connection closed with fd: 28 and Session ID: 1111505545
32  Sun Feb 8 12:52:04.700475127 2015  6  [connection.c:263] RTSP Connection from 192.168.1.2 Port 62448 (fd: 28)
33  Sun Feb 8 12:52:04.705571075 2015  6  [rtsp.c:855] Client connection with sessionID 1111505545
34  Sun Feb 8 12:52:04.705592494 2015  6  [rtsp.c:860] RTSP Client 192.168.1.2: Found by sessionID 1111505545 with fd: 28
35  Sun Feb 8 12:52:04.705606090 2015  7  [rtsp.c:903] PLAY rtsp://192.168.1.161:554/stream=1?delpids=3500&addpids=3600 RTSP/1.0
36  Sun Feb 8 12:52:04.705606090 2015  7  Connection:Close
37  Sun Feb 8 12:52:04.705606090 2015  7  CSeq:1159
38  Sun Feb 8 12:52:04.705606090 2015  7  Session:1111505545
39  Sun Feb 8 12:52:04.705606090 2015  7  --- LINE END ---
40  Sun Feb 8 12:52:04.705768737 2015  6  [rtsp.c:917] RTSP Client 192.168.1.2: Requested Connection closed with fd: 28 and Session ID: 1111505545
41  Sun Feb 8 12:52:04.705802699 2015  6  [rtsp.c:520] Play Message
42  Sun Feb 8 12:52:04.705826433 2015  7  [tune.c:520] Frontend: 1, Remove filter PID: 3500 - fd: 68 - Packet Count: 1
43  Sun Feb 8 12:52:04.705881815 2015  7  [tune.c:516] Frontend: 1, Set filter PID: 3600 - fd: 68
44  Sun Feb 8 12:52:04.796826170 2015  7  [rtsp.c:975] RTSP Client 192.168.1.2: Connection closed with fd: 28 and Session ID: 1111505545
45  Sun Feb 8 12:52:04.801946702 2015  6  [connection.c:263] RTSP Connection from 192.168.1.2 Port 62449 (fd: 28)
46  Sun Feb 8 12:52:04.807037185 2015  6  [rtsp.c:855] Client connection with sessionID 1111505545
47  Sun Feb 8 12:52:04.807057477 2015  6  [rtsp.c:860] RTSP Client 192.168.1.2: Found by sessionID 1111505545 with fd: 28
48  Sun Feb 8 12:52:04.807071144 2015  7  [rtsp.c:903] PLAY rtsp://192.168.1.161:554/stream=1?delpids=3600&addpids=3700 RTSP/1.0
49  Sun Feb 8 12:52:04.807071144 2015  7  Connection:Close
50  Sun Feb 8 12:52:04.807071144 2015  7  CSeq:1160
51  Sun Feb 8 12:52:04.807071144 2015  7  Session:1111505545
52  Sun Feb 8 12:52:04.807071144 2015  7  --- LINE END ---
53  Sun Feb 8 12:52:04.807233244 2015  6  [rtsp.c:917] RTSP Client 192.168.1.2: Requested Connection closed with fd: 28 and Session ID: 1111505545
54  Sun Feb 8 12:52:04.807266326 2015  6  [rtsp.c:520] Play Message
[... cut ...]
343  Sun Feb 8 12:52:24.477776781 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,1,0,11494,0,dvbs2,8psk,2,0.35,22000,23;pids=0,16,17,18,19,20,1270,5110,5111,5112,5113,5114,5115,5116,5117,5118,5119,8190
344  Sun Feb 8 12:52:24.477776781 2015  7  a=inactive
345  Sun Feb 8 12:52:24.477776781 2015  7  m=video 0 RTP/AVP 33
346  Sun Feb 8 12:52:24.477776781 2015  7  c=IN IP4 0.0.0.0
347  Sun Feb 8 12:52:24.477776781 2015  7  a=control:stream=1
348  Sun Feb 8 12:52:24.477776781 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=2,0,1,0,12604,0,dvbs,qpsk,2,0.35,22000,56;pids=0,16,17,18,19,20,2171,2370,5601,5602,5603,5604,5605,5606,5610,8190
349  Sun Feb 8 12:52:24.477776781 2015  7  a=sendonly

pbriesch, 2015-02-08:
I tuned Mac 1 to Hotbird and left Mac 2 on Astra. After some minutes, the stream on Mac 1 stopped. I found following error message:
````
345  Sun Feb 8 13:05:00.302661100 2015  3  [tune.c:643] FE_READ_STATUS failed: Bad file descriptor (code 9)
````
[image]

[image]
````
0  Sun Feb 8 13:04:39.286212230 2015  7  o=- 5678901234 7890123456 IN IP4 192.168.1.161
1  Sun Feb 8 13:04:39.286212230 2015  7  s=SatIPServer:1 2
2  Sun Feb 8 13:04:39.286212230 2015  7  t=0 0
3  Sun Feb 8 13:04:39.286212230 2015  7  m=video 0 RTP/AVP 33
4  Sun Feb 8 13:04:39.286212230 2015  7  c=IN IP4 0.0.0.0
5  Sun Feb 8 13:04:39.286212230 2015  7  a=control:stream=0
6  Sun Feb 8 13:04:39.286212230 2015  7  a=fmtp:33 ver=1.0;src=2;tuner=1,0,1,0,10873,1,dvbs,qpsk,2,0.35,27500,34;pids=0,16,17,18,19,20,306,308,309,8190
7  Sun Feb 8 13:04:39.286212230 2015  7  a=sendonly
8  Sun Feb 8 13:04:39.286212230 2015  7  m=video 0 RTP/AVP 33
9  Sun Feb 8 13:04:39.286212230 2015  7  c=IN IP4 0.0.0.0
10  Sun Feb 8 13:04:39.286212230 2015  7  a=control:stream=1
11  Sun Feb 8 13:04:39.286212230 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=2,0,1,0,12604,0,dvbs,qpsk,2,0.35,22000,56;pids=0,16,17,18,19,20,2171,2370,5601,5602,5603,5604,5605,5606,5610,8190
12  Sun Feb 8 13:04:39.286212230 2015  7  a=sendonly
13  Sun Feb 8 13:04:39.487924944 2015  3  [tune.c:643] FE_READ_STATUS failed: Bad file descriptor (code 9)
14  Sun Feb 8 13:04:39.741795550 2015  7  [rtsp.c:975] RTSP Client 192.168.1.107: Connection closed with fd: 30 and Session ID: 0029780030
15  Sun Feb 8 13:04:39.746923756 2015  6  [connection.c:263] RTSP Connection from 192.168.1.107 Port 50183 (fd: 30)
16  Sun Feb 8 13:04:39.752008888 2015  6  [rtsp.c:872] RTSP Client 192.168.1.107: Found by fd: 30 with Session ID: 0029780030
17  Sun Feb 8 13:04:39.752028547 2015  7  [rtsp.c:903] DESCRIBE rtsp://192.168.1.161/ RTSP/1.0
18  Sun Feb 8 13:04:39.752028547 2015  7  Connection:Close
19  Sun Feb 8 13:04:39.752028547 2015  7  CSeq:1261
20  Sun Feb 8 13:04:39.752028547 2015  7  Accept:application/sdp
21  Sun Feb 8 13:04:39.752028547 2015  7  --- LINE END ---
22  Sun Feb 8 13:04:39.752089367 2015  6  [rtsp.c:917] RTSP Client 192.168.1.107: Requested Connection closed with fd: 30 and Session ID: 0029780030
23  Sun Feb 8 13:04:39.752204104 2015  7  [rtsp.c:646] RTSP/1.0 200 OK
24  Sun Feb 8 13:04:39.752204104 2015  7  CSeq: 1261
25  Sun Feb 8 13:04:39.752204104 2015  7  Content-Type: application/sdp
26  Sun Feb 8 13:04:39.752204104 2015  7  Content-Base: rtsp://192.168.1.161/
27  Sun Feb 8 13:04:39.752204104 2015  7  Content-Length: 480
28  Sun Feb 8 13:04:39.752204104 2015  7  --- LINE END ---
29  Sun Feb 8 13:04:39.752204104 2015  7  v=0
30  Sun Feb 8 13:04:39.752204104 2015  7  o=- 5678901234 7890123456 IN IP4 192.168.1.161
31  Sun Feb 8 13:04:39.752204104 2015  7  s=SatIPServer:1 2
32  Sun Feb 8 13:04:39.752204104 2015  7  t=0 0
33  Sun Feb 8 13:04:39.752204104 2015  7  m=video 0 RTP/AVP 33
34  Sun Feb 8 13:04:39.752204104 2015  7  c=IN IP4 0.0.0.0
35  Sun Feb 8 13:04:39.752204104 2015  7  a=control:stream=0
36  Sun Feb 8 13:04:39.752204104 2015  7  a=fmtp:33 ver=1.0;src=2;tuner=1,0,1,0,10873,1,dvbs,qpsk,2,0.35,27500,34;pids=0,16,17,18,19,20,306,308,309,8190
37  Sun Feb 8 13:04:39.752204104 2015  7  a=sendonly
38  Sun Feb 8 13:04:39.752204104 2015  7  m=video 0 RTP/AVP 33
39  Sun Feb 8 13:04:39.752204104 2015  7  c=IN IP4 0.0.0.0
40  Sun Feb 8 13:04:39.752204104 2015  7  a=control:stream=1
41  Sun Feb 8 13:04:39.752204104 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=2,0,1,0,12604,0,dvbs,qpsk,2,0.35,22000,56;pids=0,16,17,18,19,20,2171,2370,5601,5602,5603,5604,5605,5606,5610,8190
42  Sun Feb 8 13:04:39.752204104 2015  7  a=sendonly
43  Sun Feb 8 13:04:39.888180579 2015  3  [tune.c:643] FE_READ_STATUS failed: Bad file descriptor (code 9)
44  Sun Feb 8 13:04:40.288455655 2015  3  [tune.c:643] FE_READ_STATUS failed: Bad file descriptor (code 9)
45  Sun Feb 8 13:04:40.688732635 2015  3  [tune.c:643] FE_READ_STATUS failed: Bad file descriptor (code 9)
46  Sun Feb 8 13:04:41.089009127 2015  3  [tune.c:643] FE_READ_STATUS failed: Bad file descriptor (code 9)
47  Sun Feb 8 13:04:41.489299278 2015  3  [tune.c:643] FE_READ_STATUS failed: Bad file descriptor (code 9)
48  Sun Feb 8 13:04:41.889576048 2015  3  [tune.c:643] FE_READ_STATUS failed: Bad file descriptor (code 9)
49  Sun Feb 8 13:04:42.289883653 2015  3  [tune.c:643] FE_READ_STATUS failed: Bad file descriptor (code 9)
50  Sun Feb 8 13:04:42.690132789 2015  3  [tune.c:643] FE_READ_STATUS failed: Bad file descriptor (code 9)
51  Sun Feb 8 13:04:43.090411252 2015  3  [tune.c:643] FE_READ_STATUS failed: Bad file descriptor (code 9)
52  Sun Feb 8 13:04:43.490687111 2015  3  [tune.c:643] FE_READ_STATUS failed: Bad file descriptor (code 9)
53  Sun Feb 8 13:04:43.890975547 2015  3  [tune.c:643] FE_READ_STATUS failed: Bad file descriptor (code 9)
54  Sun Feb 8 13:04:44.291250257 2015  3  [tune.c:643] FE_READ_STATUS failed: Bad file descriptor (code 9)
55  Sun Feb 8 13:04:44.369672522 2015  7  [rtsp.c:975] RTSP Client 192.168.1.2: Connection closed with fd: 29 and Session ID: 1526250981
56  Sun Feb 8 13:04:44.374866128 2015  6  [connection.c:263] RTSP Connection from 192.168.1.2 Port 62707 (fd: 29)
57  Sun Feb 8 13:04:44.379952518 2015  6  [rtsp.c:872] RTSP Client 192.168.1.2: Found by fd: 29 with Session ID: 1526250981
58  Sun Feb 8 13:04:44.379972439 2015  7  [rtsp.c:903] DESCRIBE rtsp://192.168.1.161/ RTSP/1.0
59  Sun Feb 8 13:04:44.379972439 2015  7  Connection:Close
[... cut ...]
345  Sun Feb 8 13:05:00.302661100 2015  3  [tune.c:643] FE_READ_STATUS failed: Bad file descriptor (code 9)
346  Sun Feb 8 13:05:00.702962861 2015  3  [tune.c:643] FE_READ_STATUS failed: Bad file descriptor (code 9)
347  Sun Feb 8 13:05:01.103242530 2015  3  [tune.c:643] FE_READ_STATUS failed: Bad file descriptor (code 9)
348  Sun Feb 8 13:05:01.503531329 2015  3  [tune.c:643] FE_READ_STATUS failed: Bad file descriptor (code 9)
349  Sun Feb 8 13:05:01.903808013 2015  3  [tune.c:643] FE_READ_STATUS failed: Bad file descriptor (code 9)

pbriesch, 2015-02-08:
Now I have the situation in which Mac 1 shows a picture but Mac 2 not. Mac 2 is connected to SATPI, however, only one fronted is attached.

[image]

````
0  Sun Feb 8 13:37:17.370708124 2015  7  Content-Base: rtsp://192.168.1.161/
1  Sun Feb 8 13:37:17.370708124 2015  7  Content-Length: 454
2  Sun Feb 8 13:37:17.370708124 2015  7  --- LINE END ---
3  Sun Feb 8 13:37:17.370708124 2015  7  v=0
4  Sun Feb 8 13:37:17.370708124 2015  7  o=- 5678901234 7890123456 IN IP4 192.168.1.161
5  Sun Feb 8 13:37:17.370708124 2015  7  s=SatIPServer:1 2
6  Sun Feb 8 13:37:17.370708124 2015  7  t=0 0
7  Sun Feb 8 13:37:17.370708124 2015  7  m=video 0 RTP/AVP 33
8  Sun Feb 8 13:37:17.370708124 2015  7  c=IN IP4 0.0.0.0
9  Sun Feb 8 13:37:17.370708124 2015  7  a=control:stream=0
10  Sun Feb 8 13:37:17.370708124 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,1,0,11362,0,dvbs2,8psk,2,0.35,22000,23;pids=
11  Sun Feb 8 13:37:17.370708124 2015  7  a=inactive
12  Sun Feb 8 13:37:17.370708124 2015  7  m=video 0 RTP/AVP 33
13  Sun Feb 8 13:37:17.370708124 2015  7  c=IN IP4 0.0.0.0
14  Sun Feb 8 13:37:17.370708124 2015  7  a=control:stream=1
15  Sun Feb 8 13:37:17.370708124 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=2,0,1,0,11362,0,dvbs2,8psk,2,0.35,22000,23;pids=0,16,17,18,19,20,6100,6110,6120,6121,6122,6123,6130,6131,6132,6170,8190
16  Sun Feb 8 13:37:17.370708124 2015  7  a=inactive
17  Sun Feb 8 13:37:18.604695707 2015  7  [rtsp.c:975] RTSP Client 192.168.1.2: Connection closed with fd: 26 and Session ID: 0753412919
18  Sun Feb 8 13:37:18.609832624 2015  6  [connection.c:263] RTSP Connection from 192.168.1.2 Port 63401 (fd: 26)
19  Sun Feb 8 13:37:18.614924194 2015  6  [rtsp.c:872] RTSP Client 192.168.1.2: Found by fd: 26 with Session ID: 0753412919
20  Sun Feb 8 13:37:18.614943332 2015  7  [rtsp.c:903] DESCRIBE rtsp://192.168.1.161/ RTSP/1.0
21  Sun Feb 8 13:37:18.614943332 2015  7  Connection:Close
22  Sun Feb 8 13:37:18.614943332 2015  7  CSeq:1022
23  Sun Feb 8 13:37:18.614943332 2015  7  Accept:application/sdp
24  Sun Feb 8 13:37:18.614943332 2015  7  --- LINE END ---
25  Sun Feb 8 13:37:18.615067871 2015  6  [rtsp.c:917] RTSP Client 192.168.1.2: Requested Connection closed with fd: 26 and Session ID: 0753412919
26  Sun Feb 8 13:37:18.615185118 2015  7  [rtsp.c:646] RTSP/1.0 200 OK
27  Sun Feb 8 13:37:18.615185118 2015  7  CSeq: 1022
28  Sun Feb 8 13:37:18.615185118 2015  7  Content-Type: application/sdp
29  Sun Feb 8 13:37:18.615185118 2015  7  Content-Base: rtsp://192.168.1.161/
30  Sun Feb 8 13:37:18.615185118 2015  7  Content-Length: 454
31  Sun Feb 8 13:37:18.615185118 2015  7  --- LINE END ---
32  Sun Feb 8 13:37:18.615185118 2015  7  v=0
33  Sun Feb 8 13:37:18.615185118 2015  7  o=- 5678901234 7890123456 IN IP4 192.168.1.161
34  Sun Feb 8 13:37:18.615185118 2015  7  s=SatIPServer:1 2
35  Sun Feb 8 13:37:18.615185118 2015  7  t=0 0
36  Sun Feb 8 13:37:18.615185118 2015  7  m=video 0 RTP/AVP 33
37  Sun Feb 8 13:37:18.615185118 2015  7  c=IN IP4 0.0.0.0
38  Sun Feb 8 13:37:18.615185118 2015  7  a=control:stream=0
39  Sun Feb 8 13:37:18.615185118 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,1,0,11362,0,dvbs2,8psk,2,0.35,22000,23;pids=
40  Sun Feb 8 13:37:18.615185118 2015  7  a=inactive
41  Sun Feb 8 13:37:18.615185118 2015  7  m=video 0 RTP/AVP 33
42  Sun Feb 8 13:37:18.615185118 2015  7  c=IN IP4 0.0.0.0
43  Sun Feb 8 13:37:18.615185118 2015  7  a=control:stream=1
44  Sun Feb 8 13:37:18.615185118 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=2,0,1,0,11362,0,dvbs2,8psk,2,0.35,22000,23;pids=0,16,17,18,19,20,6100,6110,6120,6121,6122,6123,6130,6131,6132,6170,8190
45  Sun Feb 8 13:37:18.615185118 2015  7  a=inactive
46  Sun Feb 8 13:37:18.895934045 2015  7  [rtsp.c:975] RTSP Client 192.168.1.107: Connection closed with fd: 25 and Session ID: 0735120141
47  Sun Feb 8 13:37:18.901059022 2015  6  [connection.c:263] RTSP Connection from 192.168.1.107 Port 51244 (fd: 25)
48  Sun Feb 8 13:37:18.906149416 2015  6  [rtsp.c:872] RTSP Client 192.168.1.107: Found by fd: 25 with Session ID: 0735120141
49  Sun Feb 8 13:37:18.906168798 2015  7  [rtsp.c:903] DESCRIBE rtsp://192.168.1.161/ RTSP/1.0
50  Sun Feb 8 13:37:18.906168798 2015  7  Connection:Close
51  Sun Feb 8 13:37:18.906168798 2015  7  CSeq:1096
52  Sun Feb 8 13:37:18.906168798 2015  7  Accept:application/sdp
53  Sun Feb 8 13:37:18.906168798 2015  7  --- LINE END ---
54  Sun Feb 8 13:37:18.906267794 2015  6  [rtsp.c:917] RTSP Client 192.168.1.107: Requested Connection closed with fd: 25 and Session ID: 0735120141
55  Sun Feb 8 13:37:18.906381512 2015  7  [rtsp.c:646] RTSP/1.0 200 OK
56  Sun Feb 8 13:37:18.906381512 2015  7  CSeq: 1096
57  Sun Feb 8 13:37:18.906381512 2015  7  Content-Type: application/sdp
58  Sun Feb 8 13:37:18.906381512 2015  7  Content-Base: rtsp://192.168.1.161/
59  Sun Feb 8 13:37:18.906381512 2015  7  Content-Length: 454
60  Sun Feb 8 13:37:18.906381512 2015  7  --- LINE END ---
61  Sun Feb 8 13:37:18.906381512 2015  7  v=0
62  Sun Feb 8 13:37:18.906381512 2015  7  o=- 5678901234 7890123456 IN IP4 192.168.1.161
63  Sun Feb 8 13:37:18.906381512 2015  7  s=SatIPServer:1 2
64  Sun Feb 8 13:37:18.906381512 2015  7  t=0 0
[... 260 lines cut ...]
325  Sun Feb 8 13:37:34.135896642 2015  7  Connection:Close
326  Sun Feb 8 13:37:34.135896642 2015  7  CSeq:1099
327  Sun Feb 8 13:37:34.135896642 2015  7  Accept:application/sdp
328  Sun Feb 8 13:37:34.135896642 2015  7  --- LINE END ---
[... cut ...]
343  Sun Feb 8 13:37:34.136069140 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,1,0,11362,0,dvbs2,8psk,2,0.35,22000,23;pids=
344  Sun Feb 8 13:37:34.136069140 2015  7  a=inactive
345  Sun Feb 8 13:37:34.136069140 2015  7  m=video 0 RTP/AVP 33
346  Sun Feb 8 13:37:34.136069140 2015  7  c=IN IP4 0.0.0.0
347  Sun Feb 8 13:37:34.136069140 2015  7  a=control:stream=1
348  Sun Feb 8 13:37:34.136069140 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=2,0,1,0,11362,0,dvbs2,8psk,2,0.35,22000,23;pids=0,16,17,18,19,20,6100,6110,6120,6121,6122,6123,6130,6131,6132,6170,8190
349  Sun Feb 8 13:37:34.136069140 2015  7  a=inactive

pbriesch, 2015-02-08:
After restarting SATPI I could get a stream on both Macs. Both frontends were attached. But after about 10 s the stream on frontend 1 stopped.
[image]
````
0  Sun Feb 8 13:44:58.626715090 2015  7  Session: 0205744672
1  Sun Feb 8 13:44:58.626715090 2015  7  --- LINE END ---
2  Sun Feb 8 13:44:58.626715090 2015  7  v=0
3  Sun Feb 8 13:44:58.626715090 2015  7  o=- 5678901234 7890123456 IN IP4 192.168.1.161
4  Sun Feb 8 13:44:58.626715090 2015  7  s=SatIPServer:1 2
5  Sun Feb 8 13:44:58.626715090 2015  7  t=0 0
6  Sun Feb 8 13:44:58.626715090 2015  7  m=video 0 RTP/AVP 33
7  Sun Feb 8 13:44:58.626715090 2015  7  c=IN IP4 0.0.0.0
8  Sun Feb 8 13:44:58.626715090 2015  7  a=control:stream=0
9  Sun Feb 8 13:44:58.626715090 2015  7  a=fmtp:33 ver=1.0;src=2;tuner=1,0,1,0,11054,0,dvbs,qpsk,2,0.35,27500,56;pids=0,16,17,18,19,20,570,571,572,1033,8190
10  Sun Feb 8 13:44:58.626715090 2015  7  a=sendonly
11  Sun Feb 8 13:44:58.626715090 2015  7  m=video 0 RTP/AVP 33
12  Sun Feb 8 13:44:58.626715090 2015  7  c=IN IP4 0.0.0.0
13  Sun Feb 8 13:44:58.626715090 2015  7  a=control:stream=1
14  Sun Feb 8 13:44:58.626715090 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=2,0,1,0,11836,0,dvbs,qpsk,2,0.35,27500,34;pids=0,16,17,18,19,20,84,100,101,102,103,104,105,106,2070,2171,8190
15  Sun Feb 8 13:44:58.626715090 2015  7  a=sendonly
16  Sun Feb 8 13:44:59.204209312 2015  6  [connection.c:263] RTSP Connection from 192.168.1.2 Port 63595 (fd: 38)
17  Sun Feb 8 13:44:59.209311736 2015  6  [rtsp.c:855] Client connection with sessionID 0205744672
18  Sun Feb 8 13:44:59.209331009 2015  6  [rtsp.c:860] RTSP Client 192.168.1.2: Found by sessionID 0205744672 with fd: 38
19  Sun Feb 8 13:44:59.209343904 2015  7  [rtsp.c:903] OPTIONS rtsp://192.168.1.161/ RTSP/1.0
20  Sun Feb 8 13:44:59.209343904 2015  7  Connection:Close
21  Sun Feb 8 13:44:59.209343904 2015  7  CSeq:1120
22  Sun Feb 8 13:44:59.209343904 2015  7  Session:0205744672
23  Sun Feb 8 13:44:59.209343904 2015  7  --- LINE END ---
24  Sun Feb 8 13:44:59.209506862 2015  6  [rtsp.c:917] RTSP Client 192.168.1.2: Requested Connection closed with fd: 38 and Session ID: 0205744672
25  Sun Feb 8 13:45:02.349607907 2015  7  [rtsp.c:975] RTSP Client 192.168.1.107: Connection closed with fd: 25 and Session ID: 0203094728
26  Sun Feb 8 13:45:02.356536242 2015  6  [connection.c:263] RTSP Connection from 192.168.1.107 Port 51424 (fd: 25)
27  Sun Feb 8 13:45:02.361625490 2015  6  [rtsp.c:855] Client connection with sessionID 0945628226
28  Sun Feb 8 13:45:02.361645085 2015  6  [rtsp.c:860] RTSP Client 192.168.1.107: Found by sessionID 0945628226 with fd: 25
29  Sun Feb 8 13:45:02.361658392 2015  7  [rtsp.c:903] SETUP rtsp://192.168.1.161:554/stream=1?src=1&freq=11836&pol=h&ro=0.35&msys=dvbs&mtype=qpsk&plts=off&sr=27500&fec=34 RTSP/1.0
30  Sun Feb 8 13:45:02.361658392 2015  7  Connection:Close
31  Sun Feb 8 13:45:02.361658392 2015  7  CSeq:1115
32  Sun Feb 8 13:45:02.361658392 2015  7  Transport:RTP/AVP;unicast;client_port=5000-5001
33  Sun Feb 8 13:45:02.361658392 2015  7  Session:0945628226
34  Sun Feb 8 13:45:02.361658392 2015  7  --- LINE END ---
35  Sun Feb 8 13:45:02.361865081 2015  6  [rtsp.c:917] RTSP Client 192.168.1.107: Requested Connection closed with fd: 25 and Session ID: 0945628226
36  Sun Feb 8 13:45:02.361905530 2015  6  [rtsp.c:455] Setup Message
37  Sun Feb 8 13:45:02.361914886 2015  7  [tune.c:360] Frontend: 1, Start tuning process
38  Sun Feb 8 13:45:02.361922136 2015  6  [tune.c:289] Frontend: 1, Sending DiSEqC [e0] [10] [38] [f3]
39  Sun Feb 8 13:45:02.571355541 2015  6  [tune.c:547] Frontend: 1, Waiting on lock.
40  Sun Feb 8 13:45:02.900889857 2015  6  [tune.c:558] Frontend: 1, Tuned and locked (FE status 31).
41  Sun Feb 8 13:45:02.947550725 2015  7  [rtsp.c:975] RTSP Client 192.168.1.107: Connection closed with fd: 25 and Session ID: 0203094728
42  Sun Feb 8 13:45:02.952670624 2015  6  [connection.c:263] RTSP Connection from 192.168.1.107 Port 51425 (fd: 25)
43  Sun Feb 8 13:45:02.957759449 2015  6  [rtsp.c:855] Client connection with sessionID 0945628226
44  Sun Feb 8 13:45:02.957779407 2015  6  [rtsp.c:860] RTSP Client 192.168.1.107: Found by sessionID 0945628226 with fd: 25
45  Sun Feb 8 13:45:02.957792883 2015  7  [rtsp.c:903] PLAY rtsp://192.168.1.161:554/stream=1?pids=0,16,17,18,19,20,84,100,101,102,103,104,105,106,2070,2171,8190 RTSP/1.0
46  Sun Feb 8 13:45:02.957792883 2015  7  Connection:Close
47  Sun Feb 8 13:45:02.957792883 2015  7  CSeq:1116
48  Sun Feb 8 13:45:02.957792883 2015  7  Session:0945628226
49  Sun Feb 8 13:45:02.957792883 2015  7  --- LINE END ---
50  Sun Feb 8 13:45:02.957850890 2015  6  [rtsp.c:917] RTSP Client 192.168.1.107: Requested Connection closed with fd: 25 and Session ID: 0945628226
51  Sun Feb 8 13:45:02.957888163 2015  6  [rtsp.c:520] Play Message
52  Sun Feb 8 13:45:03.643749775 2015  6  [connection.c:263] RTSP Connection from 192.168.1.2 Port 63596 (fd: 27)
53  Sun Feb 8 13:45:03.648858337 2015  6  [rtsp.c:884] RTSP Client 192.168.1.2: Found empty slot with fd: 27 giving Session ID: 0525888344
54  Sun Feb 8 13:45:03.648877495 2015  7  [rtsp.c:903] DESCRIBE rtsp://192.168.1.161/ RTSP/1.0
55  Sun Feb 8 13:45:03.648877495 2015  7  Connection:Close
56  Sun Feb 8 13:45:03.648877495 2015  7  CSeq:1113
57  Sun Feb 8 13:45:03.648877495 2015  7  Accept:application/sdp
58  Sun Feb 8 13:45:03.648877495 2015  7  --- LINE END ---
[... cut ...]
343  Sun Feb 8 13:45:31.975338461 2015  7  a=fmtp:33 ver=1.0;src=2;tuner=1,0,1,0,11054,0,dvbs,qpsk,2,0.35,27500,56;pids=0,16,17,18,19,20,570,571,572,1033,8190
344  Sun Feb 8 13:45:31.975338461 2015  7  a=sendonly
345  Sun Feb 8 13:45:31.975338461 2015  7  m=video 0 RTP/AVP 33
346  Sun Feb 8 13:45:31.975338461 2015  7  c=IN IP4 0.0.0.0
347  Sun Feb 8 13:45:31.975338461 2015  7  a=control:stream=1
348  Sun Feb 8 13:45:31.975338461 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=2,0,1,0,11836,0,dvbs,qpsk,2,0.35,27500,34;pids=0,16,17,18,19,20,84,100,101,102,103,104,105,106,2070,2171,8190
349  Sun Feb 8 13:45:31.975338461 2015  7  a=sendonly

Barracuda09, 2015-02-08:
Hi,

Many thanks for testing again... I see that there is are a lot of connections that do nothing..
but they will clear out?.. Probably I have to change this...
I have here Elgato EyeTV for android and DD-TV together running for some time now..

Thanks again,

Marc

pbriesch, 2015-02-08:
Currently I have two streams open at the same time on my Mac with EyeTV and it works fine. At the start of EyeTV it took a bit time until both frontends were attached. But since then it runs. Only 2 open connections as it should be. I didn't change the channels during the whole time. Maybe switching channels leaves connections open?
[image]
````
0  Sun Feb 8 16:33:46.742229766 2015  6  [rtsp.c:520] Play Message
1  Sun Feb 8 16:33:46.742256901 2015  7  [tune.c:516] Frontend: 1, Set filter PID: 411 - fd: 68
2  Sun Feb 8 16:33:46.742268395 2015  7  [tune.c:520] Frontend: 1, Remove filter PID: 1100 - fd: 66 - Packet Count: 1
3  Sun Feb 8 16:33:46.832428058 2015  7  [rtsp.c:975] RTSP Client 192.168.1.107: Connection closed with fd: 42 and Session ID: 0288582906
4  Sun Feb 8 16:33:46.840576218 2015  6  [connection.c:263] RTSP Connection from 192.168.1.107 Port 56358 (fd: 42)
5  Sun Feb 8 16:33:46.845655623 2015  6  [rtsp.c:855] Client connection with sessionID 0288582906
6  Sun Feb 8 16:33:46.845674047 2015  6  [rtsp.c:860] RTSP Client 192.168.1.107: Found by sessionID 0288582906 with fd: 42
7  Sun Feb 8 16:33:46.845687178 2015  7  [rtsp.c:903] PLAY rtsp://192.168.1.161:554/stream=1?delpids=411 RTSP/1.0
8  Sun Feb 8 16:33:46.845687178 2015  7  Connection:Close
9  Sun Feb 8 16:33:46.845687178 2015  7  CSeq:1284
10  Sun Feb 8 16:33:46.845687178 2015  7  Session:0288582906
11  Sun Feb 8 16:33:46.845687178 2015  7  --- LINE END ---
12  Sun Feb 8 16:33:46.845837484 2015  6  [rtsp.c:917] RTSP Client 192.168.1.107: Requested Connection closed with fd: 42 and Session ID: 0288582906
13  Sun Feb 8 16:33:46.845869795 2015  6  [rtsp.c:520] Play Message
14  Sun Feb 8 16:33:46.845880899 2015  7  [tune.c:520] Frontend: 1, Remove filter PID: 411 - fd: 68 - Packet Count: 1
15  Sun Feb 8 16:33:59.735429586 2015  7  [rtsp.c:975] RTSP Client 192.168.1.107: Connection closed with fd: 25 and Session ID: 0205744672
16  Sun Feb 8 16:33:59.740559040 2015  6  [connection.c:263] RTSP Connection from 192.168.1.107 Port 56359 (fd: 25)
17  Sun Feb 8 16:33:59.745648493 2015  6  [rtsp.c:855] Client connection with sessionID 0205744672
18  Sun Feb 8 16:33:59.745667452 2015  6  [rtsp.c:860] RTSP Client 192.168.1.107: Found by sessionID 0205744672 with fd: 25
19  Sun Feb 8 16:33:59.745680673 2015  7  [rtsp.c:903] OPTIONS rtsp://192.168.1.161/ RTSP/1.0
20  Sun Feb 8 16:33:59.745680673 2015  7  Connection:Close
21  Sun Feb 8 16:33:59.745680673 2015  7  CSeq:1042
22  Sun Feb 8 16:33:59.745680673 2015  7  Session:0205744672
23  Sun Feb 8 16:33:59.745680673 2015  7  --- LINE END ---
24  Sun Feb 8 16:33:59.745828725 2015  6  [rtsp.c:917] RTSP Client 192.168.1.107: Requested Connection closed with fd: 25 and Session ID: 0205744672
25  Sun Feb 8 16:34:00.177697929 2015  7  [http.c:245] GET web/log.html (size 1784)
26  Sun Feb 8 16:34:00.261997277 2015  7  [http.c:245] GET web/stylesheet.css (size 1463)
27  Sun Feb 8 16:34:00.265074688 2015  7  [http.c:245] GET web/loadxmldoc.js (size 521)
28  Sun Feb 8 16:34:00.338560431 2015  7  [http.c:245] GET web/grid_noise.png (size 4638)
29  Sun Feb 8 16:34:00.343266126 2015  7  [http.c:245] GET web/bg-grain.png (size 5636)
30  Sun Feb 8 16:34:18.528560915 2015  7  [http.c:245] GET web/index.html (size 4201)
31  Sun Feb 8 16:34:18.642989806 2015  7  [http.c:245] GET web/stylesheet.css (size 1463)
32  Sun Feb 8 16:34:18.643081138 2015  7  [http.c:245] GET web/loadxmldoc.js (size 521)
33  Sun Feb 8 16:34:18.707980614 2015  7  [http.c:245] GET web/grid_noise.png (size 4638)
34  Sun Feb 8 16:34:21.606553904 2015  7  [http.c:245] GET web/frontend.html (size 4963)
35  Sun Feb 8 16:34:21.689544572 2015  7  [http.c:245] GET web/stylesheet.css (size 1463)
36  Sun Feb 8 16:34:21.690577099 2015  7  [http.c:245] GET web/loadxmldoc.js (size 521)
37  Sun Feb 8 16:34:21.693346336 2015  7  [http.c:245] GET web/postxmldoc.js (size 739)
38  Sun Feb 8 16:34:21.748112009 2015  7  [http.c:245] GET web/grid_noise.png (size 4638)
39  Sun Feb 8 16:34:35.587233767 2015  7  [rtsp.c:975] RTSP Client 192.168.1.107: Connection closed with fd: 42 and Session ID: 0288582906
40  Sun Feb 8 16:34:35.592366347 2015  6  [connection.c:263] RTSP Connection from 192.168.1.107 Port 56368 (fd: 42)
41  Sun Feb 8 16:34:35.597457794 2015  6  [rtsp.c:855] Client connection with sessionID 0288582906
42  Sun Feb 8 16:34:35.597476726 2015  6  [rtsp.c:860] RTSP Client 192.168.1.107: Found by sessionID 0288582906 with fd: 42
43  Sun Feb 8 16:34:35.597489587 2015  7  [rtsp.c:903] OPTIONS rtsp://192.168.1.161/ RTSP/1.0
44  Sun Feb 8 16:34:35.597489587 2015  7  Connection:Close
45  Sun Feb 8 16:34:35.597489587 2015  7  CSeq:1285
46  Sun Feb 8 16:34:35.597489587 2015  7  Session:0288582906
47  Sun Feb 8 16:34:35.597489587 2015  7  --- LINE END ---
48  Sun Feb 8 16:34:35.597656092 2015  6  [rtsp.c:917] RTSP Client 192.168.1.107: Requested Connection closed with fd: 42 and Session ID: 0288582906
49  Sun Feb 8 16:34:49.805810846 2015  7  [rtsp.c:975] RTSP Client 192.168.1.107: Connection closed with fd: 25 and Session ID: 0205744672
50  Sun Feb 8 16:34:49.810937666 2015  6  [connection.c:263] RTSP Connection from 192.168.1.107 Port 56369 (fd: 25)
51  Sun Feb 8 16:34:49.816027423 2015  6  [rtsp.c:855] Client connection with sessionID 0205744672
52  Sun Feb 8 16:34:49.816046367 2015  6  [rtsp.c:860] RTSP Client 192.168.1.107: Found by sessionID 0205744672 with fd: 25
53  Sun Feb 8 16:34:49.816059588 2015  7  [rtsp.c:903] OPTIONS rtsp://192.168.1.161/ RTSP/1.0
[... cut ...]
344  Sun Feb 8 16:44:50.596198516 2015  6  [rtsp.c:917] RTSP Client 192.168.1.107: Requested Connection closed with fd: 25 and Session ID: 0205744672
345  Sun Feb 8 16:44:54.045349149 2015  7  [http.c:245] GET web/log.html (size 1784)
346  Sun Feb 8 16:44:54.134527542 2015  7  [http.c:245] GET web/loadxmldoc.js (size 521)
347  Sun Feb 8 16:44:54.134610152 2015  7  [http.c:245] GET web/stylesheet.css (size 1463)
348  Sun Feb 8 16:44:54.203172808 2015  7  [http.c:245] GET web/bg-grain.png (size 5636)
349  Sun Feb 8 16:44:54.203265523 2015  7  [http.c:245] GET web/grid_noise.png (size 4638)

Barracuda09, 2015-02-08:
Hi,

The connections should timeout when not used anymore. So it should clear out.
In the begin Eyetv is grabing EPG then it busy doing that.
But I will change these extra connections.

Marc

Barracuda09, 2015-04-10:
Should be fixed in the mean time

closed 2015-04-10 by Barracuda09