Barracuda09/SATPI #19: Frontend Type: Unknown 4

pbriesch, 2015-02-03:
Hi Marc,

it is probably just cosmetics: *Frontend Type: Unknown 4*. Both DVB S2 cards have a connector for an optional CI. Could it be that for this reason *Unknown* is logged? I don't have the CI's hooked up.

Paul
````
0  Tue Feb 3 22:54:02.280248376 2015  6  [satpi.c:235] --- starting ---
1  Tue Feb 3 22:54:02.280335971 2015  6  [tune.c:393] Detecting frontends in: /dev/dvb
2  Tue Feb 3 22:54:02.280447503 2015  6  [tune.c:403] Frontends found: 2
3  Tue Feb 3 22:54:02.280464738 2015  7  [tune.c:468] Open FE fd: 4 readonly 1
4  Tue Feb 3 22:54:02.280493619 2015  6  [tune.c:117] Frontend Name: STB0899 Multistandard
5  Tue Feb 3 22:54:02.280505154 2015  7  [tune.c:142] Frontend Type: Satellite (DVB-S)
6  Tue Feb 3 22:54:02.280523652 2015  7  [tune.c:146] Frontend Type: Satellite (DVB-S2)
7  Tue Feb 3 22:54:02.280543041 2015  7  [tune.c:172] Frontend Type: Unknown 4
8  Tue Feb 3 22:54:02.280561884 2015  7  [tune.c:177] Frontend Freq: 950000 Hz to 2150000 Hz
9  Tue Feb 3 22:54:02.280580674 2015  7  [tune.c:178] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
10  Tue Feb 3 22:54:02.280605131 2015  7  [tune.c:468] Open FE fd: 4 readonly 1
11  Tue Feb 3 22:54:02.280615030 2015  6  [tune.c:117] Frontend Name: STB0899 Multistandard
12  Tue Feb 3 22:54:02.280622285 2015  7  [tune.c:142] Frontend Type: Satellite (DVB-S)
13  Tue Feb 3 22:54:02.280638962 2015  7  [tune.c:146] Frontend Type: Satellite (DVB-S2)
14  Tue Feb 3 22:54:02.280655254 2015  7  [tune.c:172] Frontend Type: Unknown 4
15  Tue Feb 3 22:54:02.280673509 2015  7  [tune.c:177] Frontend Freq: 950000 Hz to 2150000 Hz
16  Tue Feb 3 22:54:02.280694910 2015  7  [tune.c:178] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
17  Tue Feb 3 22:54:02.281060249 2015  6  [connection.c:334] eth0: 192.168.1.161 [00:1d:7d:a9:eb:f3]
18  Tue Feb 3 22:54:02.281070803 2015  6  [rtsp.c:960] Setting up RTSP server
19  Tue Feb 3 22:54:02.281125080 2015  6  [http.c:483] Setting up HTTP server
20  Tue Feb 3 22:54:02.281160443 2015  6  [rtp.c:348] Setting up 8 RTP servers
21  Tue Feb 3 22:54:02.281366670 2015  6  [rtcp.c:338] Setting up 8 RTCP servers
22  Tue Feb 3 22:54:02.281583497 2015  6  [ssdp.c:170] Setting up SSDP server

Barracuda09, 2015-02-03:
No it is DSS frontend type. I will add this. Thanks for pointing this out.

Barracuda09, 2015-02-06:
Hi,

I have added this frontend type string.

Marc

pbriesch, 2015-02-07:
Hi Marc,

it works now.

````
0  Sat Feb  7 10:30:32.848554315 2015  6  [satpi.c:235] --- starting ---
1  Sat Feb  7 10:30:32.848636704 2015  6  [tune.c:401] Detecting frontends in: /dev/dvb
2  Sat Feb  7 10:30:32.848747310 2015  6  [tune.c:411] Frontends found: 2
3  Sat Feb  7 10:30:32.848764857 2015  7  [tune.c:476] Open FE fd: 4 readonly 1
4  Sat Feb  7 10:30:32.848793684 2015  6  [tune.c:117] Frontend Name: STB0899 Multistandard
5  Sat Feb  7 10:30:32.848816104 2015  7  [tune.c:146] Frontend Type: Satellite (DVB-S)
6  Sat Feb  7 10:30:32.848835261 2015  7  [tune.c:150] Frontend Type: Satellite (DVB-S2)
7  Sat Feb  7 10:30:32.848853928 2015  7  [tune.c:142] Frontend Type: DSS
8  Sat Feb  7 10:30:32.848872710 2015  7  [tune.c:181] Frontend Freq: 950000 Hz to 2150000 Hz
9  Sat Feb  7 10:30:32.848891279 2015  7  [tune.c:182] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
10  Sat Feb  7 10:30:32.848916553 2015  7  [tune.c:476] Open FE fd: 4 readonly 1
11  Sat Feb  7 10:30:32.848935201 2015  6  [tune.c:117] Frontend Name: STB0899 Multistandard
12  Sat Feb  7 10:30:32.848954751 2015  7  [tune.c:146] Frontend Type: Satellite (DVB-S)
13  Sat Feb  7 10:30:32.848972931 2015  7  [tune.c:150] Frontend Type: Satellite (DVB-S2)
14  Sat Feb  7 10:30:32.848990904 2015  7  [tune.c:142] Frontend Type: DSS
15  Sat Feb  7 10:30:32.849008886 2015  7  [tune.c:181] Frontend Freq: 950000 Hz to 2150000 Hz
16  Sat Feb  7 10:30:32.849030005 2015  7  [tune.c:182] Frontend srat: 5000000 symbols/s to 45000000 symbols/s

closed 2015-02-07 by pbriesch