Barracuda09/SATPI #20: UNKNOWN DELIVERY SYSTEM

pbriesch, 2015-02-08:
I have added a DVB-T USB stick to the server so that there are now 3 adapters. SATPI does't recognize the delivery of the DVB-T USB stick and its symbol rate is shown as 0.
Please note that I have not tested if there is DVB-T reception possible at all due to the position of the aerial.

[image]
````
0  Sun Feb 8 15:49:12.570603714 2015  6  [satpi.c:235] --- starting ---
1  Sun Feb 8 15:49:12.570690874 2015  6  [tune.c:401] Detecting frontends in: /dev/dvb
2  Sun Feb 8 15:49:12.570832361 2015  6  [tune.c:411] Frontends found: 3
3  Sun Feb 8 15:49:12.570849308 2015  7  [tune.c:476] Open FE fd: 4 readonly 1
4  Sun Feb 8 15:49:12.570875630 2015  6  [tune.c:117] Frontend Name: STB0899 Multistandard
5  Sun Feb 8 15:49:12.570900836 2015  7  [tune.c:146] Frontend Type: Satellite (DVB-S)
6  Sun Feb 8 15:49:12.570920173 2015  7  [tune.c:150] Frontend Type: Satellite (DVB-S2)
7  Sun Feb 8 15:49:12.570938952 2015  7  [tune.c:142] Frontend Type: DSS
8  Sun Feb 8 15:49:12.570957843 2015  7  [tune.c:181] Frontend Freq: 950000 Hz to 2150000 Hz
9  Sun Feb 8 15:49:12.570976319 2015  7  [tune.c:182] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
10  Sun Feb 8 15:49:12.571001079 2015  7  [tune.c:476] Open FE fd: 4 readonly 1
11  Sun Feb 8 15:49:12.571011120 2015  6  [tune.c:117] Frontend Name: STB0899 Multistandard
12  Sun Feb 8 15:49:12.571018176 2015  7  [tune.c:146] Frontend Type: Satellite (DVB-S)
13  Sun Feb 8 15:49:12.571035049 2015  7  [tune.c:150] Frontend Type: Satellite (DVB-S2)
14  Sun Feb 8 15:49:12.571053584 2015  7  [tune.c:142] Frontend Type: DSS
15  Sun Feb 8 15:49:12.571075105 2015  7  [tune.c:181] Frontend Freq: 950000 Hz to 2150000 Hz
16  Sun Feb 8 15:49:12.571093663 2015  7  [tune.c:182] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
17  Sun Feb 8 15:49:12.571117000 2015  7  [tune.c:476] Open FE fd: 4 readonly 1
18  Sun Feb 8 15:49:12.571126910 2015  6  [tune.c:117] Frontend Name: Afatech AF9013
19  Sun Feb 8 15:49:12.571508623 2015  7  [tune.c:154] Frontend Type: Terrestrial (DVB-T)
20  Sun Feb 8 15:49:12.571525046 2015  7  [tune.c:181] Frontend Freq: 174000000 Hz to 862000000 Hz
21  Sun Feb 8 15:49:12.571536334 2015  7  [tune.c:182] Frontend srat: 0 symbols/s to 0 symbols/s
22  Sun Feb 8 15:49:12.572040568 2015  6  [connection.c:336] eth0: 192.168.1.161 [00:1d:7d:a9:eb:f3]
23  Sun Feb 8 15:49:12.572050954 2015  6  [rtsp.c:1017] Setting up RTSP server
24  Sun Feb 8 15:49:12.572105863 2015  6  [http.c:490] Setting up HTTP server
25  Sun Feb 8 15:49:12.572141421 2015  6  [rtp.c:348] Setting up 8 RTP servers
26  Sun Feb 8 15:49:12.572351986 2015  6  [rtcp.c:338] Setting up 8 RTCP servers
27  Sun Feb 8 15:49:12.572571910 2015  6  [ssdp.c:170] Setting up SSDP server

Barracuda09, 2015-02-08:
Hi,

The unknown delivery system is just that there has been no setup
Message that set this.. it should maybe set as default as the dvbs.

Thanks for noticing this.. The symbol rate I don't know why.

I will look at this thanks,

Marc

pbriesch, 2015-02-08:
OK. I'm trying to test the DVB-T adapter wit VDR. EyeTV only allows SAT>IP, as the name suggests, for satellite connections.

Barracuda09, 2015-02-08:
I used DD-tv for dvb-t the new version can do also dvb-c

FuNK3Y, 2015-02-12:
Does your previous comment means that satpi now support DVB-C? If yes I would be happy to give it a try

Barracuda09, 2015-02-12:
Hi FuNK3Y,

It should work, but I have not tested it very well.
Because I do not have it myself. 
But I appreciate your time if you would give it a try.

Thanks,

Marc

FuNK3Y, 2015-02-16:
I did finally test it, and you can add DVB-C as supported. It worked perfectly.

I wanted to try DVB-T as well, since my tuner is hybrid but this did not work (I did disable DVB-C first)

328  Mon Feb 16 20:09:18.120804950 2015  6  Play Message
329  Mon Feb 16 20:09:18.120819012 2015  6  Frontend: 0, Waiting on lock.
330  Mon Feb 16 20:09:18.120826382 2015  6  Frontend: 0, Not locked yet (FE status 0)...
331  Mon Feb 16 20:09:18.172449771 2015  6  Frontend: 0, Not locked yet (FE status 0)...
332  Mon Feb 16 20:09:18.202558973 2015  6  Frontend: 0, Not locked yet (FE status 0)...
333  Mon Feb 16 20:09:18.232548077 2015  6  Frontend: 0, Not locked yet (FE status 0)...
334  Mon Feb 16 20:09:22.981121963 2015  6  Client connection with sessionID 0205744672

Is SATPI able to handle multiple tuners?

Barracuda09, 2015-02-16:
Hi,

Mmm... DVB-T should have worked (I have used an hybrid sundtek tuner at some point)
What is the setup message?

Marc

FuNK3Y, 2015-02-16:
Is this what you need:
256  Mon Feb 16 20:10:18.096666292 2015  6  Frontend: 0, Waiting on lock.
257  Mon Feb 16 20:10:18.096679047 2015  6  Frontend: 0, Not locked yet (FE status 0)...
258  Mon Feb 16 20:10:18.150511471 2015  6  Frontend: 0, Not locked yet (FE status 0)...
259  Mon Feb 16 20:10:18.179989242 2015  6  Frontend: 0, Not locked yet (FE status 0)...
260  Mon Feb 16 20:10:18.209603533 2015  6  Frontend: 0, Not locked yet (FE status 0)...
261  Mon Feb 16 20:10:22.999759354 2015  6  Client connection with sessionID 0205744672
262  Mon Feb 16 20:10:22.999834728 2015  6  RTSP Client 192.168.1.20: Found by sessionID 0205744672 with fd: 30
263  Mon Feb 16 20:10:22.999853359 2015  6  Frontend: 0, Teardown Message, gracefull: 1
264  Mon Feb 16 20:10:23.045718832 2015  6  Frontend: 0, Stop streaming to 192.168.1.20 (53208 - 53209) (Streamed 0.000 MBytes)
265  Mon Feb 16 20:10:23.189060976 2015  6  Frontend: 0, Detached from client 192.168.1.20 with session ID: 0205744672
266  Mon Feb 16 20:10:23.949744853 2015  6  RTSP Connection from 192.168.1.20 Port 34162 (fd: 27)
267  Mon Feb 16 20:10:23.955027558 2015  6  RTSP Client 192.168.1.20: Found empty slot for fd: 27
268  Mon Feb 16 20:10:23.955086107 2015  6  RTSP Client 192.168.1.20: With fd: 27 giving Session ID: 0288582906
269  Mon Feb 16 20:10:23.955100117 2015  6  Frontend: 0, With dvbt Attaching to client 192.168.1.20 as requested with session ID: 0288582906
270  Mon Feb 16 20:10:23.955164522 2015  6  Setup Message
271  Mon Feb 16 20:10:23.955454537 2015  6  Frontend: 0, Opened FE fd: 28.
272  Mon Feb 16 20:10:23.956830894 2015  6  Frontend: 0, Waiting on lock.
273  Mon Feb 16 20:10:24.008751130 2015  6  Frontend: 0, Not locked yet (FE status 0)...
274  Mon Feb 16 20:10:24.038107796 2015  6  Frontend: 0, Not locked yet (FE status 0)...
275  Mon Feb 16 20:10:24.067466208 2015  6  Frontend: 0, Not locked yet (FE status 0)...
276  Mon Feb 16 20:10:24.097075964 2015  6  Frontend: 0, Not locked yet (FE status 0)...
277  Mon Feb 16 20:10:24.122434233 2015  6  Frontend: 0, Opened DVR fd: 29.
278  Mon Feb 16 20:10:24.122541759 2015  6  Frontend: 0, Start streaming to 192.168.1.20 (57872 - 57873)
279  Mon Feb 16 20:10:24.127688332 2015  6  Client connection with sessionID 0288582906
280  Mon Feb 16 20:10:24.127724300 2015  6  RTSP Client 192.168.1.20: Found by sessionID 0288582906 with fd: 27
281  Mon Feb 16 20:10:24.127754304 2015  6  Play Message
282  Mon Feb 16 20:10:24.127775980 2015  6  Frontend: 0, Waiting on lock.
283  Mon Feb 16 20:10:24.127788453 2015  6  Frontend: 0, Not locked yet (FE status 0)...
284  Mon Feb 16 20:10:24.180665888 2015  6  Frontend: 0, Not locked yet (FE status 0)...
285  Mon Feb 16 20:10:24.210136048 2015  6  Frontend: 0, Not locked yet (FE status 0)...
286  Mon Feb 16 20:10:24.239757999 2015  6  Frontend: 0, Not locked yet (FE status 0)...
287  Mon Feb 16 20:10:24.266620390 2015  6  Frontend: 0, Start RTCP stream to 192.168.1.20 (57872 - 57873)
288  Mon Feb 16 20:10:24.266652670 2015  6  Frontend: 0, Open Frontend Monitor for 192.168.1.20 (57872 - 57873)
289  Mon Feb 16 20:10:48.986517149 2015  6  Client connection with sessionID 0288582906
290  Mon Feb 16 20:10:48.986636588 2015  6  RTSP Client 192.168.1.20: Found by sessionID 0288582906 with fd: 27
291  Mon Feb 16 20:10:48.986655625 2015  6  Frontend: 0, Teardown Message, gracefull: 1
292  Mon Feb 16 20:10:49.002323199 2015  6  Frontend: 0, Stop streaming to 192.168.1.20 (57872 - 57873) (Streamed 0.000 MBytes)
293  Mon Feb 16 20:10:49.041506483 2015  6  Frontend: 0, Detached from client 192.168.1.20 with session ID: 0288582906
294  Mon Feb 16 20:10:49.939043055 2015  6  RTSP Connection from 192.168.1.20 Port 34163 (fd: 27)
295  Mon Feb 16 20:10:49.944287398 2015  6  RTSP Client 192.168.1.20: Found empty slot for fd: 27
296  Mon Feb 16 20:10:49.944329004 2015  6  RTSP Client 192.168.1.20: With fd: 27 giving Session ID: 0671693627
297  Mon Feb 16 20:10:49.944351487 2015  6  Frontend: 0, With dvbt Attaching to client 192.168.1.20 as requested with session ID: 0671693627
298  Mon Feb 16 20:10:49.944383100 2015  6  Setup Message
299  Mon Feb 16 20:10:49.944687980 2015  6  Frontend: 0, Opened FE fd: 28.
300  Mon Feb 16 20:10:49.946481795 2015  6  Frontend: 0, Waiting on lock.
301  Mon Feb 16 20:10:50.000596540 2015  6  Frontend: 0, Not locked yet (FE status 0)...
302  Mon Feb 16 20:10:50.030698243 2015  6  Frontend: 0, Not locked yet (FE status 0)...
303  Mon Feb 16 20:10:50.060934893 2015  6  Frontend: 0, Not locked yet (FE status 0)...
304  Mon Feb 16 20:10:50.091062482 2015  6  Frontend: 0, Not locked yet (FE status 0)...
305  Mon Feb 16 20:10:50.116519769 2015  6  Frontend: 0, Opened DVR fd: 29.
306  Mon Feb 16 20:10:50.116767309 2015  6  Frontend: 0, Start RTCP stream to 192.168.1.20 (36560 - 36561)
307  Mon Feb 16 20:10:50.116903844 2015  6  Frontend: 0, Start streaming to 192.168.1.20 (36560 - 36561)
308  Mon Feb 16 20:10:50.116931760 2015  6  Frontend: 0, Open Frontend Monitor for 192.168.1.20 (36560 - 36561)
309  Mon Feb 16 20:10:50.121814569 2015  6  Client connection with sessionID 0671693627
310  Mon Feb 16 20:10:50.121838314 2015  6  RTSP Client 192.168.1.20: Found by sessionID 0671693627 with fd: 27
311  Mon Feb 16 20:10:50.121857984 2015  6  Play Message
312  Mon Feb 16 20:10:50.121871620 2015  6  Frontend: 0, Waiting on lock.
313  Mon Feb 16 20:10:50.121897197 2015  6  Frontend: 0, Not locked yet (FE status 0)...
[... cut ...]
322  Mon Feb 16 20:11:15.923253747 2015  6  RTSP Connection from 192.168.1.20 Port 34164 (fd: 27)
323  Mon Feb 16 20:11:15.928406417 2015  6  RTSP Client 192.168.1.20: Found empty slot for fd: 27
324  Mon Feb 16 20:11:15.928438494 2015  6  RTSP Client 192.168.1.20: With fd: 27 giving Session ID: 0107366313
325  Mon Feb 16 20:11:15.928460580 2015  6  Frontend: 0, With dvbt Attaching to client 192.168.1.20 as requested with session ID: 0107366313
326  Mon Feb 16 20:11:15.928491331 2015  6  Setup Message

Barracuda09, 2015-02-16:
Hi,

Yes something like that but then the debug version.
run make debug to make one.

FuNK3Y, 2015-02-17:
0  Tue Feb 17 18:56:15.020705289 2015  6  [satpi.c:259] --- starting SatPI version: 1.0.14~gd48d644 ---
1  Tue Feb 17 18:56:15.020772525 2015  6  [tune.c:437] Detecting frontends in: /dev/dvb
2  Tue Feb 17 18:56:15.020894887 2015  6  [tune.c:447] Frontends found: 1
3  Tue Feb 17 18:56:15.020910132 2015  7  [tune.c:512] Open FE fd: 4 readonly 1
4  Tue Feb 17 18:56:15.020920424 2015  6  [tune.c:152] Frontend Name: DRXK DVB-C DVB-T
5  Tue Feb 17 18:56:15.020928851 2015  7  [tune.c:197] Frontend Type: Cable (Annex A)
6  Tue Feb 17 18:56:15.020934248 2015  7  [tune.c:206] Frontend Type: Cable (Annex C)
7  Tue Feb 17 18:56:15.020939400 2015  7  [tune.c:189] Frontend Type: Terrestrial (DVB-T)
8  Tue Feb 17 18:56:15.020945027 2015  7  [tune.c:216] Frontend Freq: 47000000 Hz to 865000000 Hz
9  Tue Feb 17 18:56:15.020950340 2015  7  [tune.c:217] Frontend srat: 870000 symbols/s to 11700000 symbols/s
10  Tue Feb 17 18:56:15.021176761 2015  6  [connection.c:336] eth0: 192.168.1.112 [00:50:56:a2:83:bd]
11  Tue Feb 17 18:56:15.021245496 2015  6  [rtsp.c:1040] Setting up RTSP server
12  Tue Feb 17 18:56:15.021335858 2015  6  [http.c:491] Setting up HTTP server
13  Tue Feb 17 18:56:15.021427835 2015  6  [rtp.c:359] Setting up 8 RTP servers
14  Tue Feb 17 18:56:15.021736163 2015  6  [rtcp.c:346] Setting up 8 RTCP servers
15  Tue Feb 17 18:56:15.022180323 2015  6  [ssdp.c:204] Setting up SSDP server with BOOTID: 1
16  Tue Feb 17 18:56:36.351830216 2015  6  [connection.c:263] RTSP Connection from 192.168.1.20 Port 58227 (fd: 25)
17  Tue Feb 17 18:56:36.357063261 2015  6  [rtsp.c:888] RTSP Client 192.168.1.20: Found empty slot for fd: 25
18  Tue Feb 17 18:56:36.357098642 2015  6  [rtsp.c:908] RTSP Client 192.168.1.20: With fd: 25 giving Session ID: 1702957685
19  Tue Feb 17 18:56:36.357111458 2015  7  [rtsp.c:922] SETUP rtsp://192.168.1.112/?fe=1&freq=498&bw=8&msys=dvbt&mtype=64qam&tmode=8k&gi=132 RTSP/1.0
20  Tue Feb 17 18:56:36.357111458 2015  7  Transport: RTP/AVP;unicast;client_port=36688-36689
21  Tue Feb 17 18:56:36.357111458 2015  7  CSeq: 1
22  Tue Feb 17 18:56:36.357111458 2015  7  --- LINE END ---
23  Tue Feb 17 18:56:36.357175050 2015  6  [rtsp.c:375] Frontend: 0, With dvbt Attaching to client 192.168.1.20 as requested with session ID: 1702957685
24  Tue Feb 17 18:56:36.357240168 2015  6  [rtsp.c:457] Setup Message
25  Tue Feb 17 18:56:36.357673812 2015  7  [tune.c:512] Open FE fd: 26 readonly 0
26  Tue Feb 17 18:56:36.357761038 2015  6  [tune.c:572] Frontend: 0, Opened FE fd: 26.
27  Tue Feb 17 18:56:36.357772941 2015  7  [tune.c:396] Frontend: 0, Start tuning process
28  Tue Feb 17 18:56:36.359320824 2015  6  [tune.c:583] Frontend: 0, Waiting on lock.
29  Tue Feb 17 18:56:36.411559774 2015  6  [tune.c:597] Frontend: 0, Not locked yet (FE status 0)...
30  Tue Feb 17 18:56:36.441703935 2015  6  [tune.c:597] Frontend: 0, Not locked yet (FE status 0)...
31  Tue Feb 17 18:56:36.471708525 2015  6  [tune.c:597] Frontend: 0, Not locked yet (FE status 0)...
32  Tue Feb 17 18:56:36.501724118 2015  6  [tune.c:597] Frontend: 0, Not locked yet (FE status 0)...
33  Tue Feb 17 18:56:36.527107574 2015  6  [tune.c:615] Frontend: 0, Opened DVR fd: 27.
34  Tue Feb 17 18:56:36.532416403 2015  6  [rtsp.c:861] Client connection with sessionID 1702957685
35  Tue Feb 17 18:56:36.532468967 2015  6  [rtsp.c:866] RTSP Client 192.168.1.20: Found by sessionID 1702957685 with fd: 25
36  Tue Feb 17 18:56:36.532483617 2015  7  [rtsp.c:922] PLAY rtsp://192.168.1.112/stream=0?pids=0,1,16,17,18,57,3002,3003 RTSP/1.0
37  Tue Feb 17 18:56:36.532483617 2015  7  Session: 1702957685
38  Tue Feb 17 18:56:36.532483617 2015  7  CSeq: 2
39  Tue Feb 17 18:56:36.532483617 2015  7  --- LINE END ---
40  Tue Feb 17 18:56:36.532541428 2015  6  [rtsp.c:522] Play Message
41  Tue Feb 17 18:56:36.532551850 2015  7  [tune.c:396] Frontend: 0, Start tuning process
42  Tue Feb 17 18:56:36.532572736 2015  6  [tune.c:583] Frontend: 0, Waiting on lock.
43  Tue Feb 17 18:56:36.532585211 2015  6  [tune.c:597] Frontend: 0, Not locked yet (FE status 0)...
44  Tue Feb 17 18:56:36.587511683 2015  6  [tune.c:597] Frontend: 0, Not locked yet (FE status 0)...
45  Tue Feb 17 18:56:36.617411805 2015  6  [tune.c:597] Frontend: 0, Not locked yet (FE status 0)...
46  Tue Feb 17 18:56:36.647713158 2015  6  [tune.c:597] Frontend: 0, Not locked yet (FE status 0)...
47  Tue Feb 17 18:56:36.674388357 2015  7  [tune.c:552] Frontend: 0, Set filter PID: 0 - fd: 28
48  Tue Feb 17 18:56:36.674427168 2015  7  [tune.c:552] Frontend: 0, Set filter PID: 1 - fd: 29
49  Tue Feb 17 18:56:36.674440432 2015  7  [tune.c:552] Frontend: 0, Set filter PID: 16 - fd: 30
50  Tue Feb 17 18:56:36.674455914 2015  7  [tune.c:552] Frontend: 0, Set filter PID: 17 - fd: 31
51  Tue Feb 17 18:56:36.674489997 2015  7  [tune.c:552] Frontend: 0, Set filter PID: 18 - fd: 32
52  Tue Feb 17 18:56:36.674559059 2015  7  [tune.c:552] Frontend: 0, Set filter PID: 57 - fd: 33
53  Tue Feb 17 18:56:36.674598629 2015  7  [tune.c:552] Frontend: 0, Set filter PID: 3002 - fd: 34
54  Tue Feb 17 18:56:36.674642340 2015  7  [tune.c:552] Frontend: 0, Set filter PID: 3003 - fd: 35
55  Tue Feb 17 18:56:36.674803292 2015  6  [rtp.c:98] Frontend: 0, Start streaming to 192.168.1.20 (36688 - 36689)
56  Tue Feb 17 18:56:36.674861270 2015  6  [rtcp.c:260] Frontend: 0, Start RTCP stream to 192.168.1.20 (36688 - 36689)
57  Tue Feb 17 18:56:36.674884293 2015  6  [rtcp.c:274] Frontend: 0, Open Frontend Monitor for 192.168.1.20 (36688 - 36689)
[... cut ...]
75  Tue Feb 17 18:56:41.154067713 2015  7  [rtsp.c:922] PLAY rtsp://192.168.1.112/stream=0?pids=0,1,16,17,18,57,3002,3003 RTSP/1.0
76  Tue Feb 17 18:56:41.154067713 2015  7  Session: 1702957685
77  Tue Feb 17 18:56:41.154067713 2015  7  CSeq: 4

Here you go. Maybe the tuner needs to be switched from DVB-C to DVB-T (it can do both, but not at the same time). I can't find it anymore, but as far as I recall I needed to manually input a command to switch with TVH 3.4. I will update this post if I find something

Barracuda09, 2015-02-17:
Hi,
yes this looks ok but there is no lock. I don't know how this switching should work.
I think it should just switch by delsys parameter. If you can find out how it whas done in TVH i'm 
interested.
Thanks again for testing, but  I am in progress of rewriting SatPI in C++

Marc

FuNK3Y, 2015-02-17:
Humm I spent 30 minutes looking for it earlier without any luck. But finaly here it is:

I used dvb-fe-tool (http://www.linuxtv.org/wiki/index.php/Dvb-fe-tool) to switch the tuner mode. But in this specific case I don't think it is of any use. I did try to specify DVB-T before launching SATPI but it didn't go any better.

I also noticed that satpi correctly changes this mode. If I kill SATPI after watching DVB-C the tuner is in DVB-C mode (same thing for DVB-T).

FuNK3Y, 2015-02-22:
Can't wait to test the C++ version. Do you have an ETA for its release?

Is SATPI power savvy? Is it possible to put the tuner in standby when not used?

Barracuda09, 2015-02-22:
Hi,

No it is not heavy, I use it on a RPI 1 (in turbo mode) with 2 HD streams with little to
nothing on CPU usage. The C++ version will be finished I hope within a few days.
I am testing and smashing some bug out as I type.

Thanks for your enthusiasm. The standby I have to look into that.

Thanks,
Marc

pbriesch, 2015-02-22:
Hi Marc,

I'm looking forward, too!  :+1: 

Regards, Paul

Barracuda09, 2015-02-23:
Hi,

Here is an Dropbox link to an Pre-Alpha version, which should work I hope.

https://www.dropbox.com/sh/5zzco085spd74f9/AACLOu__oMgigcOGZOGfKefQa?dl=0

Happy Testing and thanks,

Marc

pbriesch, 2015-02-23:
Hi Marc,

I couldn't compile. Did I do something wrong?

````
pbriesch@ubuntu:~/satpi_c++$ make debug
make "BUILD=debug"
make[1]: Entering directory '/home/pbriesch/satpi_c++'
./version.sh Version.cpp > /dev/null
g++ -c -Wall -Wextra -Winit-self  -O0 -g3 -DDEBUG -fstack-protector-all Version.cpp
g++ -c -Wall -Wextra -Winit-self  -O0 -g3 -DDEBUG -fstack-protector-all Satpi.cpp
In file included from InterfaceAttr.h:24:0,
                 from Satpi.cpp:25:
InterfaceAttr.h: In member function ‘void InterfaceAttr::get_interface_properties()’:
Utils.h:27:47: error: ‘close’ was not declared in this scope
 #define CLOSE_FD(x) if (x != -1) { if (close(x) == -1) { PERROR("close error"); } x = -1; }
                                               ^
InterfaceAttr.h:65:5: note: in expansion of macro ‘CLOSE_FD’
     CLOSE_FD(fd);
     ^
Utils.h:27:47: error: ‘close’ was not declared in this scope
 #define CLOSE_FD(x) if (x != -1) { if (close(x) == -1) { PERROR("close error"); } x = -1; }
                                               ^
InterfaceAttr.h:72:5: note: in expansion of macro ‘CLOSE_FD’
     CLOSE_FD(fd);
     ^
Utils.h:27:47: error: ‘close’ was not declared in this scope
 #define CLOSE_FD(x) if (x != -1) { if (close(x) == -1) { PERROR("close error"); } x = -1; }
                                               ^
InterfaceAttr.h:92:5: note: in expansion of macro ‘CLOSE_FD’
     CLOSE_FD(fd);
     ^
Utils.h:27:47: error: ‘close’ was not declared in this scope
 #define CLOSE_FD(x) if (x != -1) { if (close(x) == -1) { PERROR("close error"); } x = -1; }
                                               ^
InterfaceAttr.h:99:5: note: in expansion of macro ‘CLOSE_FD’
     CLOSE_FD(fd);
     ^
Utils.h:27:47: error: ‘close’ was not declared in this scope
 #define CLOSE_FD(x) if (x != -1) { if (close(x) == -1) { PERROR("close error"); } x = -1; }
                                               ^
InterfaceAttr.h:102:4: note: in expansion of macro ‘CLOSE_FD’
    CLOSE_FD(fd);
    ^
Makefile:74: recipe for target 'Satpi.o' failed
make[1]: *** [Satpi.o] Error 1
make[1]: Leaving directory '/home/pbriesch/satpi_c++'
Makefile:84: recipe for target 'debug' failed
make: *** [debug] Error 2
pbriesch@ubuntu:~/satpi_c++$

Barracuda09, 2015-02-23:
Hi,

That is strange.. can you try it again?

pbriesch, 2015-02-23:
Same code?

Barracuda09, 2015-02-23:
Yes you can get it from dropbox

pbriesch, 2015-02-23:
Here we go:
````
pbriesch@ubuntu:~/satpi_c++$ make
./version.sh Version.cpp > /dev/null
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Version.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Satpi.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG ChannelData.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG StreamProperties.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Properties.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Utils.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG StringConverter.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Log.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG HttpcSocket.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG TcpSocket.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG UdpSocket.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Frontend.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG StreamClient.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Streams.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Stream.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG HttpServer.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG RtspServer.cpp
RtspServer.cpp: In member function ‘bool RtspServer::methodDescribe(const StreamClient&)’:
RtspServer.cpp:421:103: warning: format ‘%d’ expects argument of type ‘int’, but argument 4 has type ‘size_t {aka long unsigned int}’ [-Wformat=]
    snprintf(cont2, size, RTSP_DESCRIBE_CONT2, i, desc_attr.c_str(), (active) ? "sendonly" : "inactive");
                                                                                                       ^
RtspServer.cpp:421:103: warning: format ‘%d’ expects argument of type ‘int’, but argument 4 has type ‘size_t {aka long unsigned int}’ [-Wformat=]
RtspServer.cpp:442:120: warning: format ‘%zu’ expects argument of type ‘size_t’, but argument 7 has type ‘int’ [-Wformat=]
   snprintf(tmp, sizeof(tmp), RTSP_DESCRIBE, RTSP_404_ERROR, client.getCSeq(), _server_ip_addr.c_str(), 0, sessionID, "");
                                                                                                                        ^
RtspServer.cpp:442:120: warning: format ‘%zu’ expects argument of type ‘size_t’, but argument 7 has type ‘int’ [-Wformat=]
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG SsdpServer.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG RtpThread.cpp
RtpThread.cpp: In member function ‘virtual void RtpThread::threadEntry()’:
RtpThread.cpp:126:66: error: ‘read’ was not declared in this scope
    const int bytes_read = read(pfd[0].fd, _bufPtr, TS_PACKET_SIZE);
                                                                  ^
Makefile:74: recipe for target 'RtpThread.o' failed
make: *** [RtpThread.o] Error 1
pbriesch@ubuntu:~/satpi_c++$

Barracuda09, 2015-02-23:
Sorry try it again... I probably use a different compiler and/or libs...

Barracuda09, 2015-02-23:
I have to look into this... why this is different..

Thanks for trying at least,

pbriesch, 2015-02-23:
No problem. Here we go again:
````
pbriesch@ubuntu:~/satpi_c++$ g++ --version
g++ (Ubuntu 4.9.1-16ubuntu6) 4.9.1
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
````
````
pbriesch@ubuntu:~/satpi_c++$ make
./version.sh Version.cpp > /dev/null
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Version.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Satpi.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG ChannelData.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG StreamProperties.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Properties.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Utils.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG StringConverter.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Log.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG HttpcSocket.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG TcpSocket.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG UdpSocket.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Frontend.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG StreamClient.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Streams.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Stream.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG HttpServer.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG RtspServer.cpp
RtspServer.cpp: In member function ‘bool RtspServer::methodDescribe(const StreamClient&)’:
RtspServer.cpp:421:103: warning: format ‘%d’ expects argument of type ‘int’, but argument 4 has type ‘size_t {aka long unsigned int}’ [-Wformat=]
    snprintf(cont2, size, RTSP_DESCRIBE_CONT2, i, desc_attr.c_str(), (active) ? "sendonly" : "inactive");
                                                                                                       ^
RtspServer.cpp:421:103: warning: format ‘%d’ expects argument of type ‘int’, but argument 4 has type ‘size_t {aka long unsigned int}’ [-Wformat=]
RtspServer.cpp:442:120: warning: format ‘%zu’ expects argument of type ‘size_t’, but argument 7 has type ‘int’ [-Wformat=]
   snprintf(tmp, sizeof(tmp), RTSP_DESCRIBE, RTSP_404_ERROR, client.getCSeq(), _server_ip_addr.c_str(), 0, sessionID, "");
                                                                                                                        ^
RtspServer.cpp:442:120: warning: format ‘%zu’ expects argument of type ‘size_t’, but argument 7 has type ‘int’ [-Wformat=]
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG SsdpServer.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG RtpThread.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG RtcpThread.cpp
RtcpThread.cpp: In member function ‘virtual void RtcpThread::threadEntry()’:
RtcpThread.cpp:361:16: error: ‘usleep’ was not declared in this scope
   usleep(200000);
                ^
Makefile:74: recipe for target 'RtcpThread.o' failed
make: *** [RtcpThread.o] Error 1
pbriesch@ubuntu:~/satpi_c++$

Barracuda09, 2015-02-24:
Probably it is time for me to upgrade my ubuntu, will do so now.
And compile it with a newer one.

Thanks for pointing this out.

pbriesch, 2015-02-24:
Hello Marc,

I tried now to compile Satpi on my Raspberry and there were no compile errors:
````
pbriesch@camera-pi ~/satpi_c++ $ cpp --version
cpp (Debian 4.6.3-14+rpi1) 4.6.3
Copyright (C) 2011 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
````
````
pbriesch@camera-pi ~/satpi_c++ $ make
./version.sh Version.cpp > /dev/null
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Version.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Satpi.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG ChannelData.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG StreamProperties.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Properties.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Utils.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG StringConverter.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Log.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG HttpcSocket.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG TcpSocket.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG UdpSocket.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Frontend.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG StreamClient.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Streams.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Stream.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG HttpServer.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG RtspServer.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG SsdpServer.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG RtpThread.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG RtcpThread.cpp
g++  Version.o Satpi.o ChannelData.o StreamProperties.o Properties.o Utils.o StringConverter.o Log.o HttpcSocket.o TcpSocket.o UdpSocket.o Frontend.o StreamClient.o Streams.o Stream.o HttpServer.o RtspServer.o SsdpServer.o RtpThread.o RtcpThread.o -o satpi -lpthread -lrt
pbriesch@camera-pi ~/satpi_c++ $

Barracuda09, 2015-02-24:
Yes so it seems...

Barracuda09, 2015-02-24:
Hi again,

It should compile now with g++ 4.9

Marc

pbriesch, 2015-02-24:
Yes, it does compile now. Probably you got the same compiler warnings as I did:
````
pbriesch@ubuntu:~/satpi_c++$ make debug
make "BUILD=debug"
make[1]: Entering directory '/home/pbriesch/satpi_c++'
./version.sh Version.cpp > /dev/null
g++ -c -Wall -Wextra -Winit-self  -O0 -g3 -DDEBUG -fstack-protector-all Version.cpp
g++ -c -Wall -Wextra -Winit-self  -O0 -g3 -DDEBUG -fstack-protector-all Satpi.cpp
g++ -c -Wall -Wextra -Winit-self  -O0 -g3 -DDEBUG -fstack-protector-all ChannelData.cpp
g++ -c -Wall -Wextra -Winit-self  -O0 -g3 -DDEBUG -fstack-protector-all StreamProperties.cpp
g++ -c -Wall -Wextra -Winit-self  -O0 -g3 -DDEBUG -fstack-protector-all Properties.cpp
g++ -c -Wall -Wextra -Winit-self  -O0 -g3 -DDEBUG -fstack-protector-all Utils.cpp
g++ -c -Wall -Wextra -Winit-self  -O0 -g3 -DDEBUG -fstack-protector-all StringConverter.cpp
g++ -c -Wall -Wextra -Winit-self  -O0 -g3 -DDEBUG -fstack-protector-all Log.cpp
g++ -c -Wall -Wextra -Winit-self  -O0 -g3 -DDEBUG -fstack-protector-all HttpcSocket.cpp
g++ -c -Wall -Wextra -Winit-self  -O0 -g3 -DDEBUG -fstack-protector-all TcpSocket.cpp
g++ -c -Wall -Wextra -Winit-self  -O0 -g3 -DDEBUG -fstack-protector-all UdpSocket.cpp
g++ -c -Wall -Wextra -Winit-self  -O0 -g3 -DDEBUG -fstack-protector-all Frontend.cpp
g++ -c -Wall -Wextra -Winit-self  -O0 -g3 -DDEBUG -fstack-protector-all StreamClient.cpp
g++ -c -Wall -Wextra -Winit-self  -O0 -g3 -DDEBUG -fstack-protector-all Streams.cpp
g++ -c -Wall -Wextra -Winit-self  -O0 -g3 -DDEBUG -fstack-protector-all Stream.cpp
g++ -c -Wall -Wextra -Winit-self  -O0 -g3 -DDEBUG -fstack-protector-all HttpServer.cpp
g++ -c -Wall -Wextra -Winit-self  -O0 -g3 -DDEBUG -fstack-protector-all RtspServer.cpp
RtspServer.cpp: In member function ‘bool RtspServer::methodDescribe(const StreamClient&)’:
RtspServer.cpp:421:103: warning: format ‘%d’ expects argument of type ‘int’, but argument 4 has type ‘size_t {aka long unsigned int}’ [-Wformat=]
    snprintf(cont2, size, RTSP_DESCRIBE_CONT2, i, desc_attr.c_str(), (active) ? "sendonly" : "inactive");
                                                                                                       ^
RtspServer.cpp:421:103: warning: format ‘%d’ expects argument of type ‘int’, but argument 4 has type ‘size_t {aka long unsigned int}’ [-Wformat=]
RtspServer.cpp:442:120: warning: format ‘%zu’ expects argument of type ‘size_t’, but argument 7 has type ‘int’ [-Wformat=]
   snprintf(tmp, sizeof(tmp), RTSP_DESCRIBE, RTSP_404_ERROR, client.getCSeq(), _server_ip_addr.c_str(), 0, sessionID, "");
                                                                                                                        ^
RtspServer.cpp:442:120: warning: format ‘%zu’ expects argument of type ‘size_t’, but argument 7 has type ‘int’ [-Wformat=]
g++ -c -Wall -Wextra -Winit-self  -O0 -g3 -DDEBUG -fstack-protector-all SsdpServer.cpp
g++ -c -Wall -Wextra -Winit-self  -O0 -g3 -DDEBUG -fstack-protector-all RtpThread.cpp
g++ -c -Wall -Wextra -Winit-self  -O0 -g3 -DDEBUG -fstack-protector-all RtcpThread.cpp
g++  Version.o Satpi.o ChannelData.o StreamProperties.o Properties.o Utils.o StringConverter.o Log.o HttpcSocket.o TcpSocket.o UdpSocket.o Frontend.o StreamClient.o Streams.o Stream.o HttpServer.o RtspServer.o SsdpServer.o RtpThread.o RtcpThread.o -o satpi -lpthread -lrt
make[1]: Leaving directory '/home/pbriesch/satpi_c++'
pbriesch@ubuntu:~/satpi_c++$
````
Now I can test a little bit.

Barracuda09, 2015-02-24:
Well the strange thing is that I still do not see the warnings...
I upgrade my ubuntu to 14.04 with gcc 4.9 so it should be up to date .

pbriesch, 2015-02-24:
I got 4.9.1
Not sure if this makes any difference. 
````
Linux version 3.16.0-30-generic (buildd@komainu) (gcc version 4.9.1 (Ubuntu 4.9.1-16ubuntu6) ) #40-Ubuntu SMP Mon Jan 12 22:06:37 UTC 2015 (Ubuntu 3.16.0-30.40-generic 3.16.7-ckt3)

Barracuda09, 2015-02-24:
Well I seem to have this
g++ (Ubuntu 4.9.2-0ubuntu1~14.04) 4.9.2

FuNK3Y, 2015-02-24:
I did some test, and a few things are not working
- I see nothing under frontend (but in the log my frontend is detected, and it is available in TVH)
- I was unable to stream anything using DVB-C (which worked fine before)
- The ssdp discovery is still not working

Here is the log:
0  Tue Feb 24 18:53:45.482315312 2015  6  [ Satpi.cpp:244] --- starting SatPI version: 1.2 ---
1  Tue Feb 24 18:53:45.482581219 2015  6  [ InterfaceAttr.h:106] eth0: 192.168.1.112 [00:50:56:a2:83:bd]
2  Tue Feb 24 18:53:45.484144615 2015  6  [ Streams.cpp:108] Detecting frontends in: /dev/dvb
3  Tue Feb 24 18:53:45.484287548 2015  6  [ Streams.cpp:118] Frontends found: 1
4  Tue Feb 24 18:53:45.484407359 2015  6  [ Frontend.cpp:121] Frontend Name: DRXK DVB-C DVB-T
5  Tue Feb 24 18:53:45.484531268 2015  7  [ Frontend.cpp:166] Frontend Type: Cable (Annex A)
6  Tue Feb 24 18:53:45.484550625 2015  7  [ Frontend.cpp:175] Frontend Type: Cable (Annex C)
7  Tue Feb 24 18:53:45.484563910 2015  7  [ Frontend.cpp:158] Frontend Type: Terrestrial (DVB-T)
8  Tue Feb 24 18:53:45.484577138 2015  7  [ Frontend.cpp:185] Frontend Freq: 47000000 Hz to 865000000 Hz
9  Tue Feb 24 18:53:45.484590191 2015  7  [ Frontend.cpp:186] Frontend srat: 870000 symbols/s to 11700000 symbols/s
10  Tue Feb 24 18:53:45.484788049 2015  6  [ RtspServer.cpp:058] Setting up RTSP server
11  Tue Feb 24 18:53:45.485038387 2015  6  [ HttpServer.cpp:070] Setting up HTTP server
12  Tue Feb 24 18:53:45.485082167 2015  6  [ SsdpServer.cpp:107] Setting up SSDP server with BOOTID: 4
13  Tue Feb 24 18:53:47.465800841 2015  6  [ TcpSocket.cpp:147] HTTP Connection from 192.168.1.119 Port 59198 (fd: 8)
14  Tue Feb 24 18:53:47.915356054 2015  6  [ TcpSocket.cpp:147] HTTP Connection from 192.168.1.119 Port 59200 (fd: 9)
15  Tue Feb 24 18:53:47.915485939 2015  6  [ TcpSocket.cpp:147] HTTP Connection from 192.168.1.119 Port 59201 (fd: 10)
16  Tue Feb 24 18:53:47.915507640 2015  6  [ TcpSocket.cpp:147] HTTP Connection from 192.168.1.119 Port 59199 (fd: 11)
17  Tue Feb 24 18:53:47.915584649 2015  7  [ HttpServer.cpp:112] GET web/index.html (size 4927)
18  Tue Feb 24 18:53:47.915656370 2015  6  [ TcpSocket.cpp:147] HTTP Connection from 192.168.1.119 Port 59202 (fd: 12)
19  Tue Feb 24 18:53:47.915677134 2015  6  [ TcpSocket.cpp:147] HTTP Connection from 192.168.1.119 Port 59203 (fd: 13)
20  Tue Feb 24 18:53:47.933195426 2015  7  [ HttpServer.cpp:112] GET web/stylesheet.css (size 1463)
21  Tue Feb 24 18:53:47.933547733 2015  7  [ HttpServer.cpp:112] GET web/loadxmldoc.js (size 521)
22  Tue Feb 24 18:53:48.036598292 2015  7  [ HttpServer.cpp:112] GET web/grid_noise.png (size 4638)
23  Tue Feb 24 18:53:48.041888290 2015  7  [ HttpServer.cpp:112] GET web/favicon.ico (size 1858)
24  Tue Feb 24 18:53:48.460071701 2015  7  [ HttpServer.cpp:112] GET web/log.html (size 1784)
25  Tue Feb 24 18:53:48.479842844 2015  7  [ HttpServer.cpp:112] GET web/stylesheet.css (size 1463)
26  Tue Feb 24 18:53:48.479939712 2015  7  [ HttpServer.cpp:112] GET web/loadxmldoc.js (size 521)
27  Tue Feb 24 18:53:48.571582560 2015  7  [ HttpServer.cpp:112] GET web/grid_noise.png (size 4638)
28  Tue Feb 24 18:53:48.571715344 2015  7  [ HttpServer.cpp:112] GET web/bg-grain.png (size 5636)
29  Tue Feb 24 18:53:52.986298265 2015  6  [ TcpSocket.cpp:147] RTSP Connection from 192.168.1.20 Port 34692 (fd: 14)
30  Tue Feb 24 18:53:52.986417596 2015  6  [ RtspServer.cpp:069] Found some data from client 192.168.1.20: SETUP rtsp://192.168.1.112/?fe=2%s&freq=602%s&sr=6900%s&msys=dvbc%s&mtype=256qam%s&fec=35 RTSP/1.0
31  Tue Feb 24 18:53:52.986417596 2015  6  Transport: RTP/AVP;unicast;client_port=46196-46197
32  Tue Feb 24 18:53:52.986417596 2015  6  CSeq: 1
33  Tue Feb 24 18:53:52.986417596 2015  6  --- LINE END ---
34  Tue Feb 24 18:53:52.986520653 2015  6  [ Streams.cpp:226] Found: StreamID 1 - SessionID 1702957685
35  Tue Feb 24 18:53:52.986547681 2015  6  [ Stream.cpp:063] Found: StreamID 1 - StreamClient[0] with SessionID -1
36  Tue Feb 24 18:53:52.986580174 2015  7  [ RtspServer.cpp:137] Stream: 1, Parsing transport parameters...
37  Tue Feb 24 18:53:52.986720339 2015  6  [ StreamClient.cpp:055] Connection can close: 0
38  Tue Feb 24 18:53:52.986775292 2015  3  [ Frontend.cpp:053] FRONTEND DEVICE: No such file or directory (code 2)
39  Tue Feb 24 18:53:52.986792145 2015  6  [ Frontend.cpp:353] Stream: 1, Opened FE fd: -1.
40  Tue Feb 24 18:53:52.986805151 2015  7  [ Frontend.cpp:312] Stream: 1, Start tuning process...
41  Tue Feb 24 18:53:52.986821202 2015  3  [ Frontend.cpp:242] FE_SET_PROPERTY failed: Bad file descriptor (code 9)
42  Tue Feb 24 18:53:52.986834313 2015  6  [ Frontend.cpp:364] Stream: 1, Waiting on lock.
43  Tue Feb 24 18:53:53.087463036 2015  3  [ Frontend.cpp:061] DVR DEVICE: No such file or directory (code 2)
44  Tue Feb 24 18:53:53.237714238 2015  3  [ Frontend.cpp:061] DVR DEVICE: No such file or directory (code 2)
45  Tue Feb 24 18:53:53.387904639 2015  3  [ Frontend.cpp:061] DVR DEVICE: No such file or directory (code 2)
46  Tue Feb 24 18:53:53.538054055 2015  3  [ Frontend.cpp:061] DVR DEVICE: No such file or directory (code 2)
47  Tue Feb 24 18:53:53.688278151 2015  7  [ RtspServer.cpp:318] RTSP/1.0 200 OK
48  Tue Feb 24 18:53:53.688278151 2015  7  CSeq: 1
49  Tue Feb 24 18:53:53.688278151 2015  7  Session: 1702957685;timeout=60
50  Tue Feb 24 18:53:53.688278151 2015  7  Transport: RTP/AVP;unicast;client_port=46196-46197
51  Tue Feb 24 18:53:53.688278151 2015  7  com.ses.streamID: 1
[... cut ...]
58  Tue Feb 24 18:53:53.689078592 2015  6  [ Stream.cpp:068] Found: StreamID 0 - No StreamClient with SessionID 1702957685
59  Tue Feb 24 18:54:02.216865906 2015  7  [ TcpSocket.cpp:085] HTTP Client 192.168.1.119: Connection closed with fd: 10
60  Tue Feb 24 18:54:02.216960987 2015  7  [ TcpSocket.cpp:085] HTTP Client 192.168.1.119: Connection closed with fd: 12
61  Tue Feb 24 18:54:02.216989407 2015  7  [ TcpSocket.cpp:085] HTTP Client 192.168.1.119: Connection closed with fd: 13
62  Tue Feb 24 18:54:02.217330983 2015  7  [ TcpSocket.cpp:085] HTTP Client 192.168.1.119: Connection closed with fd: 9

Barracuda09, 2015-02-24:
Hi FuNK3Y,

- Yes I did not implement the web page frontend info (or config or PID list)
- the SSDP discovery is just an copy of the 'old' code
- DVB-C is another point (which I think should work) but I see in the log that it request fe=2
  -> Did you restart tvh?

Thanks to all for testing this so far,

Marc

pbriesch, 2015-02-25:
Hello Marc,

I didn't have much time to test recently. Do you have a new build for me to test?

Regards

Paul

Barracuda09, 2015-02-25:
Hi,

That is no problem. No I am still working out some issues at the moment.

pbriesch, 2015-02-25:
OK, then I'd better wait until you have something new.

Barracuda09, 2015-02-25:
Yes that would be a good idea.. I hope I have then the SSDP issue fixed and web interface implemented.

FuNK3Y, 2015-02-26:
I am pretty sure I did restart TVH prior to any of my testing. But since I am using the unstable branche with constantly changing build there might be a regression in TVH itself.

I'll do further testing with the upcoming version as I won't the possibility to touch my productive for the upcoming week

Barracuda09, 2015-02-28:
Hi,

I have fixed hopefully some issues in the 'new'  SatPI please see Dropbox link:
https://www.dropbox.com/sh/5zzco085spd74f9/AACLOu__oMgigcOGZOGfKefQa?dl=0

Fixes:
- Frontend web page added
- SSDP fixes
- other small fixes

I hoped I could test DVB-T but I am unable to get my new DVB-T(2)/C stick to work under linux :-(

But for now Happy Testing and thanks,

Marc

pbriesch, 2015-02-28:
Marc,

which stick do you have? Maybe you can provide output of dmesg, lsusb -v

Barracuda09, 2015-02-28:
Hi Paul,

It is an DVBSky T330 (HMP-Combo). It appears that it does nothing (In windows
it works and gets warm)
The firmware is dowloaded but seems to do nothing.

pbriesch, 2015-02-28:
Hi Marc,
I need to fix the power supply of my Ubuntu server before I can continue testing. Regarding your DVB-stick please check this out: http://www.linuxtv.org/wiki/index.php/TechnoTrend_TT-TVStick_CT2-4400

Paul

Barracuda09, 2015-02-28:
Hi,

Yes I have a backported media tree to 3.16 kernel. (I have had a machine with 3.19 kernel which did
not work as well. it loads the drivers and downloads the firmware but with no luck.

pbriesch, 2015-03-01:
Hello Marc,

I got this error message:

````
[    TcpSocket.cpp:085] RTSP Client 192.168.1.107: Connection closed with fd: 231
[      Streams.cpp:204] Found SessionID 1702957685 by SocketClient
[      Streams.cpp:224] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:076] StreamID 0 - No StreamClient with SessionID 1702957685 for UNKNOWN DELSYS
[       Stream.cpp:076] StreamID 1 - No StreamClient with SessionID 1702957685 for UNKNOWN DELSYS
[   HttpServer.cpp:102] GET web/config.xml
[   HttpServer.cpp:103] File not found: No such file or directory (code 2)
[   HttpServer.cpp:102] GET web/config.xml
[   HttpServer.cpp:103] File not found: No such file or directory (code 2)
[   HttpServer.cpp:102] GET web/config.xml
[   HttpServer.cpp:103] File not found: No such file or directory (code 2)
````

There is only a file called *config.html* under *web/*.

Furthermore, I had to add the execution bit *chmod +x version.h* before compiling. Maybe this is a issue with Dropbox?

Regards,

Paul

Barracuda09, 2015-03-01:
Hi Paul,

Yes that is in deed a Dropbox issue with version.h (sorry). The config.xml is no problem.. the config page is not implemented yet (SatPI generates this file)

pbriesch, 2015-03-01:
Hi Marc,

could you show in the fronted webpage the streams and corresponding clients with IP? I'm testing several clients at the same time which are switched on and off.

Barracuda09, 2015-03-01:
Yes, something what is was before?

pbriesch, 2015-03-01:
Yes, however, instead of writing client #1, client #2 or so it would be better to mention the IP address. (Maybe you already did this before and I'm confusing something with an other project)

Barracuda09, 2015-03-01:
I have some small succes with my DvbSky T330 stick... toke it from dvbsky.net. but seems only the "Proprietary" drivers work :-(

Barracuda09, 2015-03-01:
I have added the IP and sessionID to the frontend webpage (Is this what you mean?)

FuNK3Y, 2015-03-04:
Great work!

I did retest and ssdp works fine. But DVB-C is still not working, here is the log:

0  Wed Mar  4 22:59:07.547385763 2015  6  --- starting SatPI version: 1.5.6~gba2c20f ---
1  Wed Mar  4 22:59:07.547794608 2015  6  eth0: 192.168.1.112 [00:50:56:a2:83:bd]
2  Wed Mar  4 22:59:07.548889259 2015  6  Detecting frontends in: /dev/dvb
3  Wed Mar  4 22:59:07.549066401 2015  6  Frontends found: 1
4  Wed Mar  4 22:59:07.549185729 2015  6  Frontend Name: DRXK DVB-C DVB-T
5  Wed Mar  4 22:59:07.549483784 2015  6  Setting up RTSP server
6  Wed Mar  4 22:59:07.549995608 2015  6  Setting up HTTP server
7  Wed Mar  4 22:59:07.550183765 2015  6  Setting up SSDP server with BOOTID: 2
8  Wed Mar  4 22:59:07.730627599 2015  6  HTTP Connection from 192.168.1.108 Port 61069 (fd: 8)
9  Wed Mar  4 22:59:07.986262873 2015  6  HTTP Connection from 192.168.1.108 Port 61068 (fd: 9)
10  Wed Mar  4 22:59:28.069924929 2015  6  RTSP Connection from 192.168.1.20 Port 42999 (fd: 9)
11  Wed Mar  4 22:59:28.070331750 2015  6  Found StreamID x - SessionID x - Creating new SessionID: 1702957685
12  Wed Mar  4 22:59:28.070397848 2015  6  Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbc
13  Wed Mar  4 22:59:28.070887880 2015  6  Stream: 0, Opened FE fd: 10.
14  Wed Mar  4 22:59:28.073387427 2015  6  Stream: 0, Waiting on lock.
15  Wed Mar  4 22:59:28.344757525 2015  6  Stream: 0, Not locked yet (FE status 0)...
16  Wed Mar  4 22:59:28.383877466 2015  6  Stream: 0, Not locked yet (FE status 0)...
17  Wed Mar  4 22:59:28.440294643 2015  6  Stream: 0, Tuned and locked (FE status 31).
18  Wed Mar  4 22:59:28.440522682 2015  6  Stream: 0, Opened DVR fd: 11.
19  Wed Mar  4 22:59:28.440747004 2015  6  Stream: 0, Start RTP streaming to 192.168.1.20 (47948 - 47949)
20  Wed Mar  4 22:59:28.440906009 2015  6  Stream: 0, Start RTCP streaming to 192.168.1.20 (47948 - 47949)
21  Wed Mar  4 22:59:28.441469481 2015  6  Found StreamID x - SessionID: 1702957685
22  Wed Mar  4 22:59:28.441537671 2015  6  Stream: 0, StreamClient[0] with SessionID 1702957685
23  Wed Mar  4 22:59:28.441612431 2015  6  Stream: 0, Updating PID filters...
24  Wed Mar  4 22:59:32.431486009 2015  6  Found StreamID x - SessionID: 1702957685
25  Wed Mar  4 22:59:32.431605814 2015  6  Stream: 0, StreamClient[0] with SessionID 1702957685
26  Wed Mar  4 22:59:32.431637531 2015  6  Stream: 0, Teardown StreamClient[0] with SessionID 1702957685
27  Wed Mar  4 22:59:32.431959789 2015  6  Stream: 0, Stop RTP streaming to 192.168.1.20 (47948 - 47949) (Streamed 0.000 MBytes)
28  Wed Mar  4 22:59:32.432137188 2015  6  Stream: 0, Stop RTCP stream to 192.168.1.20 (47948 - 47949)
29  Wed Mar  4 22:59:32.433062756 2015  6  Found SessionID 1702957685 by SocketClient
30  Wed Mar  4 22:59:32.433113194 2015  6  Found StreamID x - SessionID: 1702957685
31  Wed Mar  4 22:59:32.433130971 2015  6  Stream: 0, StreamClient[0] with SessionID 1702957685
32  Wed Mar  4 22:59:32.433145270 2015  6  Stream: 0, Close StreamClient[0] with SessionID 1702957685
33  Wed Mar  4 22:59:35.851845850 2015  6  SAT>IP Client 192.168.1.20 : tries to discover the network, sending an reply
34  Wed Mar  4 22:59:35.851993270 2015  6  SAT>IP Client 192.168.1.20 : tries to discover the network, sending an reply
35  Wed Mar  4 22:59:35.852045886 2015  6  SAT>IP Client 192.168.1.20 : tries to discover the network, sending an reply
36  Wed Mar  4 22:59:36.103281216 2015  6  HTTP Connection from 192.168.1.20 Port 41892 (fd: 9)
37  Wed Mar  4 23:00:00.074106697 2015  6  RTSP Connection from 192.168.1.20 Port 43002 (fd: 9)
38  Wed Mar  4 23:00:00.074243775 2015  6  Found StreamID x - SessionID x - Creating new SessionID: 0205744672
39  Wed Mar  4 23:00:00.074268509 2015  6  Stream: 0, StreamClient[0] with SessionID 0205744672 for dvbc
40  Wed Mar  4 23:00:00.074621339 2015  6  Stream: 0, Opened FE fd: 10.
41  Wed Mar  4 23:00:00.076729284 2015  6  Stream: 0, Waiting on lock.
42  Wed Mar  4 23:00:00.340066717 2015  6  Stream: 0, Not locked yet (FE status 0)...
43  Wed Mar  4 23:00:00.379701534 2015  6  Stream: 0, Not locked yet (FE status 0)...
44  Wed Mar  4 23:00:00.421542401 2015  6  Stream: 0, Not locked yet (FE status 7)...
45  Wed Mar  4 23:00:00.471167010 2015  6  Stream: 0, Tuned and locked (FE status 31).
46  Wed Mar  4 23:00:00.471437458 2015  6  Stream: 0, Opened DVR fd: 11.
47  Wed Mar  4 23:00:00.471608520 2015  6  Stream: 0, Start RTP streaming to 192.168.1.20 (35422 - 35423)
48  Wed Mar  4 23:00:00.471735883 2015  6  Stream: 0, Start RTCP streaming to 192.168.1.20 (35422 - 35423)
49  Wed Mar  4 23:00:00.472372058 2015  6  Found StreamID x - SessionID: 0205744672
50  Wed Mar  4 23:00:00.472424332 2015  6  Stream: 0, StreamClient[0] with SessionID 0205744672
51  Wed Mar  4 23:00:00.472468245 2015  6  Stream: 0, Updating PID filters...
52  Wed Mar  4 23:00:06.119664984 2015  6  Found StreamID x - SessionID: 0205744672
53  Wed Mar  4 23:00:06.119776172 2015  6  Stream: 0, StreamClient[0] with SessionID 0205744672
54  Wed Mar  4 23:00:06.119806656 2015  6  Stream: 0, Teardown StreamClient[0] with SessionID 0205744672
55  Wed Mar  4 23:00:06.120487742 2015  6  Stream: 0, Stop RTP streaming to 192.168.1.20 (35422 - 35423) (Streamed 0.000 MBytes)
56  Wed Mar  4 23:00:06.120615948 2015  6  Stream: 0, Stop RTCP stream to 192.168.1.20 (35422 - 35423)
57  Wed Mar  4 23:00:06.121361499 2015  6  Found SessionID 0205744672 by SocketClient
[... cut ...]
76  Wed Mar  4 23:00:09.167763626 2015  6  Found StreamID x - SessionID: 0288582906
77  Wed Mar  4 23:00:09.167855573 2015  6  Stream: 0, StreamClient[0] with SessionID 0288582906
78  Wed Mar  4 23:00:09.167950775 2015  6  Stream: 0, Updating PID filters...
79  Wed Mar  4 23:00:15.089242406 2015  6  Found StreamID x - SessionID: 0288582906
80  Wed Mar  4 23:00:15.089315018 2015  6  Stream: 0, StreamClient[0] with SessionID 0288582906
81  Wed Mar  4 23:00:15.089391796 2015  6  Stream: 0, Updating PID filters...

Barracuda09, 2015-03-04:
Hi,

What does the Frontend web page show?

FuNK3Y, 2015-03-04:
Here it is:
[image]

Barracuda09, 2015-03-04:
So it worked before?..

FuNK3Y, 2015-03-04:
It worked before the cpp rework, but I was also using a different build of tvheadend. I will try to revert do an older version of satpi to validate if it still works

Barracuda09, 2015-03-04:
Ok.. thanks for trying... It is I can not test it here directly myself

FuNK3Y, 2015-03-04:
Okay, I did rebuild the version 1.0, and the URL is still something like that:

rtsp://192.168.1.112/?fe=2&freq=426&sr=6900&msys=dvbc&mtype=64qam RTSP/1.0

I assume that fe=2 means frontend 2, and since I have only one it shall on be more something like fe=0 or fe=1, right? If it is the case tvheadend seems to be the one to blame

Barracuda09, 2015-03-04:
Mmmm... I'm not sure if that make any differences in this case... It chooses stream/frontend 0..

FuNK3Y, 2015-03-04:
I shall try with an older build of tvheadend sometime next week.

[52 more replies]

closed 2015-03-30 by Barracuda09