Barracuda09/SATPI #21: cannel scan Panasonic TV client

9000h, 2015-02-08:
looks like the session handling is right, but the Panasonic is not able to find any channel.

Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 0 [satpi.c:235] --- starting ---
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 1 [tune.c:398] Not the preferred DVB API version, for correct function it should be 5.5 or higher
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 2 [tune.c:399] current DVB_API_VERSION: 5.4
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 3 [tune.c:401] Detecting frontends in: /dev/dvb
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 4 [tune.c:411] Frontends found: 4
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 5 [tune.c:476] Open FE fd: 4 readonly 1
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 6 [tune.c:117] Frontend Name: Montage RS6000(DVBSky)
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 7 [tune.c:146] Frontend Type: Satellite (DVB-S)
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 8 [tune.c:150] Frontend Type: Satellite (DVB-S2)
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 9 [tune.c:181] Frontend Freq: 950000 Hz to 2150000 Hz
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 10 [tune.c:182] Frontend srat: 1000000 symbols/s to 45000000 symbols/s
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 11 [tune.c:476] Open FE fd: 4 readonly 1
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 12 [tune.c:117] Frontend Name: Philips TDA10023 DVB-C
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 13 [tune.c:162] Frontend Type: Cable (Annex A)
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 14 [tune.c:176] Frontend Type: Unknown 18
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 15 [tune.c:181] Frontend Freq: 47000000 Hz to 862000000 Hz
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 16 [tune.c:182] Frontend srat: 451875 symbols/s to 7230000 symbols/s
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 17 [tune.c:476] Open FE fd: 4 readonly 1
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 18 [tune.c:117] Frontend Name: Montage RS6000(DVBSky)
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 19 [tune.c:146] Frontend Type: Satellite (DVB-S)
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 20 [tune.c:150] Frontend Type: Satellite (DVB-S2)
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 21 [tune.c:181] Frontend Freq: 950000 Hz to 2150000 Hz
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 22 [tune.c:182] Frontend srat: 1000000 symbols/s to 45000000 symbols/s
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 23 [tune.c:476] Open FE fd: 4 readonly 1
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 24 [tune.c:117] Frontend Name: DRXK DVB-C DVB-T
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 25 [tune.c:162] Frontend Type: Cable (Annex A)
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 26 [tune.c:176] Frontend Type: Unknown 18
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 27 [tune.c:154] Frontend Type: Terrestrial (DVB-T)
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 28 [tune.c:181] Frontend Freq: 47000000 Hz to 865000000 Hz
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 29 [tune.c:182] Frontend srat: 870000 symbols/s to 11700000 symbols/s
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 30 [connection.c:336] eth0: 192.168.178.133 [00:40:ca:a2:c7:9d]
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 31 [rtsp.c:1017] Setting up RTSP server
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 32 [http.c:490] Setting up HTTP server
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 33 [rtp.c:348] Setting up 8 RTP servers
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 34 [rtcp.c:338] Setting up 8 RTCP servers
Feb  8 19:54:05 yavdr2 SatIP Server[6184]: 35 [ssdp.c:170] Setting up SSDP server
Feb  8 19:54:20 yavdr2 SatIP Server[6184]: 36 [ssdp.c:256] SAT>IP Client 192.168.178.124 : contacted us sending reply fd: 24
Feb  8 19:54:20 yavdr2 SatIP Server[6184]: 37 [ssdp.c:256] SAT>IP Client 192.168.178.124 : contacted us sending reply fd: 24
Feb  8 19:54:28 yavdr2 SatIP Server[6184]: 38 [ssdp.c:256] SAT>IP Client 192.168.178.124 : contacted us sending reply fd: 24
Feb  8 19:54:28 yavdr2 SatIP Server[6184]: 39 [ssdp.c:256] SAT>IP Client 192.168.178.124 : contacted us sending reply fd: 24
Feb  8 19:54:29 yavdr2 SatIP Server[6184]: 40 [ssdp.c:256] SAT>IP Client 192.168.178.124 : contacted us sending reply fd: 24
Feb  8 19:54:29 yavdr2 SatIP Server[6184]: 41 [ssdp.c:256] SAT>IP Client 192.168.178.124 : contacted us sending reply fd: 24
Feb  8 19:54:35 yavdr2 SatIP Server[6184]: 42 [ssdp.c:256] SAT>IP Client 192.168.178.124 : contacted us sending reply fd: 24
Feb  8 19:54:35 yavdr2 SatIP Server[6184]: 43 [ssdp.c:256] SAT>IP Client 192.168.178.124 : contacted us sending reply fd: 24
Feb  8 19:54:40 yavdr2 SatIP Server[6184]: 44 [ssdp.c:256] SAT>IP Client 192.168.178.124 : contacted us sending reply fd: 24
Feb  8 19:54:40 yavdr2 SatIP Server[6184]: 45 [ssdp.c:256] SAT>IP Client 192.168.178.124 : contacted us sending reply fd: 24
Feb  8 19:54:47 yavdr2 SatIP Server[6204]: 0 [satpi.c:235] --- starting ---
Feb  8 19:54:47 yavdr2 SatIP Server[6204]: 1 [tune.c:398] Not the preferred DVB API version, for correct function it should be 5.5 or higher
Feb  8 19:54:47 yavdr2 SatIP Server[6204]: 2 [tune.c:399] current DVB_API_VERSION: 5.4
Feb  8 19:54:47 yavdr2 SatIP Server[6204]: 3 [tune.c:401] Detecting frontends in: /dev/dvb
Feb  8 19:54:47 yavdr2 SatIP Server[6204]: 4 [tune.c:411] Frontends found: 4
Feb  8 19:54:47 yavdr2 SatIP Server[6204]: 5 [tune.c:476] Open FE fd: 4 readonly 1
Feb  8 19:54:47 yavdr2 SatIP Server[6204]: 6 [tune.c:117] Frontend Name: Montage RS6000(DVBSky)
Feb  8 19:54:47 yavdr2 SatIP Server[6204]: 7 [tune.c:146] Frontend Type: Satellite (DVB-S)
Feb  8 19:54:47 yavdr2 SatIP Server[6204]: 8 [tune.c:150] Frontend Type: Satellite (DVB-S2)
Feb  8 19:54:47 yavdr2 SatIP Server[6204]: 9 [tune.c:181] Frontend Freq: 950000 Hz to 2150000 Hz
Feb  8 19:54:47 yavdr2 SatIP Server[6204]: 10 [tune.c:182] Frontend srat: 1000000 symbols/s to 45000000 symbols/s
Feb  8 19:54:47 yavdr2 SatIP Server[6204]: 11 [tune.c:476] Open FE fd: 4 readonly 1
Feb  8 19:54:47 yavdr2 SatIP Server[6204]: 12 [tune.c:117] Frontend Name: Philips TDA10023 DVB-C
Feb  8 19:54:47 yavdr2 SatIP Server[6204]: 13 [tune.c:162] Frontend Type: Cable (Annex A)
Feb  8 19:54:47 yavdr2 SatIP Server[6204]: 14 [tune.c:176] Frontend Type: Unknown 18
Feb  8 19:54:47 yavdr2 SatIP Server[6204]: 15 [tune.c:181] Frontend Freq: 47000000 Hz to 862000000 Hz
Feb  8 19:54:47 yavdr2 SatIP Server[6204]: 16 [tune.c:182] Frontend srat: 451875 symbols/s to 7230000 symbols/s
Feb  8 19:54:47 yavdr2 SatIP Server[6204]: 17 [tune.c:476] Open FE fd: 4 readonly 1
Feb  8 19:54:47 yavdr2 SatIP Server[6204]: 18 [tune.c:117] Frontend Name: Montage RS6000(DVBSky)
Feb  8 19:54:47 yavdr2 SatIP Server[6204]: 19 [tune.c:146] Frontend Type: Satellite (DVB-S)
Feb  8 19:54:47 yavdr2 SatIP Server[6204]: 20 [tune.c:150] Frontend Type: Satellite (DVB-S2)
Feb  8 19:54:47 yavdr2 SatIP Server[6204]: 21 [tune.c:181] Frontend Freq: 950000 Hz to 2150000 Hz
[... 193 lines cut ...]
Feb  8 19:57:35 yavdr2 SatIP Server[6204]: 215 CSeq: 15
Feb  8 19:57:35 yavdr2 SatIP Server[6204]: 216 Session: 1702957685
Feb  8 19:57:35 yavdr2 SatIP Server[6204]: 217 --- LINE END ---
Feb  8 19:57:35 yavdr2 SatIP Server[6204]: 218 [rtsp.c:520] Play Message
Feb  8 19:57:35 yavdr2 SatIP Server[6204]: 219 [tune.c:360] Frontend: 0, Start tuning process
Feb  8 19:57:35 yavdr2 SatIP Server[6204]: 220 [tune.c:289] Frontend: 0, Sending DiSEqC [e0] [10] [38] [f0]
Feb  8 19:57:35 yavdr2 SatIP Server[6204]: 221 [tune.c:547] Frontend: 0, Waiting on lock.
Feb  8 19:57:35 yavdr2 SatIP Server[6204]: 222 [tune.c:558] Frontend: 0, Tuned and locked (FE status 31).
Feb  8 19:57:38 yavdr2 SatIP Server[6204]: 223 [ssdp.c:256] SAT>IP Client 192.168.178.124 : contacted us sending reply fd: 37
Feb  8 19:57:38 yavdr2 SatIP Server[6204]: 224 [ssdp.c:256] SAT>IP Client 192.168.178.124 : contacted us sending reply fd: 37
Feb  8 19:57:52 yavdr2 SatIP Server[6204]: 225 [rtsp.c:855] Client connection with sessionID 1702957685
Feb  8 19:57:52 yavdr2 SatIP Server[6204]: 226 [rtsp.c:860] RTSP Client 192.168.178.124: Found by sessionID 1702957685 with fd: 24
Feb  8 19:57:52 yavdr2 SatIP Server[6204]: 227 [rtsp.c:903] TEARDOWN rtsp://192.168.178.133:554/stream=0 RTSP/1.0
Feb  8 19:57:52 yavdr2 SatIP Server[6204]: 228 CSeq: 16
Feb  8 19:57:52 yavdr2 SatIP Server[6204]: 229 Session: 1702957685
Feb  8 19:57:52 yavdr2 SatIP Server[6204]: 230 --- LINE END ---
Feb  8 19:57:52 yavdr2 SatIP Server[6204]: 231 [rtsp.c:766] Frontend: 0, Teardown Message, gracefull: 1
[... cut ...]
Feb  8 19:57:52 yavdr2 SatIP Server[6204]: 236 [tune.c:591] Frontend: 0, Remove filter PID: 17 - fd: 30 - Packet Count: 8931
Feb  8 19:57:52 yavdr2 SatIP Server[6204]: 237 [tune.c:591] Frontend: 0, Remove filter PID: 18 - fd: 31 - Packet Count: 38031
Feb  8 19:57:52 yavdr2 SatIP Server[6204]: 238 [tune.c:591] Frontend: 0, Remove filter PID: 20 - fd: 32 - Packet Count: 104
Feb  8 19:57:52 yavdr2 SatIP Server[6204]: 239 [rtsp.c:715] Frontend: 0, Detached from client 192.168.178.124 with session ID: 1702957685

Barracuda09, 2015-03-02:
Hi,

Is this issue fixed with latest commits?

Marc

Barracuda09, 2015-09-08:
I will close this for now... If anybody still has issues regarding this bug please make a new bug report.

closed 2015-09-08 by Barracuda09