Barracuda09/SATPI #22: Frontends don't always attach --> Channelscan with EyeTV on iOS didn't work

pbriesch, 2015-02-08:
Hi Marc,

This is a long post. Please take the time to scroll all the way down.

I tried to scan with EyeTV on iOS but iOS displayed: "in use by somebody else" This is the case if the frontends don't attach.
[image]

I have here two DVB-S2 frontends and one DVB-T frontend. Please note that the screenshot from the iPhone depicts the tunertype as DVB-T.  Hence EyeTV offered to perform a scan for DVB-T channels. Since no frontend was attached it did not work.
[image]
````
0  Sun Feb 8 22:24:17.948639250 2015  7  CSeq:1004
1  Sun Feb 8 22:24:17.948639250 2015  7  Accept:application/sdp
2  Sun Feb 8 22:24:17.948639250 2015  7  --- LINE END ---
3  Sun Feb 8 22:24:17.948705673 2015  6  [rtsp.c:917] RTSP Client 192.168.1.115: Requested Connection closed with fd: 24 and Session ID: 1702957685
4  Sun Feb 8 22:24:17.948835328 2015  7  [rtsp.c:646] RTSP/1.0 404 Not Found
5  Sun Feb 8 22:24:17.948835328 2015  7  CSeq: 1004
6  Sun Feb 8 22:24:17.948835328 2015  7  Content-Type: application/sdp
7  Sun Feb 8 22:24:17.948835328 2015  7  Content-Base: rtsp://192.168.1.161/
8  Sun Feb 8 22:24:17.948835328 2015  7  Content-Length: 0
9  Sun Feb 8 22:24:17.948835328 2015  7  --- LINE END ---
10  Sun Feb 8 22:24:18.454443915 2015  6  [rtsp.c:993] RTSP Client 192.168.1.115: Connection Watchdog kicked-in with fd: 26 and Session ID: 0205744672
11  Sun Feb 8 22:24:18.454471121 2015  6  [rtsp.c:768] Frontend: x, Teardown Message, gracefull: 0
12  Sun Feb 8 22:24:18.454484046 2015  3  [rtsp.c:718] Frontend: x, Not attached anymore to client 192.168.1.115 with session ID: 0205744672
13  Sun Feb 8 22:24:18.454507233 2015  3  [rtsp.c:745] close error: Bad file descriptor (code 9)
14  Sun Feb 8 22:24:22.342274976 2015  7  [rtsp.c:975] RTSP Client 192.168.1.115: Connection closed with fd: 24 and Session ID: 1702957685
15  Sun Feb 8 22:24:22.347415329 2015  6  [connection.c:263] RTSP Connection from 192.168.1.115 Port 52310 (fd: 24)
16  Sun Feb 8 22:24:22.352503239 2015  6  [rtsp.c:872] RTSP Client 192.168.1.115: Found by fd: 24 with Session ID: 1702957685
17  Sun Feb 8 22:24:22.352522168 2015  7  [rtsp.c:903] SETUP rtsp://192.168.1.161:554/?&freq=177.500000&bw=7.000000&msys=dvbt RTSP/1.0
18  Sun Feb 8 22:24:22.352522168 2015  7  Connection:Close
19  Sun Feb 8 22:24:22.352522168 2015  7  CSeq:1005
20  Sun Feb 8 22:24:22.352522168 2015  7  Transport:RTP/AVP;unicast;client_port=5000-5001
21  Sun Feb 8 22:24:22.352522168 2015  7  --- LINE END ---
22  Sun Feb 8 22:24:22.352622370 2015  6  [rtsp.c:917] RTSP Client 192.168.1.115: Requested Connection closed with fd: 24 and Session ID: 1702957685
23  Sun Feb 8 22:24:22.352666440 2015  6  [rtsp.c:455] Setup Message
24  Sun Feb 8 22:24:22.352696942 2015  3  [rtsp.c:484] Frontend: x, Setup message but no frontend attached.
25  Sun Feb 8 22:24:22.352722766 2015  3  [rtsp.c:931] RTSP Client 192.168.1.115: Setup message failed with fd: 24 and Session ID: 1702957685
26  Sun Feb 8 22:24:22.360127522 2015  7  [rtsp.c:975] RTSP Client 192.168.1.115: Connection closed with fd: 24 and Session ID: 1702957685
27  Sun Feb 8 22:24:22.566433842 2015  6  [ssdp.c:256] SAT>IP Client 192.168.1.115 : contacted us sending reply fd: 24
28  Sun Feb 8 22:24:22.628679856 2015  6  [ssdp.c:256] SAT>IP Client 192.168.1.115 : contacted us sending reply fd: 24
29  Sun Feb 8 22:24:22.636909645 2015  6  [ssdp.c:256] SAT>IP Client 192.168.1.115 : contacted us sending reply fd: 24
30  Sun Feb 8 22:24:22.980479325 2015  6  [connection.c:263] RTSP Connection from 192.168.1.115 Port 52311 (fd: 24)
31  Sun Feb 8 22:24:22.985570699 2015  6  [rtsp.c:872] RTSP Client 192.168.1.115: Found by fd: 24 with Session ID: 1702957685
32  Sun Feb 8 22:24:22.985589726 2015  7  [rtsp.c:903] DESCRIBE rtsp://192.168.1.161/ RTSP/1.0
33  Sun Feb 8 22:24:22.985589726 2015  7  Connection:Close
34  Sun Feb 8 22:24:22.985589726 2015  7  CSeq:1006
35  Sun Feb 8 22:24:22.985589726 2015  7  Accept:application/sdp
36  Sun Feb 8 22:24:22.985589726 2015  7  --- LINE END ---
37  Sun Feb 8 22:24:22.985710485 2015  6  [rtsp.c:917] RTSP Client 192.168.1.115: Requested Connection closed with fd: 24 and Session ID: 1702957685
38  Sun Feb 8 22:24:22.985836380 2015  7  [rtsp.c:646] RTSP/1.0 404 Not Found
39  Sun Feb 8 22:24:22.985836380 2015  7  CSeq: 1006
40  Sun Feb 8 22:24:22.985836380 2015  7  Content-Type: application/sdp
41  Sun Feb 8 22:24:22.985836380 2015  7  Content-Base: rtsp://192.168.1.161/
42  Sun Feb 8 22:24:22.985836380 2015  7  Content-Length: 0
43  Sun Feb 8 22:24:22.985836380 2015  7  --- LINE END ---
44  Sun Feb 8 22:24:28.009425129 2015  7  [rtsp.c:975] RTSP Client 192.168.1.115: Connection closed with fd: 24 and Session ID: 1702957685
45  Sun Feb 8 22:24:28.014566391 2015  6  [connection.c:263] RTSP Connection from 192.168.1.115 Port 52312 (fd: 24)
46  Sun Feb 8 22:24:28.019657837 2015  6  [rtsp.c:872] RTSP Client 192.168.1.115: Found by fd: 24 with Session ID: 1702957685
47  Sun Feb 8 22:24:28.019686178 2015  7  [rtsp.c:903] SETUP rtsp://192.168.1.161:554/?&freq=177.500000&bw=7.000000&msys=dvbt RTSP/1.0
48  Sun Feb 8 22:24:28.019686178 2015  7  Connection:Close
49  Sun Feb 8 22:24:28.019686178 2015  7  CSeq:1007
50  Sun Feb 8 22:24:28.019686178 2015  7  Transport:RTP/AVP;unicast;client_port=5000-5001
51  Sun Feb 8 22:24:28.019686178 2015  7  --- LINE END ---
52  Sun Feb 8 22:24:28.019839293 2015  6  [rtsp.c:917] RTSP Client 192.168.1.115: Requested Connection closed with fd: 24 and Session ID: 1702957685
53  Sun Feb 8 22:24:28.019862331 2015  6  [rtsp.c:455] Setup Message
54  Sun Feb 8 22:24:28.019881375 2015  3  [rtsp.c:484] Frontend: x, Setup message but no frontend attached.
55  Sun Feb 8 22:24:28.019907776 2015  3  [rtsp.c:931] RTSP Client 192.168.1.115: Setup message failed with fd: 24 and Session ID: 1702957685
56  Sun Feb 8 22:24:28.024972308 2015  7  [rtsp.c:975] RTSP Client 192.168.1.115: Connection closed with fd: 24 and Session ID: 1702957685
57  Sun Feb 8 22:24:28.219218828 2015  6  [ssdp.c:256] SAT>IP Client 192.168.1.115 : contacted us sending reply fd: 24
58  Sun Feb 8 22:24:28.221295241 2015  6  [connection.c:263] RTSP Connection from 192.168.1.115 Port 52313 (fd: 24)
59  Sun Feb 8 22:24:28.226386447 2015  6  [rtsp.c:872] RTSP Client 192.168.1.115: Found by fd: 24 with Session ID: 1702957685
[... 959 lines cut ...]
311  Sun Feb 8 22:54:15.973841468 2015  7  [rtsp.c:975] RTSP Client 192.168.1.115: Connection closed with fd: 25 and Session ID: 0205744672
312  Sun Feb 8 22:54:15.978978401 2015  6  [connection.c:263] RTSP Connection from 192.168.1.115 Port 53911 (fd: 25)
313  Sun Feb 8 22:54:15.984066840 2015  6  [rtsp.c:872] RTSP Client 192.168.1.115: Found by fd: 25 with Session ID: 0205744672
314  Sun Feb 8 22:54:15.984086304 2015  7  [rtsp.c:903] DESCRIBE rtsp://192.168.1.161/ RTSP/1.0
315  Sun Feb 8 22:54:15.984086304 2015  7  Connection:Close
316  Sun Feb 8 22:54:15.984086304 2015  7  CSeq:1342
317  Sun Feb 8 22:54:15.984086304 2015  7  Accept:application/sdp
318  Sun Feb 8 22:54:15.984086304 2015  7  --- LINE END ---
319  Sun Feb 8 22:54:15.984153660 2015  6  [rtsp.c:917] RTSP Client 192.168.1.115: Requested Connection closed with fd: 25 and Session ID: 0205744672
320  Sun Feb 8 22:54:15.984258464 2015  7  [rtsp.c:646] RTSP/1.0 200 OK
321  Sun Feb 8 22:54:15.984258464 2015  7  CSeq: 1342
322  Sun Feb 8 22:54:15.984258464 2015  7  Content-Type: application/sdp
323  Sun Feb 8 22:54:15.984258464 2015  7  Content-Base: rtsp://192.168.1.161/
324  Sun Feb 8 22:54:15.984258464 2015  7  Content-Length: 231
325  Sun Feb 8 22:54:15.984258464 2015  7  --- LINE END ---
326  Sun Feb 8 22:54:15.984258464 2015  7  v=0
327  Sun Feb 8 22:54:15.984258464 2015  7  o=- 5678901234 7890123456 IN IP4 192.168.1.161
328  Sun Feb 8 22:54:15.984258464 2015  7  s=SatIPServer:1 2
329  Sun Feb 8 22:54:15.984258464 2015  7  t=0 0
330  Sun Feb 8 22:54:15.984258464 2015  7  m=video 0 RTP/AVP 33
331  Sun Feb 8 22:54:15.984258464 2015  7  c=IN IP4 0.0.0.0
332  Sun Feb 8 22:54:15.984258464 2015  7  a=control:stream=0
333  Sun Feb 8 22:54:15.984258464 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,1,0,11023,0,dvbs2,qpsk,2,0.35,22000,23;pids=
334  Sun Feb 8 22:54:15.984258464 2015  7  a=inactive
335  Sun Feb 8 22:54:19.794999224 2015  7  [rtsp.c:975] RTSP Client 192.168.1.115: Connection closed with fd: 25 and Session ID: 0205744672

Barracuda09, 2015-02-09:
Hi Paul,

Just a quick question:  Did you also close EyeTV also before reseting/reboot server?

Thanks again for testing,

Marc

pbriesch, 2015-02-09:
Hi Marc,

Unfortunately, I cannot tell you the exact order. I can do some more testing for you. Please let me know in more in detail what you need exactly.

Paul

Barracuda09, 2015-02-09:
Hi,

I appreciate your time and testing.
Well order is not so important (I think) but that you also restart EyeTV (Not sure if this makes that much difference)
But my EyeTV behaves also sometimes strange then I restart than it works. 
I think it is in my case  something to do with the state of EyeTV it expects that there is some stream
or something which is no longer there.

Marc

pbriesch, 2015-02-09:
Marc,

Yes, I had to switch EyeTV on and off to make it work. I'm sure that there are some quirks in the client software we use.
Do you have have a SAT>IP server available for testing, e.g. DIGIBIT R1? I have such a device which works so far quite well. It might help to compare the network traffic of the DIGIBIT R1 with the network traffic of SATPI.

Paul

Barracuda09, 2015-02-09:
Hi,

Sorry, I have fixed some problems with dynamically allocating a frontend.
Now DVB-T should work as well again.

Marc

Barracuda09, 2015-02-09:
Sorry.. No do not have any off the shelf  SAT>IP server to test.

For your info I will see if the BOOTID has something todo with this (it is now a fixed value)

Thanks,
Marc

Barracuda09, 2015-03-02:
Hi,

Is this issue is fixed with latest commits?

Marc

Barracuda09, 2015-04-10:
Should be fixed in the mean time

closed 2015-04-10 by Barracuda09