Barracuda09/SATPI #23: No picture with VDR sat-ip-plugin - no pids sent?

pbriesch, 2015-02-10:
Hello Marc,

You previously mentioned that VDR does not send pids according to the log of SATPI. I go t the following:
````
340  Tue Feb 10 23:23:14.176666066 2015  7  m=video 0 RTP/AVP 33
341  Tue Feb 10 23:23:14.176666066 2015  7  c=IN IP4 0.0.0.0
342  Tue Feb 10 23:23:14.176666066 2015  7  a=control:stream=0
343  Tue Feb 10 23:23:14.176666066 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,0,1,0,11837.00,0,dvbs,8psk,2,0.35,27500,34;pids=
344  Tue Feb 10 23:23:14.176666066 2015  7  a=sendonly
345  Tue Feb 10 23:23:14.176666066 2015  7  m=video 0 RTP/AVP 33
346  Tue Feb 10 23:23:14.176666066 2015  7  c=IN IP4 0.0.0.0
347  Tue Feb 10 23:23:14.176666066 2015  7  a=control:stream=1
348  Tue Feb 10 23:23:14.176666066 2015  7  a=fmtp:33 ver=1.0;src=2;tuner=2,0,1,0,11526.00,0,dvbs,qpsk,2,0.35,27500,34;pids=
349  Tue Feb 10 23:23:14.176666066 2015  7  a=inactive
````
VDR does not show a picture. When I quit SATPI and start the DIGIBIT R1, I can get a stream. I did not restart the VDR. Therefore VDR must be sending pids.

Question: Does SATPI get the pids correctly? 

If SATPI does not get the pid information from VDR because of problems parsing these then the log of SATPI can't show them. 

Thanks for looking into this.

Paul

P.S.: Probably you have this link already: http://www.saunalahti.fi/~rahrenbe/vdr/satip/

Barracuda09, 2015-02-11:
Hi Paul,

Thanks for testing. I will setup Minidvblinux (I saw it is also for Raspberry PI) and
do some test myself. Never tried VDR so it will take some time to figure it out, I think.

Marc

Barracuda09, 2015-02-13:
Hi Paul,

I have made some changes to the RTSP Engine.
Did not yet test VDR myself (If put Minidvblinux on one of my PI's but been not able to make it work)
It will discover and SatPI is sending an reply and send some RTSP message.

Marc

pbriesch, 2015-02-13:
Hello Marc,

VDR was running, then I started SATPI, but no picture. Then I restarted SATPI, still no luck. Later today I will have some more time to test.

Paul

[image]
````
0  Fri Feb 13 17:38:26.535861854 2015  6  [satpi.c:259] --- starting SatPI version: 1.0.14~gd48d644 ---
1  Fri Feb 13 17:38:26.535954747 2015  6  [tune.c:437] Detecting frontends in: /dev/dvb
2  Fri Feb 13 17:38:26.536070196 2015  6  [tune.c:447] Frontends found: 2
3  Fri Feb 13 17:38:26.536087311 2015  7  [tune.c:512] Open FE fd: 4 readonly 1
4  Fri Feb 13 17:38:26.536116798 2015  6  [tune.c:152] Frontend Name: STB0899 Multistandard
5  Fri Feb 13 17:38:26.536139905 2015  7  [tune.c:181] Frontend Type: Satellite (DVB-S)
6  Fri Feb 13 17:38:26.536159229 2015  7  [tune.c:185] Frontend Type: Satellite (DVB-S2)
7  Fri Feb 13 17:38:26.536177976 2015  7  [tune.c:177] Frontend Type: DSS
8  Fri Feb 13 17:38:26.536196801 2015  7  [tune.c:216] Frontend Freq: 950000 Hz to 2150000 Hz
9  Fri Feb 13 17:38:26.536215337 2015  7  [tune.c:217] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
10  Fri Feb 13 17:38:26.536240050 2015  7  [tune.c:512] Open FE fd: 4 readonly 1
11  Fri Feb 13 17:38:26.536249898 2015  6  [tune.c:152] Frontend Name: STB0899 Multistandard
12  Fri Feb 13 17:38:26.536256955 2015  7  [tune.c:181] Frontend Type: Satellite (DVB-S)
13  Fri Feb 13 17:38:26.536273984 2015  7  [tune.c:185] Frontend Type: Satellite (DVB-S2)
14  Fri Feb 13 17:38:26.536290331 2015  7  [tune.c:177] Frontend Type: DSS
15  Fri Feb 13 17:38:26.536308421 2015  7  [tune.c:216] Frontend Freq: 950000 Hz to 2150000 Hz
16  Fri Feb 13 17:38:26.536329684 2015  7  [tune.c:217] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
17  Fri Feb 13 17:38:26.536717071 2015  6  [connection.c:336] eth0: 192.168.1.60 [00:1d:7d:a9:eb:f3]
18  Fri Feb 13 17:38:26.536727800 2015  6  [rtsp.c:1040] Setting up RTSP server
19  Fri Feb 13 17:38:26.536780983 2015  6  [http.c:491] Setting up HTTP server
20  Fri Feb 13 17:38:26.536849932 2015  6  [rtp.c:359] Setting up 8 RTP servers
21  Fri Feb 13 17:38:26.537041731 2015  6  [rtcp.c:346] Setting up 8 RTCP servers
22  Fri Feb 13 17:38:26.537282121 2015  6  [ssdp.c:204] Setting up SSDP server with BOOTID: 31
23  Fri Feb 13 17:38:30.249838464 2015  7  [http.c:245] GET web/index.html (size 4201)
24  Fri Feb 13 17:38:30.319843994 2015  7  [http.c:245] GET web/stylesheet.css (size 1463)
25  Fri Feb 13 17:38:30.321582466 2015  7  [http.c:245] GET web/loadxmldoc.js (size 521)
26  Fri Feb 13 17:38:30.366472699 2015  7  [http.c:245] GET web/grid_noise.png (size 4638)
27  Fri Feb 13 17:38:32.409201813 2015  7  [http.c:245] GET web/log.html (size 1784)
28  Fri Feb 13 17:38:32.487759672 2015  7  [http.c:245] GET web/stylesheet.css (size 1463)
29  Fri Feb 13 17:38:32.488584637 2015  7  [http.c:245] GET web/loadxmldoc.js (size 521)
30  Fri Feb 13 17:38:32.534491615 2015  7  [http.c:245] GET web/grid_noise.png (size 4638)
31  Fri Feb 13 17:38:32.535302027 2015  7  [http.c:245] GET web/bg-grain.png (size 5636)
32  Fri Feb 13 17:38:40.016622780 2015  7  [http.c:245] GET web/frontend.html (size 4963)
33  Fri Feb 13 17:38:40.095215526 2015  7  [http.c:245] GET web/stylesheet.css (size 1463)
34  Fri Feb 13 17:38:40.096272334 2015  7  [http.c:245] GET web/loadxmldoc.js (size 521)
35  Fri Feb 13 17:38:40.098988316 2015  7  [http.c:245] GET web/postxmldoc.js (size 739)
36  Fri Feb 13 17:38:40.152619483 2015  7  [http.c:245] GET web/grid_noise.png (size 4638)
37  Fri Feb 13 17:38:45.764364761 2015  7  [http.c:245] GET web/index.html (size 4201)
38  Fri Feb 13 17:38:45.841166900 2015  7  [http.c:245] GET web/stylesheet.css (size 1463)
39  Fri Feb 13 17:38:45.841896926 2015  7  [http.c:245] GET web/loadxmldoc.js (size 521)
40  Fri Feb 13 17:38:45.888719954 2015  7  [http.c:245] GET web/grid_noise.png (size 4638)
41  Fri Feb 13 17:38:47.181886553 2015  7  [http.c:245] GET web/log.html (size 1784)
42  Fri Feb 13 17:38:47.225865056 2015  7  [http.c:245] GET web/stylesheet.css (size 1463)
43  Fri Feb 13 17:38:47.225954232 2015  7  [http.c:245] GET web/loadxmldoc.js (size 521)
44  Fri Feb 13 17:38:47.275883251 2015  7  [http.c:245] GET web/grid_noise.png (size 4638)
45  Fri Feb 13 17:38:47.276902559 2015  7  [http.c:245] GET web/bg-grain.png (size 5636)
46  Fri Feb 13 17:38:56.481280438 2015  6  [ssdp.c:295] SAT>IP Client 192.168.1.21 : tries to discover the network, sending an reply fd: 27
47  Fri Feb 13 17:38:56.482976213 2015  7  [http.c:245] GET web/desc.xml (size 1678)
48  Fri Feb 13 17:38:56.483026550 2015  7  [http.c:465] Requested Connection closed by Client 4
49  Fri Feb 13 17:39:11.303309169 2015  7  [http.c:245] GET web/index.html (size 4201)
50  Fri Feb 13 17:39:11.380100659 2015  7  [http.c:245] GET web/stylesheet.css (size 1463)
51  Fri Feb 13 17:39:11.380835454 2015  7  [http.c:245] GET web/loadxmldoc.js (size 521)
52  Fri Feb 13 17:39:11.423845942 2015  7  [http.c:245] GET web/grid_noise.png (size 4638)
53  Fri Feb 13 17:39:13.497757830 2015  7  [http.c:245] GET web/index.html (size 4201)
54  Fri Feb 13 17:39:13.580276443 2015  7  [http.c:245] GET web/stylesheet.css (size 1463)
55  Fri Feb 13 17:39:13.581004020 2015  7  [http.c:245] GET web/loadxmldoc.js (size 521)
56  Fri Feb 13 17:39:13.630730563 2015  7  [http.c:245] GET web/grid_noise.png (size 4638)
57  Fri Feb 13 17:39:15.422134821 2015  7  [http.c:245] GET web/log.html (size 1784)
58  Fri Feb 13 17:39:15.492011911 2015  7  [http.c:245] GET web/loadxmldoc.js (size 521)
[... cut ...]
156  Fri Feb 13 17:43:41.081666670 2015  7  [http.c:245] GET web/grid_noise.png (size 4638)
157  Fri Feb 13 17:43:41.083013401 2015  7  [http.c:245] GET web/bg-grain.png (size 5636)
158  Fri Feb 13 17:43:46.380134348 2015  6  [ssdp.c:295] SAT>IP Client 192.168.1.21 : tries to discover the network, sending an reply fd: 31
159  Fri Feb 13 17:43:46.381710833 2015  7  [http.c:245] GET web/desc.xml (size 1678)
160  Fri Feb 13 17:43:46.381760799 2015  7  [http.c:465] Requested Connection closed by Client 4

pbriesch, 2015-02-13:
Marc,
I could test a bit more: First I started SatPi, then VDR. :-1: 
[image]
````
0  Fri Feb 13 19:30:24.508434894 2015  6  [satpi.c:259] --- starting SatPI version: 1.0.14~gd48d644 ---
1  Fri Feb 13 19:30:24.508520328 2015  6  [tune.c:437] Detecting frontends in: /dev/dvb
2  Fri Feb 13 19:30:24.508636609 2015  6  [tune.c:447] Frontends found: 2
3  Fri Feb 13 19:30:24.508653083 2015  7  [tune.c:512] Open FE fd: 4 readonly 1
4  Fri Feb 13 19:30:24.508684100 2015  6  [tune.c:152] Frontend Name: STB0899 Multistandard
5  Fri Feb 13 19:30:24.508707050 2015  7  [tune.c:181] Frontend Type: Satellite (DVB-S)
6  Fri Feb 13 19:30:24.508726288 2015  7  [tune.c:185] Frontend Type: Satellite (DVB-S2)
7  Fri Feb 13 19:30:24.508745053 2015  7  [tune.c:177] Frontend Type: DSS
8  Fri Feb 13 19:30:24.508763935 2015  7  [tune.c:216] Frontend Freq: 950000 Hz to 2150000 Hz
9  Fri Feb 13 19:30:24.508782779 2015  7  [tune.c:217] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
10  Fri Feb 13 19:30:24.508807716 2015  7  [tune.c:512] Open FE fd: 4 readonly 1
11  Fri Feb 13 19:30:24.508817961 2015  6  [tune.c:152] Frontend Name: STB0899 Multistandard
12  Fri Feb 13 19:30:24.508825117 2015  7  [tune.c:181] Frontend Type: Satellite (DVB-S)
13  Fri Feb 13 19:30:24.508842014 2015  7  [tune.c:185] Frontend Type: Satellite (DVB-S2)
14  Fri Feb 13 19:30:24.508858274 2015  7  [tune.c:177] Frontend Type: DSS
15  Fri Feb 13 19:30:24.508876353 2015  7  [tune.c:216] Frontend Freq: 950000 Hz to 2150000 Hz
16  Fri Feb 13 19:30:24.508897717 2015  7  [tune.c:217] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
17  Fri Feb 13 19:30:24.509276150 2015  6  [connection.c:336] eth0: 192.168.1.60 [00:1d:7d:a9:eb:f3]
18  Fri Feb 13 19:30:24.509287715 2015  6  [rtsp.c:1040] Setting up RTSP server
19  Fri Feb 13 19:30:24.509341869 2015  6  [http.c:491] Setting up HTTP server
20  Fri Feb 13 19:30:24.509376197 2015  6  [rtp.c:359] Setting up 8 RTP servers
21  Fri Feb 13 19:30:24.509583342 2015  6  [rtcp.c:346] Setting up 8 RTCP servers
22  Fri Feb 13 19:30:24.509820776 2015  6  [ssdp.c:204] Setting up SSDP server with BOOTID: 34
23  Fri Feb 13 19:30:38.095750043 2015  6  [ssdp.c:295] SAT>IP Client 192.168.1.21 : tries to discover the network, sending an reply fd: 26
24  Fri Feb 13 19:30:38.098523877 2015  7  [http.c:245] GET web/desc.xml (size 1678)
25  Fri Feb 13 19:30:38.098627049 2015  7  [http.c:465] Requested Connection closed by Client 3
26  Fri Feb 13 19:30:40.392049346 2015  6  [connection.c:263] RTSP Connection from 192.168.1.21 Port 53402 (fd: 26)
27  Fri Feb 13 19:30:40.397188477 2015  6  [rtsp.c:888] RTSP Client 192.168.1.21: Found empty slot for fd: 26
28  Fri Feb 13 19:30:40.397214270 2015  6  [rtsp.c:912] RTSP Client 192.168.1.21: With fd: 26 is outide Session
29  Fri Feb 13 19:30:40.397235214 2015  7  [rtsp.c:922] OPTIONS rtsp://192.168.1.60/ RTSP/1.0
30  Fri Feb 13 19:30:40.397235214 2015  7  CSeq: 1
31  Fri Feb 13 19:30:40.397235214 2015  7  User-Agent: vdr-satip/1.0.2-GIT-0-159-g4d42593 (device 0)
32  Fri Feb 13 19:30:40.397235214 2015  7  --- LINE END ---
33  Fri Feb 13 19:30:40.402374663 2015  6  [rtsp.c:878] RTSP Client 192.168.1.21: Found by fd: 26
34  Fri Feb 13 19:30:40.402395337 2015  6  [rtsp.c:908] RTSP Client 192.168.1.21: With fd: 26 giving Session ID: 1702957685
35  Fri Feb 13 19:30:40.402408159 2015  7  [rtsp.c:922] SETUP rtsp://192.168.1.60/?freq=11836&src=1&sr=27500&pol=h&fec=34&msys=dvbs RTSP/1.0
36  Fri Feb 13 19:30:40.402408159 2015  7  CSeq: 2
37  Fri Feb 13 19:30:40.402408159 2015  7  Transport: RTP/AVP;unicast;client_port=53014-53015
38  Fri Feb 13 19:30:40.402408159 2015  7  User-Agent: vdr-satip/1.0.2-GIT-0-159-g4d42593 (device 0)
39  Fri Feb 13 19:30:40.402408159 2015  7  If-Modified-Since: Thu, 01 Jan 1970 00:00:00 GMT
40  Fri Feb 13 19:30:40.402408159 2015  7  --- LINE END ---
41  Fri Feb 13 19:30:40.402482919 2015  6  [rtsp.c:410] Frontend: 0, With dvbs Attaching dynamically to client 192.168.1.21 with session ID: 1702957685
42  Fri Feb 13 19:30:40.402528501 2015  6  [rtsp.c:457] Setup Message
43  Fri Feb 13 19:30:40.402613984 2015  7  [tune.c:512] Open FE fd: 27 readonly 0
44  Fri Feb 13 19:30:40.402626096 2015  6  [tune.c:572] Frontend: 0, Opened FE fd: 27.
45  Fri Feb 13 19:30:40.402633173 2015  7  [tune.c:396] Frontend: 0, Start tuning process
46  Fri Feb 13 19:30:40.402639428 2015  6  [tune.c:325] Frontend: 0, Sending DiSEqC [e0] [10] [38] [f3]
47  Fri Feb 13 19:30:40.779625094 2015  6  [tune.c:583] Frontend: 0, Waiting on lock.
48  Fri Feb 13 19:30:41.102968947 2015  6  [tune.c:594] Frontend: 0, Tuned and locked (FE status 31).
49  Fri Feb 13 19:30:41.103139900 2015  6  [tune.c:615] Frontend: 0, Opened DVR fd: 28.
50  Fri Feb 13 19:30:41.103192300 2015  6  [rtp.c:98] Frontend: 0, Start streaming to 192.168.1.21 (53014 - 53015)
51  Fri Feb 13 19:30:41.103248265 2015  6  [rtcp.c:260] Frontend: 0, Start RTCP stream to 192.168.1.21 (53014 - 53015)
52  Fri Feb 13 19:30:41.103275074 2015  6  [rtcp.c:274] Frontend: 0, Open Frontend Monitor for 192.168.1.21 (53014 - 53015)
53  Fri Feb 13 19:30:41.103286456 2015  7  [tune.c:512] Open FE fd: 29 readonly 1
54  Fri Feb 13 19:30:41.108250021 2015  6  [rtsp.c:861] Client connection with sessionID 1702957685
55  Fri Feb 13 19:30:41.108273530 2015  6  [rtsp.c:866] RTSP Client 192.168.1.21: Found by sessionID 1702957685 with fd: 26
[... cut ...]
60  Fri Feb 13 19:30:41.108292756 2015  7  --- LINE END ---
61  Fri Feb 13 19:30:41.108375163 2015  6  [rtsp.c:522] Play Message
62  Fri Feb 13 19:30:41.108389815 2015  7  [tune.c:396] Frontend: 0, Start tuning process
63  Fri Feb 13 19:30:41.108402168 2015  6  [tune.c:325] Frontend: 0, Sending DiSEqC [e0] [10] [38] [f3]
64  Fri Feb 13 19:30:41.319625244 2015  6  [tune.c:583] Frontend: 0, Waiting on lock.
65  Fri Feb 13 19:30:41.614932705 2015  6  [tune.c:594] Frontend: 0, Tuned and locked (FE status 31).

Barracuda09, 2015-02-13:
Hi Paul,

Thanks..Mmm.. strange is this. It seems still there are no PIDs in the SETUP nor PLAY messages.

I have to setup VDR here to test it myself.

Marc

pbriesch, 2015-02-14:
Marc,

there are now new files available for download for MLD. Give it an other try.
http://www.minidvblinux.de/download/4.0.1-rpi
Paul

Barracuda09, 2015-03-02:
Hi,

Is this issue is fixed with latest commits?  (Sorry still have to check it myself though)

Marc

Barracuda09, 2015-03-09:
Hi Paul,

Just did some quick tests with Kodi, but here it seems to work with satip - vtuner. Just only the satip plugin is not doing ok, It gets the XML but then nothing (or I have to configure it better)... 
But in all it feels 'too' complicated for my tast. 
I Will look into it in more detail to try to figure it out .

pbriesch, 2015-03-09:
Marc,

let me know when I can do some more testing for you! BTW, there is now MLD out for the RPI 2!

Paul

Barracuda09, 2015-03-09:
Hi,

Yes I have seen it to (Only have one of them)... but I installed it now on a virtual machine.

Barracuda09, 2015-09-08:
I will close this for now... If anybody still has issues regarding this bug please make a new bug report.

closed 2015-09-08 by Barracuda09