Barracuda09/SATPI #24: clashing DEVICEID

pbriesch, 2015-02-10:
Marc,

I'm not sure if this is an issue. But I would like to show to you the log out during following scenario:

VDR (192.168.1.21) and DIGIBIT R1 ((192.168.1.55) )were running. Then I started SATPI (192.168.1.161)  and following message came up: 

````
23  Tue Feb 10 23:43:13.202504219 2015  6  [ssdp.c:258] SAT>IP Server 192.168.1.55: contacted us because of clashing DEVICEID
````
Then I switched the DIGIBIT R1 off and VDR lost the picture off course. After that, VDR contacted SATPI but the picture came not back.

Paul

````
0  Tue Feb 10 23:43:12.294681861 2015  6  [satpi.c:259] --- starting SatPI version: 1.0.6~g86117f3 ---
1  Tue Feb 10 23:43:12.294767355 2015  6  [tune.c:437] Detecting frontends in: /dev/dvb
2  Tue Feb 10 23:43:12.294879373 2015  6  [tune.c:447] Frontends found: 2
3  Tue Feb 10 23:43:12.294896253 2015  7  [tune.c:512] Open FE fd: 4 readonly 1
4  Tue Feb 10 23:43:12.294925002 2015  6  [tune.c:152] Frontend Name: STB0899 Multistandard
5  Tue Feb 10 23:43:12.294947407 2015  7  [tune.c:181] Frontend Type: Satellite (DVB-S)
6  Tue Feb 10 23:43:12.294966410 2015  7  [tune.c:185] Frontend Type: Satellite (DVB-S2)
7  Tue Feb 10 23:43:12.294984875 2015  7  [tune.c:177] Frontend Type: DSS
8  Tue Feb 10 23:43:12.295003594 2015  7  [tune.c:216] Frontend Freq: 950000 Hz to 2150000 Hz
9  Tue Feb 10 23:43:12.295022137 2015  7  [tune.c:217] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
10  Tue Feb 10 23:43:12.295046309 2015  7  [tune.c:512] Open FE fd: 4 readonly 1
11  Tue Feb 10 23:43:12.295056046 2015  6  [tune.c:152] Frontend Name: STB0899 Multistandard
12  Tue Feb 10 23:43:12.295063103 2015  7  [tune.c:181] Frontend Type: Satellite (DVB-S)
13  Tue Feb 10 23:43:12.295079912 2015  7  [tune.c:185] Frontend Type: Satellite (DVB-S2)
14  Tue Feb 10 23:43:12.295096061 2015  7  [tune.c:177] Frontend Type: DSS
15  Tue Feb 10 23:43:12.295114121 2015  7  [tune.c:216] Frontend Freq: 950000 Hz to 2150000 Hz
16  Tue Feb 10 23:43:12.295135469 2015  7  [tune.c:217] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
17  Tue Feb 10 23:43:12.295499924 2015  6  [connection.c:336] eth0: 192.168.1.161 [00:1d:7d:a9:eb:f3]
18  Tue Feb 10 23:43:12.295510681 2015  6  [rtsp.c:1017] Setting up RTSP server
19  Tue Feb 10 23:43:12.295565269 2015  6  [http.c:490] Setting up HTTP server
20  Tue Feb 10 23:43:12.295601107 2015  6  [rtp.c:348] Setting up 8 RTP servers
21  Tue Feb 10 23:43:12.295808301 2015  6  [rtcp.c:338] Setting up 8 RTCP servers
22  Tue Feb 10 23:43:12.296043528 2015  6  [ssdp.c:196] Setting up SSDP server with BOOTID: 14
23  Tue Feb 10 23:43:13.202504219 2015  6  [ssdp.c:258] SAT>IP Server 192.168.1.55: contacted us because of clashing DEVICEID
24  Tue Feb 10 23:43:13.602741363 2015  6  [ssdp.c:281] SAT>IP Client 192.168.1.55 : contacted us sending reply fd: 24
25  Tue Feb 10 23:43:26.645500804 2015  6  [ssdp.c:281] SAT>IP Client 192.168.1.21 : contacted us sending reply fd: 24
26  Tue Feb 10 23:43:26.646924954 2015  7  [http.c:245] GET web/desc.xml (size 1594)
27  Tue Feb 10 23:43:26.646984055 2015  7  [http.c:464] Requested Connection closed by Client 2
28  Tue Feb 10 23:44:02.230311838 2015  7  [http.c:245] GET web/index.html (size 4201)
29  Tue Feb 10 23:44:02.303239400 2015  7  [http.c:245] GET web/stylesheet.css (size 1463)
30  Tue Feb 10 23:44:02.306574399 2015  7  [http.c:245] GET web/loadxmldoc.js (size 521)
31  Tue Feb 10 23:44:02.361130167 2015  7  [http.c:245] GET web/grid_noise.png (size 4638)
32  Tue Feb 10 23:44:04.918832501 2015  7  [http.c:245] GET web/log.html (size 1784)
33  Tue Feb 10 23:44:04.993697949 2015  7  [http.c:245] GET web/stylesheet.css (size 1463)
34  Tue Feb 10 23:44:04.993978209 2015  7  [http.c:245] GET web/loadxmldoc.js (size 521)
35  Tue Feb 10 23:44:05.046369014 2015  7  [http.c:245] GET web/grid_noise.png (size 4638)
36  Tue Feb 10 23:44:05.047144516 2015  7  [http.c:245] GET web/bg-grain.png (size 5636)
37  Tue Feb 10 23:44:26.780461555 2015  6  [ssdp.c:281] SAT>IP Client 192.168.1.21 : contacted us sending reply fd: 26
38  Tue Feb 10 23:44:26.781866279 2015  7  [http.c:245] GET web/desc.xml (size 1594)
39  Tue Feb 10 23:44:26.781912571 2015  7  [http.c:464] Requested Connection closed by Client 3
40  Tue Feb 10 23:45:26.916455803 2015  6  [ssdp.c:281] SAT>IP Client 192.168.1.21 : contacted us sending reply fd: 26
41  Tue Feb 10 23:45:26.918085091 2015  7  [http.c:245] GET web/desc.xml (size 1594)
42  Tue Feb 10 23:45:26.918133008 2015  7  [http.c:464] Requested Connection closed by Client 3
43  Tue Feb 10 23:45:41.725133070 2015  7  [http.c:245] GET web/index.html (size 4201)
44  Tue Feb 10 23:45:41.811707844 2015  7  [http.c:245] GET web/stylesheet.css (size 1463)
45  Tue Feb 10 23:45:41.812393249 2015  7  [http.c:245] GET web/loadxmldoc.js (size 521)
46  Tue Feb 10 23:45:41.865481107 2015  7  [http.c:245] GET web/grid_noise.png (size 4638)
47  Tue Feb 10 23:45:43.957278772 2015  7  [http.c:245] GET web/log.html (size 1784)
48  Tue Feb 10 23:45:44.037388032 2015  7  [http.c:245] GET web/stylesheet.css (size 1463)
49  Tue Feb 10 23:45:44.038035040 2015  7  [http.c:245] GET web/loadxmldoc.js (size 521)
50  Tue Feb 10 23:45:44.090554276 2015  7  [http.c:245] GET web/grid_noise.png (size 4638)
51  Tue Feb 10 23:45:44.091325815 2015  7  [http.c:245] GET web/bg-grain.png (size 5636)
52  Tue Feb 10 23:45:46.041520016 2015  7  [http.c:245] GET web/frontend.html (size 4963)
53  Tue Feb 10 23:45:46.125796925 2015  7  [http.c:245] GET web/stylesheet.css (size 1463)
54  Tue Feb 10 23:45:46.126094957 2015  7  [http.c:245] GET web/loadxmldoc.js (size 521)
55  Tue Feb 10 23:45:46.127996659 2015  7  [http.c:245] GET web/postxmldoc.js (size 739)
56  Tue Feb 10 23:45:46.169477220 2015  7  [http.c:245] GET web/grid_noise.png (size 4638)
57  Tue Feb 10 23:45:48.773407414 2015  7  [http.c:245] GET web/index.html (size 4201)
58  Tue Feb 10 23:45:48.850648448 2015  7  [http.c:245] GET web/loadxmldoc.js (size 521)
59  Tue Feb 10 23:45:48.850732115 2015  7  [http.c:245] GET web/stylesheet.css (size 1463)
60  Tue Feb 10 23:45:48.897127235 2015  7  [http.c:245] GET web/grid_noise.png (size 4638)
61  Tue Feb 10 23:45:54.368848878 2015  7  [http.c:245] GET web/index.html (size 4201)
62  Tue Feb 10 23:45:54.448116367 2015  7  [http.c:245] GET web/stylesheet.css (size 1463)
63  Tue Feb 10 23:45:54.448433065 2015  7  [http.c:245] GET web/loadxmldoc.js (size 521)
64  Tue Feb 10 23:45:54.495749031 2015  7  [http.c:245] GET web/grid_noise.png (size 4638)
65  Tue Feb 10 23:45:58.798041487 2015  7  [http.c:245] GET web/log.html (size 1784)
66  Tue Feb 10 23:45:58.866674756 2015  7  [http.c:245] GET web/stylesheet.css (size 1463)
67  Tue Feb 10 23:45:58.867363359 2015  7  [http.c:245] GET web/loadxmldoc.js (size 521)
68  Tue Feb 10 23:45:58.910991572 2015  7  [http.c:245] GET web/grid_noise.png (size 4638)
69  Tue Feb 10 23:45:58.911808954 2015  7  [http.c:245] GET web/bg-grain.png (size 5636)
70  Tue Feb 10 23:46:27.050314111 2015  6  [ssdp.c:281] SAT>IP Client 192.168.1.21 : contacted us sending reply fd: 27
71  Tue Feb 10 23:46:27.052014810 2015  7  [http.c:245] GET web/desc.xml (size 1594)
72  Tue Feb 10 23:46:27.052060619 2015  7  [http.c:464] Requested Connection closed by Client 4
73  Tue Feb 10 23:47:27.185308139 2015  6  [ssdp.c:281] SAT>IP Client 192.168.1.21 : contacted us sending reply fd: 27
74  Tue Feb 10 23:47:27.187082040 2015  7  [http.c:245] GET web/desc.xml (size 1594)
75  Tue Feb 10 23:47:27.187125283 2015  7  [http.c:464] Requested Connection closed by Client 4

Barracuda09, 2015-02-11:
Hi Paul,

According to the specs every device should have a none clashing DEVICEID.
So the DIGIBIT R1 was on at first and it probably took 1. Then SatPI is started it will also start with
1 and it will broadcast this over the network. The  DIGIBIT R1 should defend its DEVICEID so it should
contact SatPI.

Sorry, but I don't know if this scenario should work. But I will try to setup VDR to test it myself.

Marc

9000h, 2015-03-18:
Hi,
the digibit r1 defend may not work or is not implenented.

here a snip from a short test with 2 satip servers in my net
'''
[        Satpi.cpp:243] --- starting SatPI version: 1.5.26~gbd9dab9 ---
[  InterfaceAttr.h:094] eth0: 192.168.178.126 [02:48:05:c1:fa:bd]
[      Streams.cpp:118] Current DVB_API_VERSION: 5.5
[      Streams.cpp:120] Detecting frontends in: /dev/dvb
[      Streams.cpp:130] Frontends found: 1
[     Frontend.cpp:136] Frontend Name: DRXK DVB-C DVB-T
[     Frontend.cpp:227] Frontend Type: Cable (Annex A)
[     Frontend.cpp:231] Frontend Type: Cable (Annex C)
[     Frontend.cpp:218] Frontend Type: Terrestrial (DVB-T)
[     Frontend.cpp:249] Frontend Freq: 47000000 Hz to 865000000 Hz
[     Frontend.cpp:250] Frontend srat: 870000 symbols/s to 11700000 symbols/s
[   RtspServer.cpp:059] Setting up RTSP server
[   SsdpServer.cpp:107] Setting up SSDP server with BOOTID: 1
[   HttpServer.cpp:074] Setting up HTTP server
[    TcpSocket.cpp:149] HTTP Connection from 192.168.178.124 Port 24915 (fd: 8)
[   HttpServer.cpp:167] Client: 192.168.178.124 requesed /desc.xml
[    TcpSocket.cpp:085] HTTP Client 192.168.178.124: Connection closed with fd: 8
[   SsdpServer.cpp:158] Found SAT>IP Server 192.168.178.133: with DEVICEID 6
[   SsdpServer.cpp:158] Found SAT>IP Server 192.168.178.133: with DEVICEID 6
[   SsdpServer.cpp:158] Found SAT>IP Server 192.168.178.133: with DEVICEID 6
[    TcpSocket.cpp:149] HTTP Connection from 192.168.178.130 Port 49371 (fd: 8)
[    TcpSocket.cpp:149] HTTP Connection from 192.168.178.130 Port 49372 (fd: 9)
[    TcpSocket.cpp:149] HTTP Connection from 192.168.178.130 Port 49373 (fd: 10)
[   SsdpServer.cpp:146] Found SAT>IP Server 192.168.178.135: with clashing DEVICEID 1
[   SsdpServer.cpp:146] Found SAT>IP Server 192.168.178.135: with clashing DEVICEID 1
[   SsdpServer.cpp:146] Found SAT>IP Server 192.168.178.135: with clashing DEVICEID 1
[   HttpServer.cpp:103] GET web/config.xml
[   HttpServer.cpp:104] File not found: No such file or directory (code 2)
[   HttpServer.cpp:103] GET web/config.xml
[   HttpServer.cpp:104] File not found: No such file or directory (code 2)
[   HttpServer.cpp:103] GET web/config.xml
[   HttpServer.cpp:104] File not found: No such file or directory (code 2)
[    TcpSocket.cpp:149] RTSP Connection from 192.168.178.130 Port 49374 (fd: 11)
'''

Barracuda09, 2015-03-29:
Should work now.

closed 2015-03-29 by Barracuda09