Barracuda09/SATPI #25: EyeTV on Mac "steals" connection from EyeTV on iPad

pbriesch, 2015-02-13:
Hi Marc,

Sorry for keeping you busy. I found some interesting behavior:

While running EyeTV on both Mac and iPad concurrently only one client could get a stream. As you can see in the screenshot only one fronted was attached. The second one was idle.
When I was watching a channel on the Mac the iPad complained that the server is already in use. So far so good. Then I switched to an other channel on the iPad and the stream on the Mac went away and I could watch now TV on the iPad. When I changed the channel on the Mac then, guess what, the iPad lost connection and I could watch TV on the Mac and so on.

Paul

[image]

````
0  Fri Feb 13 22:04:28.215786601 2015  7  [tune.c:556] Frontend: 0, Remove filter PID: 45 - fd: 47 - Packet Count: 1
1  Fri Feb 13 22:04:28.215812896 2015  7  [tune.c:552] Frontend: 0, Set filter PID: 459 - fd: 47
2  Fri Feb 13 22:04:28.305172733 2015  7  [rtsp.c:997] RTSP Client 192.168.1.107: Connection closed with fd: 28 and Session ID: 1702957685
3  Fri Feb 13 22:04:28.310292608 2015  6  [connection.c:263] RTSP Connection from 192.168.1.107 Port 52751 (fd: 28)
4  Fri Feb 13 22:04:28.315381372 2015  6  [rtsp.c:861] Client connection with sessionID 1702957685
5  Fri Feb 13 22:04:28.315400744 2015  6  [rtsp.c:866] RTSP Client 192.168.1.107: Found by sessionID 1702957685 with fd: 28
6  Fri Feb 13 22:04:28.315415512 2015  7  [rtsp.c:922] PLAY rtsp://192.168.1.60:554/stream=0?delpids=459&addpids=129 RTSP/1.0
7  Fri Feb 13 22:04:28.315415512 2015  7  Connection:Close
8  Fri Feb 13 22:04:28.315415512 2015  7  CSeq:1255
9  Fri Feb 13 22:04:28.315415512 2015  7  Session:1702957685
10  Fri Feb 13 22:04:28.315415512 2015  7  --- LINE END ---
11  Fri Feb 13 22:04:28.315474136 2015  6  [rtsp.c:936] RTSP Client 192.168.1.107: Requested Connection closed with fd: 28 and Session ID: 1702957685
12  Fri Feb 13 22:04:28.315511037 2015  6  [rtsp.c:522] Play Message
13  Fri Feb 13 22:04:28.315553034 2015  7  [tune.c:552] Frontend: 0, Set filter PID: 129 - fd: 48
14  Fri Feb 13 22:04:28.315569193 2015  7  [tune.c:556] Frontend: 0, Remove filter PID: 459 - fd: 47 - Packet Count: 1
15  Fri Feb 13 22:04:28.405519767 2015  7  [rtsp.c:997] RTSP Client 192.168.1.107: Connection closed with fd: 28 and Session ID: 1702957685
16  Fri Feb 13 22:04:28.410633968 2015  6  [connection.c:263] RTSP Connection from 192.168.1.107 Port 52753 (fd: 28)
17  Fri Feb 13 22:04:28.415724269 2015  6  [rtsp.c:861] Client connection with sessionID 1702957685
18  Fri Feb 13 22:04:28.415743477 2015  6  [rtsp.c:866] RTSP Client 192.168.1.107: Found by sessionID 1702957685 with fd: 28
19  Fri Feb 13 22:04:28.415758387 2015  7  [rtsp.c:922] PLAY rtsp://192.168.1.60:554/stream=0?delpids=129&addpids=1100 RTSP/1.0
20  Fri Feb 13 22:04:28.415758387 2015  7  Connection:Close
21  Fri Feb 13 22:04:28.415758387 2015  7  CSeq:1256
22  Fri Feb 13 22:04:28.415758387 2015  7  Session:1702957685
23  Fri Feb 13 22:04:28.415758387 2015  7  --- LINE END ---
24  Fri Feb 13 22:04:28.415925369 2015  6  [rtsp.c:936] RTSP Client 192.168.1.107: Requested Connection closed with fd: 28 and Session ID: 1702957685
25  Fri Feb 13 22:04:28.415958081 2015  6  [rtsp.c:522] Play Message
26  Fri Feb 13 22:04:28.415968078 2015  7  [tune.c:556] Frontend: 0, Remove filter PID: 129 - fd: 48 - Packet Count: 2
27  Fri Feb 13 22:04:28.416002050 2015  7  [tune.c:552] Frontend: 0, Set filter PID: 1100 - fd: 47
28  Fri Feb 13 22:04:28.505339226 2015  7  [rtsp.c:997] RTSP Client 192.168.1.107: Connection closed with fd: 28 and Session ID: 1702957685
29  Fri Feb 13 22:04:28.510444857 2015  6  [connection.c:263] RTSP Connection from 192.168.1.107 Port 52754 (fd: 28)
30  Fri Feb 13 22:04:28.515528398 2015  6  [rtsp.c:861] Client connection with sessionID 1702957685
31  Fri Feb 13 22:04:28.515548100 2015  6  [rtsp.c:866] RTSP Client 192.168.1.107: Found by sessionID 1702957685 with fd: 28
32  Fri Feb 13 22:04:28.515562857 2015  7  [rtsp.c:922] PLAY rtsp://192.168.1.60:554/stream=0?delpids=1100&addpids=38 RTSP/1.0
33  Fri Feb 13 22:04:28.515562857 2015  7  Connection:Close
34  Fri Feb 13 22:04:28.515562857 2015  7  CSeq:1257
35  Fri Feb 13 22:04:28.515562857 2015  7  Session:1702957685
36  Fri Feb 13 22:04:28.515562857 2015  7  --- LINE END ---
37  Fri Feb 13 22:04:28.515621361 2015  6  [rtsp.c:936] RTSP Client 192.168.1.107: Requested Connection closed with fd: 28 and Session ID: 1702957685
38  Fri Feb 13 22:04:28.515655697 2015  6  [rtsp.c:522] Play Message
39  Fri Feb 13 22:04:28.515687718 2015  7  [tune.c:552] Frontend: 0, Set filter PID: 38 - fd: 48
40  Fri Feb 13 22:04:28.515711722 2015  7  [tune.c:556] Frontend: 0, Remove filter PID: 1100 - fd: 47 - Packet Count: 1
41  Fri Feb 13 22:04:28.605195660 2015  7  [rtsp.c:997] RTSP Client 192.168.1.107: Connection closed with fd: 28 and Session ID: 1702957685
42  Fri Feb 13 22:04:28.610292588 2015  6  [connection.c:263] RTSP Connection from 192.168.1.107 Port 52756 (fd: 28)
43  Fri Feb 13 22:04:28.615385331 2015  6  [rtsp.c:861] Client connection with sessionID 1702957685
44  Fri Feb 13 22:04:28.615405787 2015  6  [rtsp.c:866] RTSP Client 192.168.1.107: Found by sessionID 1702957685 with fd: 28
45  Fri Feb 13 22:04:28.615413531 2015  7  [rtsp.c:922] PLAY rtsp://192.168.1.60:554/stream=0?delpids=38 RTSP/1.0
46  Fri Feb 13 22:04:28.615413531 2015  7  Connection:Close
47  Fri Feb 13 22:04:28.615413531 2015  7  CSeq:1258
48  Fri Feb 13 22:04:28.615413531 2015  7  Session:1702957685
49  Fri Feb 13 22:04:28.615413531 2015  7  --- LINE END ---
50  Fri Feb 13 22:04:28.615445635 2015  6  [rtsp.c:936] RTSP Client 192.168.1.107: Requested Connection closed with fd: 28 and Session ID: 1702957685
51  Fri Feb 13 22:04:28.615463358 2015  6  [rtsp.c:522] Play Message
52  Fri Feb 13 22:04:28.615471702 2015  7  [tune.c:556] Frontend: 0, Remove filter PID: 38 - fd: 48 - Packet Count: 2
53  Fri Feb 13 22:04:30.173285166 2015  7  [rtsp.c:997] RTSP Client 192.168.1.107: Connection closed with fd: 30 and Session ID: 0
54  Fri Feb 13 22:04:30.178420903 2015  6  [connection.c:263] RTSP Connection from 192.168.1.107 Port 52769 (fd: 30)
55  Fri Feb 13 22:04:30.183509239 2015  6  [rtsp.c:888] RTSP Client 192.168.1.107: Found empty slot for fd: 30
56  Fri Feb 13 22:04:30.183530735 2015  6  [rtsp.c:912] RTSP Client 192.168.1.107: With fd: 30 is outide Session
57  Fri Feb 13 22:04:30.183543826 2015  7  [rtsp.c:922] DESCRIBE rtsp://192.168.1.60/ RTSP/1.0
[... 267 lines cut ...]
325  Fri Feb 13 22:04:41.044109244 2015  7  [rtsp.c:997] RTSP Client 192.168.1.107: Connection closed with fd: 39 and Session ID: 0
326  Fri Feb 13 22:04:41.049206142 2015  6  [connection.c:263] RTSP Connection from 192.168.1.107 Port 52857 (fd: 39)
327  Fri Feb 13 22:04:41.054289469 2015  6  [rtsp.c:888] RTSP Client 192.168.1.107: Found empty slot for fd: 39
328  Fri Feb 13 22:04:41.054309393 2015  6  [rtsp.c:912] RTSP Client 192.168.1.107: With fd: 39 is outide Session
329  Fri Feb 13 22:04:41.054322117 2015  7  [rtsp.c:922] DESCRIBE rtsp://192.168.1.60/ RTSP/1.0
330  Fri Feb 13 22:04:41.054322117 2015  7  Connection:Close
331  Fri Feb 13 22:04:41.054322117 2015  7  CSeq:1292
332  Fri Feb 13 22:04:41.054322117 2015  7  Accept:application/sdp
333  Fri Feb 13 22:04:41.054322117 2015  7  --- LINE END ---
334  Fri Feb 13 22:04:41.054484692 2015  6  [rtsp.c:936] RTSP Client 192.168.1.107: Requested Connection closed with fd: 39 and Session ID: 0
335  Fri Feb 13 22:04:41.054556776 2015  7  [rtsp.c:645] RTSP/1.0 200 OK
336  Fri Feb 13 22:04:41.054556776 2015  7  CSeq: 1292
337  Fri Feb 13 22:04:41.054556776 2015  7  Content-Type: application/sdp
338  Fri Feb 13 22:04:41.054556776 2015  7  Content-Base: rtsp://192.168.1.60/
339  Fri Feb 13 22:04:41.054556776 2015  7  Content-Length: 281
340  Fri Feb 13 22:04:41.054556776 2015  7  --- LINE END ---
341  Fri Feb 13 22:04:41.054556776 2015  7  v=0
342  Fri Feb 13 22:04:41.054556776 2015  7  o=- 5678901234 7890123456 IN IP4 192.168.1.60
343  Fri Feb 13 22:04:41.054556776 2015  7  s=SatIPServer:1 2
344  Fri Feb 13 22:04:41.054556776 2015  7  t=0 0
345  Fri Feb 13 22:04:41.054556776 2015  7  m=video 0 RTP/AVP 33
346  Fri Feb 13 22:04:41.054556776 2015  7  c=IN IP4 0.0.0.0
347  Fri Feb 13 22:04:41.054556776 2015  7  a=control:stream=0
348  Fri Feb 13 22:04:41.054556776 2015  7  a=fmtp:33 ver=1.0;src=1;tuner=1,1,1,0,12188.00,0,dvbs,qpsk,2,0.35,27500,34;pids=0,16,17,18,19,20,32,33,36,44,104,105,110,163,8190
349  Fri Feb 13 22:04:41.054556776 2015  7  a=sendonly

Barracuda09, 2015-02-13:
Hi Paul,

Thanks.. No it's no problem if you keep me busy. I'm now rewriting SatPI to C++ it will (I hope) be better to manage.

Thanks again for your time,

Marc

Barracuda09, 2015-03-02:
Hi,

Is this issue is fixed with latest commits?

Marc

pbriesch, 2015-03-02:
Hi Marc,

I still need to test. Sorry for the delay, my server was down and I had some other issues to take care of.

Paul

Barracuda09, 2015-03-03:
Hi,

No problem

pbriesch, 2015-03-04:
Hello Marc,

today I did some testing with VDR and the vdr-plugin-satip. Unfortunately, no stream could be established. 
````
[     Frontend.cpp:379] Stream: 0, Opened FE fd: 9.
[     Frontend.cpp:340] Stream: 0, Start tuning process...
[     Frontend.cpp:332] Stream: 0, Sending DiSEqC [e0] [10] [38] [f2]
[     Frontend.cpp:390] Stream: 0, Waiting on lock.
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:422] Stream: 0, Opened DVR fd: 10.
[    RtpThread.cpp:089] Stream: 0, Start RTP streaming to 192.168.1.166 (50140 - 50141)
[   RtcpThread.cpp:073] Stream: 0, Start RTCP streaming to 192.168.1.166 (50140 - 50141)
[   RtspServer.cpp:346] RTSP/1.0 200 OK
CSeq: 2
Session: 1702957685;timeout=60
Transport: RTP/AVP;unicast;client_port=50140-50141
com.ses.streamID: 0
--- LINE END ---
[    TcpSocket.cpp:147] HTTP Connection from 192.168.1.107 Port 57276 (fd: 13)
[   SsdpServer.cpp:189] SAT>IP Client 192.168.1.166 : tries to discover the network, sending an reply
[    TcpSocket.cpp:147] HTTP Connection from 192.168.1.166 Port 47373 (fd: 14)
[   HttpServer.cpp:166] Client: 192.168.1.166 requesed /desc.xml
^Cpbriesch@ubuntu32:~/satpi_c++$ clear
pbriesch@ubuntu32:~/satpi_c++$ sudo ./satpi --no-daemon
[        Satpi.cpp:243] --- starting SatPI version: 1.2 ---
[        Satpi.cpp:247] streamID -1
[  InterfaceAttr.h:106] eth0: 192.168.1.169 [00:19:66:0f:8d:07]
[      Streams.cpp:120] Detecting frontends in: /dev/dvb
[      Streams.cpp:130] Frontends found: 2
[     Frontend.cpp:145] Frontend Name: STB0899 Multistandard
[     Frontend.cpp:180] Frontend Type: Satellite (DVB-S)
[     Frontend.cpp:184] Frontend Type: Satellite (DVB-S2)
[     Frontend.cpp:176] Frontend Type: DSS
[     Frontend.cpp:214] Frontend Freq: 950000 Hz to 2150000 Hz
[     Frontend.cpp:215] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
[     Frontend.cpp:145] Frontend Name: STB0899 Multistandard
[     Frontend.cpp:180] Frontend Type: Satellite (DVB-S)
[     Frontend.cpp:184] Frontend Type: Satellite (DVB-S2)
[     Frontend.cpp:176] Frontend Type: DSS
[     Frontend.cpp:214] Frontend Freq: 950000 Hz to 2150000 Hz
[     Frontend.cpp:215] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
[   SsdpServer.cpp:106] Setting up SSDP server with BOOTID: 14
[   HttpServer.cpp:073] Setting up HTTP server
[   RtspServer.cpp:058] Setting up RTSP server
[    TcpSocket.cpp:147] HTTP Connection from 192.168.1.107 Port 57334 (fd: 8)
[   SsdpServer.cpp:189] SAT>IP Client 192.168.1.166 : tries to discover the network, sending an reply
[    TcpSocket.cpp:147] HTTP Connection from 192.168.1.166 Port 47405 (fd: 9)
[   HttpServer.cpp:166] Client: 192.168.1.166 requesed /desc.xml
[    TcpSocket.cpp:147] RTSP Connection from 192.168.1.166 Port 36698 (fd: 10)
[   RtspServer.cpp:069] Data from client 192.168.1.166: TEARDOWN rtsp://192.168.1.169/stream=0 RTSP/1.0
CSeq: 3
Session: 1702957685
User-Agent: vdr-satip/2.2.0-GIT-0-171-g2c2f415 (device 0)
--- LINE END ---
[      Streams.cpp:224] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:080] StreamID 0 - No StreamClient with SessionID 1702957685
[       Stream.cpp:080] StreamID 1 - No StreamClient with SessionID 1702957685
[    TcpSocket.cpp:085] RTSP Client 192.168.1.166: Connection closed with fd: 10
[      Streams.cpp:212] Found message outside session
[    TcpSocket.cpp:085] HTTP Client 192.168.1.166: Connection closed with fd: 9
[   SsdpServer.cpp:189] SAT>IP Client 192.168.1.166 : tries to discover the network, sending an reply
[    TcpSocket.cpp:147] HTTP Connection from 192.168.1.166 Port 47409 (fd: 9)
[   HttpServer.cpp:166] Client: 192.168.1.166 requesed /desc.xml
[    TcpSocket.cpp:147] RTSP Connection from 192.168.1.166 Port 36703 (fd: 10)
[   RtspServer.cpp:069] Data from client 192.168.1.166: OPTIONS rtsp://192.168.1.169/ RTSP/1.0
CSeq: 1
User-Agent: vdr-satip/2.2.0-GIT-0-171-g2c2f415 (device 0)
[... 45 lines cut ...]
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 26)...
[   RtspServer.cpp:372] RTSP/1.0 200 OK
RTP-Info: url=rtsp://192.168.1.169/stream=0
CSeq: 3
Session: 1702957685
--- LINE END ---
[    TcpSocket.cpp:147] HTTP Connection from 192.168.1.107 Port 57355 (fd: 15)
[    TcpSocket.cpp:147] HTTP Connection from 192.168.1.107 Port 57356 (fd: 16)
[    TcpSocket.cpp:147] HTTP Connection from 192.168.1.107 Port 57357 (fd: 17)
[    TcpSocket.cpp:085] HTTP Client 192.168.1.107: Connection closed with fd: 17
[   SsdpServer.cpp:189] SAT>IP Client 192.168.1.166 : tries to discover the network, sending an reply
[   HttpServer.cpp:166] Client: 192.168.1.166 requesed /desc.xml
[       Stream.cpp:088] Stream: 0, Watchdog kicked in for StreamClient[0] with SessionID 1702957685
[       Stream.cpp:111] Stream: 0, Teardown StreamClient[0] with SessionID 1702957685
[    RtpThread.cpp:101] Stream: 0, Stop RTP streaming to 192.168.1.166 (52987 - 52988) (Streamed 0.000 MBytes)
[   RtcpThread.cpp:084] Stream: 0, Stop RTCP stream to 192.168.1.166 (52987 - 52988)
[   SsdpServer.cpp:189] SAT>IP Client 192.168.1.166 : tries to discover the network, sending an reply
[   HttpServer.cpp:166] Client: 192.168.1.166 requesed /desc.xml
[   SsdpServer.cpp:189] SAT>IP Client 192.168.1.166 : tries to discover the network, sending an reply
[   HttpServer.cpp:166] Client: 192.168.1.166 requesed /desc.xml
[   SsdpServer.cpp:189] SAT>IP Client 192.168.1.166 : tries to discover the network, sending an reply
[   HttpServer.cpp:166] Client: 192.168.1.166 requesed /desc.xml

pbriesch, 2015-03-04:
With EyeTV on the Mac it did work. However I had to restart SatPi and EyeTV so that client and server could connect.
````
a=control:stream=1
a=fmtp:33 ver=1.0;src=1;tuner=2,1,1,0,12187.00,0,dvbs,qpsk,2,0.35,27500,34;pids=
a=inactive
[    TcpSocket.cpp:085] RTSP Client 192.168.1.107: Connection closed with fd: 16
[      Streams.cpp:212] Found message outside session
[    TcpSocket.cpp:147] RTSP Connection from 192.168.1.107 Port 57454 (fd: 16)
[   RtspServer.cpp:069] Data from client 192.168.1.107: DESCRIBE rtsp://192.168.1.169/ RTSP/1.0
Connection:Close
CSeq:1007
Accept:application/sdp
--- LINE END ---
[      Streams.cpp:212] Found message outside session
[   RtspServer.cpp:107] Connection can close: 1
[   RtspServer.cpp:465] RTSP/1.0 200 OK
CSeq: 1007
Content-Type: application/sdp
Content-Base: rtsp://192.168.1.169/
Content-Length: 414
--- LINE END ---
v=0
o=- 2 3 IN IP4 192.168.1.169
s=SatIPServer:1 2,0,0
t=0 0
m=video 0 RTP/AVP 33
c=IN IP4 0.0.0.0
a=control:stream=0
a=fmtp:33 ver=1.0;src=1;tuner=1,1,1,0,12187.00,0,dvbs,qpsk,2,0.35,27500,34;pids=0,16,17,18,19,20,45,65,66,67,120,165,8190
a=sendonly
m=video 0 RTP/AVP 33
c=IN IP4 0.0.0.0
a=control:stream=1
a=fmtp:33 ver=1.0;src=1;tuner=2,1,1,0,12187.00,0,dvbs,qpsk,2,0.35,27500,34;pids=
a=inactive
[    TcpSocket.cpp:085] RTSP Client 192.168.1.107: Connection closed with fd: 16
[      Streams.cpp:212] Found message outside session
[    TcpSocket.cpp:147] RTSP Connection from 192.168.1.107 Port 57455 (fd: 16)
[   RtspServer.cpp:069] Data from client 192.168.1.107: DESCRIBE rtsp://192.168.1.169/ RTSP/1.0
Connection:Close
CSeq:1008
Accept:application/sdp
--- LINE END ---
[      Streams.cpp:212] Found message outside session
[   RtspServer.cpp:107] Connection can close: 1
[   RtspServer.cpp:465] RTSP/1.0 200 OK
CSeq: 1008
Content-Type: application/sdp
Content-Base: rtsp://192.168.1.169/
Content-Length: 414
--- LINE END ---
v=0
o=- 2 3 IN IP4 192.168.1.169
s=SatIPServer:1 2,0,0
t=0 0
m=video 0 RTP/AVP 33
c=IN IP4 0.0.0.0
a=control:stream=0
a=fmtp:33 ver=1.0;src=1;tuner=1,1,1,0,12187.00,0,dvbs,qpsk,2,0.35,27500,34;pids=0,16,17,18,19,20,45,65,66,67,120,165,8190
a=sendonly
m=video 0 RTP/AVP 33
c=IN IP4 0.0.0.0
a=control:stream=1
a=fmtp:33 ver=1.0;src=1;tuner=2,1,1,0,12187.00,0,dvbs,qpsk,2,0.35,27500,34;pids=
a=inactive

^Cpbriesch@ubuntu32:~/satpi_c++$ clear
pbriesch@ubuntu32:~/satpi_c++$ sudo ./satpi --no-daemon
[        Satpi.cpp:243] --- starting SatPI version: 1.2 ---
[        Satpi.cpp:247] streamID -1
[... 696 lines cut ...]
CSeq:1004
Accept:application/sdp
--- LINE END ---
[      Streams.cpp:212] Found message outside session
[   RtspServer.cpp:107] Connection can close: 1
[   RtspServer.cpp:465] RTSP/1.0 200 OK
CSeq: 1004
Content-Type: application/sdp
Content-Base: rtsp://192.168.1.169/
Content-Length: 431
--- LINE END ---
v=0
o=- 2 3 IN IP4 192.168.1.169
s=SatIPServer:1 2,0,0
t=0 0
m=video 0 RTP/AVP 33
c=IN IP4 0.0.0.0
a=control:stream=0
a=fmtp:33 ver=1.0;src=1;tuner=1,1,1,0,10743.00,0,dvbs,qpsk,2,0.35,22000,56;pids=0,16,17,18,19,20,400,401,402,403,404,407,408,470,2171,8190
a=sendonly
m=video 0 RTP/AVP 33
c=IN IP4 0.0.0.0
a=control:stream=1
a=fmtp:33 ver=1.0;src=1;tuner=2,1,1,0,10743.00,0,dvbs,qpsk,2,0.35,22000,56;pids=
a=inactive

Barracuda09, 2015-03-04:
Hi Paul,

Thanks for testing. Did you get the new source from git?
I see that version is 1.2

pbriesch, 2015-03-04:
No, from Dropbox. I will test the source from git now.

Barracuda09, 2015-03-04:
Yes sorry.. I have put it on git now...

pbriesch, 2015-03-04:
No change:
,,,,
[        Satpi.cpp:243] --- starting SatPI version: 1.5.6~gba2c20f ---
[  InterfaceAttr.h:106] eth0: 192.168.1.169 [00:19:66:0f:8d:07]
[      Streams.cpp:120] Detecting frontends in: /dev/dvb
[      Streams.cpp:130] Frontends found: 2
[     Frontend.cpp:145] Frontend Name: STB0899 Multistandard
[     Frontend.cpp:180] Frontend Type: Satellite (DVB-S)
[     Frontend.cpp:184] Frontend Type: Satellite (DVB-S2)
[     Frontend.cpp:176] Frontend Type: DSS
[     Frontend.cpp:214] Frontend Freq: 950000 Hz to 2150000 Hz
[     Frontend.cpp:215] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
[     Frontend.cpp:145] Frontend Name: STB0899 Multistandard
[     Frontend.cpp:180] Frontend Type: Satellite (DVB-S)
[     Frontend.cpp:184] Frontend Type: Satellite (DVB-S2)
[     Frontend.cpp:176] Frontend Type: DSS
[     Frontend.cpp:214] Frontend Freq: 950000 Hz to 2150000 Hz
[     Frontend.cpp:215] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
[   SsdpServer.cpp:106] Setting up SSDP server with BOOTID: 1
[   HttpServer.cpp:073] Setting up HTTP server
[   RtspServer.cpp:058] Setting up RTSP server
[    TcpSocket.cpp:147] RTSP Connection from 192.168.1.166 Port 40434 (fd: 8)
[   RtspServer.cpp:070] Data from client 192.168.1.166: OPTIONS rtsp://192.168.1.169/ RTSP/1.0
CSeq: 1
User-Agent: vdr-satip/2.2.0-GIT-0-171-g2c2f415 (device 0)
--- LINE END ---
[      Streams.cpp:214] Found message outside session
[       Stream.cpp:188] Stream: -1, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:195] RTSP/1.0 200 OK
CSeq: 1
Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
--- LINE END ---
[   RtspServer.cpp:070] Data from client 192.168.1.166: SETUP rtsp://192.168.1.169/?freq=11361&src=1&sr=22000&pol=h&fec=23&mtype=8psk&ro=0.35&msys=dvbs2 RTSP/1.0
CSeq: 2
Transport: RTP/AVP;unicast;client_port=46864-46865
User-Agent: vdr-satip/2.2.0-GIT-0-171-g2c2f415 (device 0)
If-Modified-Since: Thu, 01 Jan 1970 00:00:00 GMT
--- LINE END ---
[      Streams.cpp:232] Found StreamID x - SessionID x - Creating new SessionID: 1702957685
[       Stream.cpp:064] Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbs2
[       Stream.cpp:188] Stream: 0, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 0
[     Frontend.cpp:379] Stream: 0, Opened FE fd: 9.
[     Frontend.cpp:340] Stream: 0, Start tuning process...
[     Frontend.cpp:332] Stream: 0, Sending DiSEqC [e0] [10] [38] [f2]
[     Frontend.cpp:390] Stream: 0, Waiting on lock.
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:422] Stream: 0, Opened DVR fd: 10.
[    RtpThread.cpp:089] Stream: 0, Start RTP streaming to 192.168.1.166 (46864 - 46865)
[   RtcpThread.cpp:073] Stream: 0, Start RTCP streaming to 192.168.1.166 (46864 - 46865)
[   RtspServer.cpp:141] RTSP/1.0 200 OK
CSeq: 2
Session: 1702957685;timeout=60
Transport: RTP/AVP;unicast;client_port=46864-46865
com.ses.streamID: 0
--- LINE END ---
[    TcpSocket.cpp:147] HTTP Connection from 192.168.1.107 Port 60538 (fd: 13)
[    TcpSocket.cpp:147] HTTP Connection from 192.168.1.131 Port 61184 (fd: 14)
[   HttpServer.cpp:166] Client: 192.168.1.131 requesed /desc.xml
[   RtspServer.cpp:070] Data from client 192.168.1.166: TEARDOWN rtsp://192.168.1.169/stream=0 RTSP/1.0
CSeq: 3
Session: 1702957685
User-Agent: vdr-satip/2.2.0-GIT-0-171-g2c2f415 (device 0)
--- LINE END ---
[      Streams.cpp:224] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 1702957685
[... 132 lines cut ...]
[   HttpServer.cpp:103] File not found: No such file or directory (code 2)
[   HttpServer.cpp:102] GET web/config.xml
[   HttpServer.cpp:103] File not found: No such file or directory (code 2)
[   HttpServer.cpp:102] GET web/config.xml
[   HttpServer.cpp:103] File not found: No such file or directory (code 2)
[   HttpServer.cpp:102] GET web/config.xml
[   HttpServer.cpp:103] File not found: No such file or directory (code 2)
[   HttpServer.cpp:102] GET web/config.xml
[   HttpServer.cpp:103] File not found: No such file or directory (code 2)
[   HttpServer.cpp:102] GET web/config.xml
[   HttpServer.cpp:103] File not found: No such file or directory (code 2)
[   HttpServer.cpp:102] GET web/config.xml
[   HttpServer.cpp:103] File not found: No such file or directory (code 2)
[       Stream.cpp:095] Stream: 0, Watchdog kicked in for StreamClient[0] with SessionID 0205744672
[       Stream.cpp:122] Stream: 0, Teardown StreamClient[0] with SessionID 0205744672
[    RtpThread.cpp:101] Stream: 0, Stop RTP streaming to 192.168.1.166 (39746 - 39747) (Streamed 0.000 MBytes)
[   RtcpThread.cpp:084] Stream: 0, Stop RTCP stream to 192.168.1.166 (39746 - 39747)
[    TcpSocket.cpp:085] RTSP Client 192.168.1.166: Connection closed with fd: 16
[      Streams.cpp:204] Found SessionID 0205744672 by SocketClient
[      Streams.cpp:224] Found StreamID x - SessionID: 0205744672
[       Stream.cpp:081] Stream: 0, No StreamClient with SessionID 0205744672
[       Stream.cpp:081] Stream: 1, No StreamClient with SessionID 0205744672
[      Streams.cpp:260] Found no Stream/Client of interest!
[    TcpSocket.cpp:091] close error: Bad file descriptor (code 9)
[   HttpServer.cpp:102] GET web/config.xml

Barracuda09, 2015-03-04:
Hi,

Can you try to check the same channel as with EyeTV on the Mac?

pbriesch, 2015-03-04:
Marc,

first VDR, second EyeTV on Mac. EyeTV needed some time to connect. And there were many interruptions, which is not always the case. I can test more later today.

Paul

````
pbriesch@ubuntu32:~/satpi$ sudo ./satpi --no-daemon
[        Satpi.cpp:243] --- starting SatPI version: 1.5.6~gba2c20f ---
[  InterfaceAttr.h:106] eth0: 192.168.1.169 [00:19:66:0f:8d:07]
[      Streams.cpp:120] Detecting frontends in: /dev/dvb
[      Streams.cpp:130] Frontends found: 2
[     Frontend.cpp:145] Frontend Name: STB0899 Multistandard
[     Frontend.cpp:180] Frontend Type: Satellite (DVB-S)
[     Frontend.cpp:184] Frontend Type: Satellite (DVB-S2)
[     Frontend.cpp:176] Frontend Type: DSS
[     Frontend.cpp:214] Frontend Freq: 950000 Hz to 2150000 Hz
[     Frontend.cpp:215] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
[     Frontend.cpp:145] Frontend Name: STB0899 Multistandard
[     Frontend.cpp:180] Frontend Type: Satellite (DVB-S)
[     Frontend.cpp:184] Frontend Type: Satellite (DVB-S2)
[     Frontend.cpp:176] Frontend Type: DSS
[     Frontend.cpp:214] Frontend Freq: 950000 Hz to 2150000 Hz
[     Frontend.cpp:215] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
[   SsdpServer.cpp:106] Setting up SSDP server with BOOTID: 3
[   HttpServer.cpp:073] Setting up HTTP server
[   RtspServer.cpp:058] Setting up RTSP server
[    TcpSocket.cpp:147] HTTP Connection from 192.168.1.107 Port 60767 (fd: 8)
[   HttpServer.cpp:102] GET web/config.xml
[   HttpServer.cpp:103] File not found: No such file or directory (code 2)
[    TcpSocket.cpp:147] HTTP Connection from 192.168.1.107 Port 60768 (fd: 9)
[   HttpServer.cpp:102] GET web/config.xml
[   HttpServer.cpp:103] File not found: No such file or directory (code 2)
[    TcpSocket.cpp:147] RTSP Connection from 192.168.1.166 Port 40476 (fd: 10)
[   RtspServer.cpp:070] Data from client 192.168.1.166: PLAY rtsp://192.168.1.169/stream=0?freq=11361&src=1&sr=22000&pol=h&fec=23&mtype=8psk&ro=0.35&msys=dvbs2 RTSP/1.0
CSeq: 3
Session: 0205744672
User-Agent: vdr-satip/2.2.0-GIT-0-171-g2c2f415 (device 0)
--- LINE END ---
[      Streams.cpp:224] Found StreamID x - SessionID: 0205744672
[       Stream.cpp:078] Stream: 0, No StreamClient with SessionID 0205744672 for dvbs2
[       Stream.cpp:078] Stream: 1, No StreamClient with SessionID 0205744672 for dvbs2
[      Streams.cpp:260] Found no Stream/Client of interest!
[   RtspServer.cpp:070] Data from client 192.168.1.166: OPTIONS rtsp://192.168.1.169/ RTSP/1.0
CSeq: 4
Session: 0205744672
User-Agent: vdr-satip/2.2.0-GIT-0-171-g2c2f415 (device 0)
--- LINE END ---
[      Streams.cpp:224] Found StreamID x - SessionID: 0205744672
[       Stream.cpp:081] Stream: 0, No StreamClient with SessionID 0205744672
[       Stream.cpp:081] Stream: 1, No StreamClient with SessionID 0205744672
[      Streams.cpp:260] Found no Stream/Client of interest!
[    TcpSocket.cpp:085] RTSP Client 192.168.1.166: Connection closed with fd: 10
[      Streams.cpp:214] Found message outside session
[    TcpSocket.cpp:147] RTSP Connection from 192.168.1.166 Port 40477 (fd: 10)
[   RtspServer.cpp:070] Data from client 192.168.1.166: OPTIONS rtsp://192.168.1.169/ RTSP/1.0
CSeq: 1
User-Agent: vdr-satip/2.2.0-GIT-0-171-g2c2f415 (device 0)
--- LINE END ---
[      Streams.cpp:214] Found message outside session
[       Stream.cpp:188] Stream: -1, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:195] RTSP/1.0 200 OK
CSeq: 1
Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
--- LINE END ---
[   RtspServer.cpp:070] Data from client 192.168.1.166: SETUP rtsp://192.168.1.169/?freq=11361&src=1&sr=22000&pol=h&fec=23&mtype=8psk&ro=0.35&msys=dvbs2 RTSP/1.0
CSeq: 2
Transport: RTP/AVP;unicast;client_port=34808-34809
User-Agent: vdr-satip/2.2.0-GIT-0-171-g2c2f415 (device 0)
[... 1119 lines cut ...]
[     Frontend.cpp:429] Stream: 0, Updating PID filters...
[   RtspServer.cpp:167] RTSP/1.0 200 OK
RTP-Info: url=rtsp://192.168.1.169/stream=0
CSeq: 1039
Session: 1702957685
--- LINE END ---
[    TcpSocket.cpp:085] RTSP Client 192.168.1.107: Connection closed with fd: 10
[      Streams.cpp:204] Found SessionID 1702957685 by SocketClient
[      Streams.cpp:224] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 1702957685
[    TcpSocket.cpp:147] RTSP Connection from 192.168.1.107 Port 61003 (fd: 10)
[   RtspServer.cpp:070] Data from client 192.168.1.107: PLAY rtsp://192.168.1.169:554/stream=0?addpids=6300 RTSP/1.0
Connection:Close
CSeq:1040
Session:1702957685
--- LINE END ---
[      Streams.cpp:224] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 1702957685
[       Stream.cpp:188] Stream: 0, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 0
[     Frontend.cpp:340] Stream: 0, Start tuning process...
[     Frontend.cpp:332] Stream: 0, Sending DiSEqC [e0] [10] [38] [f2]
[     Frontend.cpp:390] Stream: 0, Waiting on lock.
[   HttpServer.cpp:102] GET web/config.xml
[   HttpServer.cpp:103] File not found: No such file or directory (code 2)

Barracuda09, 2015-03-04:
Hi Paul,

If I understand this is ZDF HD on astra 19.2?

if you try VLC with this
rtsp://192.168.0.109/?freq=11361&src=1&sr=22000&pol=h&fec=23&mtype=8psk&ro=0.35&msys=dvbs2&pids=0,6110,6100,6110,6130

Barracuda09, 2015-03-04:
With this I am able to see ZDF HD in VLC

pbriesch, 2015-03-04:
Hi Marc,

yes, it is ZDF HD and it works with VLC, but just for about 15 s and then it stops.
````
--- LINE END ---
[      Streams.cpp:232] Found StreamID x - SessionID x - Creating new SessionID: 1702957685
[       Stream.cpp:064] Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbs
[       Stream.cpp:188] Stream: 0, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 0
[     Frontend.cpp:379] Stream: 0, Opened FE fd: 9.
[     Frontend.cpp:340] Stream: 0, Start tuning process...
[     Frontend.cpp:332] Stream: 0, Sending DiSEqC [e0] [10] [38] [f3]
[    TcpSocket.cpp:147] HTTP Connection from 192.168.1.107 Port 64901 (fd: 10)
[     Frontend.cpp:390] Stream: 0, Waiting on lock.
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:401] Stream: 0, Tuned and locked (FE status 31).
[     Frontend.cpp:422] Stream: 0, Opened DVR fd: 11.
[    RtpThread.cpp:089] Stream: 0, Start RTP streaming to 192.168.1.166 (47232 - 47233)
[   RtcpThread.cpp:073] Stream: 0, Start RTCP streaming to 192.168.1.166 (47232 - 47233)
[   RtspServer.cpp:141] RTSP/1.0 200 OK
CSeq: 2
Session: 1702957685;timeout=60
Transport: RTP/AVP;unicast;client_port=47232-47233
com.ses.streamID: 0
--- LINE END ---
[    TcpSocket.cpp:147] HTTP Connection from 192.168.1.131 Port 64641 (fd: 14)
[   HttpServer.cpp:166] Client: 192.168.1.131 requesed /desc.xml
[    TcpSocket.cpp:147] RTSP Connection from 192.168.1.107 Port 64902 (fd: 15)
[   RtspServer.cpp:070] Data from client 192.168.1.107: OPTIONS rtsp://192.168.1.169/?freq=11361&src=1&sr=22000&pol=h&fec=23&mtype=8psk&ro=0.35&msys=dvbs2&pids=0,6110,6100,6110,6130 RTSP/1.0
CSeq: 2
User-Agent: LibVLC/2.2.0 (LIVE555 Streaming Media v2014.07.25)
--- LINE END ---
[      Streams.cpp:232] Found StreamID x - SessionID x - Creating new SessionID: 0205744672
[       Stream.cpp:064] Stream: 1, StreamClient[0] with SessionID 0205744672 for dvbs2
[       Stream.cpp:188] Stream: 1, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:195] RTSP/1.0 200 OK
CSeq: 2
Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
Session: 0205744672
--- LINE END ---
[   RtspServer.cpp:070] Data from client 192.168.1.107: DESCRIBE rtsp://192.168.1.169/?freq=11361&src=1&sr=22000&pol=h&fec=23&mtype=8psk&ro=0.35&msys=dvbs2&pids=0,6110,6100,6110,6130 RTSP/1.0
CSeq: 3
User-Agent: LibVLC/2.2.0 (LIVE555 Streaming Media v2014.07.25)
Accept: application/sdp
--- LINE END ---
[      Streams.cpp:204] Found SessionID 0205744672 by SocketClient
[      Streams.cpp:224] Found StreamID x - SessionID: 0205744672
[       Stream.cpp:078] Stream: 0, No StreamClient with SessionID 0205744672 for dvbs2
[       Stream.cpp:064] Stream: 1, StreamClient[0] with SessionID 0205744672 for dvbs2
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:260] RTSP/1.0 200 OK
CSeq: 3
Content-Type: application/sdp
Content-Base: rtsp://192.168.1.169/
Content-Length: 391
Session: 0205744672
--- LINE END ---
v=0
o=- 2 3 IN IP4 192.168.1.169
s=SatIPServer:1 2,0,0
t=0 0
m=video 0 RTP/AVP 33
c=IN IP4 0.0.0.0
a=control:stream=0
a=fmtp:33 ver=1.0;src=1;tuner=1,1,1,0,12148.00,0,dvbs,64qam,2,0.35,27500,34;pids=
a=sendonly
m=video 0 RTP/AVP 33
c=IN IP4 0.0.0.0
a=control:stream=1
[... 265 lines cut ...]
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 1702957685
[       Stream.cpp:188] Stream: 0, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 0
[   RtspServer.cpp:195] RTSP/1.0 200 OK
CSeq: 5
Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
Session: 1702957685
--- LINE END ---
[       Stream.cpp:095] Stream: 1, Watchdog kicked in for StreamClient[0] with SessionID 0288582906
[       Stream.cpp:122] Stream: 1, Teardown StreamClient[0] with SessionID 0288582906
[    RtpThread.cpp:101] Stream: 1, Stop RTP streaming to 192.168.1.107 (62090 - 62091) (Streamed 238.644 MBytes)
[   RtcpThread.cpp:084] Stream: 1, Stop RTCP stream to 192.168.1.107 (62090 - 62091)
[     Frontend.cpp:462] Stream: 1, Remove filter PID: 0000 - fd: 018 - Packet Count: 756
[     Frontend.cpp:462] Stream: 1, Remove filter PID: 6100 - fd: 019 - Packet Count: 755
[     Frontend.cpp:462] Stream: 1, Remove filter PID: 6110 - fd: 020 - Packet Count: 647797
[     Frontend.cpp:462] Stream: 1, Remove filter PID: 6130 - fd: 021 - Packet Count: 15117
[    TcpSocket.cpp:085] RTSP Client 192.168.1.107: Connection closed with fd: 14
[      Streams.cpp:204] Found SessionID 0288582906 by SocketClient
[      Streams.cpp:224] Found StreamID x - SessionID: 0288582906
[       Stream.cpp:081] Stream: 0, No StreamClient with SessionID 0288582906
[       Stream.cpp:081] Stream: 1, No StreamClient with SessionID 0288582906
[      Streams.cpp:260] Found no Stream/Client of interest!
[    TcpSocket.cpp:091] close error: Bad file descriptor (code 9)
[   SsdpServer.cpp:189] SAT>IP Client 192.168.1.166 : tries to discover the network, sending an reply
[   HttpServer.cpp:166] Client: 192.168.1.166 requesed /desc.xml

Barracuda09, 2015-03-04:
Yes that it will only work for a little while is "normal" because VLC does not send "keep alive" messages.
I see that the FE Status 26 and not 31 that is a strange... I have the status 31 locked and signal

Barracuda09, 2015-03-04:
I think now what the problem is... VDR wait probably on RTCP packets with a Lock status before sending PIDs... And SatPI only sends RTCP packets lock status when it is locked and have signal.

Barracuda09, 2015-03-04:
But it is stranger that it receives data on the PIDs.... mmmm...

pbriesch, 2015-03-04:
Marc, are you following the discussions of *minisatip* on GitHub? There has been a lot of testing as well to make sure different clients could connect.

Barracuda09, 2015-03-04:
Yes I know.. but I think it should work...  I'm stubborn   :-)

pbriesch, 2015-03-04:
:-)

What about VLC. Why does it stop after some short time?

Barracuda09, 2015-03-04:
Well VLC does not send "Keep Alive" message... I had before did not check the timeout.. but that is not fail save..

Barracuda09, 2015-04-10:
Should be fixed in the mean time

closed 2015-04-10 by Barracuda09