Barracuda09/SATPI #26: compile issue

9000h, 2015-03-04:
it's on a banana pi
Linux bananas 3.4.103 #1 SMP PREEMPT Fri Oct 17 00:32:24 CST 2014 armv7l GNU/Linux

root@bananas:/usr/src/satip/SATPI-master# make
./version.sh Version.cpp > /dev/null
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Version.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Satpi.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG ChannelData.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG StreamProperties.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Properties.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG StringConverter.cpp
StringConverter.cpp: In static member function 'static const char* StringConverter::transmode_to_string(int)':
StringConverter.cpp:323:8: error: 'TRANSMISSION_MODE_C1' was not declared in this scope
StringConverter.cpp:325:8: error: 'TRANSMISSION_MODE_C3780' was not declared in this scope
Makefile:73: recipe for target 'StringConverter.o' failed
make: *** [StringConverter.o] Error 1
root@bananas:/usr/src/satip/SATPI-master#

Barracuda09, 2015-03-04:
So that means it has DVB api >= 5.5?

9000h, 2015-03-05:
from the headers
/usr/include/linux/dvb/version.h:#define DVB_API_VERSION 5
/usr/include/linux/dvb/version.h:#define DVB_API_VERSION_MINOR 5

Barracuda09, 2015-03-05:
Hi,

I commit  some changes regarding this issue.

Barracuda09, 2015-03-14:
This should be fixed.

closed 2015-03-14 by Barracuda09

9000h, 2015-03-14:
I made probably a mistake, and other people may also.
I have media-build under /usr/src/media-build but the standard kernel headers are old, may be it's a good idea to use configure to build the Makefile with the option to specify the the location of the right headers.