Barracuda09/SATPI #27: IP_ADD_MEMBERSHIP: No such device (code 19)

pbriesch, 2015-03-05:
Hello Marc,

today I have played a little bit with my RPI 2. I put a DVB-T stick on it, a WLAN-dongle, and put it on the balcony to get better reception. Apart from the fact that it didn't work with the *vdr-plugin-satip* I got following message:

    IP_ADD_MEMBERSHIP: No such device (code 19)

Could it be that there are now two network interfaces eth0 and wlan0 of which eth0 has no IP-adresses.

Here is the log and the output of *ifconfig*

Paul
````
0  Thu Mar  5 17:52:57.711164649 2015  6  --- starting SatPI version: 1.5.6~gba2c20f ---
1  Thu Mar  5 17:52:57.713744370 2015  3  ioctl SIOCGIFHWADDR: Cannot assign requested address (code 99)
2  Thu Mar  5 17:52:57.718625533 2015  3  Not the preferred DVB API version, for correct function it should be 5.5 or higher
3  Thu Mar  5 17:52:57.720436978 2015  3  current DVB_API_VERSION: 5.4
4  Thu Mar  5 17:52:57.721371815 2015  6  Detecting frontends in: /dev/dvb
5  Thu Mar  5 17:52:57.722691441 2015  6  Frontends found: 1
6  Thu Mar  5 17:52:57.724008463 2015  6  Frontend Name: Afatech AF9013
7  Thu Mar  5 17:52:57.726278811 2015  6  Setting up RTSP server
8  Thu Mar  5 17:52:57.726626048 2015  6  Setting up HTTP server
9  Thu Mar  5 17:52:57.728474941 2015  6  Setting up SSDP server with BOOTID: 5
10  Thu Mar  5 17:52:57.729688578 2015  3  IP_ADD_MEMBERSHIP: No such device (code 19)
11  Thu Mar  5 17:52:57.918534458 2015  6  RTSP Connection from 192.168.1.21 Port 50054 (fd: 8)
12  Thu Mar  5 17:52:57.919106328 2015  6  Found message outside session
13  Thu Mar  5 17:52:57.923042654 2015  6  Found StreamID x - SessionID x - Creating new SessionID: 1702957685
14  Thu Mar  5 17:52:57.923450672 2015  6  Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbt
15  Thu Mar  5 17:52:57.925091077 2015  6  Stream: 0, Opened FE fd: 9.
16  Thu Mar  5 17:52:58.006177472 2015  6  Stream: 0, Waiting on lock.
17  Thu Mar  5 17:52:58.029051682 2015  6  Stream: 0, Not locked yet (FE status 0)...
18  Thu Mar  5 17:52:58.054445666 2015  6  Stream: 0, Not locked yet (FE status 0)...
19  Thu Mar  5 17:52:58.080142200 2015  6  Stream: 0, Not locked yet (FE status 0)...
20  Thu Mar  5 17:52:58.105541964 2015  6  Stream: 0, Not locked yet (FE status 0)...
21  Thu Mar  5 17:52:58.131554225 2015  6  Stream: 0, Opened DVR fd: 10.
22  Thu Mar  5 17:52:58.132128283 2015  6  Stream: 0, Start RTP streaming to 192.168.1.21 (33865 - 33866)
23  Thu Mar  5 17:52:58.132575155 2015  6  Stream: 0, Start RTCP streaming to 192.168.1.21 (33865 - 33866)
24  Thu Mar  5 17:52:59.266145999 2015  6  HTTP Connection from 192.168.1.107 Port 53537 (fd: 13)
25  Thu Mar  5 17:53:32.034971897 2015  6  Found StreamID x - SessionID: 1702957685
26  Thu Mar  5 17:53:32.036320013 2015  6  Stream: 0, StreamClient[0] with SessionID 1702957685
27  Thu Mar  5 17:53:32.037301256 2015  6  Stream: 0, Teardown StreamClient[0] with SessionID 1702957685
28  Thu Mar  5 17:53:32.038616454 2015  6  Stream: 0, Stop RTP streaming to 192.168.1.21 (33865 - 33866) (Streamed 0.000 MBytes)
29  Thu Mar  5 17:53:32.039393584 2015  6  Stream: 0, Stop RTCP stream to 192.168.1.21 (33865 - 33866)
30  Thu Mar  5 17:53:32.046243952 2015  6  Found SessionID 1702957685 by SocketClient
31  Thu Mar  5 17:53:32.046831864 2015  6  Found StreamID x - SessionID: 1702957685
32  Thu Mar  5 17:53:32.047305506 2015  6  Stream: 0, StreamClient[0] with SessionID 1702957685
33  Thu Mar  5 17:53:32.047727274 2015  6  Stream: 0, Close StreamClient[0] with SessionID 1702957685
````

````
pi@pi2 ~/satpi $ ifconfig
eth0      Link encap:Ethernet  HWaddr b8:27:eb:08:36:55
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:8 errors:0 dropped:0 overruns:0 frame:0
          TX packets:8 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:1104 (1.0 KiB)  TX bytes:1104 (1.0 KiB)

wlan0     Link encap:Ethernet  HWaddr 80:1f:02:f7:33:0d
          inet addr:192.168.1.158  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:15127 errors:0 dropped:44 overruns:0 frame:0
          TX packets:68829 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:2337552 (2.2 MiB)  TX bytes:86012513 (82.0 MiB)

pi@pi2 ~/satpi $

Barracuda09, 2015-03-05:
Hi Paul,

That could be very well the case... I have never tried it.. Interesting situation.

pbriesch, 2015-03-05:
Marc,

would it be a big deal to support DVB API 5.4?

````
Not the preferred DVB API version, for correct function it should be 5.5 or higher
current DVB_API_VERSION: 5.4

Barracuda09, 2015-03-05:
Hi,
It will work it's only that it will give this 'warning'
lower dan 5 will give compiler error

pbriesch, 2015-03-05:
Well, if it works you might want to get rid of the warning or explain a bit more what potential problems could be there.

Barracuda09, 2015-03-05:
Yes your right.. I will remove/change this.

Barracuda09, 2015-03-05:
I changed the warning, it will only be visible in debug. Is this good enough you think?

pbriesch, 2015-03-05:
If the program works well then the user has no reason to run it in debug. If there is a problem the user will run the program in debug and will see the warning. This is much better than before. However, the user still doesn't know which problems he could get from not running 5.5 or higher. Maybe there are problems which are not related to the API version?
It is a little bit difficult for me to explain: If you know that it will work with 5.4 (or even 5.3 ?) then there does not need to be a warning at all. If you think there could be some issues it would be great to be a little bit more specific if possible :-)

Barracuda09, 2015-03-05:
Well there are some defines which are not in lower the 5.5 API.. Which I define myself.. which I hope is sufficient.

Barracuda09, 2015-03-05:
I still am not able to make VDR run... :-(
I have installed MiniDVBLinux on an PI but I am unable to make any logic of it all.
At some point it connects to SatPI and does some kind of scan... but why and how I don't know yet,

pbriesch, 2015-03-05:
It could be that currently there are some problems with the RPI build.

After you boot from the freshly installed image on the SD-card you have to install MLD onto the SD-card. Then the SD-card will be formatted and MLD will be installed. Then you have to reboot. If you do not install on SD-card, MLD will always run in memory and changes will be lost after a restart.

There are three possibilities to run SAT>IP on MLD:

1. vdr-plugin-satip (see picture, right column). This plugin searches for the SAT>IP server. However, currently SatPi and the plugin don't work well together.

2. satip/vtuner (left column) This works better with SatPi. It should find SatPi during the installation. However, there is a menu entry to set the IP of the server and the number of vtuners. Click the column system, then the entry TV, from there click satip.

3. Actually, 1 and 2 together at the same time should work as well but may not make much sense.

If you still have problems please let me know. I can pull a working image from my SD-card for you and put it into my Dropbox.

Barracuda09, 2015-03-05:
Well maybe my problem is I use my PI headless,, so I like to use it with Kodi..

pbriesch, 2015-03-05:
I don't quite understand your setup with Kodi. But here is the missing picture:
[image]

Barracuda09, 2015-03-05:
I have no screen connected to the PI... I use kodi to watch live tv with TVH at the moment. So I thought VDR had the same possibility.

pbriesch, 2015-03-05:
I haven't played much with Kodi. However, there is vdr-plugin-xvdr and there is Kodi in the MLD distribution.
````
xvdr
LiveTV Integration, EPG-Darstellung, Verwaltung von Timern und Zugriff auf Aufnahmen
des VDR in XBMC*

lars18th, 2015-03-06:
Hi,

Related to the problem with support for API 5.4 the solution is close! See this thread: https://github.com/Barracuda09/SATPI/issues/28

Barracuda09, 2015-03-06:
Hi Paul,

I have fix the multiple eth/wlan interface issue (I hope). Also an RTCP buffer overrun (maybe VDR is also happy now).
If you like and have time you can test it.

Thanks and regards,

Marc

pbriesch, 2015-03-06:
Hello Marc,

I did run a test. However, no stream was established with *vdr-plugin-satpi*. Here is the log:
````
pi@pi2 ~/satpi $ sudo ./satpi --no-daemon
[        Satpi.cpp:243] --- starting SatPI version: 1.5.9~g6264947 ---
[  InterfaceAttr.h:094] wlan0: 192.168.1.158 [80:1f:02:f7:33:0d]
[      Streams.cpp:116] Not the preferred DVB API version, for correct function it should be 5.5 or higher
[      Streams.cpp:118] Current DVB_API_VERSION: 5.4
[      Streams.cpp:120] Detecting frontends in: /dev/dvb
[      Streams.cpp:130] Frontends found: 1
[     Frontend.cpp:145] Frontend Name: Afatech AF9013
[     Frontend.cpp:188] Frontend Type: Terrestrial (DVB-T)
[     Frontend.cpp:214] Frontend Freq: 174000000 Hz to 862000000 Hz
[     Frontend.cpp:215] Frontend srat: 0 symbols/s to 0 symbols/s
[   RtspServer.cpp:058] Setting up RTSP server
[   HttpServer.cpp:073] Setting up HTTP server
[   SsdpServer.cpp:106] Setting up SSDP server with BOOTID: 20
[    TcpSocket.cpp:148] RTSP Connection from 192.168.1.21 Port 33417 (fd: 8)
[   RtspServer.cpp:070] Data from client 192.168.1.21: OPTIONS rtsp://192.168.1.158/ RTSP/1.0
CSeq: 1
User-Agent: vdr-satip/2.2.0-GIT-0-174-g78e1f48 (device 0)
--- LINE END ---
[      Streams.cpp:214] Found message outside session
[       Stream.cpp:203] Stream: -1, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:195] RTSP/1.0 200 OK
CSeq: 1
Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
--- LINE END ---
[   RtspServer.cpp:070] Data from client 192.168.1.21: SETUP rtsp://192.168.1.158/?freq=546&bw=8&gi=14&fec=23&mtype=16qam&msys=dvbt&tmode=8k RTSP/1.0
CSeq: 2
Transport: RTP/AVP;unicast;client_port=33071-33072
User-Agent: vdr-satip/2.2.0-GIT-0-174-g78e1f48 (device 0)
If-Modified-Since: Thu, 01 Jan 1970 00:00:00 GMT
--- LINE END ---
[      Streams.cpp:232] Found StreamID x - SessionID x - Creating new SessionID: 1702957685
[       Stream.cpp:064] Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbt
[       Stream.cpp:203] Stream: 0, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 0
[     Frontend.cpp:379] Stream: 0, Opened FE fd: 9.
[     Frontend.cpp:340] Stream: 0, Start tuning process...
[     Frontend.cpp:390] Stream: 0, Waiting on lock.
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:422] Stream: 0, Opened DVR fd: 10.
[    RtpThread.cpp:089] Stream: 0, Start RTP streaming to 192.168.1.21 (33071 - 33072)
[   RtcpThread.cpp:073] Stream: 0, Start RTCP streaming to 192.168.1.21 (33071 - 33072)
[   RtspServer.cpp:141] RTSP/1.0 200 OK
CSeq: 2
Session: 1702957685;timeout=60
Transport: RTP/AVP;unicast;client_port=33071-33072
com.ses.streamID: 0
--- LINE END ---
[    TcpSocket.cpp:148] HTTP Connection from 192.168.1.107 Port 55362 (fd: 13)
[   SsdpServer.cpp:189] SAT>IP Client 192.168.1.21 : tries to discover the network, sending an reply
[    TcpSocket.cpp:148] HTTP Connection from 192.168.1.21 Port 37940 (fd: 14)
[   HttpServer.cpp:166] Client: 192.168.1.21 requesed /desc.xml
[   RtspServer.cpp:070] Data from client 192.168.1.21: PLAY rtsp://192.168.1.158/stream=0?freq=674&bw=8&gi=14&fec=23&mtype=16qam&msys=dvbt&tmode=8k RTSP/1.0
CSeq: 3
Session: 1702957685
User-Agent: vdr-satip/2.2.0-GIT-0-174-g78e1f48 (device 0)
--- LINE END ---
[      Streams.cpp:224] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:064] Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbt
[       Stream.cpp:203] Stream: 0, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 0
[     Frontend.cpp:340] Stream: 0, Start tuning process...
[... 123 lines cut ...]
User-Agent: vdr-satip/2.2.0-GIT-0-174-g78e1f48 (device 0)
If-Modified-Since: Thu, 01 Jan 1970 00:00:00 GMT
--- LINE END ---
[      Streams.cpp:232] Found StreamID x - SessionID x - Creating new SessionID: 0288582906
[       Stream.cpp:064] Stream: 0, StreamClient[0] with SessionID 0288582906 for dvbt
[       Stream.cpp:203] Stream: 0, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 0
[     Frontend.cpp:379] Stream: 0, Opened FE fd: 9.
[     Frontend.cpp:340] Stream: 0, Start tuning process...
[     Frontend.cpp:390] Stream: 0, Waiting on lock.
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:422] Stream: 0, Opened DVR fd: 10.
[    RtpThread.cpp:089] Stream: 0, Start RTP streaming to 192.168.1.21 (33071 - 33072)
[   RtcpThread.cpp:073] Stream: 0, Start RTCP streaming to 192.168.1.21 (33071 - 33072)
[   RtspServer.cpp:141] RTSP/1.0 200 OK
CSeq: 2
Session: 0288582906;timeout=60
Transport: RTP/AVP;unicast;client_port=33071-33072
com.ses.streamID: 0
--- LINE END ---
````
Regards, Paul

pbriesch, 2015-03-06:
Here  the client is now *vdr-satip/vtuner* and I attached the ethernet cable. No stream was established as well:
````
pi@pi2 ~/satpi $ sudo ./satpi --no-daemon
[        Satpi.cpp:243] --- starting SatPI version: 1.5.9~g6264947 ---
[  InterfaceAttr.h:094] eth0: 192.168.1.166 [b8:27:eb:08:36:55]
[      Streams.cpp:116] Not the preferred DVB API version, for correct function it should be 5.5 or higher
[      Streams.cpp:118] Current DVB_API_VERSION: 5.4
[      Streams.cpp:120] Detecting frontends in: /dev/dvb
[      Streams.cpp:130] Frontends found: 1
[     Frontend.cpp:145] Frontend Name: Afatech AF9013
[     Frontend.cpp:188] Frontend Type: Terrestrial (DVB-T)
[     Frontend.cpp:214] Frontend Freq: 174000000 Hz to 862000000 Hz
[     Frontend.cpp:215] Frontend srat: 0 symbols/s to 0 symbols/s
[   RtspServer.cpp:058] Setting up RTSP server
[   SsdpServer.cpp:106] Setting up SSDP server with BOOTID: 22
[   HttpServer.cpp:073] Setting up HTTP server
[    TcpSocket.cpp:148] RTSP Connection from 192.168.1.21 Port 44915 (fd: 8)
[   RtspServer.cpp:070] Data from client 192.168.1.21: PLAY rtsp://192.168.1.166/stream=0?freq=546&bw=8&gi=14&fec=23&mtype=16qam&msys=dvbt&tmode=8k RTSP/1.0
CSeq: 3
Session: 0288582906
User-Agent: vdr-satip/2.2.0-GIT-0-174-g78e1f48 (device 0)
--- LINE END ---
[      Streams.cpp:224] Found StreamID x - SessionID: 0288582906
[       Stream.cpp:078] Stream: 0, No StreamClient with SessionID 0288582906 for dvbt
[      Streams.cpp:260] Found no Stream/Client of interest!
[   RtspServer.cpp:070] Data from client 192.168.1.21: OPTIONS rtsp://192.168.1.166/ RTSP/1.0
CSeq: 4
Session: 0288582906
User-Agent: vdr-satip/2.2.0-GIT-0-174-g78e1f48 (device 0)
--- LINE END ---
[      Streams.cpp:224] Found StreamID x - SessionID: 0288582906
[       Stream.cpp:081] Stream: 0, No StreamClient with SessionID 0288582906
[      Streams.cpp:260] Found no Stream/Client of interest!
[    TcpSocket.cpp:085] RTSP Client 192.168.1.21: Connection closed with fd: 8
[      Streams.cpp:214] Found message outside session
[    TcpSocket.cpp:148] RTSP Connection from 192.168.1.21 Port 44916 (fd: 8)
[   RtspServer.cpp:070] Data from client 192.168.1.21: OPTIONS rtsp://192.168.1.166/ RTSP/1.0
CSeq: 1
User-Agent: vdr-satip/2.2.0-GIT-0-174-g78e1f48 (device 0)
--- LINE END ---
[      Streams.cpp:214] Found message outside session
[       Stream.cpp:203] Stream: -1, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:195] RTSP/1.0 200 OK
CSeq: 1
Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
--- LINE END ---
[   RtspServer.cpp:070] Data from client 192.168.1.21: SETUP rtsp://192.168.1.166/?freq=546&bw=8&gi=14&fec=23&mtype=16qam&msys=dvbt&tmode=8k RTSP/1.0
CSeq: 2
Transport: RTP/AVP;unicast;client_port=33071-33072
User-Agent: vdr-satip/2.2.0-GIT-0-174-g78e1f48 (device 0)
If-Modified-Since: Thu, 01 Jan 1970 00:00:00 GMT
--- LINE END ---
[      Streams.cpp:232] Found StreamID x - SessionID x - Creating new SessionID: 1702957685
[       Stream.cpp:064] Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbt
[       Stream.cpp:203] Stream: 0, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 0
[     Frontend.cpp:379] Stream: 0, Opened FE fd: 9.
[     Frontend.cpp:340] Stream: 0, Start tuning process...
[     Frontend.cpp:390] Stream: 0, Waiting on lock.
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:422] Stream: 0, Opened DVR fd: 10.
[    RtpThread.cpp:089] Stream: 0, Start RTP streaming to 192.168.1.21 (33071 - 33072)
[   RtcpThread.cpp:073] Stream: 0, Start RTCP streaming to 192.168.1.21 (33071 - 33072)
[   RtspServer.cpp:141] RTSP/1.0 200 OK
CSeq: 2
Session: 1702957685;timeout=60
[... 59 lines cut ...]
CSeq: 2
Transport: RTP/AVP;unicast;client_port=33071-33072
User-Agent: vdr-satip/2.2.0-GIT-0-174-g78e1f48 (device 0)
If-Modified-Since: Thu, 01 Jan 1970 00:00:00 GMT
--- LINE END ---
[      Streams.cpp:232] Found StreamID x - SessionID x - Creating new SessionID: 0205744672
[       Stream.cpp:064] Stream: 0, StreamClient[0] with SessionID 0205744672 for dvbt
[       Stream.cpp:203] Stream: 0, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 0
[     Frontend.cpp:379] Stream: 0, Opened FE fd: 9.
[     Frontend.cpp:340] Stream: 0, Start tuning process...
[     Frontend.cpp:390] Stream: 0, Waiting on lock.
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:404] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:422] Stream: 0, Opened DVR fd: 10.
[    RtpThread.cpp:089] Stream: 0, Start RTP streaming to 192.168.1.21 (33071 - 33072)
[   RtcpThread.cpp:073] Stream: 0, Start RTCP streaming to 192.168.1.21 (33071 - 33072)
[   RtspServer.cpp:141] RTSP/1.0 200 OK
CSeq: 2
Session: 0205744672;timeout=60
Transport: RTP/AVP;unicast;client_port=33071-33072
com.ses.streamID: 0
--- LINE END ---

Barracuda09, 2015-03-07:
Hi Paul,

Many many thanks for trying again.... I really think I have to try it my self now..
The strange thing I see in the log is that it does not give any  FE status (Lock, signal)

Marc

Barracuda09, 2015-03-10:
Hi Paul,

Is this issue fix? regarding the IP_ADD_MEMBERSHIP.

pbriesch, 2015-03-10:
Yes, it is. Thanks.

Paul

closed 2015-03-10 by pbriesch