Barracuda09/SATPI #28: Add support for DVB API 5.4

lars18th, 2015-03-06:
Hi,

In order to add support for DVB API 5.4 (several recent Linux drivers only support up to 5.4) you only need to manually check tuner type if the ioctl for FE_GET_INFO command fails. This is the only current requirement for API 5.5.

So, if you like some inspiration you can review my patch for minisatip (similar project) here: http://github.com/catalinii/minisatip/issues/98#issuecomment-76725502

I'm sure you can adapt to your code in some minutes! :)
Regards!

Barracuda09, 2015-03-06:
Hi lars18th,

Thanks I will look into this.. but did you perhaps mean FE_GET_PROPERTY?

Marc

Barracuda09, 2015-03-06:
Hi,

I have bubble check what I run for API here.. But is is 5.4 as well. I do not have these problems.

Regards,

Marc

lars18th, 2015-03-09:
Hi Barracuda,

Yes, I'm speaking about FE_GET_PROPERTY, not FE_GET_INFO. Sorry for that!

Related to the problem, your code don't work with my current tuners (USB DVB-T using API 5.4). This is the output with the current version (MAC changed):

$ ./satpi --no-daemon
--- starting SatPI version: -unknown ---
eth0: 192.168.1.100 [00:00:00:00:00:00]
Detecting frontends in: /dev/dvb
Frontends found: 2
Frontend Name: Afatech AF9013 DVB-T
ioctl FE_GET_PROPERTY: Invalid argument (code 22)
Frontend Name: Afatech AF9013 DVB-T
ioctl FE_GET_PROPERTY: Invalid argument (code 22)
Setting up RTSP server
Setting up SSDP server with BOOTID: 1

As you can see the code of line #159 in file "Frontend.cpp" (https://github.com/Barracuda09/SATPI/blob/e929fadbb3097e571f989c4e1fe1805544984459/Frontend.cpp#L159) fails when you use this CMD ioctl. This command is supported ONLY in DVB-API 5.5, and not in API 5.4 But if you use devices with only one Tunning Space, you're fine if using API 5.4 to determine the tuner type. You only need to add some code when this call fails, and check the tuner type using the old API. With this simple change you can support API>=5.5 AND API=5.4. You can as example in my patch for the other similar project.

Please, update you code as It's easy! Several systems are still running with 5.4 and not all device drivers are compatible with 5.5.

I hope you can do it! :)
Best!

Barracuda09, 2015-03-09:
Hi,

Well I was under the impression that FE_GET_PROPERTY was already in 5.4?

Barracuda09, 2015-03-09:
Ok.. the problem is DTV_ENUM_DELSYS which is not in 5.4

lars18th, 2015-03-09:
Hi,

Yes, the COMMAND not found in 5.5 is DTV_ENUM_DELSYS... sorry (several posts in one day!).
But the line that you need to check is the IF in #159, because you execute the ioctl with the CMD in this line! The CMD is in the parameter set at line #153.

I hope you soon update your code with 5.4 support! :)

Barracuda09, 2015-03-09:
Hi,

Yes thanks.. I will do so. If I have fixed this can you check it then?

Marc

lars18th, 2015-03-10:
Hi Marc,

Yes, please, fix it and I try in my system!
I like to start to use your program soon.

Regards!

Barracuda09, 2015-03-10:
Hi,

I have tried to fix this. If you like and have time you can test it.

Thanks

lars18th, 2015-03-10:
Hi,

Cloned last version and compile it. Comments:

1) You need to change Frontend.cpp#232, because the new definition of SYS_DVBC_ANNEX_C breaks the compilation (I changed it for ANNEX_A for compiling because I don't use C or A at time).

2) The program starts with this info:

--- starting SatPI version: -unknown ---
eth0: 192.168.1.10 [00:00:00:00:00:00]
Detecting frontends in: /dev/dvb
Frontends found: 2
Frontend Name: Afatech AF9013 DVB-T
Frontend Name: Afatech AF9013 DVB-T
Setting up SSDP server with BOOTID: 5
Setting up HTTP server
Setting up RTSP server

But when I use any SAT>IP Client, the log shows:

Found StreamID x - SessionID x - Creating new SessionID: 0671693627
Stream: 0, StreamClient[0] with SessionID 0671693627 for dvbt
Stream: 0, Opened FE fd: 9.
Stream: 0, Waiting on lock.
Stream: 0, Not locked yet (FE status 0)...
Stream: 0, Not locked yet (FE status 0)...
Stream: 0, Not locked yet (FE status 0)...
Stream: 0, Not locked yet (FE status 0)...
Stream: 0, Not locked yet (FE status 0)...
Stream: 0, Not locked yet (FE status 0)...
Stream: 0, Not locked yet (FE status 0)...
Stream: 0, Not locked yet (FE status 0)...
Stream: 0, Opened DVR fd: 10.
Stream: 0, Start RTP streaming to 192.168.1.20 (52042 - 52043)
Stream: 0, Start RTCP streaming to 192.168.1.20 (52042 - 52043)
Found StreamID x - SessionID: 0671693627
Stream: 0, StreamClient[0] with SessionID 0671693627
Stream: 0, Waiting on lock.
Stream: 0, Not locked yet (FE status 0)...
Stream: 0, Not locked yet (FE status 0)...
Stream: 0, Not locked yet (FE status 0)...
Stream: 0, Not locked yet (FE status 0)...
Stream: 0, Not locked yet (FE status 0)...
Stream: 0, Not locked yet (FE status 0)...
Stream: 0, Not locked yet (FE status 0)...
Stream: 0, Not locked yet (FE status 0)...
Stream: 0, Updating PID filters...
Found StreamID x - SessionID: 0671693627
Stream: 0, StreamClient[0] with SessionID 0671693627
Stream: 0, Teardown StreamClient[0] with SessionID 0671693627
Stream: 0, Stop RTP streaming to 192.168.1.20 (52042 - 52043) (Streamed 0.000 MBytes)
Stream: 0, Stop RTCP stream to 192.168.1.20 (52042 - 52043)
Found SessionID 0671693627 by SocketClient
Found StreamID x - SessionID: 0671693627
Stream: 0, StreamClient[0] with SessionID 0671693627
Stream: 0, Close StreamClient[0] with SessionID 0671693627

And no data received by the client. :-(

But, see the info of "Streamed 0.000 MBytes"... I feel the problem is that you aren't reading data from the dvb device!

I hope you can fix it!
Regards!

Barracuda09, 2015-03-10:
Hi,

Can you make an debug version and post again the log?

Barracuda09, 2015-03-10:
And what is the compiler output with SYS_DVBC_ANNEX_C then?

lars18th, 2015-03-10:
Hi Marc,

> And what is the compiler output with SYS_DVBC_ANNEX_C then?

$ uname -a
Linux debian 3.2.0-4-amd64 #1 SMP Debian 3.2.63-2+deb7u1 x86_64 GNU/Linux

$ git clone --depth=1 git://github.com/Barracuda09/SATPI
Cloning into 'SATPI'...

$ cd SATPI
$ make
./version.sh Version.cpp > /dev/null
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Version.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Satpi.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG ChannelData.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG StreamProperties.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Properties.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG StringConverter.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Log.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG HttpcSocket.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG TcpSocket.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG UdpSocket.cpp
g++ -c -Wall -Wextra -Winit-self  -O2 -s -DNDEBUG Frontend.cpp
Frontend.cpp: In member function ‘bool Frontend::setFrontendInfo()’:
Frontend.cpp:232:24: error: invalid conversion from ‘int’ to ‘fe_delivery_system_t {aka fe_delivery_system}’ [-fpermissive]
Frontend.cpp: In member function ‘bool Frontend::tune(int, const ChannelData&)’:
Frontend.cpp:280:3: warning: case value ‘18’ not in enumerated type ‘const fe_delivery_system_t {aka const fe_delivery_system}’ [-Wswitch]
make: *** [Frontend.o] Error 1

The problem is the definition of SYS_DVBC_ANNEX_C:
in DVB-API 5.4 is only declared SYS_DVBC_ANNEX_AC and SYS_DVBC_ANNEX_B

Regards!

Barracuda09, 2015-03-10:
Hi,

Yes Thanks... I whipped up an old DVB-API and fixed it...

lars18th, 2015-03-10:
Hi Marc,

> Can you make an debug version and post again the log?

Compiled with "make debug"...

[        Satpi.cpp:243] --- starting SatPI version: -unknown ---
[  InterfaceAttr.h:094] eth0: 192.168.1.10 [00:00:00:00:00:00]
[      Streams.cpp:116] Not the preferred DVB API version, for correct function it should be 5.5 or higher
[      Streams.cpp:118] Current DVB_API_VERSION: 5.4
[      Streams.cpp:120] Detecting frontends in: /dev/dvb
[      Streams.cpp:130] Frontends found: 2
[     Frontend.cpp:145] Frontend Name: Afatech AF9013 DVB-T
[     Frontend.cpp:161] Unable to enumerate the delivery systems, retrying via old API Call
[     Frontend.cpp:217] Frontend Type: Terrestrial (DVB-T)
[     Frontend.cpp:241] Frontend Freq: 174000000 Hz to 860000000 Hz
[     Frontend.cpp:242] Frontend srat: 0 symbols/s to 0 symbols/s
[     Frontend.cpp:145] Frontend Name: Afatech AF9013 DVB-T
[     Frontend.cpp:161] Unable to enumerate the delivery systems, retrying via old API Call
[     Frontend.cpp:217] Frontend Type: Terrestrial (DVB-T)
[     Frontend.cpp:241] Frontend Freq: 174000000 Hz to 860000000 Hz
[     Frontend.cpp:242] Frontend srat: 0 symbols/s to 0 symbols/s
[   SsdpServer.cpp:107] Setting up SSDP server with BOOTID: 4
[   HttpServer.cpp:074] Setting up HTTP server
[   RtspServer.cpp:059] Setting up RTSP server
[    TcpSocket.cpp:148] RTSP Connection from 192.168.1.20 Port 58107 (fd: 8)
[    TcpSocket.cpp:085] RTSP Client 192.168.1.20: Connection closed with fd: 8
[      Streams.cpp:212] Found message outside session
[    TcpSocket.cpp:148] RTSP Connection from 192.168.1.20 Port 58108 (fd: 8)
[    TcpSocket.cpp:085] RTSP Client 192.168.1.20: Connection closed with fd: 8
[      Streams.cpp:212] Found message outside session
[    TcpSocket.cpp:148] RTSP Connection from 192.168.1.20 Port 58109 (fd: 8)
[   RtspServer.cpp:071] Data from client 192.168.1.20: OPTIONS rtsp://192.168.1.10:554/ RTSP/1.0
CSeq: 1
--- LINE END ---
[      Streams.cpp:212] Found message outside session
[       Stream.cpp:203] Stream: -1, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:196] RTSP/1.0 200 OK
CSeq: 1
Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
--- LINE END ---
[   RtspServer.cpp:071] Data from client 192.168.1.20: SETUP rtsp://192.168.1.10:554/?freq=858&msys=dvbt&bw=8 RTSP/1.0
CSeq: 2
Transport: RTP/AVP;unicast;client_port=52014-52015
--- LINE END ---
[      Streams.cpp:230] Found StreamID x - SessionID x - Creating new SessionID: 1702957685
[       Stream.cpp:064] Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbt
[       Stream.cpp:203] Stream: 0, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 0
[     Frontend.cpp:407] Stream: 0, Opened FE fd: 9.
[     Frontend.cpp:368] Stream: 0, Start tuning process...
[     Frontend.cpp:418] Stream: 0, Waiting on lock.
[     Frontend.cpp:432] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:432] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:432] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:432] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:432] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:432] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:432] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:432] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:450] Stream: 0, Opened DVR fd: 10.
[    RtpThread.cpp:090] Stream: 0, Start RTP streaming to 192.168.1.20 (52014 - 52015)
[   RtcpThread.cpp:074] Stream: 0, Start RTCP streaming to 192.168.1.20 (52014 - 52015)
[   RtspServer.cpp:142] RTSP/1.0 200 OK
CSeq: 2
Session: 1702957685;timeout=60
Transport: RTP/AVP;unicast;client_port=52014-52015
com.ses.streamID: 0
--- LINE END ---
[... 23 lines cut ...]
Session: 1702957685
--- LINE END ---
[   RtspServer.cpp:071] Data from client 192.168.1.20: TEARDOWN rtsp://192.168.1.10:554/stream=0 RTSP/1.0
CSeq: 4
Session: 1702957685
--- LINE END ---
[      Streams.cpp:222] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 1702957685
[ StreamClient.cpp:043] Connection can close: 1
[       Stream.cpp:125] Stream: 0, Teardown StreamClient[0] with SessionID 1702957685
[    RtpThread.cpp:101] Stream: 0, Stop RTP streaming to 192.168.1.20 (52014 - 52015) (Streamed 0.000 MBytes)
[   RtcpThread.cpp:084] Stream: 0, Stop RTCP stream to 192.168.1.20 (52014 - 52015)
[     Frontend.cpp:490] Stream: 0, Remove filter PID: 0000 - fd: 013 - Packet Count: 0
[   RtspServer.cpp:284] RTSP/1.0 200 OK
CSeq: 4
Session: 1702957685
--- LINE END ---
[    TcpSocket.cpp:085] RTSP Client 192.168.1.20: Connection closed with fd: 8
[      Streams.cpp:202] Found SessionID 1702957685 by SocketClient
[      Streams.cpp:222] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 1702957685
[       Stream.cpp:110] Stream: 0, Close StreamClient[0] with SessionID 1702957685

I feel the problem is the tunning process: Are you using some 5.5 specific commands for it?
For sure, the same frequency works with the same client and minisatip as SAT>IP server!

Barracuda09, 2015-03-10:
Hi,

If you like and have some time could you try the last commit?

Thanks

lars18th, 2015-03-10:
Hi Marc,

Compìled and tested... result: no go!

Here the relevant part of the log (compiled with debug):

[      Streams.cpp:222] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 1702957685
[       Stream.cpp:203] Stream: 0, Parsing transport parameters...
[       Stream.cpp:317] Not supported delivery system
[ StreamClient.cpp:043] Connection can close: 0
[     Frontend.cpp:374] Stream: 0, Start tuning process...
[     Frontend.cpp:424] Stream: 0, Waiting on lock.
[     Frontend.cpp:438] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:438] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:438] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:438] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:438] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:438] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:438] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:438] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:463] Stream: 0, Updating PID filters...
[     Frontend.cpp:480] Stream: 0, Set filter PID: 0000 - fd: 013

The message "Not supported delivery system" is new. I suggest to include in the debug the characteristics of detected tuners. Now I see at start:

[      Streams.cpp:116] Not the preferred DVB API version, for correct function it should be 5.5 or higher
[      Streams.cpp:118] Current DVB_API_VERSION: 5.4
[      Streams.cpp:120] Detecting frontends in: /dev/dvb
[      Streams.cpp:130] Frontends found: 2
[     Frontend.cpp:145] Frontend Name: Afatech AF9013 DVB-T
[     Frontend.cpp:161] Unable to enumerate the delivery systems, retrying via old API Call
[     Frontend.cpp:217] Frontend Type: Terrestrial (DVB-T)
[     Frontend.cpp:245] Frontend Freq: 174000000 Hz to 860000000 Hz
[     Frontend.cpp:246] Frontend srat: 0 symbols/s to 0 symbols/s
[     Frontend.cpp:145] Frontend Name: Afatech AF9013 DVB-T
[     Frontend.cpp:161] Unable to enumerate the delivery systems, retrying via old API Call
[     Frontend.cpp:217] Frontend Type: Terrestrial (DVB-T)
[     Frontend.cpp:245] Frontend Freq: 174000000 Hz to 860000000 Hz
[     Frontend.cpp:246] Frontend srat: 0 symbols/s to 0 symbols/s

And both "devices" are the same USB DUAL DVB-T tuner.

I hope you can finally fix today! :)

Barracuda09, 2015-03-10:
Hi,

Thanks again for trying.. But could you also give the SETUP message?

lars18th, 2015-03-10:
Hi Marc,

> But could you also give the SETUP message?

I do another clone, compile with "debug", and here the relevant log that you request:

[   RtspServer.cpp:071] Data from client 192.168.210.101: OPTIONS rtsp://192.168.1.10:554/ RTSP/1.0
[   RtspServer.cpp:071] Data from client 192.168.210.101: SETUP rtsp://192.168.1.10:554/?freq=634&msys=dvbt&bw=8&pids=0 RTSP/1.0
[   RtspServer.cpp:071] Data from client 192.168.210.101: PLAY rtsp://192.168.1.10:554/stream=0?addpids=18 RTSP/1.0
RTP-Info: url=rtsp://192.168.1.10/stream=0
[   RtspServer.cpp:071] Data from client 192.168.210.101: PLAY rtsp://192.168.1.10:554/stream=0?addpids=1001,1101,0,1000,1200 RTSP/1.0
RTP-Info: url=rtsp://192.168.1.10/stream=0
[   RtspServer.cpp:071] Data from client 192.168.210.101: PLAY rtsp://192.168.1.10:554/stream=0?delpids=1001,1101,0,1000,1200 RTSP/1.0
RTP-Info: url=rtsp://192.168.1.10/stream=0
[   RtspServer.cpp:071] Data from client 192.168.210.101: PLAY rtsp://192.168.1.10:554/stream=0?delpids=18 RTSP/1.0
RTP-Info: url=rtsp://192.168.1.10/stream=0
[   RtspServer.cpp:071] Data from client 192.168.210.101: TEARDOWN rtsp://192.168.1.10:554/stream=0 RTSP/1.0

As you see "msys=" parameter is only in the "tune" message (aka SETUP), and the problem is that you requiere this parameter in each command (and is not requiered, as when the tune is done, you can modify the 'stream' without more info about the tunning).

I hope you can finally fix the problem!
Moreover, I suggest this patch for more debug info:

--- ../SATPI/Stream.cpp 2015-03-10 15:17:41.745383980 -0400
+++ ./Stream.cpp        2015-03-10 15:04:59.013419723 -0400
@@ -215,6 +215,7 @@
                setSymbolRate(intVal * 1000);
        }
        if ((msys = StringConverter::getMSYSParameter(msg, method)) != SYS_UNDEFINED) {
+               SI_LOG_DEBUG("getMSYSParameter:setDeliverySystem(%d)",msys);
                setDeliverySystem(msys);
        }
        if (StringConverter::getStringParameter(msg, method, "pol=", strVal) == true) {
@@ -318,7 +319,7 @@
                                setModulationType(QAM_AUTO);
                                break;
                        default:
-                               SI_LOG_ERROR("Not supported delivery system");
+                               SI_LOG_ERROR("Not supported delivery system (value=%d)",msys);
                                break;
                }
        }

It helps me to see that the problem isn't related to the code processing "msys=" parameter but the lack of the parameter in que RTSP message! :)

Barracuda09, 2015-03-10:
Hi,

Did you still see the message "Not supported delivery system"?

pbriesch, 2015-03-10:
Hi Marc,

I do see this message: [       Stream.cpp:321] Not supported delivery system
````
[      Streams.cpp:225] Found StreamID x - SessionID: 0205744672
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 0205744672
[       Stream.cpp:204] Stream: 0, Parsing transport parameters...
[       Stream.cpp:321] Not supported delivery system
[ StreamClient.cpp:043] Connection can close: 0
[     Frontend.cpp:384] Stream: 0, Start tuning process...
[     Frontend.cpp:376] Stream: 0, Sending DiSEqC [e0] [10] [38] [f2]
^Cpbriesch@ubuntu32:~/satpi$ sudo ./satpi --no-daemon
[        Satpi.cpp:243] --- starting SatPI version: 1.5.14~g91c11d8 ---
[  InterfaceAttr.h:094] eth0: 192.168.1.169 [00:19:66:0f:8d:07]
[      Streams.cpp:118] Current DVB_API_VERSION: 5.10
[      Streams.cpp:120] Detecting frontends in: /dev/dvb
[      Streams.cpp:130] Frontends found: 2
[     Frontend.cpp:145] Frontend Name: STB0899 Multistandard
[     Frontend.cpp:213] Frontend Type: Satellite (DVB-S)
[     Frontend.cpp:217] Frontend Type: Satellite (DVB-S2)
[     Frontend.cpp:209] Frontend Type: DSS
[     Frontend.cpp:252] Frontend Freq: 950000 Hz to 2150000 Hz
[     Frontend.cpp:253] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
[     Frontend.cpp:145] Frontend Name: STB0899 Multistandard
[     Frontend.cpp:213] Frontend Type: Satellite (DVB-S)
[     Frontend.cpp:217] Frontend Type: Satellite (DVB-S2)
[     Frontend.cpp:209] Frontend Type: DSS
[     Frontend.cpp:252] Frontend Freq: 950000 Hz to 2150000 Hz
[     Frontend.cpp:253] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
[   SsdpServer.cpp:107] Setting up SSDP server with BOOTID: 25
[   HttpServer.cpp:074] Setting up HTTP server
[   RtspServer.cpp:059] Setting up RTSP server
[    TcpSocket.cpp:148] RTSP Connection from 192.168.1.21 Port 53098 (fd: 8)
[   RtspServer.cpp:071] Data from client 192.168.1.21: OPTIONS rtsp://192.168.1.169:554/ RTSP/1.0
CSeq: 1
--- LINE END ---

pbriesch, 2015-03-10:
Now I'm getting an error message I haven't seen before: 
**[   RtcpThread.cpp:123] FE_READ_STATUS failed: Inappropriate ioctl for device (code 25)**
````
v=0
o=- 2 3 IN IP4 192.168.1.169
s=SatIPServer:1 2,0,0
t=0 0
m=video 0 RTP/AVP 33
c=IN IP4 0.0.0.0
a=control:stream=0
a=fmtp:33 ver=1.0;src=1;tuner=1,2,1,0,11082.00,0,dvbs2,8psk,0,0.20,22000,34;pids=
a=inactive
m=video 0 RTP/AVP 33
c=IN IP4 0.0.0.0
a=control:stream=1
a=fmtp:33 ver=1.0;src=1;tuner=2,2,1,0,11493.00,0,dvbs2,8psk,0,0.35,22000,23;pids=0,16,17,18,20,54,5100,5101,5102,5103,5104,5105,5106
a=sendonly
[   RtcpThread.cpp:123] FE_READ_STATUS failed: Inappropriate ioctl for device (code 25)
[   RtcpThread.cpp:123] FE_READ_STATUS failed: Inappropriate ioctl for device (code 25)
[   RtcpThread.cpp:123] FE_READ_STATUS failed: Inappropriate ioctl for device (code 25)
[   RtcpThread.cpp:123] FE_READ_STATUS failed: Inappropriate ioctl for device (code 25)
[   RtcpThread.cpp:123] FE_READ_STATUS failed: Inappropriate ioctl for device (code 25)
[   RtcpThread.cpp:123] FE_READ_STATUS failed: Inappropriate ioctl for device (code 25)
[   RtcpThread.cpp:123] FE_READ_STATUS failed: Inappropriate ioctl for device (code 25)
[   RtcpThread.cpp:123] FE_READ_STATUS failed: Inappropriate ioctl for device (code 25)
[    TcpSocket.cpp:085] RTSP Client 192.168.1.107: Connection closed with fd: 11
[      Streams.cpp:215] Found message outside session
[       Stream.cpp:110] Stream: -1, Close StreamClient[0] with SessionID -1
[    TcpSocket.cpp:148] RTSP Connection from 192.168.1.107 Port 62675 (fd: 11)
[   RtspServer.cpp:071] Data from client 192.168.1.107: DESCRIBE rtsp://192.168.1.169/ RTSP/1.0
Connection:Close
CSeq:1320
Accept:application/sdp
--- LINE END ---
[      Streams.cpp:215] Found message outside session
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:261] RTSP/1.0 200 OK
CSeq: 1320
Content-Type: application/sdp
Content-Base: rtsp://192.168.1.169/
Content-Length: 426
--- LINE END ---
v=0
o=- 2 3 IN IP4 192.168.1.169
s=SatIPServer:1 2,0,0
t=0 0
m=video 0 RTP/AVP 33
c=IN IP4 0.0.0.0
a=control:stream=0
a=fmtp:33 ver=1.0;src=1;tuner=1,2,1,0,11082.00,0,dvbs2,8psk,0,0.20,22000,34;pids=
a=inactive
m=video 0 RTP/AVP 33
c=IN IP4 0.0.0.0
a=control:stream=1
a=fmtp:33 ver=1.0;src=1;tuner=2,2,1,0,11493.00,0,dvbs2,8psk,0,0.35,22000,23;pids=0,16,17,18,20,54,5100,5101,5102,5103,5104,5105,5106
a=sendonly
[   RtcpThread.cpp:123] FE_READ_STATUS failed: Inappropriate ioctl for device (code 25)
[   RtcpThread.cpp:123] FE_READ_STATUS failed: Inappropriate ioctl for device (code 25)

Barracuda09, 2015-03-11:
Hi Paul,

Just saw this just now, but what were you doing at this point.
What programs were you using.

Marc

lars18th, 2015-03-11:
Hi Marc,

> Did you still see the message "Not supported delivery system"?

Sure! The bug still here. Please, revise your code: you check in ALL requests for the "msys" parameter, but this is not the case... when the request is using the "stream" parameter you need to reuse the data related to this stream, instead to receive more info inside the request.

FYI, I'm using serveral DVBViewer based SAT>IP clients for testing (DVBViewer Pro, GE, DD-TV).

pbriesch, 2015-03-11:
Hello Marc,

> What programs were you using.

According to the IP (192.168.1.107) I must have used EyeTV on the Mac.

Anyway, I have tested the new commit with VDR with vtuner and now see new error messages: 
`[   RtcpThread.cpp:122] FE_READ_STATUS failed: Bad file descriptor (code 9)`
EyeTV on Mac works. But same new error message. Please find each log below.

Paul

**VDR with vtuner:**
````
0  Wed Mar 11 13:26:27.972772523 2015  6  [        Satpi.cpp:243] --- starting SatPI version: 1.5.15~ga60aac8 ---
1  Wed Mar 11 13:26:27.974451293 2015  6  [  InterfaceAttr.h:094] eth0: 192.168.1.169 [00:19:66:0f:8d:07]
2  Wed Mar 11 13:26:27.981419300 2015  7  [      Streams.cpp:118] Current DVB_API_VERSION: 5.10
3  Wed Mar 11 13:26:27.982466848 2015  6  [      Streams.cpp:120] Detecting frontends in: /dev/dvb
4  Wed Mar 11 13:26:27.983174083 2015  6  [      Streams.cpp:130] Frontends found: 2
5  Wed Mar 11 13:26:27.984176662 2015  6  [     Frontend.cpp:145] Frontend Name: STB0899 Multistandard
6  Wed Mar 11 13:26:27.984896982 2015  7  [     Frontend.cpp:213] Frontend Type: Satellite (DVB-S)
7  Wed Mar 11 13:26:27.985331116 2015  7  [     Frontend.cpp:217] Frontend Type: Satellite (DVB-S2)
8  Wed Mar 11 13:26:27.985793728 2015  7  [     Frontend.cpp:209] Frontend Type: DSS
9  Wed Mar 11 13:26:27.986193129 2015  7  [     Frontend.cpp:252] Frontend Freq: 950000 Hz to 2150000 Hz
10  Wed Mar 11 13:26:27.986608406 2015  7  [     Frontend.cpp:253] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
11  Wed Mar 11 13:26:27.987082493 2015  6  [     Frontend.cpp:145] Frontend Name: STB0899 Multistandard
12  Wed Mar 11 13:26:27.987472130 2015  7  [     Frontend.cpp:213] Frontend Type: Satellite (DVB-S)
13  Wed Mar 11 13:26:27.988064824 2015  7  [     Frontend.cpp:217] Frontend Type: Satellite (DVB-S2)
14  Wed Mar 11 13:26:27.988495192 2015  7  [     Frontend.cpp:209] Frontend Type: DSS
15  Wed Mar 11 13:26:27.988813760 2015  7  [     Frontend.cpp:252] Frontend Freq: 950000 Hz to 2150000 Hz
16  Wed Mar 11 13:26:27.989317138 2015  7  [     Frontend.cpp:253] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
17  Wed Mar 11 13:26:27.990440513 2015  6  [   SsdpServer.cpp:107] Setting up SSDP server with BOOTID: 36
18  Wed Mar 11 13:26:27.991536702 2015  6  [   HttpServer.cpp:074] Setting up HTTP server
19  Wed Mar 11 13:26:27.992648866 2015  6  [   RtspServer.cpp:059] Setting up RTSP server
20  Wed Mar 11 13:26:28.711136670 2015  6  [    TcpSocket.cpp:148] HTTP Connection from 192.168.1.107 Port 51985 (fd: 8)
21  Wed Mar 11 13:26:29.479114739 2015  6  [    TcpSocket.cpp:148] RTSP Connection from 192.168.1.21 Port 42703 (fd: 9)
22  Wed Mar 11 13:26:29.480043440 2015  7  [   RtspServer.cpp:071] Data from client 192.168.1.21: OPTIONS rtsp://192.168.1.169:554/ RTSP/1.0
23  Wed Mar 11 13:26:29.480043440 2015  7  CSeq: 1
24  Wed Mar 11 13:26:29.480043440 2015  7  --- LINE END ---
25  Wed Mar 11 13:26:29.481150259 2015  6  [      Streams.cpp:215] Found message outside session
26  Wed Mar 11 13:26:29.481524952 2015  7  [ StreamClient.cpp:043] Connection can close: 1
27  Wed Mar 11 13:26:29.481914131 2015  7  [   RtspServer.cpp:196] RTSP/1.0 200 OK
28  Wed Mar 11 13:26:29.481914131 2015  7  CSeq: 1
29  Wed Mar 11 13:26:29.481914131 2015  7  Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
30  Wed Mar 11 13:26:29.481914131 2015  7  --- LINE END ---
31  Wed Mar 11 13:26:29.483540508 2015  7  [   RtspServer.cpp:071] Data from client 192.168.1.21: SETUP rtsp://192.168.1.169/?src=1&freq=12174.9&pol=h&msys=dvbs&sr=27500&fec=34&pids=173,18,20,0,17,16,146,84 RTSP/1.0
32  Wed Mar 11 13:26:29.483540508 2015  7  CSeq: 2
33  Wed Mar 11 13:26:29.483540508 2015  7  Transport: RTP/AVP;unicast;client_port=45538-45539
34  Wed Mar 11 13:26:29.483540508 2015  7  --- LINE END ---
35  Wed Mar 11 13:26:29.485121837 2015  6  [      Streams.cpp:233] Found StreamID x - SessionID x - Creating new SessionID: 1702957685
36  Wed Mar 11 13:26:29.485497515 2015  6  [       Stream.cpp:064] Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbs
37  Wed Mar 11 13:26:29.485894950 2015  7  [       Stream.cpp:197] Stream: 0, Parsing transport parameters...
38  Wed Mar 11 13:26:29.486962555 2015  7  [ StreamClient.cpp:043] Connection can close: 0
39  Wed Mar 11 13:26:29.487881260 2015  6  [     Frontend.cpp:423] Stream: 0, Opened FE fd: 10.
40  Wed Mar 11 13:26:29.488390796 2015  7  [     Frontend.cpp:384] Stream: 0, Start tuning process...
41  Wed Mar 11 13:26:29.488723105 2015  6  [     Frontend.cpp:376] Stream: 0, Sending DiSEqC [e0] [10] [38] [f3]
42  Wed Mar 11 13:26:29.866141096 2015  6  [     Frontend.cpp:434] Stream: 0, Waiting on lock.
43  Wed Mar 11 13:26:29.866304338 2015  6  [     Frontend.cpp:448] Stream: 0, Not locked yet (FE status 0)...
44  Wed Mar 11 13:26:30.364672439 2015  6  [     Frontend.cpp:448] Stream: 0, Not locked yet (FE status 0)...
45  Wed Mar 11 13:26:30.389941214 2015  6  [     Frontend.cpp:448] Stream: 0, Not locked yet (FE status 0)...
46  Wed Mar 11 13:26:30.415186726 2015  6  [     Frontend.cpp:448] Stream: 0, Not locked yet (FE status 0)...
47  Wed Mar 11 13:26:30.441151717 2015  6  [     Frontend.cpp:448] Stream: 0, Not locked yet (FE status 0)...
48  Wed Mar 11 13:26:30.467070848 2015  6  [     Frontend.cpp:448] Stream: 0, Not locked yet (FE status 0)...
49  Wed Mar 11 13:26:30.493021732 2015  6  [     Frontend.cpp:448] Stream: 0, Not locked yet (FE status 0)...
[... cut ...]
345  Wed Mar 11 13:38:19.143317651 2015  6  [       Stream.cpp:067] Stream: 1, StreamClient[0] with SessionID 0205744672
346  Wed Mar 11 13:38:19.143851207 2015  7  [ StreamClient.cpp:043] Connection can close: 1
347  Wed Mar 11 13:38:19.144448935 2015  6  [       Stream.cpp:123] Stream: 1, Teardown StreamClient[0] with SessionID 0205744672
348  Wed Mar 11 13:38:19.182866623 2015  3  [   RtcpThread.cpp:122] FE_READ_STATUS failed: Bad file descriptor (code 9)
349  Wed Mar 11 13:38:19.198593968 2015  3  [   RtcpThread.cpp:122] FE_READ_STATUS failed: Bad file descriptor (code 9)

Barracuda09, 2015-03-11:
Hi Paul,

Yes sorry...sorry.. I did not test this.. It should work now again.

pbriesch, 2015-03-11:
No problem. Bad file descriptor (code 9) is gone.

Regards, Paul

Barracuda09, 2015-03-11:
Did another fix for attribute describe string for dvb-s(2) (cross my fingers)

Barracuda09, 2015-03-11:
Again another fix for attribute describe string for dvb-s(2) (still fingers crossed)

pbriesch, 2015-03-11:
Congratulations Marc!

VDR with vtuner seems to work fine. Will test VDR with ver-plugin-satip later.

Paul

**VDR with vtuner**
````
0  Wed Mar 11 16:00:36.413652842 2015  6  [        Satpi.cpp:243] --- starting SatPI version: 1.5.20~gd1d1d0c ---
1  Wed Mar 11 16:00:36.415418623 2015  6  [  InterfaceAttr.h:094] eth0: 192.168.1.169 [00:19:66:0f:8d:07]
2  Wed Mar 11 16:00:36.422209691 2015  7  [      Streams.cpp:118] Current DVB_API_VERSION: 5.10
3  Wed Mar 11 16:00:36.423264778 2015  6  [      Streams.cpp:120] Detecting frontends in: /dev/dvb
4  Wed Mar 11 16:00:36.424050670 2015  6  [      Streams.cpp:130] Frontends found: 2
5  Wed Mar 11 16:00:36.424952615 2015  6  [     Frontend.cpp:145] Frontend Name: STB0899 Multistandard
6  Wed Mar 11 16:00:36.425540696 2015  7  [     Frontend.cpp:213] Frontend Type: Satellite (DVB-S)
7  Wed Mar 11 16:00:36.426005975 2015  7  [     Frontend.cpp:217] Frontend Type: Satellite (DVB-S2)
8  Wed Mar 11 16:00:36.426429554 2015  7  [     Frontend.cpp:209] Frontend Type: DSS
9  Wed Mar 11 16:00:36.426747610 2015  7  [     Frontend.cpp:252] Frontend Freq: 950000 Hz to 2150000 Hz
10  Wed Mar 11 16:00:36.427046518 2015  7  [     Frontend.cpp:253] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
11  Wed Mar 11 16:00:36.427613353 2015  6  [     Frontend.cpp:145] Frontend Name: STB0899 Multistandard
12  Wed Mar 11 16:00:36.428253682 2015  7  [     Frontend.cpp:213] Frontend Type: Satellite (DVB-S)
13  Wed Mar 11 16:00:36.428726381 2015  7  [     Frontend.cpp:217] Frontend Type: Satellite (DVB-S2)
14  Wed Mar 11 16:00:36.429178704 2015  7  [     Frontend.cpp:209] Frontend Type: DSS
15  Wed Mar 11 16:00:36.429594614 2015  7  [     Frontend.cpp:252] Frontend Freq: 950000 Hz to 2150000 Hz
16  Wed Mar 11 16:00:36.429913235 2015  7  [     Frontend.cpp:253] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
17  Wed Mar 11 16:00:36.431024083 2015  6  [   SsdpServer.cpp:107] Setting up SSDP server with BOOTID: 44
18  Wed Mar 11 16:00:36.432029629 2015  6  [   HttpServer.cpp:074] Setting up HTTP server
19  Wed Mar 11 16:00:36.432754157 2015  6  [   RtspServer.cpp:059] Setting up RTSP server
20  Wed Mar 11 16:00:37.631067172 2015  6  [    TcpSocket.cpp:148] HTTP Connection from 192.168.1.107 Port 60287 (fd: 8)
21  Wed Mar 11 16:00:38.331495104 2015  6  [    TcpSocket.cpp:148] RTSP Connection from 192.168.1.21 Port 43091 (fd: 9)
22  Wed Mar 11 16:00:38.332431450 2015  7  [   RtspServer.cpp:071] Data from client 192.168.1.21: OPTIONS rtsp://192.168.1.169:554/ RTSP/1.0
23  Wed Mar 11 16:00:38.332431450 2015  7  CSeq: 1
24  Wed Mar 11 16:00:38.332431450 2015  7  --- LINE END ---
25  Wed Mar 11 16:00:38.333536658 2015  6  [      Streams.cpp:215] Found message outside session
26  Wed Mar 11 16:00:38.333913542 2015  7  [ StreamClient.cpp:043] Connection can close: 1
27  Wed Mar 11 16:00:38.334490559 2015  7  [   RtspServer.cpp:196] RTSP/1.0 200 OK
28  Wed Mar 11 16:00:38.334490559 2015  7  CSeq: 1
29  Wed Mar 11 16:00:38.334490559 2015  7  Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
30  Wed Mar 11 16:00:38.334490559 2015  7  --- LINE END ---
31  Wed Mar 11 16:00:38.336550501 2015  7  [   RtspServer.cpp:071] Data from client 192.168.1.21: SETUP rtsp://192.168.1.169/?src=1&freq=10743.0&pol=h&msys=dvbs&sr=22000&fec=56&pids=18,401,20,0,17,16,400,402,403,407,408,404 RTSP/1.0
32  Wed Mar 11 16:00:38.336550501 2015  7  CSeq: 2
33  Wed Mar 11 16:00:38.336550501 2015  7  Transport: RTP/AVP;unicast;client_port=46469-46470
34  Wed Mar 11 16:00:38.336550501 2015  7  --- LINE END ---
35  Wed Mar 11 16:00:38.338437408 2015  6  [      Streams.cpp:233] Found StreamID x - SessionID x - Creating new SessionID: 1702957685
36  Wed Mar 11 16:00:38.339049886 2015  6  [       Stream.cpp:064] Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbs
37  Wed Mar 11 16:00:38.339604749 2015  7  [       Stream.cpp:197] Stream: 0, Parsing transport parameters...
38  Wed Mar 11 16:00:38.341054303 2015  7  [ StreamClient.cpp:043] Connection can close: 0
39  Wed Mar 11 16:00:38.341926447 2015  6  [     Frontend.cpp:423] Stream: 0, Opened FE fd: 10.
40  Wed Mar 11 16:00:38.342909900 2015  7  [     Frontend.cpp:384] Stream: 0, Start tuning process...
41  Wed Mar 11 16:00:38.343788235 2015  6  [     Frontend.cpp:376] Stream: 0, Sending DiSEqC [e0] [10] [38] [f2]
42  Wed Mar 11 16:00:38.718135309 2015  6  [     Frontend.cpp:434] Stream: 0, Waiting on lock.
43  Wed Mar 11 16:00:38.718309815 2015  6  [     Frontend.cpp:448] Stream: 0, Not locked yet (FE status 0)...
44  Wed Mar 11 16:00:39.025126687 2015  6  [     Frontend.cpp:445] Stream: 0, Tuned and locked (FE status 31).
45  Wed Mar 11 16:00:39.025567629 2015  6  [     Frontend.cpp:466] Stream: 0, Opened DVR fd: 11.
46  Wed Mar 11 16:00:39.025611955 2015  6  [     Frontend.cpp:473] Stream: 0, Updating PID filters...
47  Wed Mar 11 16:00:39.037187432 2015  7  [     Frontend.cpp:490] Stream: 0, Set filter PID: 0000 - fd: 012
48  Wed Mar 11 16:00:39.037806469 2015  7  [     Frontend.cpp:490] Stream: 0, Set filter PID: 0016 - fd: 013
49  Wed Mar 11 16:00:39.038258696 2015  7  [     Frontend.cpp:490] Stream: 0, Set filter PID: 0017 - fd: 014
50  Wed Mar 11 16:00:39.038969222 2015  7  [     Frontend.cpp:490] Stream: 0, Set filter PID: 0018 - fd: 015
51  Wed Mar 11 16:00:39.039563931 2015  7  [     Frontend.cpp:490] Stream: 0, Set filter PID: 0020 - fd: 016
[... cut ...]
284  Wed Mar 11 16:00:39.579666744 2015  7  [     Frontend.cpp:494] Stream: 0, Remove filter PID: 0500 - fd: 017 - Packet Count: 1
285  Wed Mar 11 16:00:39.580339222 2015  7  [     Frontend.cpp:490] Stream: 0, Set filter PID: 0600 - fd: 017
286  Wed Mar 11 16:00:39.581074054 2015  7  [   RtspServer.cpp:168] RTSP/1.0 200 OK
287  Wed Mar 11 16:00:39.581074054 2015  7  RTP-Info: url=rtsp://192.168.1.169/stream=0
288  Wed Mar 11 16:00:39.581074054 2015  7  CSeq: 17
289  Wed Mar 11 16:00:39.581074054 2015  7  Session: 1702957685
290  Wed Mar 11 16:00:39.581074054 2015  7  --- LINE END ---

lars18th, 2015-03-11:
Hi Marc,

I really appreciate your efforts! However, no go with last commit using DVB-T. Here the FULL log:

[        Satpi.cpp:243] --- starting SatPI version: -unknown ---
[  InterfaceAttr.h:094] eth0: 192.168.1.10 [00:00:00:00:00:00]
[      Streams.cpp:116] Not the preferred DVB API version, for correct function it should be 5.5 or higher
[      Streams.cpp:118] Current DVB_API_VERSION: 5.4
[      Streams.cpp:120] Detecting frontends in: /dev/dvb
[      Streams.cpp:130] Frontends found: 2
[     Frontend.cpp:145] Frontend Name: Afatech AF9013 DVB-T
[     Frontend.cpp:161] Unable to enumerate the delivery systems, retrying via old API Call
[     Frontend.cpp:221] Frontend Type: Terrestrial (DVB-T)
[     Frontend.cpp:252] Frontend Freq: 174000000 Hz to 860000000 Hz
[     Frontend.cpp:253] Frontend srat: 0 symbols/s to 0 symbols/s
[     Frontend.cpp:145] Frontend Name: Afatech AF9013 DVB-T
[     Frontend.cpp:161] Unable to enumerate the delivery systems, retrying via old API Call
[     Frontend.cpp:221] Frontend Type: Terrestrial (DVB-T)
[     Frontend.cpp:252] Frontend Freq: 174000000 Hz to 860000000 Hz
[     Frontend.cpp:253] Frontend srat: 0 symbols/s to 0 symbols/s
[   SsdpServer.cpp:107] Setting up SSDP server with BOOTID: 3
[   HttpServer.cpp:074] Setting up HTTP server
[   RtspServer.cpp:059] Setting up RTSP server
[    TcpSocket.cpp:148] RTSP Connection from 192.168.1.20 Port 63031 (fd: 8)
[   RtspServer.cpp:071] Data from client 192.168.1.20: OPTIONS rtsp://192.168.1.10:554/ RTSP/1.0
CSeq: 1
--- LINE END ---
[      Streams.cpp:215] Found message outside session
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:196] RTSP/1.0 200 OK
CSeq: 1
Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
--- LINE END ---
[   RtspServer.cpp:071] Data from client 192.168.1.20: SETUP rtsp://192.168.1.10:554/?freq=634&msys=dvbt&bw=8&pids=0 RTSP/1.0
CSeq: 2
Transport: RTP/AVP;unicast;client_port=52042-52043
--- LINE END ---
[      Streams.cpp:233] Found StreamID x - SessionID x - Creating new SessionID: 1702957685
[       Stream.cpp:064] Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbt
[       Stream.cpp:197] Stream: 0, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 0
[     Frontend.cpp:423] Stream: 0, Opened FE fd: 9.
[     Frontend.cpp:384] Stream: 0, Start tuning process...
[     Frontend.cpp:434] Stream: 0, Waiting on lock.
[     Frontend.cpp:448] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:448] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:448] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:448] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:448] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:448] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:448] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:448] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:466] Stream: 0, Opened DVR fd: 10.
[     Frontend.cpp:473] Stream: 0, Updating PID filters...
[     Frontend.cpp:490] Stream: 0, Set filter PID: 0000 - fd: 011
[    RtpThread.cpp:092] Stream: 0, Start  RTP stream to 192.168.1.20:52042
[   RtcpThread.cpp:073] Stream: 0, Start RTCP stream to 192.168.1.20:52043
[   RtspServer.cpp:142] RTSP/1.0 200 OK
CSeq: 2
Session: 1702957685;timeout=60
Transport: RTP/AVP;unicast;client_port=52042-52043
com.ses.streamID: 0
--- LINE END ---
[   RtspServer.cpp:071] Data from client 192.168.1.20: PLAY rtsp://192.168.1.10:554/stream=0?addpids=18 RTSP/1.0
CSeq: 3
Session: 1702957685
--- LINE END ---
[      Streams.cpp:225] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 1702957685
[       Stream.cpp:197] Stream: 0, Parsing transport parameters...
[... 114 lines cut ...]
Session: 1702957685
--- LINE END ---
[      Streams.cpp:225] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 1702957685
[ StreamClient.cpp:043] Connection can close: 1
[       Stream.cpp:123] Stream: 0, Teardown StreamClient[0] with SessionID 1702957685
[   RtcpThread.cpp:159] ver=1.1;tuner=1,97,0,0,634.00,8,dvbt,auto,UNKNOWN MODTYPE,auto,auto,0,0,0;pids=
[   RtcpThread.cpp:159] ver=1.1;tuner=1,97,0,0,634.00,8,dvbt,auto,UNKNOWN MODTYPE,auto,auto,0,0,0;pids=
[     ThreadBase.h:075] Thread did not stop within timeout?
[    RtpThread.cpp:102] Stream: 0, Stop  RTP stream to 192.168.1.20:52042 (Streamed 0.000 MBytes)
[     ThreadBase.h:075] Thread did not stop within timeout?
[   RtcpThread.cpp:083] Stream: 0, Stop RTCP stream to 192.168.1.20:52043
[   RtspServer.cpp:284] RTSP/1.0 200 OK
CSeq: 7
Session: 1702957685
--- LINE END ---
[    TcpSocket.cpp:085] RTSP Client 192.168.1.20: Connection closed with fd: 8
[      Streams.cpp:205] Found SessionID 1702957685 by SocketClient
[      Streams.cpp:225] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 1702957685
[       Stream.cpp:115] Stream: 0, Close StreamClient[0] with SessionID 1702957685

I feel you solved the problem with "msys=" parameter, but the lines showing UNKNOWN MODTYPE perhaps indicate that you don't correct tune the DVB-T device !!

What more I can do?

Barracuda09, 2015-03-11:
Hi lars18th,

Yes you can try to add modulation type because it will be default now QAM_AUTO.
The UNKNOWN MODTYPE  is because I forget the QAM_AUTO in the switch table.
But I don't think this is a problem.
Further more I hope I can get my DVB-T stick to work to try it myself.

Marc

lars18th, 2015-03-12:
Hi Marc,

Here log of last version:

[        Satpi.cpp:243] --- starting SatPI version: -unknown ---
[  InterfaceAttr.h:094] eth0: 192.168.1.10 [00:00:00:00:00:00]
[      Streams.cpp:116] Not the preferred DVB API version, for correct function it should be 5.5 or higher
[      Streams.cpp:118] Current DVB_API_VERSION: 5.4
[      Streams.cpp:120] Detecting frontends in: /dev/dvb
[      Streams.cpp:130] Frontends found: 2
[     Frontend.cpp:145] Frontend Name: Afatech AF9013 DVB-T
[     Frontend.cpp:161] Unable to enumerate the delivery systems, retrying via old API Call
[     Frontend.cpp:221] Frontend Type: Terrestrial (DVB-T)
[     Frontend.cpp:252] Frontend Freq: 174000000 Hz to 860000000 Hz
[     Frontend.cpp:253] Frontend srat: 0 symbols/s to 0 symbols/s
[     Frontend.cpp:145] Frontend Name: Afatech AF9013 DVB-T
[     Frontend.cpp:161] Unable to enumerate the delivery systems, retrying via old API Call
[     Frontend.cpp:221] Frontend Type: Terrestrial (DVB-T)
[     Frontend.cpp:252] Frontend Freq: 174000000 Hz to 860000000 Hz
[     Frontend.cpp:253] Frontend srat: 0 symbols/s to 0 symbols/s
[   SsdpServer.cpp:107] Setting up SSDP server with BOOTID: 2
[   HttpServer.cpp:074] Setting up HTTP server
[   RtspServer.cpp:059] Setting up RTSP server
[   SsdpServer.cpp:158] Found SAT>IP Server 192.168.1.80: with DEVICEID 80
[   SsdpServer.cpp:158] Found SAT>IP Server 192.168.1.80: with DEVICEID 80
[    TcpSocket.cpp:148] RTSP Connection from 192.168.1.10 Port 62683 (fd: 8)
[   RtspServer.cpp:071] Data from client 192.168.1.10: OPTIONS rtsp://192.168.1.10:554/ RTSP/1.0
CSeq: 1
--- LINE END ---
[      Streams.cpp:215] Found message outside session
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:196] RTSP/1.0 200 OK
CSeq: 1
Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
--- LINE END ---
[   RtspServer.cpp:071] Data from client 192.168.1.10: SETUP rtsp://192.168.1.10:554/?freq=538&msys=dvbt&bw=8&pids=0 RTSP/1.0
CSeq: 2
Transport: RTP/AVP;unicast;client_port=52042-52043
--- LINE END ---
[      Streams.cpp:233] Found StreamID x - SessionID x - Creating new SessionID: 1702957685
[       Stream.cpp:064] Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbt
[       Stream.cpp:197] Stream: 0, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 0
[     Frontend.cpp:423] Stream: 0, Opened FE fd: 9.
[     Frontend.cpp:384] Stream: 0, Start tuning process...
[     Frontend.cpp:434] Stream: 0, Waiting on lock.
[     Frontend.cpp:448] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:448] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:448] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:448] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:448] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:448] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:448] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:448] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:466] Stream: 0, Opened DVR fd: 10.
[     Frontend.cpp:473] Stream: 0, Updating PID filters...
[     Frontend.cpp:490] Stream: 0, Set filter PID: 0000 - fd: 011
[    RtpThread.cpp:092] Stream: 0, Start  RTP stream to 192.168.1.10:52042
[   RtcpThread.cpp:073] Stream: 0, Start RTCP stream to 192.168.1.10:52043
[   RtspServer.cpp:142] RTSP/1.0 200 OK
CSeq: 2
Session: 1702957685;timeout=60
Transport: RTP/AVP;unicast;client_port=52042-52043
com.ses.streamID: 0
--- LINE END ---
[   RtspServer.cpp:071] Data from client 192.168.1.10: PLAY rtsp://192.168.1.10:554/stream=0?addpids=18 RTSP/1.0
CSeq: 3
Session: 1702957685
--- LINE END ---
[      Streams.cpp:225] Found StreamID x - SessionID: 1702957685
[... 111 lines cut ...]
--- LINE END ---
[   RtspServer.cpp:071] Data from client 192.168.1.10: TEARDOWN rtsp://192.168.1.10:554/stream=0 RTSP/1.0
CSeq: 7
Session: 1702957685
--- LINE END ---
[      Streams.cpp:225] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 1702957685
[ StreamClient.cpp:043] Connection can close: 1
[       Stream.cpp:123] Stream: 0, Teardown StreamClient[0] with SessionID 1702957685
[    RtpThread.cpp:102] Stream: 0, Stop  RTP stream to 192.168.1.10:52042 (Streamed 0.000 MBytes)
[   RtcpThread.cpp:083] Stream: 0, Stop RTCP stream to 192.168.1.10:52043
[   RtspServer.cpp:284] RTSP/1.0 200 OK
CSeq: 7
Session: 1702957685
--- LINE END ---
[    TcpSocket.cpp:085] RTSP Client 192.168.1.10: Connection closed with fd: 8
[      Streams.cpp:205] Found SessionID 1702957685 by SocketClient
[      Streams.cpp:225] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 1702957685
[       Stream.cpp:115] Stream: 0, Close StreamClient[0] with SessionID 1702957685

Seems that the bug is on "Not locked yet (FE status 0)". Can you put code for more verbose log on tuning process?

I hope you can fix soon... it's annoying because minisatip it's working! :(
Regards!

pbriesch, 2015-03-12:
Hi Marc,

no sweat. Let me know if I can do some more testing for you.

Kind regards,

Paul

Barracuda09, 2015-03-12:
Hi Paul,

Sorry I don't think I fully know what you are meaning.... I guess VDR is still not working?

Barracuda09, 2015-03-12:
Hi lars18th,

Can you try VLC and give more known transport parameters for your freq like:
rtsp://10.201.2.191:554/?freq=674&msys=dvbt&mtype=64qam&bw=8&fec=12&tmode=8k&gi=14&pids=0

VLC will work only for a moment (because of a time-out) but you can give any kind of parameter you like.

pbriesch, 2015-03-12:
*No sweat* means something like don't get under pressure :-)
Regarding VDR I need to perform more testing.

lars18th, 2015-03-12:
Hi Marc,

> Can you try VLC and give more known transport parameters for your freq like:

I try this command:
rtsp://192.168.1.10:554/?freq=538&msys=dvbt&mtype=64qam&bw=8&fec=23&tmode=8k&gi=14&pids=all

Result with current version of satpi:

```
[        Satpi.cpp:243] --- starting SatPI version: -unknown ---
[  InterfaceAttr.h:094] eth0: 192.168.1.10 [00:00:00:00:00:00]
[      Streams.cpp:116] Not the preferred DVB API version, for correct function it should be 5.5 or higher
[      Streams.cpp:118] Current DVB_API_VERSION: 5.4
[      Streams.cpp:120] Detecting frontends in: /dev/dvb
[      Streams.cpp:130] Frontends found: 2
[     Frontend.cpp:145] Frontend Name: Afatech AF9013 DVB-T
[     Frontend.cpp:161] Unable to enumerate the delivery systems, retrying via old API Call
[     Frontend.cpp:221] Frontend Type: Terrestrial (DVB-T)
[     Frontend.cpp:252] Frontend Freq: 174000000 Hz to 860000000 Hz
[     Frontend.cpp:253] Frontend srat: 0 symbols/s to 0 symbols/s
[     Frontend.cpp:145] Frontend Name: Afatech AF9013 DVB-T
[     Frontend.cpp:161] Unable to enumerate the delivery systems, retrying via old API Call
[     Frontend.cpp:221] Frontend Type: Terrestrial (DVB-T)
[     Frontend.cpp:252] Frontend Freq: 174000000 Hz to 860000000 Hz
[     Frontend.cpp:253] Frontend srat: 0 symbols/s to 0 symbols/s
[   SsdpServer.cpp:107] Setting up SSDP server with BOOTID: 5
[   HttpServer.cpp:074] Setting up HTTP server
[   RtspServer.cpp:059] Setting up RTSP server
[    TcpSocket.cpp:148] RTSP Connection from 192.168.1.10 Port 64847 (fd: 8)
[   RtspServer.cpp:071] Data from client 192.168.1.10: OPTIONS rtsp://192.168.1.10:554/?freq=538&msys=dvbt&mtype=64qam&bw=8&fec
=23&tmode=8k&gi=14&pids=all RTSP/1.0
CSeq: 2
User-Agent: LibVLC/2.1.3 (LIVE555 Streaming Media v2014.01.21)
--- LINE END ---
[      Streams.cpp:233] Found StreamID x - SessionID x - Creating new SessionID: 1702957685
[       Stream.cpp:064] Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbt
[       Stream.cpp:197] Stream: 0, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:196] RTSP/1.0 200 OK
CSeq: 2
Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
Session: 1702957685
--- LINE END ---
[   RtspServer.cpp:071] Data from client 192.168.1.10: DESCRIBE rtsp://192.168.1.10:554/?freq=538&msys=dvbt&mtype=64qam&bw=8&fe
c=23&tmode=8k&gi=14&pids=all RTSP/1.0
CSeq: 3
User-Agent: LibVLC/2.1.3 (LIVE555 Streaming Media v2014.01.21)
Accept: application/sdp
--- LINE END ---
[      Streams.cpp:205] Found SessionID 1702957685 by SocketClient
[      Streams.cpp:225] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:064] Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbt
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:261] RTSP/1.0 200 OK
CSeq: 3
Content-Type: application/sdp
Content-Base: rtsp://192.168.1.10/
Content-Length: 213
Session: 1702957685
--- LINE END ---
v=0
o=- 2 3 IN IP4 192.168.1.10
s=SatIPServer:1 0,2,0
t=0 0
m=video 0 RTP/AVP 33
c=IN IP4 0.0.0.0
a=control:stream=0
a=fmtp:33 ver=1.1;tuner=1,0,0,0,538.00,8,dvbt,8k,64qam,14,23,0,0,0;pids=0
a=inactive
[... 57 lines cut ...]
CSeq: 6
User-Agent: LibVLC/2.1.3 (LIVE555 Streaming Media v2014.01.21)
Session: 1702957685
--- LINE END ---
[      Streams.cpp:225] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 1702957685
[ StreamClient.cpp:043] Connection can close: 1
[       Stream.cpp:123] Stream: 0, Teardown StreamClient[0] with SessionID 1702957685
[    RtpThread.cpp:102] Stream: 0, Stop  RTP stream to 192.168.1.10:57616 (Streamed 0.000 MBytes)
[   RtcpThread.cpp:083] Stream: 0, Stop RTCP stream to 192.168.1.10:57617
[     Frontend.cpp:505] Stream: 0, Remove filter PID: 0000 - fd: 011 - Packet Count: 0
[   RtspServer.cpp:284] RTSP/1.0 200 OK
CSeq: 6
Session: 1702957685
--- LINE END ---
[    TcpSocket.cpp:085] RTSP Client 192.168.1.10: Connection closed with fd: 8
[      Streams.cpp:205] Found SessionID 1702957685 by SocketClient
[      Streams.cpp:225] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 1702957685
[       Stream.cpp:115] Stream: 0, Close StreamClient[0] with SessionID 1702957685
```

With the minisatip the result is OK.

Are you sure that the tuning request is using commands compatible with API 5.4 ?

Barracuda09, 2015-03-12:
Hi lars18th,

Yes I would think so that I use the correct commands...

lars18th, 2015-03-12:
Hi Marc,

> Yes I would think so that I use the correct commands...

Then, please, add more prints of debug info around the tuning process. I'll check it and perhaps we can found where is the problem.

:(

Barracuda09, 2015-03-12:
Hi lars18th,

Yes.. But what should that be I have added the debug desc string. That shows the parameters are correctly parsed.
The only thing maybe is worth trying is adding something like this in Frontend.cpp line 312:

// get all the pending events....
for (;;) {
  struct dvb_frontend_event dfe; 
  if (ioctl (fd_fe, FE_GET_EVENT, &dfe) == -1) {
     break;
  }
}

but i'm not sure it will make any difference at all....

Marc

lars18th, 2015-03-13:
Hi Marc,

I found two errors in the tuning process with the current version:

Frontend.cpp#279:
- The value of "channel.freq" is at input in range, so you don't need to multiply it!
- If I change this line "to FILL_PROP(DTV_FREQUENCY,         channel.freq);" then my tuner gets signal! :)
- I found this adding this line: "SI_LOG_DEBUG("In Fronted::tune: SYS_DVBT, freq=%ul",channel.freq);"
- Perhaps line #298 also needs to be changed (I don't know, I can't test DVB-C)

Frontend.cpp#451:
- The time you sleep for first tuning is now 8*25ms... I feel you like to wait until 2sec., then the correct value is usleep(250000); so you need to add one more '0' :)

However, with these changes I can't receive the TS. The tuner adquires signal, but the client (for example VLC using RTSP call) never receives data. Can you please add more debug info for checking the data readed from the device? I like to confirm if the program reads data or not.

I hope that you can finally fix this! ;)

Barracuda09, 2015-03-13:
Hi lars18th,

That is a bit strange that you don't need the *1000 because I was under the impression that it should
be in Hz except DVB-S(2) which in in KHz.
My DVB-T tuner is working with this..
You should according the SAT>IP specs give the freq in MHz the first translation to KHz is done when interpreting the transport params. Then the last translation is done when tuning..
DVB-C is also tuning correct.
So I have to check this all.

lars18th, 2015-03-13:
Hi Marc,

Yes, it's annoynig. For this reason I put this line in the code"SI_LOG_DEBUG("In Fronted::tune: SYS_DVBT, freq=%ul",channel.freq);" and the result is that the value is in range. I check it with this request using VLC: "rtsp://192.168.1.20:554/?freq=538&msys=dvbt&pids=all" I suspect that you do the translation in another point of the code. Moreover, remember that the SAT>IP specification accepts "freq" in different formats (538.000 is also valid).

I suggest to put more LOG info in the code. At times is very useful for view where is the problem!
And, please, think that I always testing before with minisatip, so if it's working with my tuner, then it could also work with satpi. :)

Barracuda09, 2015-03-13:
I don't understand the problem just jet.. My DVB-T stick is working.. But what is the range then?

lars18th, 2015-03-16:
Hi,

> I don't understand the problem just jet.. My DVB-T stick is working.. But what is the range then?

Using last version from two days ago:

VLC command: rtsp://192.168.210.17:554/?freq=538&msys=dvbt&pids=all

Then the debug is:

```
[      Streams.cpp:225] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 1702957685
[ StreamClient.cpp:043] Connection can close: 0
[     Frontend.cpp:401] Stream: 0, Start tuning process...
[     Frontend.cpp:451] Stream: 0, Waiting on lock.
[     Frontend.cpp:465] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:465] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:465] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:465] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:465] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:465] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:465] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:465] Stream: 0, Not locked yet (FE status 0)...
[   RtspServer.cpp:168] RTSP/1.0 200 OK
```

If I change Frontend.cpp as line #288:
```
			FILL_PROP(DTV_FREQUENCY,         channel.freq ); // * 1000UL);
			SI_LOG_DEBUG("In Fronted::tune: SYS_DVBT, freq=%lu",channel.freq);
```

Then log is:
```
[      Streams.cpp:205] Found SessionID 1702957685 by SocketClient
[      Streams.cpp:225] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 1702957685
[ StreamClient.cpp:043] Connection can close: 0
[     Frontend.cpp:440] Stream: 0, Opened FE fd: 9.
[     Frontend.cpp:401] Stream: 0, Start tuning process...
[     Frontend.cpp:287] In Fronted::tune: SYS_DVBT, freq=538000
[     Frontend.cpp:331] FE_SET_PROPERTY failed: Invalid argument (code 22)
[     Frontend.cpp:451] Stream: 0, Waiting on lock.
[     Frontend.cpp:462] Stream: 0, Tuned and locked (FE status 31).
[     Frontend.cpp:483] Stream: 0, Opened DVR fd: 10.
[     Frontend.cpp:490] Stream: 0, Updating PID filters...
[     Frontend.cpp:507] Stream: 0, Set filter PID: 8192 - fd: 011
[    RtpThread.cpp:092] Stream: 0, Start  RTP stream to 192.168.1.20:63720
[   RtcpThread.cpp:073] Stream: 0, Start RTCP stream to 192.168.1.20:63721
[   RtspServer.cpp:142] RTSP/1.0 200 OK
```

Please, let me know if you need more info.

lars18th, 2015-03-16:
Hi,

> My DVB-T stick is working..

Please, can you write the log with your DVB-T stick in debug mode?

And, your DVB-T stich is working with "minisatip"?

I'm really confused with the tuning problem... In minisatip the code converts also to Hz, because the DVB-API specifies to use Hz for DVB-T and DVB-C, instead DVB-S that uses KHz. So, yes, your code is right... but I don't know if my drivers are wrong! In which system you test your program? In my case is a x86-64 PC,

Barracuda09, 2015-03-16:
Hi,

You don't need to change Frontend.cpp.. I think you can try change in ChannelData.cpp line 67 to:
bandwidth = 8000000;

lars18th, 2015-03-16:
Hi,

You're right! If I remove the "* 1000UL" multiply, then the dmesg shows:
```
[252201.632642] DVB: adapter 0 frontend 0 frequency 538000 out of range (174000000..860000000)
```
So your code is right. Also, with the "correct" frequency, with the "Not locked yet (FE status 0)" message, the dmesg command shows:
```
[252272.987144] af9013: invalid bw or clock
[252275.208533] af9013: invalid bw or clock
[252288.302009] af9013: invalid bw or clock
[252290.803181] af9013: invalid bw or clock
```
But the bandwith is correct (8MHz) as shows the log:
```
[   RtcpThread.cpp:159] ver=1.1;tuner=1,123,0,0,538.00,8,dvbt, , , , ,0,0,0;pids=all
```

I'll do more tests!

lars18th, 2015-03-16:
Hi,

> I think you can try change in ChannelData.cpp line 67 to: bandwidth = 8000000;

YES! YES! :)
With this I can tune!!! Finally!!!

However, the logs shows:
```
[   RtcpThread.cpp:159] ver=1.1;tuner=1,126,1,0,538.00,UNKNOWN BANDWIDTH,dvbt, , , , ,0,0,0;pids=all
```

I don't know the reason, the bandwith that I need to use is 8MHz.

Barracuda09, 2015-03-16:
Hi lars18th,

Ok I know what I have todo... Many thank you persisting in this issue..
I will fix this..

Marc

lars18th, 2015-03-16:
Thank you! You're fine! :)

Barracuda09, 2015-03-19:
Hi lars18th,

I have committed some changes regarding the DVB-T

lars18th, 2015-03-24:
Hi,

Only to confirm: with version of today (2015-03-24), it works with my DVB-T USB 2.0 dual tuner and API 5.4. Test with these clients:

DD-TV
TransEdit 4 from DVBViewer
VLC with call "rtsp://"

Great work!

Barracuda09, 2015-03-25:
Will close this issue then. Many thanks.

closed 2015-03-25 by Barracuda09