Barracuda09/SATPI #29: Web interface not working if not started from the same folder

FuNK3Y, 2015-03-09:
I noticed something strange.

If I start satpi from its folder like ./satpi the webinterface works fine.

But, if on the other hand I am located in another folder and start it with its full path like /root/SATPI/satpi the web interface fail to start.

lars18th, 2015-03-10:
Hi,

Yes, satip uses as the working dir the current directory (not only for web interface, but for all files). I don't know if this is by design or a bug.

Barracuda09, 2015-03-14:
Hi,

It is in deed per design that it is like this... I have to see if it needs to change.

FuNK3Y, 2015-03-14:
From my perspective, this is not a problem at all; it is an understandable design choice.

But, if you want things this way, you should end the application properly if it is not started from its directory. It is misleading to have it working partially.

Barracuda09, 2015-03-14:
Yes you are correct. I have to handle it more clear for the user.

Barracuda09, 2015-03-21:
Hi FuNK3Y,

I have done some changes regarding this issue. It should not matter which/were you start SatPI

Marc

Barracuda09, 2015-03-29:
Should be fixed now

closed 2015-03-29 by Barracuda09