Barracuda09/SATPI #30: msys=dvbs troubles (API 5.4)

lars18th, 2015-03-16:
Hi,

I do some tests using vtuner for SAT>IP (project http://code.google.com/p/satip/). This isn't compatible at time with SATPI, but I found a trouble with DVB-S transponders.

After start SATPI with only one DVB-S2 vtuner device, then any request using "msys=dvbs" fails... indicating that no tuner is available. But if I try to tune using "msys=dvbs2" then the code accepts the request.

Please, can you do more tests with DVB-API 5.4 and DVB-S/S2 tuners?
My current system is Debian 7.7 x86-64, that uses API 5.4

Barracuda09, 2015-03-19:
Hi lars18th,

I have on my Raspberry PI API 5.4 with 2 DVB-S/S2 tuners which work as expected.
What is the log showing when SatPI is started?

Marc

lars18th, 2015-03-24:
Hi Barracuda,

Your DVB-S/S2 tuners are INTERNAL (PCI or USB)?

I'm speaking about use a physical DVB-S2 SAT>IP, with "vtuner" DVB driver installed, and execute in the same machine the SATPI server.

You can check (if you don't have any real SAT>IP server) with this: in some Linux PC, install the vtuner, configured for use your SATPI server running in your Pi. Then try to execute the SATPI in the Linux PC.

That's all!

lars18th, 2015-03-25:
Hi Barracude,

Please, can you explain more about that?
Regards!

Barracuda09, 2015-03-25:
Hi,

- I have 2 DVB-S2 usb sticks connected to my PI.

- The SatPI log when you use vtuner.

Barracuda09, 2015-09-08:
I will close this for now... If anybody still has issues regarding this bug please make a new bug report.

closed 2015-09-08 by Barracuda09