Barracuda09/SATPI #32: No Lock on DVB-S2 channels

pbriesch, 2015-03-21:
Hi Marc,

first of all, the new Webif is nice and I'm glad to see that SATPI moves on!
I just tried out EyeTV on my Mac and found out that I cannot get a lock on DVB-S2 channels, just on DVB-S. 

Thanks for looking into this.

Paul

````
pbriesch@ubuntu32:~/satpi$ sudo ./satpi --no-daemon
[        Satpi.cpp:247] --- starting SatPI version: 1.5.50~gbddad2b ---
[  InterfaceAttr.h:094] eth0: 192.168.1.169 [00:19:66:0f:8d:07]
[      Streams.cpp:115] Current DVB_API_VERSION: 5.10
[      Streams.cpp:117] Detecting frontends in: /dev/dvb
[      Streams.cpp:135] Frontends found: 2
[     Frontend.cpp:144] Frontend Name: STB0899 Multistandard
[     Frontend.cpp:218] Frontend Type: Satellite (DVB-S)
[     Frontend.cpp:222] Frontend Type: Satellite (DVB-S2)
[     Frontend.cpp:214] Frontend Type: DSS
[     Frontend.cpp:257] Frontend Freq: 950000 Hz to 2150000 Hz
[     Frontend.cpp:258] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
[     Frontend.cpp:144] Frontend Name: STB0899 Multistandard
[     Frontend.cpp:218] Frontend Type: Satellite (DVB-S)
[     Frontend.cpp:222] Frontend Type: Satellite (DVB-S2)
[     Frontend.cpp:214] Frontend Type: DSS
[     Frontend.cpp:257] Frontend Freq: 950000 Hz to 2150000 Hz
[     Frontend.cpp:258] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
[   SsdpServer.cpp:108] Setting up SSDP server with BOOTID: 72
[   HttpServer.cpp:075] Setting up HTTP server
[   RtspServer.cpp:059] Setting up RTSP server
[    TcpSocket.cpp:149] HTTP Connection from 192.168.1.107 Port 60667 (fd: 8)
[    TcpSocket.cpp:149] HTTP Connection from 192.168.1.107 Port 60668 (fd: 9)
[    TcpSocket.cpp:149] HTTP Connection from 192.168.1.107 Port 52558 (fd: 10)
[   HttpServer.cpp:178] Client: 192.168.1.107 requesed /desc.xml
[    TcpSocket.cpp:149] RTSP Connection from 192.168.1.107 Port 60669 (fd: 11)
[   RtspServer.cpp:072] Data from client 192.168.1.107: DESCRIBE rtsp://192.168.1.169/ RTSP/1.0
Connection:Close
CSeq:1001
Accept:application/sdp
--- LINE END ---
[      Streams.cpp:232] Found message outside session
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:258] RTSP/1.0 404 Not Found
CSeq: 1001
Content-Type: application/sdp
Content-Base: rtsp://192.168.1.169/
Content-Length: 0
--- LINE END ---
[    TcpSocket.cpp:085] RTSP Client 192.168.1.107: Connection closed with fd: 11
[      Streams.cpp:232] Found message outside session
[       Stream.cpp:127] Stream: -1, Close StreamClient[0] with SessionID -1
[    TcpSocket.cpp:149] RTSP Connection from 192.168.1.107 Port 60671 (fd: 11)
[   RtspServer.cpp:072] Data from client 192.168.1.107: SETUP rtsp://192.168.1.169:554/?src=1&freq=11361&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=off&sr=22000&fec=23 RTSP/1.0
Connection:Close
CSeq:1001
Transport:RTP/AVP;unicast;client_port=5002-5003
--- LINE END ---
[      Streams.cpp:250] Found StreamID x - SessionID x - Creating new SessionID: 1702957685
[       Stream.cpp:064] Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbs2
[       Stream.cpp:207] Stream: 0, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 0
[     Frontend.cpp:451] Stream: 0, Opened FE fd: 0.
[     Frontend.cpp:400] Stream: 0, Start tuning process...
[     Frontend.cpp:392] Stream: 0, Sending DiSEqC [e0] [10] [38] [f2]
[     Frontend.cpp:462] Stream: 0, Waiting on lock...
[     Frontend.cpp:476] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:476] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:476] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:476] Stream: 0, Not locked yet (FE status 26)...
[... 858 lines cut ...]
[      Streams.cpp:232] Found message outside session
[       Stream.cpp:127] Stream: -1, Close StreamClient[0] with SessionID -1
[    TcpSocket.cpp:149] RTSP Connection from 192.168.1.107 Port 60705 (fd: 11)
[   RtspServer.cpp:072] Data from client 192.168.1.107: DESCRIBE rtsp://192.168.1.169/ RTSP/1.0
Connection:Close
CSeq:1009
Accept:application/sdp
--- LINE END ---
[      Streams.cpp:232] Found message outside session
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:258] RTSP/1.0 200 OK
CSeq: 1009
Content-Type: application/sdp
Content-Base: rtsp://192.168.1.169/
Content-Length: 274
--- LINE END ---
v=0
o=- 2 3 IN IP4 192.168.1.169
s=SatIPServer:1 2,0,0
t=0 0
m=video 0 RTP/AVP 33
c=IN IP4 0.0.0.0
a=control:stream=0
a=fmtp:33 ver=1.0;src=1;tuner=1,1,1,0,11953.00,h,dvbs,qpsk,off,0.35,27500,34;pids=0,16,17,18,19,20,650,660,670,671,672,675,680,950,8190
a=sendonly

Barracuda09, 2015-03-21:
Hi Paul,

The new WebUI is from contributor 'issue'. Is this a new problem from last commits?

pbriesch, 2015-03-21:
>Is this a new problem from last commits?
I think so. I need to perform some more testing tomorrow.

Barracuda09, 2015-03-21:
I can tune to the first SETUP message string in VLC (Astra 19,2e)

pbriesch, 2015-03-21:
Could you please paste the URL for VLC for me?

Barracuda09, 2015-03-21:
There is no PID but it locks here it is
rtsp://192.168.0.109:554/?src=1&freq=11361&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=off&sr=22000&fec=23

pbriesch, 2015-03-21:
I downloaded the latest version of VLC (3.0.0-git Vetinari (Intel 64bit)), changed the IP accordingly but cannot get a stream with VLC. Current DVB_API_VERSION: 5.10

`````
pbriesch@ubuntu32:~/satpi$ sudo ./satpi --no-daemon
[        Satpi.cpp:247] --- starting SatPI version: 1.5.50~gbddad2b ---
[  InterfaceAttr.h:094] eth0: 192.168.1.169 [00:19:66:0f:8d:07]
[      Streams.cpp:115] Current DVB_API_VERSION: 5.10
[      Streams.cpp:117] Detecting frontends in: /dev/dvb
[      Streams.cpp:135] Frontends found: 2
[     Frontend.cpp:144] Frontend Name: STB0899 Multistandard
[     Frontend.cpp:218] Frontend Type: Satellite (DVB-S)
[     Frontend.cpp:222] Frontend Type: Satellite (DVB-S2)
[     Frontend.cpp:214] Frontend Type: DSS
[     Frontend.cpp:257] Frontend Freq: 950000 Hz to 2150000 Hz
[     Frontend.cpp:258] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
[     Frontend.cpp:144] Frontend Name: STB0899 Multistandard
[     Frontend.cpp:218] Frontend Type: Satellite (DVB-S)
[     Frontend.cpp:222] Frontend Type: Satellite (DVB-S2)
[     Frontend.cpp:214] Frontend Type: DSS
[     Frontend.cpp:257] Frontend Freq: 950000 Hz to 2150000 Hz
[     Frontend.cpp:258] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
[   SsdpServer.cpp:108] Setting up SSDP server with BOOTID: 80
[   HttpServer.cpp:075] Setting up HTTP server
[   RtspServer.cpp:059] Setting up RTSP server
[    TcpSocket.cpp:149] HTTP Connection from 192.168.1.107 Port 63899 (fd: 8)
[    TcpSocket.cpp:149] HTTP Connection from 192.168.1.107 Port 63900 (fd: 9)
[    TcpSocket.cpp:149] RTSP Connection from 192.168.1.107 Port 63901 (fd: 10)
[   RtspServer.cpp:072] Data from client 192.168.1.107: OPTIONS rtsp://192.168.1.169:554/?src=1&freq=11361&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=off&sr=22000&fec=23 RTSP/1.0
CSeq: 2
User-Agent: LibVLC/3.0.0-git (LIVE555 Streaming Media v2014.07.25)
--- LINE END ---
[      Streams.cpp:250] Found StreamID x - SessionID x - Creating new SessionID: 1702957685
[       Stream.cpp:064] Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbs2
[       Stream.cpp:207] Stream: 0, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:193] RTSP/1.0 200 OK
CSeq: 2
Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
Session: 1702957685
--- LINE END ---
[   RtspServer.cpp:072] Data from client 192.168.1.107: DESCRIBE rtsp://192.168.1.169:554/?src=1&freq=11361&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=off&sr=22000&fec=23 RTSP/1.0
CSeq: 3
User-Agent: LibVLC/3.0.0-git (LIVE555 Streaming Media v2014.07.25)
Accept: application/sdp
--- LINE END ---
[      Streams.cpp:222] Found SessionID 1702957685 by SocketClient
[      Streams.cpp:242] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:064] Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbs2
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:258] RTSP/1.0 200 OK
CSeq: 3
Content-Type: application/sdp
Content-Base: rtsp://192.168.1.169/
Content-Length: 222
Session: 1702957685
--- LINE END ---
v=0
o=- 2 3 IN IP4 192.168.1.169
s=SatIPServer:1 2,0,0
t=0 0
m=video 0 RTP/AVP 33
c=IN IP4 0.0.0.0
a=control:stream=0
a=fmtp:33 ver=1.0;src=1;tuner=1,0,0,0,11361.00,h,dvbs2,8psk,off,0.35,22000,23;pids=
a=inactive
[   RtspServer.cpp:072] Data from client 192.168.1.107: SETUP rtsp://192.168.1.169/stream=0 RTSP/1.0
CSeq: 4
User-Agent: LibVLC/3.0.0-git (LIVE555 Streaming Media v2014.07.25)
Transport: RTP/AVP;unicast;client_port=60218-60219
--- LINE END ---
[... 25 lines cut ...]
[   RtspServer.cpp:072] Data from client 192.168.1.107: PLAY rtsp://192.168.1.169/ RTSP/1.0
CSeq: 5
User-Agent: LibVLC/3.0.0-git (LIVE555 Streaming Media v2014.07.25)
Session: 1702957685
Range: npt=0.000-
--- LINE END ---
[      Streams.cpp:242] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 1702957685
[ StreamClient.cpp:043] Connection can close: 0
[     Frontend.cpp:400] Stream: 0, Start tuning process...
[     Frontend.cpp:392] Stream: 0, Sending DiSEqC [e0] [10] [38] [f2]
[     Frontend.cpp:462] Stream: 0, Waiting on lock...
[     Frontend.cpp:476] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:476] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:476] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:476] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:476] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:476] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:476] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:476] Stream: 0, Not locked yet (FE status 26)...
[   RtspServer.cpp:165] RTSP/1.0 200 OK
RTP-Info: url=rtsp://192.168.1.169/stream=0
CSeq: 5
Session: 1702957685
--- LINE END ---

pbriesch, 2015-03-24:
Hi Marc,

with VLC it is the same problem. It used to work before.

Regards, Paul

Barracuda09, 2015-03-24:
Hi paul,

Could you (if you like and have time) try to find out were it still worked?

Thanks,

Marc

Barracuda09, 2015-03-24:
Hi paul,

You can do this with these commands:

Show a log of commits use:
  git log

outputs something like:
commit d1d1d0c3169c952f359a52227e7d96659584271a
Author: Billy Everyteen <Somebody@Somewere.com>
Date:   Wed Mar 11 15:33:45 2015 +0100

    Initialize ChannelData with each new freq

Then check it out with:
  git checkout d1d1d0c3169c952f359a52227e7d96659584271a

pbriesch, 2015-03-24:
Thanks for this information :-)

pbriesch, 2015-03-24:
Hi Marc,

I think I nailed it down. VLC was tested on a DVB-S2 channel:

````
commit dcf9e7bdc11609cb5b954075936ec2d52fa4ba5c
Author: Marc <m.a.postema@alice.nl>
Date:   Mon Mar 23 21:32:02 2015 +0100

    Fix weird uninitialized File Descriptor bug

commit bddad2b778dfc44b21883005476c99e0b075ef2d
Merge: 9b86db5 2a2fa7e
Author: Marc <m.a.postema@alice.nl>
Date:   Sat Mar 21 22:21:23 2015 +0100

    Merge pull request #31 from issue/master

    Nice WebUI from issue

commit 2a2fa7e8b229f38d18554140e08de2e38439e4a9
Author: Baris <baris@rbk-esports.eu>
Date:   Sat Mar 21 22:09:09 2015 +0100

    add missing javascript files

commit 21e6bb8d532473cb24dd1193e1d35653db669c89  EyeTV DVB-S2 does not work, VLC has no lock
Merge: ac257c0 9b86db5
Author: Baris <baris@rbk-esports.eu>
Date:   Sat Mar 21 20:45:12 2015 +0100

    Merge remote-tracking branch 'upstream/master'

commit 9b86db57ec2693b1b1ad30e0df1659d0f6b5984d  EyeTV DVB-S2 does not work, VLC has no lock
Author: Marc <m.a.postema@alice.nl>
Date:   Sat Mar 21 20:23:57 2015 +0100

    Some small and large fixes

commit ac257c05f0945db9df86f6d43b3844ba1be721af  EyeTV DVB-S2 works, VLC has no lock
Merge: 1d514e1 3a32541
Author: Baris <baris@rbk-esports.eu>
Date:   Sat Mar 21 14:47:35 2015 +0100

    Merge remote-tracking branch 'upstream/master'

commit 3a325411cb5990277aa0aac9cf69747db468eb7b
Author: Marc <m.a.postema@alice.nl>
Date:   Sat Mar 21 12:19:10 2015 +0100

    Add framework for config page

commit 1d514e1e03131bf72080e08dc0b690222a1202e9
Author: Baris <baris@rbk-esports.eu>
Date:   Sat Mar 21 10:59:48 2015 +0100

    "fix" STOP

commit 68f7a72972cdb11c7cd407ff42563fc480b5dcbc
Author: Baris <baris@rbk-esports.eu>
Date:   Sat Mar 21 09:30:15 2015 +0100

    add autoscroll to log

commit b56baa314b418c36f81e22605d36903384e8eaf6
Merge: 5cbee33 3d4fb83
Author: Baris <baris@rbk-esports.eu>
Date:   Sat Mar 21 09:04:38 2015 +0100

    Merge remote-tracking branch 'upstream/master'

commit 3d4fb83b935540e35e41bc4187a18affc1e37fff  EyeTV DVB-S2 works, VLC has no lock
Author: Marc <m.a.postema@alice.nl>
Date:   Thu Mar 19 22:00:14 2015 +0100

    Update README.md

commit 5cbee339a4f9d228e407df76754dacdb1acbfedd  EyeTV DVB-S2 works, VLC has no lock
Merge: d93d5b0 bd9dab9
Author: Baris <baris@rbk-esports.eu>
Date:   Wed Mar 18 20:21:09 2015 +0100

    Merge remote-tracking branch 'upstream/master'

commit bd9dab9f0dfc81aa8ce01ce6ebb5bd8d501674b5
Author: Billy Everyteen <m.a.postema@alice.nl>
Date:   Tue Mar 17 12:58:57 2015 +0100

    DVB-T Fixes and Flush buffers on channel change

commit d93d5b0f65d289181377ed042a01399b8033ce9a  EyeTV DVB-S2 works, VLC works
Merge: 04d996a 764ef59
Author: Baris <baris@rbk-esports.eu>
Date:   Sat Mar 14 15:41:32 2015 +0100

    Merge remote-tracking branch 'upstream/master'

commit 764ef595ce98d4a75c7afc0d2f17f2e9bb423503  EyeTV DVB-S2 works, VLC works
Author: Marc <m.a.postema@alice.nl>
Date:   Sat Mar 14 12:51:46 2015 +0100

    Update README.md

commit 14fdafa29bb0c960d629fcff46fb50a6a4ad3eef
Author: Billy Everyteen <m.a.postema@alice.nl>
Date:   Fri Mar 13 17:00:08 2015 +0100

    Add pids=all and DTV_STREAM_ID and clear pending events and some changes

commit 04d996a9a9ba4b00144a37713fb6f4ec79131611
Merge: e5ebd7d 135e431
Author: Baris <baris@rbk-esports.eu>
Date:   Thu Mar 12 12:05:31 2015 +0100

    Merge remote-tracking branch 'upstream/master'

commit 135e4310c44d09d1ea15c182965c076694bafd3d  EyeTV DVB-S2 works, VLC works
Author: Marc <m.a.postema@alice.nl>
Date:   Wed Mar 11 23:06:49 2015 +0100

    Fix thread locking issue

Barracuda09, 2015-03-24:
Hi Paul,

Many Many thanks. I will try to see what has changed.

Barracuda09, 2015-03-24:
Hi Paul,

I did revert some change that were not "needed".

pbriesch, 2015-03-24:
Marc,

EyeTV on DVB-S2 channels still doesn't work. VLC neither. See attached log which shows switching from a SD to a HD channel with EyeTV:
````
m=video 0 RTP/AVP 33
c=IN IP4 0.0.0.0
a=control:stream=1
a=fmtp:33 ver=1.0;src=1;tuner=2,2,1,0,11493.00,h,dvbs2,8psk, ,0.35,22000,23;pids=
a=sendonly
[    TcpSocket.cpp:085] HTTP Client 192.168.1.131: Connection closed with fd: 14
[    TcpSocket.cpp:085] RTSP Client 192.168.1.107: Connection closed with fd: 18
[      Streams.cpp:232] Found message outside session
[       Stream.cpp:127] Stream: -1, Close StreamClient[0] with SessionID -1
[    TcpSocket.cpp:149] RTSP Connection from 192.168.1.107 Port 49168 (fd: 14)
[   RtspServer.cpp:072] Data from client 192.168.1.107: DESCRIBE rtsp://192.168.1.169/ RTSP/1.0
Connection:Close
CSeq:1005
Accept:application/sdp
--- LINE END ---
[      Streams.cpp:232] Found message outside session
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:258] RTSP/1.0 200 OK
CSeq: 1005
Content-Type: application/sdp
Content-Base: rtsp://192.168.1.169/
Content-Length: 375
--- LINE END ---
v=0
o=- 2 3 IN IP4 192.168.1.169
s=SatIPServer:1 2,0,0
t=0 0
m=video 0 RTP/AVP 33
c=IN IP4 0.0.0.0
a=control:stream=0
a=fmtp:33 ver=1.0;src=1;tuner=1,1,1,0,11493.00,h,dvbs2,8psk, ,0.35,22000,23;pids=
a=sendonly
m=video 0 RTP/AVP 33
c=IN IP4 0.0.0.0
a=control:stream=1
a=fmtp:33 ver=1.0;src=1;tuner=2,2,1,0,11493.00,h,dvbs2,8psk, ,0.35,22000,23;pids=
a=sendonly
[    TcpSocket.cpp:085] RTSP Client 192.168.1.107: Connection closed with fd: 13
[      Streams.cpp:232] Found message outside session
[       Stream.cpp:127] Stream: -1, Close StreamClient[0] with SessionID -1
[    TcpSocket.cpp:149] RTSP Connection from 192.168.1.107 Port 49169 (fd: 13)
[   RtspServer.cpp:072] Data from client 192.168.1.107: SETUP rtsp://192.168.1.169:554/?src=1&freq=11836&pol=h&ro=0.35&msys=dvbs&mtype=qpsk&plts=off&sr=27500&fec=34 RTSP/1.0
Connection:Close
CSeq:1008
Transport:RTP/AVP;unicast;client_port=5000-5001
--- LINE END ---
[      Streams.cpp:250] Found StreamID x - SessionID x - Creating new SessionID: 0787658847
[      Streams.cpp:278] Found no Stream/Client of interest!
[    TcpSocket.cpp:085] RTSP Client 192.168.1.107: Connection closed with fd: 13
[      Streams.cpp:232] Found message outside session
[       Stream.cpp:127] Stream: -1, Close StreamClient[0] with SessionID -1
[    TcpSocket.cpp:149] RTSP Connection from 192.168.1.107 Port 49170 (fd: 13)
[   RtspServer.cpp:072] Data from client 192.168.1.107: DESCRIBE rtsp://192.168.1.169/ RTSP/1.0
Connection:Close
CSeq:1009
Accept:application/sdp
--- LINE END ---
[      Streams.cpp:232] Found message outside session
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:258] RTSP/1.0 200 OK
CSeq: 1009
Content-Type: application/sdp
Content-Base: rtsp://192.168.1.169/
Content-Length: 375
--- LINE END ---
v=0
[... 959 lines cut ...]
[     Frontend.cpp:530] Stream: 0, Remove filter PID: 6132 - fd: 024 - Packet Count: 0
[     Frontend.cpp:530] Stream: 0, Remove filter PID: 6170 - fd: 025 - Packet Count: 0
[    RtpThread.cpp:117] Stream: 0, Restart RTP stream to 192.168.1.107:5000
[   RtspServer.cpp:141] RTSP/1.0 200 OK
CSeq: 1030
Session: 2092596303;timeout=60
Transport: RTP/AVP;unicast;client_port=5000-5001
com.ses.streamID: 0
--- LINE END ---
[    TcpSocket.cpp:085] RTSP Client 192.168.1.107: Connection closed with fd: 8
[      Streams.cpp:222] Found SessionID 2092596303 by SocketClient
[      Streams.cpp:242] Found StreamID x - SessionID: 2092596303
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 2092596303
[    TcpSocket.cpp:149] RTSP Connection from 192.168.1.107 Port 49202 (fd: 8)
[   RtspServer.cpp:072] Data from client 192.168.1.107: PLAY rtsp://192.168.1.169:554/stream=0?pids=0,16,17,18,19,20,6100,6110,6120,6121,6122,6123,6130,6131,6132,6170,8190 RTSP/1.0
Connection:Close
CSeq:1031
Session:2092596303
--- LINE END ---
[      Streams.cpp:242] Found StreamID x - SessionID: 2092596303
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 2092596303
[       Stream.cpp:207] Stream: 0, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 0
[     Frontend.cpp:401] Stream: 0, Start tuning process...
[     Frontend.cpp:393] Stream: 0, Sending DiSEqC [e0] [10] [38] [f2]

Barracuda09, 2015-03-25:
Hi Paul,

Could you please try something for me? Try this in VLC and see if it works:

rtsp://192.168.0.109:554/?src=1&freq=11361&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=6100,6110,6122,6130

pbriesch, 2015-03-25:
Hi Marc,

it does not work:
````
pbriesch@ubuntu32:~/satpi$ sudo ./satpi --no-daemon
[sudo] password for pbriesch:
[        Satpi.cpp:247] --- starting SatPI version: 1.5.52~g07051ef ---
[  InterfaceAttr.h:094] eth0: 192.168.1.169 [00:19:66:0f:8d:07]
[      Streams.cpp:115] Current DVB_API_VERSION: 5.10
[      Streams.cpp:117] Detecting frontends in: /dev/dvb
[      Streams.cpp:135] Frontends found: 2
[     Frontend.cpp:145] Frontend Name: STB0899 Multistandard
[     Frontend.cpp:219] Frontend Type: Satellite (DVB-S)
[     Frontend.cpp:223] Frontend Type: Satellite (DVB-S2)
[     Frontend.cpp:215] Frontend Type: DSS
[     Frontend.cpp:258] Frontend Freq: 950000 Hz to 2150000 Hz
[     Frontend.cpp:259] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
[     Frontend.cpp:145] Frontend Name: STB0899 Multistandard
[     Frontend.cpp:219] Frontend Type: Satellite (DVB-S)
[     Frontend.cpp:223] Frontend Type: Satellite (DVB-S2)
[     Frontend.cpp:215] Frontend Type: DSS
[     Frontend.cpp:258] Frontend Freq: 950000 Hz to 2150000 Hz
[     Frontend.cpp:259] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
[   HttpServer.cpp:075] Setting up HTTP server
[   RtspServer.cpp:059] Setting up RTSP server
[   SsdpServer.cpp:108] Setting up SSDP server with BOOTID: 40
[    TcpSocket.cpp:149] HTTP Connection from 192.168.1.114 Port 49391 (fd: 8)
[   HttpServer.cpp:178] Client: 192.168.1.114 requesed /desc.xml
[    TcpSocket.cpp:085] HTTP Client 192.168.1.114: Connection closed with fd: 8
[    TcpSocket.cpp:149] RTSP Connection from 192.168.1.107 Port 49767 (fd: 8)
[   RtspServer.cpp:072] Data from client 192.168.1.107: OPTIONS rtsp://192.168.1.169:554/?src=1&freq=11361&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=6100,6110,6122,6130
rtsp://192.168.1.169:554/?src=1&freq=11361&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=6100,6110,6122,6130
 RTSP/1.0
CSeq: 2
User-Agent: LibVLC/3.0.0-git (LIVE555 Streaming Media v2014.07.25)
--- LINE END ---
[      Streams.cpp:250] Found StreamID x - SessionID x - Creating new SessionID: 1702957685
[       Stream.cpp:064] Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbs2
[       Stream.cpp:207] Stream: 0, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:193] RTSP/1.0 200 OK
CSeq: 2
Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
Session: 1702957685
--- LINE END ---
[   RtspServer.cpp:072] Data from client 192.168.1.107: DESCRIBE rtsp://192.168.1.169:554/?src=1&freq=11361&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=6100,6110,6122,6130
rtsp://192.168.1.169:554/?src=1&freq=11361&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=6100,6110,6122,6130
 RTSP/1.0
CSeq: 3
User-Agent: LibVLC/3.0.0-git (LIVE555 Streaming Media v2014.07.25)
Accept: application/sdp
--- LINE END ---
[      Streams.cpp:222] Found SessionID 1702957685 by SocketClient
[      Streams.cpp:242] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:064] Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbs2
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:258] RTSP/1.0 200 OK
CSeq: 3
Content-Type: application/sdp
Content-Base: rtsp://192.168.1.169/
Content-Length: 240
Session: 1702957685
--- LINE END ---
v=0
o=- 2 3 IN IP4 192.168.1.169
s=SatIPServer:1 2,0,0
t=0 0
m=video 0 RTP/AVP 33
c=IN IP4 0.0.0.0
a=control:stream=0
[... 292 lines cut ...]
Range: npt=0.000-
--- LINE END ---
[      Streams.cpp:242] Found StreamID x - SessionID: 0288582906
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 0288582906
[ StreamClient.cpp:043] Connection can close: 0
[     Frontend.cpp:401] Stream: 0, Start tuning process...
[     Frontend.cpp:393] Stream: 0, Sending DiSEqC [e0] [10] [38] [f2]
[     Frontend.cpp:463] Stream: 0, Waiting on lock...
[     Frontend.cpp:477] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:477] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:477] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:477] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:477] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:477] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:477] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:477] Stream: 0, Not locked yet (FE status 26)...
[   RtspServer.cpp:165] RTSP/1.0 200 OK
RTP-Info: url=rtsp://192.168.1.169/stream=0
CSeq: 5
Session: 0288582906
--- LINE END ---
[    TcpSocket.cpp:085] RTSP Client 192.168.1.107: Connection closed with fd: 8
[      Streams.cpp:222] Found SessionID 0288582906 by SocketClient
[      Streams.cpp:242] Found StreamID x - SessionID: 0288582906
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 0288582906

pbriesch, 2015-03-25:
Could there be a DiSEqC issue?

Barracuda09, 2015-03-25:
Hi,

Is there something strange in dmesg?

pbriesch, 2015-03-25:
Yes, there is!!
````
pbriesch@ubuntu32:~$ dmesg
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.16.0-33-generic (buildd@brownie) (gcc version 4.9.1 (Ubuntu 4.9.1-16ubuntu6) ) #44-Ubuntu SMP Thu Mar 12 12:25:57 UTC 2015 (Ubuntu 3.16.0-33.44-generic 3.16.7-ckt7)
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   NSC Geode by NSC
[    0.000000]   Cyrix CyrixInstead
[    0.000000]   Centaur CentaurHauls
[    0.000000]   Transmeta GenuineTMx86
[    0.000000]   Transmeta TransmetaCPU
[    0.000000]   UMC UMC UMC UMC
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003bfeffff] usable
[    0.000000] BIOS-e820: [mem 0x000000003bff0000-0x000000003bff7fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000003bff8000-0x000000003bffffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffee0000-0x00000000ffefffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] Notice: NX (Execute Disable) protection missing in CPU!
[    0.000000] Legacy DMI 2.3 present.
[    0.000000] DMI: American Megatrends Inc. K7S41GX/K7S41GX, BIOS P2.80 10/18/2006
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x3bff0 max_arch_pfn = 0x1000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-DFFFF write-protect
[    0.000000]   E0000-EFFFF uncachable
[    0.000000]   F0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 000000000 mask FE0000000 write-back
[    0.000000]   1 base 020000000 mask FF0000000 write-back
[    0.000000]   2 base 030000000 mask FF8000000 write-back
[    0.000000]   3 base 038000000 mask FFC000000 write-back
[    0.000000]   4 disabled
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[    0.000000]   7 base 0D0000000 mask FFC000000 write-combining
[    0.000000] x86 PAT enabled: cpu 0, old 0x7040600070406, new 0x7010600070106
[    0.000000] found SMP MP-table at [mem 0x000fbc70-0x000fbc7f] mapped at [c00fbc70]
[    0.000000] Scanning 1 areas for low memory corruption
[    0.000000] initial memory mapped: [mem 0x00000000-0x021fffff]
[    0.000000] Base memory trampoline at [c009b000] 9b000 size 16384
[    0.000000] init_memory_mapping: [mem 0x00000000-0x000fffff]
[    0.000000]  [mem 0x00000000-0x000fffff] page 4k
[    0.000000] init_memory_mapping: [mem 0x37400000-0x375fffff]
[    0.000000]  [mem 0x37400000-0x375fffff] page 2M
[    0.000000] init_memory_mapping: [mem 0x34000000-0x373fffff]
[    0.000000]  [mem 0x34000000-0x373fffff] page 2M
[    0.000000] init_memory_mapping: [mem 0x00100000-0x33ffffff]
[    0.000000]  [mem 0x00100000-0x001fffff] page 4k
[    0.000000]  [mem 0x00200000-0x33ffffff] page 2M
[    0.000000] init_memory_mapping: [mem 0x37600000-0x377fdfff]
[    0.000000]  [mem 0x37600000-0x377fdfff] page 4k
[    0.000000] BRK [0x01c08000, 0x01c08fff] PGTABLE
[    0.000000] BRK [0x01c09000, 0x01c09fff] PGTABLE
[    0.000000] RAMDISK: [mem 0x35826000-0x36c0afff]
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x000FAA40 000014 (v00 AMI   )
[... 614 lines cut ...]
[sudo] password for pbriesch:

Disk /dev/sda: 18,7 GiB, 20020396032 bytes, 39102336 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x442ffe1e

Device     Boot    Start      End  Sectors  Size Id Type
/dev/sda1  *        2048 37136383 37134336 17,7G 83 Linux
/dev/sda2       37138430 39100415  1961986  958M  5 Extended
/dev/sda5       37138432 39100415  1961984  958M 82 Linux swap / Solaris

Disk /dev/sdb: 37,3 GiB, 40020664320 bytes, 78165360 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x2b23dc36

Device     Boot Start      End  Sectors  Size Id Type
/dev/sdb1  *       63 78140159 78140097 37,3G  7 HPFS/NTFS/exFAT

pbriesch@ubuntu32:~$

pbriesch, 2015-03-25:
Marc,

I need a new hard drive. Just installed smartmontools.... Thanks for pointing this out.

Paul

Barracuda09, 2015-03-25:
Ok.. yes so it seems.. But it is not related to tuning problems I think.

Barracuda09, 2015-03-25:
I have looked back into logs you have added.. also for VDR (which did DVB-S2) there was no look also.
So maybe the tuning problem was there always?
But I still do not get why it does not work. You use just an universal LNB?

pbriesch, 2015-03-25:
I use a multiswitch and 2 LNBs. One for ASTRA 19.2E and one for Hotbird.

Barracuda09, 2015-03-25:
Hi,

could you try this (changed pol):
rtsp://192.168.1.169:554/?src=1&freq=11361&pol=v&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=6100,6110,6122,6130

pbriesch, 2015-03-25:
No change.
,,,,
pbriesch@ubuntu32:~/satpi$ sudo ./satpi --no-daemon 
[sudo] password for pbriesch: 
[        Satpi.cpp:247] --- starting SatPI version: 1.5.52~g07051ef ---
[  InterfaceAttr.h:094] eth0: 192.168.1.169 [00:19:66:0f:8d:07]
[      Streams.cpp:115] Current DVB_API_VERSION: 5.10
[      Streams.cpp:117] Detecting frontends in: /dev/dvb
[      Streams.cpp:135] Frontends found: 2
[     Frontend.cpp:145] Frontend Name: STB0899 Multistandard
[     Frontend.cpp:219] Frontend Type: Satellite (DVB-S)
[     Frontend.cpp:223] Frontend Type: Satellite (DVB-S2)
[     Frontend.cpp:215] Frontend Type: DSS
[     Frontend.cpp:258] Frontend Freq: 950000 Hz to 2150000 Hz
[     Frontend.cpp:259] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
[     Frontend.cpp:145] Frontend Name: STB0899 Multistandard
[     Frontend.cpp:219] Frontend Type: Satellite (DVB-S)
[     Frontend.cpp:223] Frontend Type: Satellite (DVB-S2)
[     Frontend.cpp:215] Frontend Type: DSS
[     Frontend.cpp:258] Frontend Freq: 950000 Hz to 2150000 Hz
[     Frontend.cpp:259] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
[   SsdpServer.cpp:108] Setting up SSDP server with BOOTID: 41
[   HttpServer.cpp:075] Setting up HTTP server
[   RtspServer.cpp:059] Setting up RTSP server
[    TcpSocket.cpp:149] HTTP Connection from 192.168.1.114 Port 65186 (fd: 8)
[   HttpServer.cpp:178] Client: 192.168.1.114 requesed /desc.xml
[    TcpSocket.cpp:149] HTTP Connection from 192.168.1.131 Port 52309 (fd: 9)
[   HttpServer.cpp:178] Client: 192.168.1.131 requesed /desc.xml
[    TcpSocket.cpp:149] RTSP Connection from 192.168.1.111 Port 57483 (fd: 10)
[   RtspServer.cpp:072] Data from client 192.168.1.111: OPTIONS rtsp://192.168.1.169:554/?src=1&freq=11361&pol=v&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=6100,6110,6122,6130 RTSP/1.0
CSeq: 2
User-Agent: LibVLC/2.1.5 (LIVE555 Streaming Media v2014.05.27)
--- LINE END ---
[      Streams.cpp:250] Found StreamID x - SessionID x - Creating new SessionID: 1702957685
[       Stream.cpp:064] Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbs2
[       Stream.cpp:207] Stream: 0, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:193] RTSP/1.0 200 OK
CSeq: 2
Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
Session: 1702957685
--- LINE END ---
[   RtspServer.cpp:072] Data from client 192.168.1.111: DESCRIBE rtsp://192.168.1.169:554/?src=1&freq=11361&pol=v&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=6100,6110,6122,6130 RTSP/1.0
CSeq: 3
User-Agent: LibVLC/2.1.5 (LIVE555 Streaming Media v2014.05.27)
Accept: application/sdp
--- LINE END ---
[      Streams.cpp:222] Found SessionID 1702957685 by SocketClient
[      Streams.cpp:242] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:064] Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbs2
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:258] RTSP/1.0 200 OK
CSeq: 3
Content-Type: application/sdp
Content-Base: rtsp://192.168.1.169/
Content-Length: 240
Session: 1702957685
--- LINE END ---
v=0
o=- 2 3 IN IP4 192.168.1.169
s=SatIPServer:1 2,0,0
t=0 0
m=video 0 RTP/AVP 33
c=IN IP4 0.0.0.0
a=control:stream=0
a=fmtp:33 ver=1.0;src=1;tuner=1,0,0,0,11361.00,v,dvbs2,8psk,on,0.35,22000,23;pids=6100,6110,6122,6130
a=inactive
[   RtspServer.cpp:072] Data from client 192.168.1.111: SETUP rtsp://192.168.1.169/stream=0 RTSP/1.0
CSeq: 4
User-Agent: LibVLC/2.1.5 (LIVE555 Streaming Media v2014.05.27)
[... 268 lines cut ...]
[     Frontend.cpp:526] Stream: 0, Set filter PID: 6130 - fd: 013
[    RtpThread.cpp:095] Stream: 0, Start   RTP stream to 192.168.1.111:64556
[   RtcpThread.cpp:074] Stream: 0, Start  RTCP stream to 192.168.1.111:64557
[   RtspServer.cpp:141] RTSP/1.0 200 OK
CSeq: 4
Session: 0288582906;timeout=60
Transport: RTP/AVP;unicast;client_port=64556-64557
com.ses.streamID: 0
--- LINE END ---
[   RtspServer.cpp:072] Data from client 192.168.1.111: PLAY rtsp://192.168.1.169/ RTSP/1.0
CSeq: 5
User-Agent: LibVLC/2.1.5 (LIVE555 Streaming Media v2014.05.27)
Session: 0288582906
Range: npt=0.000-
--- LINE END ---
[      Streams.cpp:242] Found StreamID x - SessionID: 0288582906
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 0288582906
[ StreamClient.cpp:043] Connection can close: 0
[     Frontend.cpp:401] Stream: 0, Start tuning process...
[     Frontend.cpp:393] Stream: 0, Sending DiSEqC [e0] [10] [38] [f0]
[     Frontend.cpp:463] Stream: 0, Waiting on lock...
[     Frontend.cpp:477] Stream: 0, Not locked yet (FE status 0)...
[     Frontend.cpp:477] Stream: 0, Not locked yet (FE status 10)...
[     Frontend.cpp:477] Stream: 0, Not locked yet (FE status 0)...
[

Barracuda09, 2015-03-25:
And this:
rtsp://192.168.1.169:554/?src=1&freq=11361&pol=h&msys=dvbs2&mtype=8psk&sr=22000&pids=6100,6110,6122,6130

Barracuda09, 2015-03-27:
Hi Paul,

Would you please try this (it should use more 'auto' parameters):
rtsp://192.168.1.169:554/?src=1&freq=11361&pol=h&msys=dvbs2&mtype=8psk&sr=22000&pids=6100,6110,6122,6130

Thanks for testing so far,

Marc

pbriesch, 2015-03-27:
Hello Marc,

again, no luck:
````
root@ubuntu32:/home/pbriesch/satpi# ./satpi --no-daemon
[        Satpi.cpp:247] --- starting SatPI version: 1.5.52~g07051ef ---
[  InterfaceAttr.h:094] eth0: 192.168.1.169 [00:19:66:0f:8d:07]
[      Streams.cpp:115] Current DVB_API_VERSION: 5.10
[      Streams.cpp:117] Detecting frontends in: /dev/dvb
[      Streams.cpp:135] Frontends found: 2
[     Frontend.cpp:145] Frontend Name: STB0899 Multistandard
[     Frontend.cpp:219] Frontend Type: Satellite (DVB-S)
[     Frontend.cpp:223] Frontend Type: Satellite (DVB-S2)
[     Frontend.cpp:215] Frontend Type: DSS
[     Frontend.cpp:258] Frontend Freq: 950000 Hz to 2150000 Hz
[     Frontend.cpp:259] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
[     Frontend.cpp:145] Frontend Name: STB0899 Multistandard
[     Frontend.cpp:219] Frontend Type: Satellite (DVB-S)
[     Frontend.cpp:223] Frontend Type: Satellite (DVB-S2)
[     Frontend.cpp:215] Frontend Type: DSS
[     Frontend.cpp:258] Frontend Freq: 950000 Hz to 2150000 Hz
[     Frontend.cpp:259] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
[   SsdpServer.cpp:108] Setting up SSDP server with BOOTID: 49
[   HttpServer.cpp:075] Setting up HTTP server
[   RtspServer.cpp:059] Setting up RTSP server
[    TcpSocket.cpp:149] RTSP Connection from 192.168.1.107 Port 54986 (fd: 8)
[   RtspServer.cpp:072] Data from client 192.168.1.107: OPTIONS rtsp://192.168.1.169:554/?src=1&freq=11361&pol=h&msys=dvbs2&mtype=8psk&sr=22000&pids=6100,6110,6122,6130 RTSP/1.0
CSeq: 2
User-Agent: LibVLC/3.0.0-git (LIVE555 Streaming Media v2014.07.25)
--- LINE END ---
[      Streams.cpp:250] Found StreamID x - SessionID x - Creating new SessionID: 1702957685
[       Stream.cpp:064] Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbs2
[       Stream.cpp:207] Stream: 0, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:193] RTSP/1.0 200 OK
CSeq: 2
Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
Session: 1702957685
--- LINE END ---
[   RtspServer.cpp:072] Data from client 192.168.1.107: DESCRIBE rtsp://192.168.1.169:554/?src=1&freq=11361&pol=h&msys=dvbs2&mtype=8psk&sr=22000&pids=6100,6110,6122,6130 RTSP/1.0
CSeq: 3
User-Agent: LibVLC/3.0.0-git (LIVE555 Streaming Media v2014.07.25)
Accept: application/sdp
--- LINE END ---
[      Streams.cpp:222] Found SessionID 1702957685 by SocketClient
[      Streams.cpp:242] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:064] Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbs2
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:258] RTSP/1.0 200 OK
CSeq: 3
Content-Type: application/sdp
Content-Base: rtsp://192.168.1.169/
Content-Length: 235
Session: 1702957685
--- LINE END ---
v=0
o=- 2 3 IN IP4 192.168.1.169
s=SatIPServer:1 2,0,0
t=0 0
m=video 0 RTP/AVP 33
c=IN IP4 0.0.0.0
a=control:stream=0
a=fmtp:33 ver=1.0;src=1;tuner=1,0,0,0,11361.00,h,dvbs2,8psk, , ,22000, ;pids=6100,6110,6122,6130
a=inactive
[   RtspServer.cpp:072] Data from client 192.168.1.107: SETUP rtsp://192.168.1.169/stream=0 RTSP/1.0
CSeq: 4
User-Agent: LibVLC/3.0.0-git (LIVE555 Streaming Media v2014.07.25)
Transport: RTP/AVP;unicast;client_port=59400-59401
--- LINE END ---
[      Streams.cpp:222] Found SessionID 1702957685 by SocketClient
[... 158 lines cut ...]
[ StreamClient.cpp:043] Connection can close: 0
[     Frontend.cpp:401] Stream: 0, Start tuning process...
[     Frontend.cpp:393] Stream: 0, Sending DiSEqC [e0] [10] [38] [f2]
[     Frontend.cpp:463] Stream: 0, Waiting on lock...
[     Frontend.cpp:477] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:477] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:477] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:477] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:477] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:477] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:477] Stream: 0, Not locked yet (FE status 26)...
[     Frontend.cpp:477] Stream: 0, Not locked yet (FE status 26)...
[   RtspServer.cpp:165] RTSP/1.0 200 OK
RTP-Info: url=rtsp://192.168.1.169/stream=0
CSeq: 5
Session: 0205744672
--- LINE END ---
[    TcpSocket.cpp:085] RTSP Client 192.168.1.107: Connection closed with fd: 9
[      Streams.cpp:222] Found SessionID 0205744672 by SocketClient
[      Streams.cpp:242] Found StreamID x - SessionID: 0205744672
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 0205744672
````
Thanks for looking into this issue!

Paul

DJJo14, 2015-03-27:
Change all "AUTO" to Specified options in ChannelData.cpp compile and try again. Works for my cheap dvb-t adapters.

pbriesch, 2015-03-27:
Hi DJJo14,

could you please explain this a little bit more?

Thanks, Paul

DJJo14, 2015-03-27:
Change, FEC_AUTO, ROLLOFF_AUTO, INVERSION_AUTO, PILOT_AUTO, TRANSMISSION_MODE_AUTO, GUARD_INTERVAL_AUTO, HIERARCHY_AUTO to the settings of youre mux. 
example FEC_1_2 , ROLLOFF_35 , INVERSION_ON and so on.
find the variables here http://sourcecodebrowser.com/kaffeine/1.0~pre3/frontend_8h.html
run "make clean"
"make debuge"
this workt for my adapters, the firmware of these adapters is not right.

Barracuda09, 2015-03-27:
Hi DJJo14,

This should not be necessary only as fallback. You should give it normally as 'tuning' parameters.

Barracuda09, 2015-03-28:
Hi Paul,

I have found maybe some fix (have to find why).
But if you like and have time you can try it.

Thanks for you support and testing.

Regards,
Marc

Barracuda09, 2015-03-28:
@pbriesch 
You were right there is probably a DiSEqC problem.
I have installed an raspberry Pi2 with DVB-API 5.10 and there I have also a problem. But with this
'fix' it's oke.
Very strange I have to investigate this more.

pbriesch, 2015-03-28:
Hi Marc,

I still don't get a lock.
````
pbriesch@ubuntu32:~/satpi$ sudo ./satpi --no-daemon
[        Satpi.cpp:247] --- starting SatPI version: 1.5.54~g8c2f3de ---
[  InterfaceAttr.h:094] eth0: 192.168.1.169 [00:19:66:0f:8d:07]
[      Streams.cpp:115] Current DVB_API_VERSION: 5.10
[      Streams.cpp:117] Detecting frontends in: /dev/dvb
[      Streams.cpp:135] Frontends found: 2
[     Frontend.cpp:145] Frontend Name: STB0899 Multistandard
[     Frontend.cpp:219] Frontend Type: Satellite (DVB-S)
[     Frontend.cpp:223] Frontend Type: Satellite (DVB-S2)
[     Frontend.cpp:215] Frontend Type: DSS
[     Frontend.cpp:258] Frontend Freq: 950000 Hz to 2150000 Hz
[     Frontend.cpp:259] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
[     Frontend.cpp:145] Frontend Name: STB0899 Multistandard
[     Frontend.cpp:219] Frontend Type: Satellite (DVB-S)
[     Frontend.cpp:223] Frontend Type: Satellite (DVB-S2)
[     Frontend.cpp:215] Frontend Type: DSS
[     Frontend.cpp:258] Frontend Freq: 950000 Hz to 2150000 Hz
[     Frontend.cpp:259] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
[   SsdpServer.cpp:108] Setting up SSDP server with BOOTID: 53
[   HttpServer.cpp:075] Setting up HTTP server
[   RtspServer.cpp:059] Setting up RTSP server
[    TcpSocket.cpp:149] RTSP Connection from 192.168.1.107 Port 53225 (fd: 8)
[   RtspServer.cpp:072] Data from client 192.168.1.107: DESCRIBE rtsp://192.168.1.169/ RTSP/1.0
Connection:Close
CSeq:1068
Accept:application/sdp
--- LINE END ---
[      Streams.cpp:232] Found message outside session
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:258] RTSP/1.0 404 Not Found
CSeq: 1068
Content-Type: application/sdp
Content-Base: rtsp://192.168.1.169/
Content-Length: 0
--- LINE END ---
[    TcpSocket.cpp:149] RTSP Connection from 192.168.1.107 Port 53226 (fd: 9)
[   RtspServer.cpp:072] Data from client 192.168.1.107: DESCRIBE rtsp://192.168.1.169/ RTSP/1.0
Connection:Close
CSeq:1079
Accept:application/sdp
--- LINE END ---
[      Streams.cpp:232] Found message outside session
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:258] RTSP/1.0 404 Not Found
CSeq: 1079
Content-Type: application/sdp
Content-Base: rtsp://192.168.1.169/
Content-Length: 0
--- LINE END ---
[    TcpSocket.cpp:149] RTSP Connection from 192.168.1.107 Port 53228 (fd: 10)
[    TcpSocket.cpp:149] RTSP Connection from 192.168.1.107 Port 53227 (fd: 11)
[   RtspServer.cpp:072] Data from client 192.168.1.107: DESCRIBE rtsp://192.168.1.169/ RTSP/1.0
Connection:Close
CSeq:1079
Accept:application/sdp
--- LINE END ---
[      Streams.cpp:232] Found message outside session
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:258] RTSP/1.0 404 Not Found
CSeq: 1079
Content-Type: application/sdp
Content-Base: rtsp://192.168.1.169/
Content-Length: 0
--- LINE END ---
[   RtspServer.cpp:072] Data from client 192.168.1.107: DESCRIBE rtsp://192.168.1.169/ RTSP/1.0
Connection:Close
[... 255 lines cut ...]
Content-Type: application/sdp
Content-Base: rtsp://192.168.1.169/
Content-Length: 222
--- LINE END ---
v=0
o=- 2 3 IN IP4 192.168.1.169
s=SatIPServer:1 2,0,0
t=0 0
m=video 0 RTP/AVP 33
c=IN IP4 0.0.0.0
a=control:stream=0
a=fmtp:33 ver=1.0;src=1;tuner=1,0,0,0,11361.00,h,dvbs2,8psk,off,0.35,22000,23;pids=
a=inactive
[   RtspServer.cpp:072] Data from client 192.168.1.107: PLAY rtsp://192.168.1.169:554/stream=0?pids=0,16,17,18,19,20,6100,6110,6120,6121,6122,6123,6130,6131,6132,6170,8190 RTSP/1.0
Connection:Close
CSeq:1072
Session:1702957685
--- LINE END ---
[      Streams.cpp:242] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 1702957685
[       Stream.cpp:209] Stream: 0, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 0
[     Frontend.cpp:401] Stream: 0, Start tuning process...
[     Frontend.cpp:393] Stream: 0, Sending DiSEqC [e0] [10] [38] [f2]
^Cpbriesch@ubuntu32:~/satpi$

Barracuda09, 2015-03-30:
Hi Paul,

I have looked at "STB0899 Multistandard" tuning problems, and it seems it has sometimes difficulties
getting a lock on signals. see:
http://www.vdr-portal.de/board1-news/board101-news-archiv/99603-tester-gesucht-verbessertes-dvb-s-tuning-mit-stb0899-satkarten-mantis-bridge-u-a/

https://patchwork.linuxtv.org/patch/4352/

What type of drivers are you using? s2-liplianin or v4l-dvb

pbriesch, 2015-03-30:
Hi Marc,

thanks for researching this issue. I need to find out what driver is in use. However, I'm afraid to tell you that I do not have this problem with minisatip and my tests show that DVB-S2 has worked before mit SatPi. Please let me know if I can do some more specific tests for you.

Thanks, again. Paul

Barracuda09, 2015-03-30:
Yes I Know.. but for some strange reason I can not seem to see what broke it at this point  :-(

pbriesch, 2015-03-30:
Marc,
don't let this issue stop your other developments. I can live with it for now. Sometimes it is better to get some distance to an issue and look at it at at later time
Paul

Barracuda09, 2015-04-01:
Hi Paul,

I have changed the locking statement. If you like and have some time left you can try it.

Thanks,

Marc

pbriesch, 2015-04-01:
Marc, it works now! 

:-) 

How did you fix the issue? I still have to try VDR.

Paul

````
pbriesch@ubuntu32:~/satpi$ sudo ./satpi --no-daemon 
[        Satpi.cpp:247] --- starting SatPI version: 1.5.56~gab67788 ---
[  InterfaceAttr.h:094] eth0: 192.168.1.169 [00:19:66:0f:8d:07]
[      Streams.cpp:115] Current DVB_API_VERSION: 5.10
[      Streams.cpp:117] Detecting frontends in: /dev/dvb
[      Streams.cpp:135] Frontends found: 2
[     Frontend.cpp:147] Frontend Name: STB0899 Multistandard
[     Frontend.cpp:221] Frontend Type: Satellite (DVB-S)
[     Frontend.cpp:225] Frontend Type: Satellite (DVB-S2)
[     Frontend.cpp:217] Frontend Type: DSS
[     Frontend.cpp:260] Frontend Freq: 950000 Hz to 2150000 Hz
[     Frontend.cpp:261] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
[     Frontend.cpp:147] Frontend Name: STB0899 Multistandard
[     Frontend.cpp:221] Frontend Type: Satellite (DVB-S)
[     Frontend.cpp:225] Frontend Type: Satellite (DVB-S2)
[     Frontend.cpp:217] Frontend Type: DSS
[     Frontend.cpp:260] Frontend Freq: 950000 Hz to 2150000 Hz
[     Frontend.cpp:261] Frontend srat: 5000000 symbols/s to 45000000 symbols/s
[   SsdpServer.cpp:108] Setting up SSDP server with BOOTID: 60
[   HttpServer.cpp:075] Setting up HTTP server
[   RtspServer.cpp:059] Setting up RTSP server
[   SsdpServer.cpp:166] SAT>IP Server 192.168.1.55: contacted us because of clashing DEVICEID 1
[   SsdpServer.cpp:182] Changing BOOTID to: 61
[   SsdpServer.cpp:191] SAT>IP Client 192.168.1.55 : tries to discover the network, sending an reply
[    TcpSocket.cpp:149] RTSP Connection from 192.168.1.111 Port 54446 (fd: 8)
[   RtspServer.cpp:072] Data from client 192.168.1.111: TEARDOWN rtsp://192.168.1.169:554/stream=0 RTSP/1.0
Connection:Close
CSeq:1017
Session:1702957685
--- LINE END ---
[      Streams.cpp:242] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:081] Stream: 0, No StreamClient with SessionID 1702957685
[       Stream.cpp:081] Stream: 1, No StreamClient with SessionID 1702957685
[      Streams.cpp:278] Found no Stream/Client of interest!
[    TcpSocket.cpp:085] RTSP Client 192.168.1.111: Connection closed with fd: 8
[      Streams.cpp:232] Found message outside session
[    TcpSocket.cpp:149] HTTP Connection from 192.168.1.111 Port 61823 (fd: 8)
[   HttpServer.cpp:178] Client: 192.168.1.111 requesed /desc.xml
[    TcpSocket.cpp:149] RTSP Connection from 192.168.1.111 Port 54462 (fd: 9)
[   RtspServer.cpp:072] Data from client 192.168.1.111: DESCRIBE rtsp://192.168.1.169/ RTSP/1.0
Connection:Close
CSeq:1001
Accept:application/sdp
--- LINE END ---
[      Streams.cpp:232] Found message outside session
[ StreamClient.cpp:043] Connection can close: 1
[   RtspServer.cpp:258] RTSP/1.0 404 Not Found
CSeq: 1001
Content-Type: application/sdp
Content-Base: rtsp://192.168.1.169/
Content-Length: 0
--- LINE END ---
[    TcpSocket.cpp:085] RTSP Client 192.168.1.111: Connection closed with fd: 9
[      Streams.cpp:232] Found message outside session
[       Stream.cpp:129] Stream: -1, Close StreamClient[0] with SessionID -1
[    TcpSocket.cpp:149] RTSP Connection from 192.168.1.111 Port 54463 (fd: 9)
[   RtspServer.cpp:072] Data from client 192.168.1.111: SETUP rtsp://192.168.1.169:554/?src=1&freq=11362&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=off&sr=22000&fec=23 RTSP/1.0
Connection:Close
CSeq:1002
Transport:RTP/AVP;unicast;client_port=5002-5003
--- LINE END ---
[... 339 lines cut ...]
CSeq: 1028
Session: 1702957685
--- LINE END ---
[    TcpSocket.cpp:085] RTSP Client 192.168.1.111: Connection closed with fd: 9
[      Streams.cpp:222] Found SessionID 1702957685 by SocketClient
[      Streams.cpp:242] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 1702957685
[    TcpSocket.cpp:149] RTSP Connection from 192.168.1.111 Port 54566 (fd: 9)
[   RtspServer.cpp:072] Data from client 192.168.1.111: PLAY rtsp://192.168.1.169:554/stream=0?delpids=6400 RTSP/1.0
Connection:Close
CSeq:1029
Session:1702957685
--- LINE END ---
[      Streams.cpp:242] Found StreamID x - SessionID: 1702957685
[       Stream.cpp:067] Stream: 0, StreamClient[0] with SessionID 1702957685
[       Stream.cpp:209] Stream: 0, Parsing transport parameters...
[ StreamClient.cpp:043] Connection can close: 0
[     Frontend.cpp:530] Stream: 0, Updating PID filters...
[     Frontend.cpp:551] Stream: 0, Remove filter PID: 6400 - fd: 032 - Packet Count: 1
[   RtspServer.cpp:165] RTSP/1.0 200 OK
RTP-Info: url=rtsp://192.168.1.169/stream=0
CSeq: 1029
Session: 1702957685
--- LINE END ---
^Cpbriesch@ubuntu32:~/satpi$

Barracuda09, 2015-04-01:
Hi Paul,

Well the 'problem' was in the tune locking logic (see when the tuner is locked).

closed 2015-04-02 by pbriesch