Barracuda09/SATPI #33: HTTP Streaming support

lars18th, 2015-03-24:
Hi,

Last version is great! Howerver, I suggest to add HTTP Streaming support.

For example, at time, using VLC you can tune & receive the TS using a call like:

rtsp://192.168.210.17:554/?freq=538&msys=dvbt&pids=all

But, it will be also very interesting to support http calls like:

http://192.168.210.17:8875/?freq=538&msys=dvbt&pids=all

With this support, other programs, that not necessary understands the SAT>IP protocol, can receive (and tune) the TS. I'm testing this in the minisatip project, and it's very useful.

Please, can you consider to add this functionality?

Barracuda09, 2015-03-24:
Hi lars18th,

Yes thanks you are right. I have to lock into this.

Marc

Barracuda09, 2015-12-19:
I have added this feature. Will close this defect.

closed 2015-12-19 by Barracuda09