Barracuda09/SATPI #34: SATPI does not stream anything after ~1 day

FuNK3Y, 2015-04-01:
Hi Marc,

I noticed that SATPI stop streaming after ~1 day. During the day SATPI is mostly idle, about 2h of TV plus the EPG scanning (occuring everyday at 00:04 & 12:04).

This issue is pretty strange, as killing & restarting SATPI does not solve it, only a full restart of my cubox clear the error out.This does not seems to be driver/firmware related, as w_scan works fine when SATPI don't.

My best guess would be that the some parameter of the tuner (counters, status, ...) might over time get values that does not suit SATPI's expectations.

Here is the log:
0  Wed Apr  1 23:46:17.084133398 2015  6  [        Satpi.cpp:247] --- starting SatPI version: 1.5.56~gab67788 ---
1  Wed Apr  1 23:46:17.085467135 2015  6  [  InterfaceAttr.h:094] eth0: 192.168.1.22 [d0:63:b4:00:6f:fa]
2  Wed Apr  1 23:46:17.086566524 2015  7  [      Streams.cpp:113] Not the preferred DVB API version, for correct function it should be 5.5 or higher
3  Wed Apr  1 23:46:17.086661862 2015  7  [      Streams.cpp:115] Current DVB_API_VERSION: 5.4
4  Wed Apr  1 23:46:17.086732866 2015  6  [      Streams.cpp:117] Detecting frontends in: /dev/dvb
5  Wed Apr  1 23:46:17.089678350 2015  6  [      Streams.cpp:135] Frontends found: 2
6  Wed Apr  1 23:46:17.090209046 2015  6  [     Frontend.cpp:147] Frontend Name: DRXK DVB-C DVB-T
7  Wed Apr  1 23:46:17.090294718 2015  7  [     Frontend.cpp:247] Frontend Type: Cable (Annex C)
8  Wed Apr  1 23:46:17.090355388 2015  7  [     Frontend.cpp:256] Frontend Type: Unknown 18
9  Wed Apr  1 23:46:17.090412391 2015  7  [     Frontend.cpp:229] Frontend Type: Terrestrial (DVB-T)
10  Wed Apr  1 23:46:17.090471728 2015  7  [     Frontend.cpp:260] Frontend Freq: 47000000 Hz to 865000000 Hz
11  Wed Apr  1 23:46:17.090529398 2015  7  [     Frontend.cpp:261] Frontend srat: 870000 symbols/s to 11700000 symbols/s
12  Wed Apr  1 23:46:17.090614736 2015  6  [     Frontend.cpp:147] Frontend Name: DRXK DVB-C DVB-T
13  Wed Apr  1 23:46:17.090680739 2015  7  [     Frontend.cpp:247] Frontend Type: Cable (Annex C)
14  Wed Apr  1 23:46:17.090735076 2015  7  [     Frontend.cpp:256] Frontend Type: Unknown 18
15  Wed Apr  1 23:46:17.090834748 2015  7  [     Frontend.cpp:229] Frontend Type: Terrestrial (DVB-T)
16  Wed Apr  1 23:46:17.090891418 2015  7  [     Frontend.cpp:260] Frontend Freq: 47000000 Hz to 865000000 Hz
17  Wed Apr  1 23:46:17.090945088 2015  7  [     Frontend.cpp:261] Frontend srat: 870000 symbols/s to 11700000 symbols/s
18  Wed Apr  1 23:46:17.091359445 2015  6  [   RtspServer.cpp:059] Setting up RTSP server
19  Wed Apr  1 23:46:17.091630128 2015  6  [   HttpServer.cpp:075] Setting up HTTP server
20  Wed Apr  1 23:46:17.092005482 2015  6  [   SsdpServer.cpp:108] Setting up SSDP server with BOOTID: 12
21  Wed Apr  1 23:46:17.141704523 2015  6  [    TcpSocket.cpp:149] HTTP Connection from 192.168.1.119 Port 59908 (fd: 8)
22  Wed Apr  1 23:46:17.341457745 2015  6  [    TcpSocket.cpp:149] HTTP Connection from 192.168.1.119 Port 59910 (fd: 9)
23  Wed Apr  1 23:46:17.347433071 2015  6  [    TcpSocket.cpp:149] HTTP Connection from 192.168.1.119 Port 59911 (fd: 10)
24  Wed Apr  1 23:46:17.347735088 2015  6  [    TcpSocket.cpp:149] HTTP Connection from 192.168.1.119 Port 59912 (fd: 11)
25  Wed Apr  1 23:46:17.355869531 2015  6  [    TcpSocket.cpp:149] HTTP Connection from 192.168.1.119 Port 59913 (fd: 12)
26  Wed Apr  1 23:46:17.431427983 2015  6  [    TcpSocket.cpp:149] HTTP Connection from 192.168.1.119 Port 59909 (fd: 13)
27  Wed Apr  1 23:46:26.647211324 2015  6  [   SsdpServer.cpp:191] SAT>IP Client 192.168.1.20 : tries to discover the network, sending an reply
28  Wed Apr  1 23:46:26.647515341 2015  6  [   SsdpServer.cpp:191] SAT>IP Client 192.168.1.20 : tries to discover the network, sending an reply
29  Wed Apr  1 23:46:26.647799689 2015  6  [   SsdpServer.cpp:191] SAT>IP Client 192.168.1.20 : tries to discover the network, sending an reply
30  Wed Apr  1 23:46:26.898895396 2015  6  [    TcpSocket.cpp:149] HTTP Connection from 192.168.1.20 Port 59384 (fd: 14)
31  Wed Apr  1 23:46:26.899156075 2015  7  [   HttpServer.cpp:178] Client: 192.168.1.20 requesed /desc.xml
32  Wed Apr  1 23:46:26.901820204 2015  7  [    TcpSocket.cpp:085] HTTP Client 192.168.1.20: Connection closed with fd: 14
33  Wed Apr  1 23:46:42.006462860 2015  6  [    TcpSocket.cpp:149] RTSP Connection from 192.168.1.20 Port 51744 (fd: 14)
34  Wed Apr  1 23:46:42.006698540 2015  7  [   RtspServer.cpp:072] Data from client 192.168.1.20: SETUP rtsp://192.168.1.22/?fe=3&freq=354&sr=6900&msys=dvbc&mtype=256qam&fec=35 RTSP/1.0
35  Wed Apr  1 23:46:42.006698540 2015  7  Transport: RTP/AVP;unicast;client_port=37080-37081
36  Wed Apr  1 23:46:42.006698540 2015  7  CSeq: 1
37  Wed Apr  1 23:46:42.006698540 2015  7  --- LINE END ---
38  Wed Apr  1 23:46:42.007270572 2015  6  [      Streams.cpp:250] Found StreamID x - SessionID x - Creating new SessionID: 1702957685
39  Wed Apr  1 23:46:42.007442914 2015  6  [       Stream.cpp:064] Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbc
40  Wed Apr  1 23:46:42.007622924 2015  7  [       Stream.cpp:209] Stream: 0, Parsing transport parameters...
41  Wed Apr  1 23:46:42.009242345 2015  7  [ StreamClient.cpp:043] Connection can close: 0
42  Wed Apr  1 23:46:42.009585697 2015  6  [     Frontend.cpp:475] Stream: 0, Opened /dev/dvb/adapter0/frontend0 fd: 15
43  Wed Apr  1 23:46:42.009729371 2015  7  [     Frontend.cpp:410] Stream: 0, Start tuning process...
44  Wed Apr  1 23:46:42.010401740 2015  6  [     Frontend.cpp:486] Stream: 0, Waiting on lock...
45  Wed Apr  1 23:46:42.010534748 2015  6  [     Frontend.cpp:500] Stream: 0, Not locked yet (FE status 0x0)...
46  Wed Apr  1 23:46:42.184504563 2015  6  [     Frontend.cpp:500] Stream: 0, Not locked yet (FE status 0x0)...
47  Wed Apr  1 23:46:42.240883969 2015  6  [     Frontend.cpp:500] Stream: 0, Not locked yet (FE status 0x0)...
48  Wed Apr  1 23:46:42.303012023 2015  6  [     Frontend.cpp:497] Stream: 0, Tuned and locked (FE status 0x1F).
49  Wed Apr  1 23:46:42.303522717 2015  6  [     Frontend.cpp:518] Stream: 0, Opened /dev/dvb/adapter0/dvr0 fd: 16.
50  Wed Apr  1 23:46:42.304020411 2015  6  [     Frontend.cpp:530] Stream: 0, Updating PID filters...
51  Wed Apr  1 23:46:42.304957462 2015  6  [    RtpThread.cpp:095] Stream: 0, Start   RTP stream to 192.168.1.20:37080
52  Wed Apr  1 23:46:42.305298147 2015  6  [   RtcpThread.cpp:073] Stream: 0, Start  RTCP stream to 192.168.1.20:37081
53  Wed Apr  1 23:46:42.305446822 2015  7  [   RtspServer.cpp:141] RTSP/1.0 200 OK
54  Wed Apr  1 23:46:42.305446822 2015  7  CSeq: 1
55  Wed Apr  1 23:46:42.305446822 2015  7  Session: 1702957685;timeout=60
56  Wed Apr  1 23:46:42.305446822 2015  7  Transport: RTP/AVP;unicast;client_port=37080-37081
57  Wed Apr  1 23:46:42.305446822 2015  7  com.ses.streamID: 0
58  Wed Apr  1 23:46:42.305446822 2015  7  --- LINE END ---
59  Wed Apr  1 23:46:42.306811896 2015  7  [   RtspServer.cpp:072] Data from client 192.168.1.20: PLAY rtsp://192.168.1.22/stream=0?addpids=0,1,16,17,18,208,209,210,219,228,238 RTSP/1.0
60  Wed Apr  1 23:46:42.306811896 2015  7  Session: 1702957685
[... 133 lines cut ...]
194  Wed Apr  1 23:46:54.309961493 2015  6  [      Streams.cpp:242] Found StreamID x - SessionID: 0205744672
195  Wed Apr  1 23:46:54.310093500 2015  6  [       Stream.cpp:081] Stream: 0, No StreamClient with SessionID 0205744672
196  Wed Apr  1 23:46:54.310310512 2015  6  [       Stream.cpp:067] Stream: 1, StreamClient[0] with SessionID 0205744672
197  Wed Apr  1 23:46:54.310436852 2015  6  [       Stream.cpp:129] Stream: 1, Close StreamClient[0] with SessionID 0205744672

Here is what TVH thinks of it:
2015-04-01 23:46:48.000 mpegts: 354MHz in Cablecom - tuning on SAT>IP DVB-C Tuner #4 (192.168.1.22)
2015-04-01 23:46:48.000 subscription: 0001: "HTTP" subscribing on "RTS Un HD", weight: 100, adapter: "SAT>IP DVB-C Tuner #4 (192.168.1.22)", network: "Cablecom", mux: "354MHz", provider: "upc", service: "RTS Un HD", profile="pass", hostname="192.168.1.119", username="", client="VLC/2.1.5 LibVLC/2.1.5"
2015-04-01 23:46:58.000 webui: Couldn't start streaming /stream/channel/5e5d62222d01b16f97b770cb62be5481, No input detected

Here is an example of w_scan:
314000: sr6900 (time: 07:44) (time: 07:45) signal ok:
        QAM_256  f = 314000 kHz S6900C999
Info: no data from NIT(actual)
322000: sr6900 (time: 07:58) (time: 07:59) signal ok:
        QAM_256  f = 322000 kHz S6900C999
Info: no data from NIT(actual)
330000: sr6900 (time: 08:11) (time: 08:12) signal ok:
        QAM_256  f = 330000 kHz S6900C999
Info: no data from NIT(actual)
338000: sr6900 (time: 08:24) (time: 08:25) signal ok:
        QAM_256  f = 338000 kHz S6900C999
Info: no data from NIT(actual)
346000: sr6900 (time: 08:38) (time: 08:39) signal ok:
        QAM_256  f = 346000 kHz S6900C999

Barracuda09, 2015-04-02:
Hi FuNK3Y,

Thanks for this report and testing.

I have some questions:
- Is this a problem from the last time?
- A full restart of the cubox, is a power cycle or reboot

SatPI sees a lock on the signal (FE status 0x1F)  But indeed no pids receive data.:
```
48  Wed Apr 1 23:46:42.303012023 2015  6  [ Frontend.cpp:497] Stream: 0, Tuned and locked (FE status 0x1F).
```
And w_scan sees indeed also an lock but saying also:
```
Info: no data from NIT(actual)     (Don't realy know what this means)
```

FuNK3Y, 2015-04-02:
I did some more testing, and as you mentioned the problem lies within the driver.

I tested with tvheadend as well and it only detected half of the tuners & so on. A reboot (not a power cycle) solves it.

As I do have 2 time the same tuner, I took one away; I'll see if it helps.

SATPI is clearly not responsible for this one :+1:

closed 2015-04-02 by FuNK3Y