Barracuda09/SATPI #36: OSCAM - Does it work?

FuNK3Y, 2015-07-25:
Checking the latest version I found oscam related config. This is a very interesting feature I would like to test

Does it works? How shall it be used?

Thanks you!

Barracuda09, 2015-08-10:
Hi,

Sorry for this late reply, I was testing this, but due to less free time I did not finish it yet.
I hope someone can help me out with this some day.
So far SatPI is working out for me (that was at the beginning the main reason I started this project,
but I hope someone else has found use for it as well)

Regards,

Marc

Barracuda09, 2015-10-17:
Hello,

I have implemented something that works-out in my situation.

See for more details: https://github.com/Barracuda09/SATPI/wiki/4.-OSCam-DVBAPI-Protocol

Regards,

Marc

holli73, 2015-12-23:
hello, 

i tried it but it looks like no request / connect is send to my oscam server?
what are the settings used in the dvbapi conf within oscam?

i could only provide the ip + port but the status is not going to be active?

any idea how to troubleshoot

thanks
holli

Barracuda09, 2015-12-23:
Hi,

You have to hit enter when you typed in the IP and Port number.
Hope this works out for you.

Thanks for trying and regards,

Marc

holli73, 2015-12-23:
hello,

this did the trick but still not working - example for sky

2015/12/23 10:08:47 7213B264 c   (dvbapi) Demuxer 0 ecmpid 0 CAID: 09C4 ECM_PID: 1B6C PROVID: 000000 
2015/12/23 10:08:47 7213B264 c   (dvbapi) Demuxer 0 ecmpid 1 CAID: 098C ECM_PID: 1A6C PROVID: 000000 
2015/12/23 10:08:47 7213B264 c   (dvbapi) Demuxer 0 ecmpid 2 CAID: 09AF ECM_PID: 1F6C PROVID: 000000 
2015/12/23 10:08:47 7213B264 c   (dvbapi) Demuxer 0 stream Reserved(type: 00 pid: 000b length: 6)
2015/12/23 10:08:47 7213B264 c   (dvbapi) Demuxer 0 found 3 ECMpids and 1 STREAMpids in caPMT
2015/12/23 10:08:47 7213B264 c   (dvbapi) Demuxer 0 serving srvid 0022 (Sky Atlantic) on adapter 0000 camask 0001 index 0000 pmtpid 0000
2015/12/23 10:08:47 7213B264 c   (dvbapi) Demuxer 0 trying to descramble PID 0 CAID 09C4 PROVID 000000 ECMPID 1B6C ANY CHID PMTPID 0000 VPID 0000
2015/12/23 10:08:48 7213B264 c      (ecm) anonymous (09C4@000000/019E/0022/B3:5F572A28801A3EA840D1A116B2917014:0F06000000:): found (117 ms) by holliskyv13 - Sky Atlantic
2015/12/23 10:08:48 7213B264 c      (ecm) anonymous (09C4@000000/019E/0022/B3:A731D7CE4528C3E710105015DF5F8B6B:0F06000000:): found (119 ms) by holliskyv13 - Sky Atlantic
2015/12/23 10:08:52 4BE840B4 c      (ecm) hollitriplexwz (1843@000000/0000/2E9B/89:66F55B3A2226F7871DF3A5547480CB8B): found (372 ms) by hdplus - Super RTL HD
2015/12/23 10:08:58 7213B264 c      (ecm) anonymous (09C4@000000/019E/0022/B3:643567C4135F02E1921016272D227740:0F06000000:): found (116 ms) by holliskyv13 - Sky Atlantic
2015/12/23 10:09:02 7213B264 c   (dvbapi)   FF 
2015/12/23 10:09:02 7213B264 c   (dvbapi) Unknown socket command received: 0xFFFFFFFF
2015/12/23 10:09:02 7213B264 c   (dvbapi) Unknown command:
2015/12/23 10:09:02 7213B264 c   (dvbapi)   FF 
2015/12/23 10:09:02 7213B264 c   (dvbapi) Unknown socket command received: 0xFFFFFFFF
2015/12/23 10:09:02 7213B264 c   (dvbapi) Unknown command:
2015/12/2

or ORF f.ex.:
2015/12/23 10:15:55 7213B264 c   (dvbapi) Client connected: 'SatPI 1.5.118~g6cf174' (protocol version = 2)
2015/12/23 10:15:56 7213B264 c   (dvbapi) Demuxer 0 ecmpid 0 CAID: 0648 ECM_PID: 0078 PROVID: 000000 
2015/12/23 10:15:56 7213B264 c   (dvbapi) Demuxer 0 ecmpid 1 CAID: 0650 ECM_PID: 007A PROVID: 000000 
2015/12/23 10:15:56 7213B264 c   (dvbapi) Demuxer 0 ecmpid 2 CAID: 0D95 ECM_PID: 010E PROVID: 000000 
2015/12/23 10:15:56 7213B264 c   (dvbapi) Demuxer 0 ecmpid 3 CAID: 0D98 ECM_PID: 0110 PROVID: 000000 
2015/12/23 10:15:56 7213B264 c   (dvbapi) Demuxer 0 ecmpid 4 CAID: 0500 ECM_PID: 01CD PROVID: 051900 
2015/12/23 10:15:56 7213B264 c   (dvbapi) Demuxer 0 ecmpid 5 CAID: 09C4 ECM_PID: 01E0 PROVID: 000000 
2015/12/23 10:15:56 7213B264 c   (dvbapi) Demuxer 0 ecmpid 6 CAID: 098C ECM_PID: 01EA PROVID: 000000 
2015/12/23 10:15:56 7213B264 c   (dvbapi) Demuxer 0 stream Reserved(type: 00 pid: 000b length: 6)
2015/12/23 10:15:56 7213B264 c   (dvbapi) Demuxer 0 found 7 ECMpids and 1 STREAMpids in caPMT
2015/12/23 10:15:56 7213B264 c   (dvbapi) Demuxer 0 serving srvid 132F (0648@000000:132F unknown) on adapter 0000 camask 0001 index 0000 pmtpid 0000
2015/12/23 10:15:56 7213B264 c   (dvbapi) Demuxer 0 trying to descramble PID 2 CAID 0D95 PROVID 000000 ECMPID 010E ANY CHID PMTPID 0000 VPID 0000

my oscam version 11164 - from the satpi log it looks like:

257  Wed Dec 23 10:16:08.8995 2015  7  [        src/DvbapiClient.cpp:632] Stream: 0, Receive data  len: 5  cmd: 6F2A
258  Wed Dec 23 10:16:08.8996 2015  6  [    src/StreamProperties.cpp:078] Stream: 0, Clear ECM PID: 270  demux: 0  filter: 1
259  Wed Dec 23 10:16:08.8996 2015  7  [            src/Frontend.cpp:447] Stream: 0, Updating frontend...
260  Wed Dec 23 10:16:08.8996 2015  6  [            src/Frontend.cpp:544] Stream: 0, Updating PID filters...
261  Wed Dec 23 10:16:08.9001 2015  7  [            src/Frontend.cpp:468] Stream: 0, Updating frontend (Finished)
262  Wed Dec 23 10:16:08.9001 2015  7  [        src/DvbapiClient.cpp:632] Stream: -128, Receive data  len: 6  cmd: 1010E
263  Wed Dec 23 10:16:08.9001 2015  7  [        src/DvbapiClient.cpp:683] Stream: 0, Receive unexpected data
264  Wed Dec 23 10:16:08.9001 2015  7  00 00 6F 2A 00 00 01 01 0E 

any idea - is it possible to prove a dvbapi user somewhere?

thanks
holli

Barracuda09, 2015-12-23:
Hi,

Could you provide a full SatPI log? From first tune and some key changes
and stopping the stream.

Thanks in advance,

Marc

lars18th, 2015-12-28:
Hi,

I like to check it, and I compiled with "make debug LIBDVBCSA=yes". The problem is when I like to configure the Oscam server: I go to the Web GUI in the "Configure" page and the Oscam section is here. However, I can't SAVE/COMMIT the changes. No Save button, and when I change some value (for example, the port) it resets to default after 3 seconds.

Please, can you explain you to configure? I can manually edit some configure.xml file?
Thank you!

Barracuda09, 2015-12-28:
Hi,
you have to hit enter to 'save'

lars18th, 2015-12-28:
Hi @Barracuda09 ,

> you have to hit enter to 'save'

Quite annoying, but it works. Thank you!
However, it ins't persistent. Please, can you implement some CONFIG file?

Barracuda09, 2016-05-20:
It is working as far I know, I will close this issue.
If anyone has an problem please make a new issue.

closed 2016-05-20 by Barracuda09