Barracuda09/SATPI #37: working version of SatPI ?

zz64, 2015-11-18:
Hi, i have successfully built 1.5 version with openwrt mips32 for brcm63xx, but i can't play any frequency :(
trying to play:
rtsp://192.168.3.57:554/?freq=11240&sr=27500&pol=v&msys=dvbs&pids=all
or
rtsp://192.168.3.57:554/?freq=11240&sr=27500&pol=v&msys=dvbs&pids=0,247,248,249
daemon log here : http://pastebin.com/raw.php?i=6s90ftpa

[image]

i try to build latest master,but fail :(
http://pastebin.com/raw.php?i=isCJqzcx
any idea ?
Thanks.

Barracuda09, 2015-11-18:
Hi,

If you build without LIBDVBCSA=yes

zz64, 2015-11-18:
Ok, i try this next..
but please, 1.5 version (from releases) not work ?
Thanks.

Barracuda09, 2015-11-18:
Sorry.. maybe not.. master should work.

zz64, 2015-11-18:
without DVBCSA the master built ok :)

Barracuda09, 2015-11-18:
Ok.. but this should work also 'make debug LIBDVBCSA=yes'

zz64, 2015-11-18:
Work but 98% Continuity errors :\
i think is not ready yes ?

[image]

[image]

Barracuda09, 2015-11-18:
Well it should work okay. what is the load of the processor?

zz64, 2015-11-18:
not over 50%
[image]

zz64, 2015-11-18:
0 Continuity errors using tvheadend (4.1) on this hardware (with same channel).

Barracuda09, 2015-11-18:
How did you start SatPI?

zz64, 2015-11-18:
satpi in rootfs
the "web" folder found in /satpi/web

zz64, 2015-11-18:
using tvheadend on same hardware:

[image]

Barracuda09, 2015-11-18:
What does syslog say?

because logd was also 12%

zz64, 2015-11-18:
http://pastebin.com/raw.php?i=FzdAdrbg

Barracuda09, 2015-11-18:
Okay there is the source of the problem.. why it is doing this I’m not sure
yet.

You stopped TvHeadend I assume.

Barracuda09, 2015-11-18:
Could you remove in RtpThread.cpp line 266 (PERROR("send RTP");)
and check again?

zz64, 2015-11-18:
openwrt tree need patch, i can't modify source manually.
can you provide patch please ?

Barracuda09, 2015-11-18:
Hope this works

--- a/src/RtpThread.cpp
+++ b/src/RtpThread.cpp
@@ -263,6 +263,6 @@ void RtpThread::sendRtpPacket(RtpPacketBuffer &buffer,
const StreamClient &clien
        // send the RTP packet
        if (sendto(_socket_fd, rtpBuffer, size + RTP_HEADER_LEN,
MSG_DONTWAIT,
                          (struct sockaddr *)&client.getRtpSockAddr(),
sizeof(client.getRtpSockAddr())) == -1) {
-               PERROR("send RTP");
+//             PERROR("send RTP");
        }
 }

zz64, 2015-11-18:
i can't copy paste from your reply.
can you attach "raw txt" please ?

Barracuda09, 2015-11-18:
This Okay

[temp.patch.txt](https://github.com/Barracuda09/SATPI/files/37884/temp.patch.txt)

zz64, 2015-11-18:
Applying ./patches/01-rtp-thread-remove-send-rtp.patch using plaintext: 
patching file src/RtpThread.cpp
Hunk #1 FAILED at 263.
1 out of 1 hunk FAILED -- saving rejects to file src/RtpThread.cpp.rej
Patch failed!  Please fix ./patches/01-rtp-thread-remove-send-rtp.patch!

zz64, 2015-11-18:
ok, i make a patch myself.

zz64, 2015-11-18:
patch ok, i build firmware now.

zz64, 2015-11-18:
same problem :/

syslog: http://pastebin.com/raw.php?i=kz6kw8zY

[image]

Barracuda09, 2015-11-18:
Sorry could you try to change in RtpThread.cpp same place:

	// send the RTP packet
	if (sendto(_socket_fd, rtpBuffer, size + RTP_HEADER_LEN, MSG_DONTWAIT,
			  (struct sockaddr *)&client.getRtpSockAddr(), sizeof(client.getRtpSockAddr())) == -1) {
		PERROR("send RTP");
	}

change MSG_DONTWAIT to MSG_NOSIGNAL 

The packets get not send because it does not wait for completion.

Barracuda09, 2015-11-19:
Hi,

Can you please give me the results of these cmd:
cat /proc/sys/net/core/wmem_default
cat /proc/sys/net/core/wmem_max

Thanks for you efforts

zz64, 2015-11-19:
Hi,
1) changing MSG_DONTWAIT to MSG_NOSIGNAL not solve the problem.

2) #cat /proc/sys/net/core/wmem_default
163840
cat /proc/sys/net/core/wmem_max
163840

please, if you want quick progress, you must add a "test branch". then you fix what seems solving the issue in this branch.
I will use this branch for testing. no patch no modification...

Barracuda09, 2015-11-19:
Hi,

Thanks for your support.
Can you try a quick test by incrementing the buffer with:
sudo sysctl -w net.core.wmem_default=524288

zz64, 2015-11-19:
root@nb6media:~# sysctl -w net.core.wmem_default=524288
net.core.wmem_default = 524288
root@nb6media:~# cat /proc/sys/net/core/wmem_default
524288

no change :\

syslog: http://pastebin.com/raw.php?i=ad6aHC8r

Barracuda09, 2015-11-19:
sorry.. there is also an max value..
sudo sysctl -w net.core.wmem_max=524288

zz64, 2015-11-19:
root@nb6media:/satpi# sysctl -w net.core.wmem_default=524288
net.core.wmem_default = 524288
root@nb6media:/satpi# cat /proc/sys/net/core/wmem_max
524288

no change :\

Barracuda09, 2015-11-19:
There is also an DVR buffer size in the frontend webpage... try incrementing this...
else I have no idea at the moment what it could be.

I run it on a RPI2 with no problems.. just like it is in master.

Again thanks for you efforts for now.

zz64, 2015-11-19:
incrementing "DVR buffer size" no solve... :\

i can't compile SatPI with uClibc
log: http://pastebin.com/raw.php?i=QYjNYfkz

any solution to get SatPI compile with uClibc ? (because is the default library used in OpenWrt distrib).

Thanks.

Barracuda09, 2015-11-19:
Hi,

Sorry.. not out of the box. But you can try to remove:
pthread_setname_np(_thread, _name.c_str());

and try again. to see what is missing more from uClibc.

zz64, 2015-11-19:
compile ok without pthread_setname_np...

here my contribution: http://dl.free.fr/iDa30Zqrw
SatPI Makefile for OpenWRT
please add it in your repository. (because "dl.free.fr" delete files after 30 days)

Thanks

Barracuda09, 2015-11-19:
Hi, Thanks for you contribution. But I can not downloaded it... Sorry I do not understand the France
language.

zz64, 2015-11-19:
can't attach file here (on github only image, txt, pdf... files allowed).
can you provide a place where i put this file ?

Barracuda09, 2015-11-19:
I have extracted the file (google translate is my friend).. Many thanks I will look how to put it into
the repo.

zz64, 2015-11-22:
Hi, I think I found the anomaly. when i launch SatPI i get message in kernel log:
``
ttusb2: there might have been an error during control message transfer. (rlen = 4, was 3)
``
full log: http://pastebin.com/raw.php?i=GV414XuQ

(this error not found when starting tvheadend)
log: http://pastebin.com/raw.php?i=TumuZN1P

i think the SatPI puts a "wrong parameter" in the kernel drivers ?
can we fix this problem ?

kernel information:
``
root@nb6media:/# uname -a

Linux nb6media 3.10.49 #5 Tue Nov 10 16:27:23 CET 2015 mips GNU/Linux
``

Thanks.

Barracuda09, 2015-11-22:
Hi,

Thanks for your support so far!.
And if you tune to an frequency. Do you see more of these message?

zz64, 2015-11-22:
No, only when first launch

zz64, 2015-11-22:
is not dvbapi version related ?

Barracuda09, 2015-11-22:
That could be the case.. you can find the version somewhere in 'linux/dvb/version.h'

Barracuda09, 2015-11-22:
But it shows up 'correctly' in the frontend.html..

zz64, 2015-11-22:
``
linux-brcm63xx_generic/linux-3.10.49$ find . -type d -name dvb
./Documentation/DocBook/media/dvb
./Documentation/dvb
./include/config/dvb
./include/uapi/linux/dvb
./user_headers/include/linux/dvb

linux-brcm63xx_generic/linux-3.10.49$ cat  ./include/uapi/linux/dvb/version.h
/*
 * version.h
 *
 * Copyright (C) 2000 Holger Waechtler <holger@convergence.de>
 *                    for convergence integrated media GmbH
 *
 * This program is free software; you can redistribute it and/or
 * modify it under the terms of the GNU Lesser General Public License
 * as published by the Free Software Foundation; either version 2.1
 * of the License, or (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU Lesser General Public License
 * along with this program; if not, write to the Free Software
 * Foundation, Inc., 59 Temple Place - Suite 330, Boston, MA  02111-1307, USA.
 *
 */

#ifndef _DVBVERSION_H_
#define _DVBVERSION_H_

#define DVB_API_VERSION 5
#define DVB_API_VERSION_MINOR 10

#endif /*_DVBVERSION_H_*/
``

zz64, 2015-11-22:
the dvb include's is in "./include/uapi/linux/dvb" Not "linux/dvb/version.h"

Barracuda09, 2015-11-22:
Yes.. do you have by any chance another DVB card you could try?

And try making a debug version of SatPI, It gives some more info.

zz64, 2015-11-22:
no, i have only this :\

Barracuda09, 2015-11-22:
Well It should work with this one too...

Barracuda09, 2015-12-19:
Hi Again,

How does it work with HTTP streaming from resent changes?

Thanks for testing,

Marc

Barracuda09, 2016-03-26:
I will close this issue for now. If someone sees this, please make a new issue.

closed 2016-03-26 by Barracuda09

Barracuda09, 2016-10-25:
Hi @zz64 

Sorry, could you try to 'upload'  the SatPI Makefile for OpenWRT again?

I lost it unfortunately, thanks for your support up until now.

Regards,

Marc