Barracuda09/SATPI #38: Delivery System: UNKNOWN DELSYS

hansp01, 2015-12-11:
Hello,

i've a Silicon Labs Si2168 dual Tuner (DVB-C and DVB-T/2). System says Unknown DELSYS.
Here is a screenshot:

[image]

Barracuda09, 2015-12-11:
Hi,

This is just the begin/default. If you tune to an channel it will change.

maxm87, 2015-12-11:
Hi,

please, can you explain how to tune to a channel? its dvbt

btw, tvheadend says:
satips: SAT>IP server announces an empty tuner list to a client 192.168.140.71 (missing tuner settings - global config)

Barracuda09, 2015-12-11:
Hi,

What does http://ip.of.your.box:8875/desc.xml say?

maxm87, 2015-12-11:
<root configId="0"><specVersion><major>1</major><minor>1</minor></specVersion><device><deviceType>urn:ses-com:device:SatIPServer:1</deviceType><friendlyName>SatPI Server</friendlyName><manufacturer>Marc Postema</manufacturer><manufacturerURL>https://github.com/Barracuda09/SATPI</manufacturerURL><modelDescription>SatPI is a SAT>IP Server for Linux</modelDescription><modelName>SatPI Server</modelName>
<modelNumber>1.5.108~g22b1f17</modelNumber>
<modelURL>https://github.com/Barracuda09/SATPI</modelURL><serialNumber>1S81A31231000007</serialNumber>
<UDN>uuid:50c958a8-e839-4b96-b7ae-b827eb890a05</UDN>
<iconList><icon><mimetype>image/png</mimetype><width>48</width><height>48</height><depth>24</depth><url>/icons/sm.png</url></icon><icon><mimetype>image/png</mimetype><width>120</width><height>120</height><depth>24</depth><url>/icons/lr.png</url></icon><icon><mimetype>image/jpeg</mimetype><width>48</width><height>48</height><depth>24</depth><url>/icons/sm.jpg</url></icon><icon><mimetype>image/jpeg</mimetype><width>120</width><height>120</height><depth>24</depth><url>/icons/lr.jpg</url></icon></iconList><presentationURL>/</presentationURL>
<satip:X_SATIPCAP>DVBS2-0,DVBT-1,DVBT2-1,DVBC-1</satip:X_SATIPCAP>

<satip:X_SATIPM3U>channellist.m3u</satip:X_SATIPM3U></
device></root>

maxm87, 2015-12-11:
[desc.txt](https://github.com/Barracuda09/SATPI/files/59893/desc.txt)

Barracuda09, 2015-12-11:
it is not empty... what do you see it tvheadend tab  'Tv adapters'

maxm87, 2015-12-11:
[image]

you should know, tvheadend and satpi are on the same machine.. the adapter is the native one attached to the rpi

Barracuda09, 2015-12-11:
You disabled this 'direct' adapter and then restart SatPI

maxm87, 2015-12-11:
okay, i disabled the adapter in tvheadend and restarted SatPI. Tvheadend does not see any new adapters..

Barracuda09, 2015-12-12:
Hi,

Can you try running SatPI on an different machine... If this works for you
I have an idea what it could be.

Thanks for testing.

maxm87, 2015-12-12:
Ok, tvheadend sees now the satpi server and the adapter. i configured the network applied it to the adapter and scanned the network. Scanning failed.

Tvheadend log:

mpegts: 506MHz in dvbt - tuning on SAT>IP DVB-T Tuner #1 (192.168.140.71)

subscription: 0011: "scan" subscribing to mux "506MHz", weight: 5, adapter: "SAT>IP DVB-T Tuner #1 (192.168.140.71)", network: "dvbt", service: "Raw PID Subscription"

mpegts: 506MHz in dvbt - scan no data, failed

subscription: 0011: "scan" unsubscribing

satpi log:

327  Sat Dec 12 17:23:36.0266 2015  6  Stream: 0, Waiting on lock...
328  Sat Dec 12 17:23:36.0444 2015  6  Stream: 0, Not locked yet   (FE status 0x0)...
329  Sat Dec 12 17:23:36.0958 2015  6  Stream: 0, Not locked yet   (FE status 0x0)...
330  Sat Dec 12 17:23:36.1474 2015  6  Stream: 0, Not locked yet   (FE status 0x0)...
331  Sat Dec 12 17:23:36.1988 2015  6  Stream: 0, Not locked yet   (FE status 0x0)...
332  Sat Dec 12 17:23:39.1138 2015  6  Found StreamID 0 - SessionID 0205744672
333  Sat Dec 12 17:23:39.1145 2015  6  Stream: 0, StreamClient[0] with SessionID 0205744672
334  Sat Dec 12 17:23:39.1151 2015  6  Stream: 0, Teardown StreamClient[0] with SessionID 0205744672
335  Sat Dec 12 17:23:49.3247 2015  6  Stream: 0, Stop RTCP stream
336  Sat Dec 12 17:23:49.3361 2015  6  Stream: 0, Destroy RTP stream
337  Sat Dec 12 17:23:49.3365 2015  3  DMX_STOP: Bad file descriptor (code 9)
338  Sat Dec 12 17:23:49.3366 2015  3  DMX_STOP: Bad file descriptor (code 9)
339  Sat Dec 12 17:23:49.3370 2015  3  DMX_STOP: Bad file descriptor (code 9)
340  Sat Dec 12 17:23:49.3371 2015  3  DMX_STOP: Bad file descriptor (code 9)
341  Sat Dec 12 17:23:49.3458 2015  6  Found SessionID 0205744672 by SocketClient
342  Sat Dec 12 17:23:49.3464 2015  6  Found StreamID x - SessionID: 0205744672
343  Sat Dec 12 17:23:49.3467 2015  6  Stream: 0, StreamClient[0] with SessionID 0205744672
344  Sat Dec 12 17:23:49.3468 2015  6  Stream: 0, Close StreamClient[0] with SessionID 0205744672
345  Sat Dec 12 17:24:08.8343 2015  6  Found message outside session
346  Sat Dec 12 17:24:08.8350 2015  6  Found message outside session
347  Sat Dec 12 17:24:08.8353 2015  6  Found message outside session
348  Sat Dec 12 17:24:08.8355 2015  6  Found message outside session
349  Sat Dec 12 17:24:08.8357 2015  6  Found message outside session

Barracuda09, 2015-12-12:
Could you make debug version.. and give the full log?

Thanks

maxm87, 2015-12-12:
here it is.

[new  2.txt](https://github.com/Barracuda09/SATPI/files/60209/new.2.txt)

Barracuda09, 2015-12-12:
Are the tuning parameters Ok?

maxm87, 2015-12-12:
i dont know. how can i check?

Barracuda09, 2015-12-12:
Well.. what channel do you expect to receive..

maxm87, 2015-12-12:
I'm scanning muxes for services from 506 till 778 MHz for DVB-T in Germany (Berlin) with tvheadend.

Barracuda09, 2015-12-12:
You don't see any services appear? they fail all?

maxm87, 2015-12-12:
Yes, they fail all. SatPI says ath the same time:

src/Frontend.cpp:399] Stream: 0, Start tuning process...
104  Sat Dec 12 19:11:33.7861 2015  6  [            src/Frontend.cpp:489] Stream: 0, Waiting on lock...
105  Sat Dec 12 19:11:33.8019 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
106  Sat Dec 12 19:11:33.8531 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
107  Sat Dec 12 19:11:33.9046 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
108  Sat Dec 12 19:11:33.9559 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...

Barracuda09, 2015-12-12:
Yes.. but that is only one frequency that failed to tune..
Maybe because of wrong or not used frequency or wrong modulation type etc.

maxm87, 2015-12-12:
i get the above lines several times..  tvh scans 9 muxes. all failed

Barracuda09, 2015-12-12:
Which pre-defined muxes did you use?

maxm87, 2015-12-12:
Germany-de-Berlin

Barracuda09, 2015-12-12:
You don't know any frequency that should work?

maxm87, 2015-12-12:
i've got not much idea on how to find out which channel has which frequency. i can view all my channels with the adapter directly attached to tvheadend.. here i can watch prosieben at 658mhz

Barracuda09, 2015-12-12:
Ok.. You still use it on the same machine

maxm87, 2015-12-12:
No, satpi runs at raspberry and tvheadend on syno nas

Barracuda09, 2015-12-12:
The antenna cable is the same also then?

maxm87, 2015-12-12:
my test environment is following:

for testing satpi i'm using satpi on a rpi2 and a tvheadend on a syno nas. dvbt adapter is attached to rpi. on the rpi is a tvheadend installed but disabled.

when im not testing im activating tvheadend on rpi and i can watch tv. satpi is disabled on rpi.

i've got one dvb adapter with one cable to one antenna.

Barracuda09, 2015-12-12:
TvHeadend Disabled means not running?

maxm87, 2015-12-12:
correct

Barracuda09, 2015-12-12:
Do you know VLC

maxm87, 2015-12-12:
yes

Barracuda09, 2015-12-12:
Can you try opening an network stream with vlc like this:
http://192.168.140.71:8875/?fe=1&freq=658&bw=8&msys=dvbt&mtype=16qam&tmode=8k&gi=18

Barracuda09, 2015-12-12:
For pids use this:
http://192.168.140.71:8875/?fe=1&freq=658&bw=8&msys=dvbt&mtype=16qam&tmode=8k&gi=18&pids=all

maxm87, 2015-12-12:
Port 8875 is correct?

Barracuda09, 2015-12-12:
Should be yes

Barracuda09, 2015-12-12:
for http

maxm87, 2015-12-12:
here is the log output:

25  Sun Dec 13 00:18:23.7830 2015  6  [           src/TcpSocket.cpp:154] HTTP Connection from 192.168.140.62 Port 59029 (fd: 8)
26  Sun Dec 13 00:18:23.7855 2015  7  [         src/HttpcServer.cpp:143] HTTP Stream data from client 192.168.140.62: GET /?fe=1&freq=658&bw=8&msys=dvbt&mtype=16qam&tmode=8k&gi=18 HTTP/1.1
27  Sun Dec 13 00:18:23.7855 2015  7  Host: 192.168.140.71:8875
28  Sun Dec 13 00:18:23.7855 2015  7  User-Agent: Mozilla/5.0 (Windows NT 10.0; WOW64; rv:42.0) Gecko/20100101 Firefox/42.0
29  Sun Dec 13 00:18:23.7855 2015  7  Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
30  Sun Dec 13 00:18:23.7855 2015  7  Accept-Language: de,en-US;q=0.7,en;q=0.3
31  Sun Dec 13 00:18:23.7855 2015  7  Accept-Encoding: gzip, deflate
32  Sun Dec 13 00:18:23.7863 2015  6  [             src/Streams.cpp:280] Found StreamID 0 - SessionID 1702957685
33  Sun Dec 13 00:18:23.7865 2015  6  [              src/Stream.cpp:079] Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbt
34  Sun Dec 13 00:18:23.7866 2015  7  [              src/Stream.cpp:291] Stream: 0, Parsing transport parameters...
35  Sun Dec 13 00:18:23.7884 2015  7  [              src/Stream.cpp:298] Stream: 0, New frequency requested, clearing channel data...
36  Sun Dec 13 00:18:23.7900 2015  7  [              src/Stream.cpp:473] Stream: 0, Parsing transport parameters (Finished)
37  Sun Dec 13 00:18:23.7903 2015  7  [        src/StreamClient.cpp:046] Connection can close: 1
38  Sun Dec 13 00:18:23.7908 2015  7  [            src/Frontend.cpp:447] Stream: 0, Updating frontend...
39  Sun Dec 13 00:18:23.7912 2015  6  [            src/Frontend.cpp:478] Stream: 0, Opened /dev/dvb/adapter0/frontend0 fd: 9
40  Sun Dec 13 00:18:23.7913 2015  7  [            src/Frontend.cpp:399] Stream: 0, Start tuning process...
41  Sun Dec 13 00:18:24.4331 2015  6  [            src/Frontend.cpp:489] Stream: 0, Waiting on lock...
42  Sun Dec 13 00:18:24.4477 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
43  Sun Dec 13 00:18:24.4991 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
44  Sun Dec 13 00:18:24.5500 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
45  Sun Dec 13 00:18:24.6015 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
46  Sun Dec 13 00:18:24.7018 2015  7  [            src/Frontend.cpp:399] Stream: 0, Start tuning process...
47  Sun Dec 13 00:18:24.7019 2015  6  [            src/Frontend.cpp:489] Stream: 0, Waiting on lock...
48  Sun Dec 13 00:18:24.7170 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
49  Sun Dec 13 00:18:24.7679 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
50  Sun Dec 13 00:18:24.8193 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
51  Sun Dec 13 00:18:24.8703 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
52  Sun Dec 13 00:18:24.9705 2015  7  [            src/Frontend.cpp:399] Stream: 0, Start tuning process...
53  Sun Dec 13 00:18:24.9707 2015  6  [            src/Frontend.cpp:489] Stream: 0, Waiting on lock...
54  Sun Dec 13 00:18:24.9849 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
55  Sun Dec 13 00:18:25.0359 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
56  Sun Dec 13 00:18:25.0868 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
57  Sun Dec 13 00:18:25.1383 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
58  Sun Dec 13 00:18:25.2385 2015  7  [            src/Frontend.cpp:399] Stream: 0, Start tuning process...
59  Sun Dec 13 00:18:25.2387 2015  6  [            src/Frontend.cpp:489] Stream: 0, Waiting on lock...
60  Sun Dec 13 00:18:25.2533 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
61  Sun Dec 13 00:18:25.3042 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
62  Sun Dec 13 00:18:25.3557 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
63  Sun Dec 13 00:18:25.4066 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
64  Sun Dec 13 00:18:25.5069 2015  7  [         src/HttpcServer.cpp:177] HTTP/1.1 200 OK
65  Sun Dec 13 00:18:25.5069 2015  7  Server: SatPI WebServer v0.1
66  Sun Dec 13 00:18:25.5069 2015  7  Location: 
67  Sun Dec 13 00:18:25.5069 2015  7  CSeq: 0
68  Sun Dec 13 00:18:25.5069 2015  7  cache-control: no-cache
69  Sun Dec 13 00:18:25.5069 2015  7  Content-Type: video/MP2T
70  Sun Dec 13 00:18:25.5069 2015  7  --- LINE END ---

maxm87, 2015-12-12:
with vlc:

153  Sun Dec 13 00:23:19.2604 2015  7  [         src/HttpcServer.cpp:143] HTTP Stream data from client 192.168.140.62: GET /?fe=1&freq=658&bw=8&msys=dvbt&mtype=16qam&tmode=8k&gi=18&pids=all HTTP/1.1
154  Sun Dec 13 00:23:19.2604 2015  7  Host: 192.168.140.71:8875
155  Sun Dec 13 00:23:19.2604 2015  7  User-Agent: VLC/2.2.1 LibVLC/2.2.1
156  Sun Dec 13 00:23:19.2604 2015  7  Range: bytes=0-
157  Sun Dec 13 00:23:19.2604 2015  7  Connection: close
158  Sun Dec 13 00:23:19.2604 2015  7  Icy-MetaData: 1
159  Sun Dec 13 00:23:19.2604 2015  7  --- LINE END ---
160  Sun Dec 13 00:23:19.2635 2015  6  [             src/Streams.cpp:280] Found StreamID 0 - SessionID 0288582906
161  Sun Dec 13 00:23:19.2640 2015  6  [              src/Stream.cpp:079] Stream: 0, StreamClient[1] with SessionID 0288582906 for dvbt
162  Sun Dec 13 00:23:19.2644 2015  7  [              src/Stream.cpp:291] Stream: 0, Parsing transport parameters...
163  Sun Dec 13 00:23:19.2673 2015  7  [              src/Stream.cpp:298] Stream: 0, New frequency requested, clearing channel data...
164  Sun Dec 13 00:23:19.2753 2015  7  [              src/Stream.cpp:473] Stream: 0, Parsing transport parameters (Finished)
165  Sun Dec 13 00:23:19.2761 2015  7  [        src/StreamClient.cpp:046] Connection can close: 1
166  Sun Dec 13 00:23:19.2769 2015  7  [            src/Frontend.cpp:447] Stream: 0, Updating frontend...
167  Sun Dec 13 00:23:19.2779 2015  6  [            src/Frontend.cpp:478] Stream: 0, Opened /dev/dvb/adapter0/frontend0 fd: 15
168  Sun Dec 13 00:23:19.2780 2015  7  [            src/Frontend.cpp:399] Stream: 0, Start tuning process...
169  Sun Dec 13 00:23:19.2853 2015  6  [            src/Frontend.cpp:489] Stream: 0, Waiting on lock...
170  Sun Dec 13 00:23:19.2854 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
171  Sun Dec 13 00:23:19.3365 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
172  Sun Dec 13 00:23:19.3876 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
173  Sun Dec 13 00:23:19.4388 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
174  Sun Dec 13 00:23:19.5397 2015  7  [            src/Frontend.cpp:399] Stream: 0, Start tuning process...
175  Sun Dec 13 00:23:19.5404 2015  6  [            src/Frontend.cpp:489] Stream: 0, Waiting on lock...
176  Sun Dec 13 00:23:19.5409 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
177  Sun Dec 13 00:23:19.5923 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
178  Sun Dec 13 00:23:19.6438 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
179  Sun Dec 13 00:23:19.6954 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
180  Sun Dec 13 00:23:19.7961 2015  7  [            src/Frontend.cpp:399] Stream: 0, Start tuning process...
181  Sun Dec 13 00:23:19.7968 2015  6  [            src/Frontend.cpp:489] Stream: 0, Waiting on lock...
182  Sun Dec 13 00:23:19.8147 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
183  Sun Dec 13 00:23:19.8662 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
184  Sun Dec 13 00:23:19.9177 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
185  Sun Dec 13 00:23:19.9692 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
186  Sun Dec 13 00:23:20.0700 2015  7  [            src/Frontend.cpp:399] Stream: 0, Start tuning process...
187  Sun Dec 13 00:23:20.0707 2015  6  [            src/Frontend.cpp:489] Stream: 0, Waiting on lock...
188  Sun Dec 13 00:23:20.0882 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
189  Sun Dec 13 00:23:20.1397 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
190  Sun Dec 13 00:23:20.1911 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
191  Sun Dec 13 00:23:20.2426 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
192  Sun Dec 13 00:23:20.3433 2015  7  [         src/HttpcServer.cpp:177] HTTP/1.1 200 OK
193  Sun Dec 13 00:23:20.3433 2015  7  Server: SatPI WebServer v0.1
194  Sun Dec 13 00:23:20.3433 2015  7  Location: 
195  Sun Dec 13 00:23:20.3433 2015  7  CSeq: 0
196  Sun Dec 13 00:23:20.3433 2015  7  cache-control: no-cache
197  Sun Dec 13 00:23:20.3433 2015  7  Content-Type: video/MP2T
198  Sun Dec 13 00:23:20.3433 2015  7  --- LINE END ---

Barracuda09, 2015-12-12:
what does dmesg say

maxm87, 2015-12-12:
looks good.

[dmesg.txt](https://github.com/Barracuda09/SATPI/files/60293/dmesg.txt)

Barracuda09, 2015-12-12:
Could you try this (removed mtype)
http://192.168.140.71:8875/?fe=1&freq=658&bw=8&msys=dvbt&tmode=8k&gi=18&pids=all

maxm87, 2015-12-12:
nope

25  Sun Dec 13 00:49:54.1194 2015  6  [           src/TcpSocket.cpp:154] HTTP Connection from 192.168.140.62 Port 61816 (fd: 8)
26  Sun Dec 13 00:49:54.1279 2015  7  [         src/HttpcServer.cpp:143] HTTP Stream data from client 192.168.140.62: GET /?fe=1&freq=658&bw=8&msys=dvbt&tmode=8k&gi=18&pids=all HTTP/1.1
27  Sun Dec 13 00:49:54.1279 2015  7  Host: 192.168.140.71:8875
28  Sun Dec 13 00:49:54.1279 2015  7  User-Agent: VLC/2.2.1 LibVLC/2.2.1
29  Sun Dec 13 00:49:54.1279 2015  7  Range: bytes=0-
30  Sun Dec 13 00:49:54.1279 2015  7  Connection: close
31  Sun Dec 13 00:49:54.1279 2015  7  Icy-MetaData: 1
32  Sun Dec 13 00:49:54.1279 2015  7  --- LINE END ---
33  Sun Dec 13 00:49:54.1287 2015  6  [             src/Streams.cpp:280] Found StreamID 0 - SessionID 1702957685
34  Sun Dec 13 00:49:54.1289 2015  6  [              src/Stream.cpp:079] Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbt
35  Sun Dec 13 00:49:54.1290 2015  7  [              src/Stream.cpp:291] Stream: 0, Parsing transport parameters...
36  Sun Dec 13 00:49:54.1315 2015  7  [              src/Stream.cpp:298] Stream: 0, New frequency requested, clearing channel data...
37  Sun Dec 13 00:49:54.1391 2015  7  [              src/Stream.cpp:473] Stream: 0, Parsing transport parameters (Finished)
38  Sun Dec 13 00:49:54.1394 2015  7  [        src/StreamClient.cpp:046] Connection can close: 1
39  Sun Dec 13 00:49:54.1401 2015  7  [            src/Frontend.cpp:447] Stream: 0, Updating frontend...
40  Sun Dec 13 00:49:54.1408 2015  6  [            src/Frontend.cpp:478] Stream: 0, Opened /dev/dvb/adapter0/frontend0 fd: 9
41  Sun Dec 13 00:49:54.1411 2015  7  [            src/Frontend.cpp:399] Stream: 0, Start tuning process...
42  Sun Dec 13 00:49:54.7905 2015  6  [            src/Frontend.cpp:489] Stream: 0, Waiting on lock...
43  Sun Dec 13 00:49:54.7907 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
44  Sun Dec 13 00:49:54.8421 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
45  Sun Dec 13 00:49:54.8931 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
46  Sun Dec 13 00:49:54.9441 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
47  Sun Dec 13 00:49:55.0444 2015  7  [            src/Frontend.cpp:399] Stream: 0, Start tuning process...
48  Sun Dec 13 00:49:55.0446 2015  6  [            src/Frontend.cpp:489] Stream: 0, Waiting on lock...
49  Sun Dec 13 00:49:55.0447 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
50  Sun Dec 13 00:49:55.0961 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
51  Sun Dec 13 00:49:55.1471 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
52  Sun Dec 13 00:49:55.1981 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
53  Sun Dec 13 00:49:55.2984 2015  7  [            src/Frontend.cpp:399] Stream: 0, Start tuning process...
54  Sun Dec 13 00:49:55.2985 2015  6  [            src/Frontend.cpp:489] Stream: 0, Waiting on lock...
55  Sun Dec 13 00:49:55.2986 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
56  Sun Dec 13 00:49:55.3495 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
57  Sun Dec 13 00:49:55.4005 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
58  Sun Dec 13 00:49:55.4515 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
59  Sun Dec 13 00:49:55.5518 2015  7  [            src/Frontend.cpp:399] Stream: 0, Start tuning process...
60  Sun Dec 13 00:49:55.5520 2015  6  [            src/Frontend.cpp:489] Stream: 0, Waiting on lock...
61  Sun Dec 13 00:49:55.5520 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
62  Sun Dec 13 00:49:55.6030 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
63  Sun Dec 13 00:49:55.6540 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
64  Sun Dec 13 00:49:55.7049 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
65  Sun Dec 13 00:49:55.8052 2015  7  [         src/HttpcServer.cpp:177] HTTP/1.1 200 OK
66  Sun Dec 13 00:49:55.8052 2015  7  Server: SatPI WebServer v0.1
67  Sun Dec 13 00:49:55.8052 2015  7  Location: 
68  Sun Dec 13 00:49:55.8052 2015  7  CSeq: 0
69  Sun Dec 13 00:49:55.8052 2015  7  cache-control: no-cache
70  Sun Dec 13 00:49:55.8052 2015  7  Content-Type: video/MP2T
71  Sun Dec 13 00:49:55.8052 2015  7  --- LINE END ---
72  Sun Dec 13 00:50:05.6753 2015  7  [            src/SocketAttr.h:069] Set Timeout (fd: 10)
73  Sun Dec 13 00:50:05.6756 2015  6  [           src/TcpSocket.cpp:154] HTTP Connection from 192.168.140.62 Port 61818 (fd: 10)
74  Sun Dec 13 00:50:05.7669 2015  7  [            src/SocketAttr.h:069] Set Timeout (fd: 11)
75  Sun Dec 13 00:50:05.7672 2015  6  [           src/TcpSocket.cpp:154] HTTP Connection from 192.168.140.62 Port 61819 (fd: 11)
76  Sun Dec 13 00:50:05.7674 2015  7  [            src/SocketAttr.h:069] Set Timeout (fd: 12)
77  Sun Dec 13 00:50:05.7675 2015  6  [           src/TcpSocket.cpp:154] HTTP Connection from 192.168.140.62 Port 61820 (fd: 12)
[... cut ...]
79  Sun Dec 13 00:50:05.7788 2015  6  [           src/TcpSocket.cpp:154] HTTP Connection from 192.168.140.62 Port 61821 (fd: 13)
80  Sun Dec 13 00:50:05.7807 2015  7  [            src/SocketAttr.h:069] Set Timeout (fd: 14)
81  Sun Dec 13 00:50:05.7808 2015  6  [           src/TcpSocket.cpp:154] HTTP Connection from 192.168.140.62 Port 61822 (fd: 14)
82  Sun Dec 13 00:50:05.7829 2015  7  [            src/SocketAttr.h:069] Set Timeout (fd: 15)
83  Sun Dec 13 00:50:05.7830 2015  6  [           src/TcpSocket.cpp:154] HTTP Connection from 192.168.140.62 Port 61823 (fd: 15)

Barracuda09, 2015-12-12:
This this added fec

http://192.168.140.71:8875/?fe=1&freq=658&bw=8&msys=dvbt&fec=23&mtype=16qam&tmode=8k&gi=18&pids=all

maxm87, 2015-12-12:
no, sorry:

32  Sun Dec 13 00:59:22.3372 2015  6  [           src/TcpSocket.cpp:154] HTTP Connection from 192.168.140.62 Port 61977 (fd: 14)
33  Sun Dec 13 00:59:22.3491 2015  7  [         src/HttpcServer.cpp:143] HTTP Stream data from client 192.168.140.62: GET /?fe=1&freq=658&bw=8&msys=dvbt&fec=23&mtype=16qam&tmode=8k&gi=18&pids=all HTTP/1.1
34  Sun Dec 13 00:59:22.3491 2015  7  Host: 192.168.140.71:8875
35  Sun Dec 13 00:59:22.3491 2015  7  User-Agent: VLC/2.2.1 LibVLC/2.2.1
36  Sun Dec 13 00:59:22.3491 2015  7  Range: bytes=0-
37  Sun Dec 13 00:59:22.3491 2015  7  Connection: close
38  Sun Dec 13 00:59:22.3491 2015  7  Icy-MetaData: 1
39  Sun Dec 13 00:59:22.3491 2015  7  --- LINE END ---
40  Sun Dec 13 00:59:22.3499 2015  6  [             src/Streams.cpp:280] Found StreamID 0 - SessionID 1702957685
41  Sun Dec 13 00:59:22.3501 2015  6  [              src/Stream.cpp:079] Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbt
42  Sun Dec 13 00:59:22.3502 2015  7  [              src/Stream.cpp:291] Stream: 0, Parsing transport parameters...
43  Sun Dec 13 00:59:22.3521 2015  7  [              src/Stream.cpp:298] Stream: 0, New frequency requested, clearing channel data...
44  Sun Dec 13 00:59:22.3597 2015  7  [              src/Stream.cpp:473] Stream: 0, Parsing transport parameters (Finished)
45  Sun Dec 13 00:59:22.3600 2015  7  [        src/StreamClient.cpp:046] Connection can close: 1
46  Sun Dec 13 00:59:22.3607 2015  7  [            src/Frontend.cpp:447] Stream: 0, Updating frontend...
47  Sun Dec 13 00:59:22.3612 2015  6  [            src/Frontend.cpp:478] Stream: 0, Opened /dev/dvb/adapter0/frontend0 fd: 15
48  Sun Dec 13 00:59:22.3614 2015  7  [            src/Frontend.cpp:399] Stream: 0, Start tuning process...
49  Sun Dec 13 00:59:23.0228 2015  6  [            src/Frontend.cpp:489] Stream: 0, Waiting on lock...
50  Sun Dec 13 00:59:23.0230 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
51  Sun Dec 13 00:59:23.0740 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
52  Sun Dec 13 00:59:23.1251 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
53  Sun Dec 13 00:59:23.1766 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
54  Sun Dec 13 00:59:23.2769 2015  7  [            src/Frontend.cpp:399] Stream: 0, Start tuning process...
55  Sun Dec 13 00:59:23.2772 2015  6  [            src/Frontend.cpp:489] Stream: 0, Waiting on lock...
56  Sun Dec 13 00:59:23.2772 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
57  Sun Dec 13 00:59:23.3282 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
58  Sun Dec 13 00:59:23.3797 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
59  Sun Dec 13 00:59:23.4312 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
60  Sun Dec 13 00:59:23.5315 2015  7  [            src/Frontend.cpp:399] Stream: 0, Start tuning process...
61  Sun Dec 13 00:59:23.5318 2015  6  [            src/Frontend.cpp:489] Stream: 0, Waiting on lock...
62  Sun Dec 13 00:59:23.5318 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
63  Sun Dec 13 00:59:23.5828 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
64  Sun Dec 13 00:59:23.6339 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
65  Sun Dec 13 00:59:23.6850 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
66  Sun Dec 13 00:59:23.7854 2015  7  [            src/Frontend.cpp:399] Stream: 0, Start tuning process...
67  Sun Dec 13 00:59:23.7857 2015  6  [            src/Frontend.cpp:489] Stream: 0, Waiting on lock...
68  Sun Dec 13 00:59:23.8004 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
69  Sun Dec 13 00:59:23.8519 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
70  Sun Dec 13 00:59:23.9029 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
71  Sun Dec 13 00:59:23.9540 2015  6  [            src/Frontend.cpp:503] Stream: 0, Not locked yet   (FE status 0x0)...
72  Sun Dec 13 00:59:24.0544 2015  7  [         src/HttpcServer.cpp:177] HTTP/1.1 200 OK
73  Sun Dec 13 00:59:24.0544 2015  7  Server: SatPI WebServer v0.1
74  Sun Dec 13 00:59:24.0544 2015  7  Location: 
75  Sun Dec 13 00:59:24.0544 2015  7  CSeq: 0
76  Sun Dec 13 00:59:24.0544 2015  7  cache-control: no-cache
77  Sun Dec 13 00:59:24.0544 2015  7  Content-Type: video/MP2T
78  Sun Dec 13 00:59:24.0544 2015  7  --- LINE END ---

Barracuda09, 2015-12-13:
Strange and interesting..... I have the same chipset in my dvb-t stick and here it works ok..

maxm87, 2015-12-13:
raspberry pi too?

Barracuda09, 2015-12-13:
Yes.. pi2 with 2 dvb-s2 and 1 dvb-t stick

maxm87, 2015-12-13:
with rasbian on it?

Barracuda09, 2015-12-13:
yes on older version:
Linux raspberrypi 3.18.9-v7+ #772 SMP PREEMPT Sun Mar 22 15:19:51 GMT 2015 armv7l GNU/Linux

maxm87, 2015-12-13:
a output of w_scan:

pi@raspberrypi ~ $ sudo w_scan -f t -c DE >> /home/pi/wscan.txt
w_scan version 20120605 (compiled for DVB API 5.4)
using settings for GERMANY
DVB aerial
DVB-T Europe
scan type TERRESTRIAL, channellist 4
output format vdr-1.6
WARNING: could not guess your codepage. Falling back to 'UTF-8'
output charset 'UTF-8', use -C <charset> to override
Info: using DVB adapter auto detection.
        /dev/dvb/adapter0/frontend0 -> TERRESTRIAL "Silicon Labs Si2168": very good :-))

Using TERRESTRIAL frontend (adapter /dev/dvb/adapter0/frontend0)
-_-_-_-_ Getting frontend capabilities-_-_-_-_
Using DVB API 5.a
frontend 'Silicon Labs Si2168' supports
DVB-T2
INVERSION_AUTO
QAM_AUTO
TRANSMISSION_MODE_AUTO
GUARD_INTERVAL_AUTO
HIERARCHY_AUTO
FEC_AUTO
FREQ (55.00MHz ... 862.00MHz)
-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_
Scanning 7MHz frequencies...
177500: (time: 00:00)
184500: (time: 00:02)
191500: (time: 00:05)
198500: (time: 00:07)
205500: (time: 00:10)
212500: (time: 00:12)
219500: (time: 00:15)
226500: (time: 00:17)
Scanning 8MHz frequencies...
474000: (time: 00:20)
482000: (time: 00:22)
490000: (time: 00:25)
498000: (time: 00:27)
506000: (time: 00:30) (time: 00:31) signal ok:
        QAM_AUTO f = 506000 kHz I999B8C999D999T999G999Y999
        updating transponder:
           (QAM_AUTO f = 506000 kHz I999B8C999D999T999G999Y999) 0x0000
        to (QAM_16   f = 506000 kHz I999B8C23D0T8G8Y0) 0x405A
514000: (time: 00:44)
522000: (time: 00:46) (time: 00:47) signal ok:
        QAM_AUTO f = 522000 kHz I999B8C999D999T999G999Y999
        new transponder:
           (QAM_16   f = 514000 kHz I999B8C23D0T8G4Y0) 0x415A
        new transponder:
           (QAM_64   f = 474000 kHz I999B8C23D0T8G4Y0) 0x415A
^C
ERROR: interrupted by SIGINT, dumping partial result...
dumping lists (0 services)
Done.

Barracuda09, 2015-12-25:
Hi,

I have Increase the tuning 'timeout' Hope this solves you issues.

Thanks for testing and regards,

Marc

Barracuda09, 2016-03-24:
I will close this issue for now, If someone is having problems please make an new issue.

closed 2016-03-24 by Barracuda09