Barracuda09/SATPI #40: Not tunning DVB-S transponders with vtuner-satip and DVB-S2 source

lars18th, 2015-12-28:
Hi,

At time I use the "vtuner-satip" driver for manage a remote (and buggy) SAT>IP DVB-S2 server (one tuner only, ALi implementation). But when I try to tune a DVB-S transponder the tunning fails.

See the log:

```
0  Mon Dec 28 04:52:19.1183 2015  6  --- starting SatPI version: 1.5.0~unknown ---
1  Mon Dec 28 04:52:19.1187 2015  6  Number of processors online: 2
2  Mon Dec 28 04:52:19.1188 2015  6  Default network buffer size: 56 KBytes
3  Mon Dec 28 04:52:19.1190 2015  6  eth0: 192.168.210.17 [00:0c:29:36:4c:1b]
4  Mon Dec 28 04:52:19.1191 2015  6  Detecting frontends in: /dev/dvb
5  Mon Dec 28 04:52:19.1241 2015  6  Frontends found: 3
6  Mon Dec 28 04:52:19.1244 2015  6  Frontend Name: Afatech AF9013 DVB-T
7  Mon Dec 28 04:52:19.1245 2015  6  Frontend Name: Afatech AF9013 DVB-T
8  Mon Dec 28 04:52:19.1246 2015  6  Frontend Name: vTuner proxyFE DVB-S2
9  Mon Dec 28 04:52:19.1251 2015  6  Setting up RTSP server
10  Mon Dec 28 04:52:19.1254 2015  6  Setting up HTTP server
11  Mon Dec 28 04:52:19.1258 2015  6  Setting up SSDP server with BOOTID: 3
12  Mon Dec 28 04:52:22.2584 2015  6  RTSP Connection from 192.168.1.31 Port 49888 (fd: 8)
13  Mon Dec 28 04:52:22.2731 2015  6  Found message outside session
14  Mon Dec 28 04:52:22.2753 2015  6  Found StreamID x - SessionID x - Creating new SessionID: 1702957685
15  Mon Dec 28 04:52:22.2754 2015  6  Stream: 0, Not capable of handling msys=dvbs
16  Mon Dec 28 04:52:22.2755 2015  6  Stream: 1, Not capable of handling msys=dvbs
17  Mon Dec 28 04:52:22.2755 2015  6  Stream: 2, Not capable of handling msys=dvbs
18  Mon Dec 28 04:52:22.2755 2015  3  Found no Stream/Client of interest!
```
For sure, the same driver works with other tools for DVB-S/S2.

Please, can you fix this issue?

lars18th, 2015-12-28:
More info,

In the "frontend.html" status page I see "Channel Info" with "Delivery System" = "UNKNOWN DELSYS". It's this normal?

Barracuda09, 2015-12-28:
I see the problem. vTuner only reports that it is an DVB-S2 and you try DVB-S

Barracuda09, 2015-12-28:
I'm not sure if DVB-S2 can always do DVB-S as well?

lars18th, 2015-12-28:
Hi @Barracuda09 ,

Yes, DVB-S2 don't implies DVB-S.
But with DVB-API < 5.5 you only have ONE tunning space. Then with API 5.4, you can receive only DVB-S2 or DVB-S as supported. And this case, you can assume that DVB-S2 includes DVB-S.

So, please, only make the change when using API 5.1-5.4... in this case, and only in this, if you like to tune DVB-S on a DVB-S2 device try it. Only if the tunning fails, then send the error.

You agree?

Barracuda09, 2015-12-28:
Ok I can agree on this, andI will look into this.

Thanks for your input!

lars18th, 2015-12-28:
Perfect! I feel it can be easy to support, only some ifs when API isn't 5.5.

Thank you for your effort! :)

Barracuda09, 2015-12-29:
Hi,

I have changed the handling of the delivery system for API older the 5.4.
It assumes now that DVB-S2/T2 can handle S/T too

Hope this fix you issue. Thanks for reporting this (again).

Regards,

Marc

lars18th, 2015-12-30:
Hi @Barracuda09 ,

Checked, tested and working. ;-)
Thank you!

Barracuda09, 2015-12-30:
Ok will do so.. Thanks for testing!

closed 2015-12-30 by Barracuda09