Barracuda09/SATPI #41: HTTP Streaming support with multiple pids call

lars18th, 2015-12-28:
Hi,

Related to the #33 enhancement, I like to comment that it's working great. Thank you!

However, I have one request: better suppot for multiple HTTP calls. I explain this "non-standard" request mode:

- I have a client that receives the TS stream over HTTP. It requests the stream using the standarized SAT>IP HTTP transport. It does requests like:

```
GET /?msys=dvbs2&freq=11377&pol=v&sr=22000&pids=0,16,17,18,610,620 HTTP/1.1
```

- Also, this client hast the capability of CHANGE PIDS on-the-fly over the same HTTP connection. Then in any time when it receives the TS stream over the HTTP socket, it can request a new "GET" command, like:

```
GET /?msys=dvbs2&freq=11377&pol=v&sr=22000&pids=0,16,17,18,610,620,640,810,820,840 HTTP/1.1
```

- In this case the expected behaviour is that the server interprets the new request, it changes the pid filtering, and CONTINUES the streaming over the HTTP socket.

The good news, is that SATPI has some support of this "non-standard" HTTP requests. It processes the new GET requests, but it's not perfect. The main problem is that it sends (as indicated by the HTTP standard) the RESPONSE to the request. Then it sends a header (as writed in the log):

```
src/HttpcServer.cpp:177] HTTP/1.1 200 OK
```

But, as you're streaming a TS you don't like to touch the TS. Then, the problem is the insertion of the HTTP HEADER in each response WHEN STREAMING (the first header is mandatory to start to receive the stream).

Then I suggest to change, or include some option, for disable the subsequent headers in the response. Can be this implemented?
I hope you agree!

Barracuda09, 2015-12-28:
Yes I can agree on this. I will look into this also.

lars18th, 2015-12-28:
Great! This will be very interesting to use SATPI as a full HTTP server... I'll test it broadly when you implement it... for me it's a very powerfull capability.

Thank you for take into account our suggestions!

lars18th, 2019-03-07:
Hi @Barracuda09 ,

Any update on this?