Barracuda09/SATPI #42: Enahnce configuration

lars18th, 2015-12-28:
Hi,

I suggest to increase the "Configure" part of the Web GUI:

- Add "Listening HTTP port" for changing from default "8875" to some other.
- Add the option to select the used (enabled/disabled) Frontends (at time the option is in the page, but no option for "save" the current status)
- Fix "SAVE" button (at time it don't appears in the page).

You agree?

Barracuda09, 2015-12-28:
Hi,

First of all I like to say there is no way at the moment the changes you make will be save to disk.
I think a 'Save' button would be more logical indeed to except values (and save to disk later).

What do you mean by edit satip:X_SATIPCAP

The rest I understand and will look into it.

Thanks for your input!

lars18th, 2015-12-28:
Hi @Barracuda09 ,

> What do you mean by edit satip:X_SATIPCAP
>
> The rest I understand and will look into it.

See my 'desc.xml':

```
<satip:X_SATIPCAP xmlns:satip="urn:ses-com:satip">DVBS2-1,DVBT-2,DVBT2-0,DVBC-0</satip:X_SATIPCAP>
```

In my case I have one DVB-S2 SAT>IP server using vtuner-satip, and two USB DVB-T sources. But, when I "disable" the DVB-T sources, the SATIPCAP don't change. So, I like to tweak it, for instance setting: "only 1 DVB-S2 tuner" for example.

Perhaps you can add the option of manually configure the number of different tuners. You agree?

goerdi, 2016-08-01:
Hi !
It should be a perfect thing that you can at least load a config file..... new config on a restart is hard thing :(

ciao gerd

Barracuda09, 2016-08-02:
Yes, this is in the making that is that the 'Frontend' configs/data will be saved.

The rest is already saved in an xml file.

Barracuda09, 2016-08-27:
I have added saving and reloading of the frontend config data.

The satip:X_SATIPCAP is now adjusted according to the 'enabled' frontends as well.

As an side note saving is done with every change in the page. So no save button at the time.

Barracuda09, 2019-03-15:
Most of this issue is fixed, only the save button in web ui is still not added.

Will make an new issue for this.

closed 2019-03-15 by Barracuda09