Barracuda09/SATPI #45: 100% cpu usage on 2 processes

joolzg, 2016-01-20:
Just built a clean version for the 1st time and did the ./satpi

htop shows me 2 satpis running at 99.9 and 100% cpu usage

Barracuda09, 2016-01-20:
Hi,

A few questions:
- how did you build  SatPI
- what or is it streaming
- what is the log showing
- what system do you use

Thanks for trying!

joolzg, 2016-01-20:
Build using the the Makefile

not even got that far all I did was run it and check the web page was available before going any further.

Its a clean machine with a quad tbs dvb-s/s2 card in

Barracuda09, 2016-01-20:
Make debug or just make?

What is the log say?

het volgende:

joolzg, 2016-01-21:
just did a make and

./satpi

Where is the log

Barracuda09, 2016-01-21:
First of all are you running as root?
Second if you have a problem you should do a make debug it has more info
for me.

The log is in de web page (limited) or in syslog.

Thanks for testing.

Barracuda09, 2016-03-26:
I will close this issue for now. If someone has problems please make a new issue.

closed 2016-03-26 by Barracuda09