Barracuda09/SATPI #46: SCR Support

goerdi, 2016-02-03:
Hi !
is Unicable Support planned ?

ciao Gerd

Barracuda09, 2016-02-03:
Hi,

I would like to implement this.

goerdi, 2016-02-05:
I guess therefore you need a config file..... :)

Barracuda09, 2016-07-01:
Hi @goerdi 

I have done some work on this. You could try it out if you like.
It is still a work in progress but should work at this point (only PIN is not implemented yet)

Regards,

Marc

goerdi, 2016-08-01:
Hi !  i'll try but i'm tired to restore config every time... so i need some command line options o a config file...

Ciao Gerd

Barracuda09, 2016-08-02:
Yes I know it can be annoying at times.
I am working on an solution, but sorry my time limited at the moment.

Thanks for giving this a try.

Marc

goerdi, 2016-08-02:
Hi maybe you store ist in the xml of the Webserver

Ciao Gerd

Gesendet mit AquaMail für Android
http://www.aqua-mail.com

Barracuda09, 2016-08-26:
Hi @goerdi 

The config is saved now and reloaded by startup again. 
So it should be more easy to test it now.

Thanks for giving this a try,
Marc

Barracuda09, 2018-09-12:
I will close this issue for now, as it is fixed for some time.

Please make an new issue if there are problems regarding SCR support.

closed 2018-09-12 by Barracuda09