Barracuda09/SATPI #48: Not working with DVBViewer Pro

bbtimx, 2016-05-12:
Hey

In the Log I see my DVBViewer Pro connects to SATIP but I get no Picture ..
After a while I get an memory access error!

Barracuda09, 2016-05-12:
Hi,

Yes thanks for pointing this out.. I Will fix this.
DId not try this out for some time.

Barracuda09, 2016-05-12:
Please give it a try

bbtimx, 2016-05-13:
Hey,

It´s working fine now. Thanks a lot :)

But the auto search only shows SatIP Server entry after 20s of waiting and the UDP TCP selcet disappear.

[image]

[image]

Barracuda09, 2016-10-08:
Initial report is fixed. Will close this for now.

closed 2016-10-08 by Barracuda09