Barracuda09/SATPI #50: Vu+ SAT>IP client support?

rezna11, 2016-10-08:
Hi, I run SATPI on raspberry, Vu+ finds it but when I try to scan channels on a satellite, nothing happens. Is  scanning function supported by SATPI? Streaming a concrete channel in VLC works, but I need to scan the frequencies first, without that, there is no channel list...
Thanks

Barracuda09, 2016-10-08:
Hi @rezna11 ,

Yes, scanning is supported. Could you give me a log of SatPI when you try to scan?

Thanks,

Marc

Barracuda09, 2016-10-08:
Hi @rezna11 ,

You could please try it again with latest commit?

Thanks for giving it a try,

Marc

rezna11, 2016-10-08:
well that's the problem, there is nothing in the log - the only thing is the discovery process - the Vu+ and Dreambox both find the SATPI server but when I try to tune in channels, nothing new is shown in the SATPI log. This worked with tvheadend, so I'd say the Vu+/Dreambox side is OK

Barracuda09, 2016-10-08:
Could you try new commit

Barracuda09, 2016-10-08:
Did you build a debug version?

rezna11, 2016-10-08:
I did build a new version but not debug. I will try debug.

rezna11, 2016-10-08:
ok my bad - the new build didn't run because I thought "sudo killall -9 satpi" would end satpi but no, it didn't so the ports were blocked... after rebooting raspberry, I was able to scan and stream but there was another problem, after a short while, when trying to scan other transponders, the log went crazy and it was impossible to scan new frequencies... here's part of the log:

`0  Sat Oct  8 18:28:02.0333 2016  7  Location: 
1  Sat Oct  8 18:28:02.0333 2016  7  CSeq: 3529
2  Sat Oct  8 18:28:02.0333 2016  7  cache-control: no-cache
3  Sat Oct  8 18:28:02.0333 2016  7  Content-Type: video/MP2T
4  Sat Oct  8 18:28:02.0333 2016  7  --- LINE END ---
5  Sat Oct  8 18:28:02.0492 2016  7  [                          src/HttpcServer.cpp:141] RTSP Stream data from client 192.168.30.216 Port 52121: SETUP rtsp://192.168.30.247:554/?pids=17 RTSP/1.0
6  Sat Oct  8 18:28:02.0492 2016  7  CSeq: 3530
7  Sat Oct  8 18:28:02.0492 2016  7  Transport: RTP/AVP;unicast;client_port=45079-45080
8  Sat Oct  8 18:28:02.0492 2016  7  --- LINE END ---
9  Sat Oct  8 18:28:02.0499 2016  6  [                        src/StreamManager.cpp:167] Found StreamID x - SessionID x - Creating new SessionID: 2105829603
10  Sat Oct  8 18:28:02.0501 2016  6  [                               src/Stream.cpp:190] Stream: 1, Not capable of handling msys=UNKNOWN DELSYS
11  Sat Oct  8 18:28:02.0503 2016  6  [                               src/Stream.cpp:190] Stream: 2, Not capable of handling msys=UNKNOWN DELSYS
12  Sat Oct  8 18:28:02.0505 2016  3  [                        src/StreamManager.cpp:195] Found no Stream/Client of interest!
13  Sat Oct  8 18:28:02.0506 2016  7  [                          src/HttpcServer.cpp:175] RTSP/1.0 503 Service Unavailable
14  Sat Oct  8 18:28:02.0506 2016  7  Server: SatPI WebServer v0.1
15  Sat Oct  8 18:28:02.0506 2016  7  Location: 
16  Sat Oct  8 18:28:02.0506 2016  7  CSeq: 3530
17  Sat Oct  8 18:28:02.0506 2016  7  cache-control: no-cache
18  Sat Oct  8 18:28:02.0506 2016  7  Content-Type: video/MP2T
19  Sat Oct  8 18:28:02.0506 2016  7  --- LINE END ---
20  Sat Oct  8 18:28:02.0710 2016  7  [                          src/HttpcServer.cpp:141] RTSP Stream data from client 192.168.30.216 Port 52121: SETUP rtsp://192.168.30.247:554/?pids=17 RTSP/1.0
21  Sat Oct  8 18:28:02.0710 2016  7  CSeq: 3531
22  Sat Oct  8 18:28:02.0710 2016  7  Transport: RTP/AVP;unicast;client_port=45079-45080
23  Sat Oct  8 18:28:02.0710 2016  7  --- LINE END ---
24  Sat Oct  8 18:28:02.0717 2016  6  [                        src/StreamManager.cpp:167] Found StreamID x - SessionID x - Creating new SessionID: 1354574427
25  Sat Oct  8 18:28:02.0721 2016  6  [                               src/Stream.cpp:190] Stream: 1, Not capable of handling msys=UNKNOWN DELSYS
26  Sat Oct  8 18:28:02.0723 2016  6  [                               src/Stream.cpp:190] Stream: 2, Not capable of handling msys=UNKNOWN DELSYS
27  Sat Oct  8 18:28:02.0724 2016  3  [                        src/StreamManager.cpp:195] Found no Stream/Client of interest!
28  Sat Oct  8 18:28:02.0726 2016  7  [                          src/HttpcServer.cpp:175] RTSP/1.0 503 Service Unavailable
29  Sat Oct  8 18:28:02.0726 2016  7  Server: SatPI WebServer v0.1
30  Sat Oct  8 18:28:02.0726 2016  7  Location: 
31  Sat Oct  8 18:28:02.0726 2016  7  CSeq: 3531
32  Sat Oct  8 18:28:02.0726 2016  7  cache-control: no-cache
33  Sat Oct  8 18:28:02.0726 2016  7  Content-Type: video/MP2T
34  Sat Oct  8 18:28:02.0726 2016  7  --- LINE END ---
35  Sat Oct  8 18:28:02.0924 2016  7  [                          src/HttpcServer.cpp:141] RTSP Stream data from client 192.168.30.216 Port 52121: SETUP rtsp://192.168.30.247:554/?pids=17 RTSP/1.0
36  Sat Oct  8 18:28:02.0924 2016  7  CSeq: 3532
37  Sat Oct  8 18:28:02.0924 2016  7  Transport: RTP/AVP;unicast;client_port=45079-45080
38  Sat Oct  8 18:28:02.0924 2016  7  --- LINE END ---
39  Sat Oct  8 18:28:02.0931 2016  6  [                        src/StreamManager.cpp:167] Found StreamID x - SessionID x - Creating new SessionID: 1740892522
40  Sat Oct  8 18:28:02.0934 2016  6  [                               src/Stream.cpp:190] Stream: 1, Not capable of handling msys=UNKNOWN DELSYS
41  Sat Oct  8 18:28:02.0937 2016  6  [                               src/Stream.cpp:190] Stream: 2, Not capable of handling msys=UNKNOWN DELSYS
42  Sat Oct  8 18:28:02.0938 2016  3  [                        src/StreamManager.cpp:195] Found no Stream/Client of interest!
43  Sat Oct  8 18:28:02.0940 2016  7  [                          src/HttpcServer.cpp:175] RTSP/1.0 503 Service Unavailable
44  Sat Oct  8 18:28:02.0940 2016  7  Server: SatPI WebServer v0.1
45  Sat Oct  8 18:28:02.0940 2016  7  Location: 
46  Sat Oct  8 18:28:02.0940 2016  7  CSeq: 3532
47  Sat Oct  8 18:28:02.0940 2016  7  cache-control: no-cache
48  Sat Oct  8 18:28:02.0940 2016  7  Content-Type: video/MP2T
49  Sat Oct  8 18:28:02.0940 2016  7  --- LINE END ---
50  Sat Oct  8 18:28:02.1064 2016  7  [                          src/HttpcServer.cpp:141] RTSP Stream data from client 192.168.30.216 Port 52121: SETUP rtsp://192.168.30.247:554/?pids=17 RTSP/1.0
51  Sat Oct  8 18:28:02.1064 2016  7  CSeq: 3533
52  Sat Oct  8 18:28:02.1064 2016  7  Transport: RTP/AVP;unicast;client_port=45079-45080
53  Sat Oct  8 18:28:02.1064 2016  7  --- LINE END ---
54  Sat Oct  8 18:28:02.1071 2016  6  [                        src/StreamManager.cpp:167] Found StreamID x - SessionID x - Creating new SessionID: 1797702436
[... cut ...]
342  Sat Oct  8 18:28:02.4520 2016  3  [                        src/StreamManager.cpp:195] Found no Stream/Client of interest!
343  Sat Oct  8 18:28:02.4522 2016  7  [                          src/HttpcServer.cpp:175] RTSP/1.0 503 Service Unavailable
344  Sat Oct  8 18:28:02.4522 2016  7  Server: SatPI WebServer v0.1
345  Sat Oct  8 18:28:02.4522 2016  7  Location: 
346  Sat Oct  8 18:28:02.4522 2016  7  CSeq: 3552
347  Sat Oct  8 18:28:02.4522 2016  7  cache-control: no-cache
348  Sat Oct  8 18:28:02.4522 2016  7  Content-Type: video/MP2T
349  Sat Oct  8 18:28:02.4522 2016  7  --- LINE END ---
`

rezna11, 2016-10-08:
it looks like this always happens in case the last transponder fails to lock signal, so the last OK looking message in the log, before it just loops the errors above, is this -  and after that, only satpi restart helps. It is also interesting that locking the transponder fails since it's a channel which I just scanned in using satpi

24  Sat Oct  8 18:38:23.6104 2016  7  [                          src/HttpcServer.cpp:141] RTSP Stream data from client 192.168.30.216 Port 52139: SETUP rtsp://192.168.30.247:554/?src=1&freq=11072&pol=v&msys=dvbs&sr=3300&fec=34&pids=none RTSP/1.0
25  Sat Oct  8 18:38:23.6104 2016  7  CSeq: 1
26  Sat Oct  8 18:38:23.6104 2016  7  Transport: RTP/AVP;unicast;client_port=45079-45080
27  Sat Oct  8 18:38:23.6104 2016  7  --- LINE END ---
28  Sat Oct  8 18:38:23.6110 2016  6  [                        src/StreamManager.cpp:167] Found StreamID x - SessionID x - Creating new SessionID: 1702957685
29  Sat Oct  8 18:38:23.6112 2016  6  [                               src/Stream.cpp:201] Stream: 0, StreamClient[0] with SessionID 1702957685 for dvbs
30  Sat Oct  8 18:38:23.6114 2016  6  [                   src/input/dvb/Frontend.cpp:309] Stream: 0, Parsing transport parameters...
31  Sat Oct  8 18:38:23.6165 2016  7  [                   src/input/dvb/Frontend.cpp:317] Stream: 0, New frequency requested, clearing channel data...
32  Sat Oct  8 18:38:23.6176 2016  7  [                   src/input/dvb/Frontend.cpp:493] Stream: 0, Parsing transport parameters (Finished)
33  Sat Oct  8 18:38:23.6178 2016  7  [                         src/StreamClient.cpp:046] Connection can close: 0
34  Sat Oct  8 18:38:23.6180 2016  7  [                               src/Stream.cpp:258] Stream: 0, Found Streaming type: RTSP
35  Sat Oct  8 18:38:23.6186 2016  6  [                   src/input/dvb/Frontend.cpp:497] Stream: 0, Updating frontend...
36  Sat Oct  8 18:38:23.6189 2016  6  [                   src/input/dvb/Frontend.cpp:858] Stream: 0, Opened /dev/dvb/adapter0/frontend0 fd: 15
37  Sat Oct  8 18:38:23.6191 2016  6  [              src/input/dvb/delivery/DVBS.cpp:105] Stream: 0, Start tuning process for DVB-S(2)...
38  Sat Oct  8 18:38:23.6192 2016  6  [      src/input/dvb/delivery/DiSEqcSwitch.cpp:085] Stream: 0, Sending DiSEqC [e0] [10] [38] [f0]
39  Sat Oct  8 18:38:23.9278 2016  6  [                   src/input/dvb/Frontend.cpp:869] Stream: 0, Waiting on lock...
40  Sat Oct  8 18:38:23.9280 2016  6  [                   src/input/dvb/Frontend.cpp:883] Stream: 0, Not locked yet   (FE status 0x0)...
41  Sat Oct  8 18:38:24.2048 2016  6  [                   src/input/dvb/Frontend.cpp:883] Stream: 0, Not locked yet   (FE status 0x0)...
42  Sat Oct  8 18:38:24.3557 2016  6  [                   src/input/dvb/Frontend.cpp:883] Stream: 0, Not locked yet   (FE status 0x0)...
43  Sat Oct  8 18:38:24.5065 2016  6  [                   src/input/dvb/Frontend.cpp:883] Stream: 0, Not locked yet   (FE status 0x0)...
44  Sat Oct  8 18:38:24.8068 2016  6  [              src/input/dvb/delivery/DVBS.cpp:105] Stream: 0, Start tuning process for DVB-S(2)...
45  Sat Oct  8 18:38:24.8070 2016  6  [      src/input/dvb/delivery/DiSEqcSwitch.cpp:085] Stream: 0, Sending DiSEqC [e0] [10] [38] [f0]
46  Sat Oct  8 18:38:25.0917 2016  6  [                   src/input/dvb/Frontend.cpp:869] Stream: 0, Waiting on lock...
47  Sat Oct  8 18:38:25.0918 2016  6  [                   src/input/dvb/Frontend.cpp:883] Stream: 0, Not locked yet   (FE status 0x0)...
48  Sat Oct  8 18:38:25.3422 2016  6  [                   src/input/dvb/Frontend.cpp:883] Stream: 0, Not locked yet   (FE status 0x0)...
49  Sat Oct  8 18:38:25.4930 2016  6  [                   src/input/dvb/Frontend.cpp:883] Stream: 0, Not locked yet   (FE status 0x0)...
50  Sat Oct  8 18:38:25.6438 2016  6  [                   src/input/dvb/Frontend.cpp:883] Stream: 0, Not locked yet   (FE status 0x0)...
51  Sat Oct  8 18:38:25.9442 2016  6  [              src/input/dvb/delivery/DVBS.cpp:105] Stream: 0, Start tuning process for DVB-S(2)...
52  Sat Oct  8 18:38:25.9443 2016  6  [      src/input/dvb/delivery/DiSEqcSwitch.cpp:085] Stream: 0, Sending DiSEqC [e0] [10] [38] [f0]
53  Sat Oct  8 18:38:26.2290 2016  6  [                   src/input/dvb/Frontend.cpp:869] Stream: 0, Waiting on lock...
54  Sat Oct  8 18:38:26.2292 2016  6  [                   src/input/dvb/Frontend.cpp:883] Stream: 0, Not locked yet   (FE status 0x0)...
55  Sat Oct  8 18:38:26.4822 2016  6  [                   src/input/dvb/Frontend.cpp:883] Stream: 0, Not locked yet   (FE status 0x0)...
56  Sat Oct  8 18:38:26.6331 2016  6  [                   src/input/dvb/Frontend.cpp:883] Stream: 0, Not locked yet   (FE status 0x0)...
57  Sat Oct  8 18:38:26.7840 2016  6  [                   src/input/dvb/Frontend.cpp:883] Stream: 0, Not locked yet   (FE status 0x0)...
58  Sat Oct  8 18:38:27.0843 2016  6  [              src/input/dvb/delivery/DVBS.cpp:105] Stream: 0, Start tuning process for DVB-S(2)...
59  Sat Oct  8 18:38:27.0844 2016  6  [      src/input/dvb/delivery/DiSEqcSwitch.cpp:085] Stream: 0, Sending DiSEqC [e0] [10] [38] [f0]
60  Sat Oct  8 18:38:27.3693 2016  6  [                   src/input/dvb/Frontend.cpp:869] Stream: 0, Waiting on lock...
61  Sat Oct  8 18:38:27.3695 2016  6  [                   src/input/dvb/Frontend.cpp:883] Stream: 0, Not locked yet   (FE status 0x0)...
[... cut ...]
71  Sat Oct  8 18:38:28.2251 2016  7  [                     src/socket/TcpSocket.cpp:090] RTSP Client 192.168.30.216:52139 Connection closed with fd: 14
72  Sat Oct  8 18:38:28.2252 2016  6  [                        src/StreamManager.cpp:140] Found SessionID 1702957685 by SocketClient
73  Sat Oct  8 18:38:28.2253 2016  6  [                        src/StreamManager.cpp:158] Found StreamID x - SessionID: 1702957685
74  Sat Oct  8 18:38:28.2254 2016  6  [                               src/Stream.cpp:204] Stream: 0, StreamClient[0] with SessionID 1702957685

Barracuda09, 2016-10-09:
Hi @rezna11 ,

I can not say anything with these logs.. I like to see the log at the point that it is OK and then fails.

In dmesg you can get the complete log, and please attach the log as txt here.

Thanks again,

Marc

rezna11, 2016-10-09:
I did a test now, a nice example, but can you explain where is the log file?

Barracuda09, 2016-10-09:
Yes sorry, it is in syslog

Barracuda09, 2016-10-09:
you can find it here /var/log/syslog

rezna11, 2016-10-09:
OK so in my test, which is shown in the attached log, this was done:
1) searched for SAT>IP server on Dreambox, it found SATPI, good
2) went to scan in first frequency - went fine
3) switched to one of the newly found channels - all good
4) then I went to the menu to scan another frequency - nothing, search timed out
5) then I tried to again watch the channel which was previously found - but it didn't work
6) I restarted SATPI and tried to watch the channel again without restarting the Dreambox, it was working 
Log:
[debug.gz](https://github.com/Barracuda09/SATPI/files/518278/debug.gz)

Barracuda09, 2016-10-09:
Ok... there is something funky going on in this log.

Thanks for this report, I will check it out,

Marc

Barracuda09, 2016-10-09:
Hi @rezna11 

Could you try and give mea log from first start of SatPI and one tune?

Many thanks,

Marc

Barracuda09, 2016-10-09:
By the way are you sure that the log is of an DEBUG version??

It seems that there is missing a lot of info which I expect in an debug version.

rezna11, 2016-10-09:
how do I verify I have the debug version?
by the way the log I posted should include what you're asking - start and tune... or are you looking for something more specific?

Barracuda09, 2016-10-09:
Yes start of SatPI and one tune (This is not in the log)

You could do:
make clean
make debug

rezna11, 2016-10-09:
ok then I'm sure it is debug but I copied wrong file earlier, the rigth one added now - the last test starts at Oct  9 20:52:13
[syslog.gz](https://github.com/Barracuda09/SATPI/files/518552/syslog.gz)

Barracuda09, 2016-10-10:
Which client did you use?

rezna11, 2016-10-10:
Dreambox 820HD

Barracuda09, 2016-10-10:
Hi @rezna11 ,

This part looks to me a bit strange from the client:

	Oct  9 20:53:30 raspberrypi SatPI[4071]: [                          src/HttpcServer.cpp:141] RTSP Stream data from client 192.168.30.216 Port 37965: PLAY rtsp://192.168.30.247:554/stream=0?delpids=20 RTSP/1.0
	Oct  9 20:53:30 raspberrypi SatPI[4071]: CSeq: 73
	Oct  9 20:53:30 raspberrypi SatPI[4071]: Session: 1702957685
	Oct  9 20:53:30 raspberrypi SatPI[4071]: --- LINE END ---
	Oct  9 20:53:30 raspberrypi SatPI[4071]: [                        src/StreamManager.cpp:179] Found StreamID 0 - SessionID 1702957685
	Oct  9 20:53:30 raspberrypi SatPI[4071]: [                               src/Stream.cpp:204] Stream: 0, StreamClient[0] with SessionID 1702957685
	Oct  9 20:53:30 raspberrypi SatPI[4071]: [                   src/input/dvb/Frontend.cpp:309] Stream: 0, Parsing transport parameters...
	Oct  9 20:53:30 raspberrypi SatPI[4071]: [                   src/input/dvb/Frontend.cpp:493] Stream: 0, Parsing transport parameters (Finished)
	Oct  9 20:53:30 raspberrypi SatPI[4071]: [                         src/StreamClient.cpp:046] Connection can close: 0
	Oct  9 20:53:30 raspberrypi SatPI[4071]: [                   src/input/dvb/Frontend.cpp:497] Stream: 0, Updating frontend...
	Oct  9 20:53:30 raspberrypi SatPI[4071]: [                   src/input/dvb/Frontend.cpp:925] Stream: 0, Updating PID filters...
	Oct  9 20:53:30 raspberrypi SatPI[4071]: [                   src/input/dvb/Frontend.cpp:947] Stream: 0, Remove filter PID: 0020 - fd: 012 - Packet Count: 1
	Oct  9 20:53:30 raspberrypi SatPI[4071]: [                   src/input/dvb/Frontend.cpp:517] Stream: 0, Updating frontend (Finished)
	Oct  9 20:53:30 raspberrypi SatPI[4071]: [                          src/HttpcServer.cpp:175] RTSP/1.0 200 OK
	Oct  9 20:53:30 raspberrypi SatPI[4071]: RTP-Info: url=rtsp://192.168.30.247/stream=0
	Oct  9 20:53:30 raspberrypi SatPI[4071]: CSeq: 73
	Oct  9 20:53:30 raspberrypi SatPI[4071]: Session: 1702957685
	Oct  9 20:53:30 raspberrypi SatPI[4071]: Range: npt=0.000-
	Oct  9 20:53:30 raspberrypi SatPI[4071]: --- LINE END ---
	Oct  9 20:53:30 raspberrypi SatPI[4071]: [                     src/socket/TcpSocket.cpp:090] RTSP Client 192.168.30.216:37965 Connection closed with fd: 10
	Oct  9 20:53:30 raspberrypi SatPI[4071]: [                        src/StreamManager.cpp:140] Found SessionID 1702957685 by SocketClient
	Oct  9 20:53:30 raspberrypi SatPI[4071]: [                        src/StreamManager.cpp:158] Found StreamID x - SessionID: 1702957685
	Oct  9 20:53:30 raspberrypi SatPI[4071]: [                               src/Stream.cpp:204] Stream: 0, StreamClient[0] with SessionID 1702957685
	Oct  9 20:53:34 raspberrypi SatPI[4071]: [                    src/socket/SocketAttr.cpp:126] RTSP Connection from 192.168.30.216 Port 37970 (fd: 10)
	Oct  9 20:53:34 raspberrypi SatPI[4071]: [                          src/HttpcServer.cpp:141] RTSP Stream data from client 192.168.30.216 Port 37970: SETUP rtsp://192.168.30.247:554/?src=2&freq=11177&pol=v&msys=dvbs2&sr=13300&fec=56&ro=0.20&mtype=8psk&pids=none RTSP/1.0
	Oct  9 20:53:34 raspberrypi SatPI[4071]: CSeq: 1
	Oct  9 20:53:34 raspberrypi SatPI[4071]: Transport: RTP/AVP;unicast;client_port=46693-46694
	Oct  9 20:53:34 raspberrypi SatPI[4071]: --- LINE END ---
	Oct  9 20:53:34 raspberrypi SatPI[4071]: [                        src/StreamManager.cpp:167] Found StreamID x - SessionID x - Creating new SessionID: 0205744672
	Oct  9 20:53:34 raspberrypi SatPI[4071]: [                               src/Stream.cpp:190] Stream: 1, Not capable of handling msys=dvbs2
	Oct  9 20:53:34 raspberrypi SatPI[4071]: [                               src/Stream.cpp:190] Stream: 2, Not capable of handling msys=dvbs2
	Oct  9 20:53:34 raspberrypi SatPI[4071]: [                        src/StreamManager.cpp:195] Found no Stream/Client of interest!
	Oct  9 20:53:34 raspberrypi SatPI[4071]: [                          src/HttpcServer.cpp:175] RTSP/1.0 503 Service Unavailable
	Oct  9 20:53:34 raspberrypi SatPI[4071]: Server: SatPI WebServer v0.1
	Oct  9 20:53:34 raspberrypi SatPI[4071]: Location:
	Oct  9 20:53:34 raspberrypi SatPI[4071]: CSeq: 1
	Oct  9 20:53:34 raspberrypi SatPI[4071]: cache-control: no-cache
	Oct  9 20:53:34 raspberrypi SatPI[4071]: Content-Type: video/MP2T
	Oct  9 20:53:34 raspberrypi SatPI[4071]: --- LINE END ---

Here you can see that the client is closing the connection (port 37965).

Then 4 sec. later a new setup is requested with an new connection (port 37970).
This is assumed to be a new stream request, but because there is only one dvb card it can not find any so it returns 503. But then the client keeps on nagging after this.

The client should have send a teardown request if it is finished to release the resource before closing port 37965.

rezna11, 2016-10-10:
thanks for looking into this, so it's a client bug - but is it possible to assume each new connection from a client means the previous one should be closed, if there is just one DVB card anyway?

Barracuda09, 2016-10-10:
Well according to the specs you should close an "session" by giving an TEARDOWN request.
However you can close an TCP connection within on open session, but that does not mean you can close the session.

So what happens here it is that SatPI is waiting for stream 0 to close by a TEARDOWN (Or an time-out occurred). And so the new TCP connection is requesting an SETUP without session id, so it assumes an new session is starting, but does not find the correct resource so it send 503.

Problem: why there is no time-out - This I have to check.

Barracuda09, 2017-08-04:
Hi @rezna11

Is this still an issue, at this point in time?
If you like and have time could please test this?

Thanks

Barracuda09, 2017-12-14:
I will close this issue, as #56  is also confirmed to be fixed.
If there are any problems, please make a new issue describing the problem.

closed 2017-12-14 by Barracuda09