Barracuda09/SATPI #51: SAT>IP server app for Android ?

bluepuma77, 2016-12-17:
Hi, have you thought about creating a DVB2IP Android app using SAT>IP protocol on the basis of this project? It seems more and more Android TV-boxes with DVB-Tuner(s) show up, but a good Android app for TV networking like SAT>IP server or tvheadend server is still missing. And if using SAT>IP, you even have a standard that some TVs already implement.

Some Andoid box examples:
WeTek Play TV
Videostrong K1 Plus T2 S2
VideoStrong KII Pro Android Set-top Box with DVB-T2 & DVB-S2 Tuners
Xtreamer mxV Plus DVB-T2 and mvX Pro DVB-T2/DVB-S2 Combo TV Boxes
Geniatech ATV1960 Octa-core Android TV Box Comes with a Dual TV Tuner
http://www.cnx-software.com/tag/dvb/

Barracuda09, 2016-12-23:
Hi @bluepuma77 

I have not thought about this, due to the fact that I do not have that much experience
with building android apps. So sorry this will be no option in the near future.
Unless someone is taking up the challenge I will help out as much as possible.

Regards,

Marc

lars18th, 2017-01-17:
Hi Bluepuema,

This isn't the way for SAT>IP support!
Instead of a non-optimized Android App, these boxes need to have **native SAT>IP support**. It's like the internal tuner, but using the network. This is the same way taken by ancient (non-smart) STBs. As I know, some of the manufacturers are considering native SAT>IP support.

Regards.