Barracuda09/SATPI #52: Documentation update for DVB-T

yglodt, 2017-02-18:
It would be great to add the parameters needed to open a DVB-T stream in vlc.

E.g. for the following:

[image]

Barracuda09, 2017-02-18:
Hi @yglodt 

I think it should be something like this
```
http://ip.of.your.box:8875/?freq=522&bw=8&msys=dvbt&tmode=8k&mtype=64qam&gi=116&fec=23&pids=0,1,16,17,18,....
```

pids you have to add which apply to the channel you like to see or use pids=all

Regards,

Marc

yglodt, 2017-02-18:
Yes this works! Thanks! To play back a 1080p-channel on my Pi 1+ it takes like 90% cpu and it's jerky. Probably the old Pi is just too weak :)

Barracuda09, 2017-02-19:
Yes, I use a Pi 2. That has more power.

closed 2017-02-19 by yglodt