Barracuda09/SATPI #53: Can not play input ts file

papachappa, 2017-04-24:
Hello and greeting!
Found your satpi utility on github. But could you please help me. I am trying to input TS File on satip and then play him stream via vlc in ubuntu machine.
As i understand satpi can input ts files, so i run server, then in terminal i typed 
 
vlc http://ipwhereserverrun:8875/?msys=file&uri=/path/to/test.ts

In web frontend status i can see in Stream info Session ID and IP of the stream. Then i try to play my ts file on the same machine:
vlc http://ipwhereserverrun:8875/?msys=dvbs&freq=2003&pids=0,851
But vlc return error "Open of http://ipwhereserverrun:8875/?msys=dvbs&freq=2003&pids=0,851 failed"
What am i doing wrong? Please help.

Barracuda09, 2017-04-28:
Hi,

At first sorry for the late reply, and thanks for trying SatPI.

The first question:
``vlc http://ipwhereserverrun:8875/?msys=file&uri=/path/to/test.ts``
``The uri is relative to the SatPI bin (At the moment).``

The second:
``vlc http://ipwhereserverrun:8875/?msys=dvbs&freq=2003&pids=0,851``
I do not know what you are trying to do here, watch a DVB-S channel?

The request is not complete to do this, and it should be something like this (maybe):
``vlc http://ipwhereserverrun:8875/?src=1&freq=11836&pol=h&ro=0.35&msys=dvbs&mtype=qpsk&plts=off&sr=27500&fec=34&pids=0,851``

You have to specify more parameter to tune.

Regards,

Marc Postema

papachappa, 2017-04-28:
Hello, Marc!
Thank you for your reply. About first question - move mpeg ts file to satpi bin directory and execute "vlc  http://ipoftheserver:8875/?msys=file&uri=Astra1_DVB-S1_12604MHz_hh_22000.ts ". Server sees the stream on frontend 0

Frontend Info
Stream Index 0
Enable
Attached yes
Name TS Reader
Path Astra1_DVB-S1_12604MHz_hh_22000.ts
Stream Info
Owner 172.31.14.154
Session ID 0279714759
RTP packet count 259181
RTP streamed (MB) 325.281
Stream Configuration
RTCP Signal Update Freq
About second question - yes i am trying to get dbv-s channel from that file  Astra1_DVB-S1_12604MHz_hh_22000.ts

I execute "vlc rtsp://ipoftheserver:554/?src=0&freq=12604&pol=h&ro=0.35&msys=dvbs&mtype=qpsk&plts=off&sr=22000&fec=56&pids=all"
Error occured 

Then i execute "http://172.31.14.154:8875/?src=0&freq=12604&pol=h&ro=0.35&msys=dvbs&mtype=qpsk&plts=off&sr=22000&fec=56&pids=all"
Same error

Logs:
Fri Apr 28 11:59:16.0280 2017  6  --- starting SatPI version: 1.5.191~gf82493a ---
Fri Apr 28 11:59:16.0281 2017  6  Number of processors online: 4
Fri Apr 28 11:59:16.0282 2017  6  Default network buffer size: 104 KBytes
Fri Apr 28 11:59:16.0283 2017  6  enp4s0: 172.31.14.154 [d8:cb:8a:ef:b8:f9]
Fri Apr 28 11:59:16.0284 2017  6  Current DVB_API_VERSION: 5.10
Fri Apr 28 11:59:16.0286 2017  6  Detecting frontends in: /dev/dvb
Fri Apr 28 11:59:16.0286 2017  6  Frontends found: 0
Fri Apr 28 11:59:16.0286 2017  6  Setting up TS Reader using path: .
Fri Apr 28 11:59:16.0288 2017  6  Setting up TS Streamer
Fri Apr 28 11:59:16.0289 2017  6  Setting SSDP annouce interval to: 60 Sec
 Fri Apr 28 11:59:16.0292 2017  6  Setting up SSDP server with BOOTID: 14
 Fri Apr 28 11:59:16.0292 2017  6  Setting up RTSP server
 Fri Apr 28 11:59:16.0292 2017  6  Setting up HTTP server
 Fri Apr 28 11:59:16.2263 2017  6  HTTP Connection from 172.31.14.154 Port 43484 (fd: 8)
 Fri Apr 28 11:59:16.5536 2017  6  HTTP Connection from 172.31.14.138 Port 58843 (fd: 9)
 Fri Apr 28 11:59:16.5714 2017  6  HTTP Connection from 172.31.14.143 Port 43984 (fd: 10)
 Fri Apr 28 11:59:16.5964 2017  6  HTTP Connection from 172.31.14.156 Port 43411 (fd: 11)
 Fri Apr 28 11:59:16.6019 2017  6  HTTP Connection from 172.31.14.150 Port 41529 (fd: 12)
 Fri Apr 28 11:59:18.9743 2017  6  HTTP Connection from 172.31.14.152 Port 51701 (fd: 9)
 Fri Apr 28 11:59:19.6992 2017  6  HTTP Connection from 172.31.14.154 Port 43610 (fd: 10)
 Fri Apr 28 11:59:19.6995 2017  6  HTTP Connection from 172.31.14.154 Port 43612 (fd: 11)
 Fri Apr 28 11:59:19.6995 2017  6  HTTP Connection from 172.31.14.154 Port 43614 (fd: 12)
 Fri Apr 28 11:59:19.6996 2017  6  HTTP Connection from 172.31.14.154 Port 43616 (fd: 13)
 Fri Apr 28 11:59:19.6998 2017  6  HTTP Connection from 172.31.14.154 Port 43618 (fd: 14)
 Fri Apr 28 11:59:35.9698 2017  6  HTTP Connection from 172.31.14.154 Port 44512 (fd: 15)
 Fri Apr 28 11:59:35.9714 2017  6  Found StreamID x - SessionID x - Creating new SessionID: 0279714759
 Fri Apr 28 11:59:35.9714 2017  6  Stream: 0, StreamClient[0] with SessionID 0279714759 for file
 Fri Apr 28 11:59:35.9715 2017  6  Stream: 0, TS Reader using path: Astra1_DVB-S1_12604MHz_hh_22000.ts
 Fri Apr 28 11:59:35.9715 2017  6  Stream: 0, HTTP set network buffer size: 25650 KBytes
 Fri Apr 28 11:59:35.9716 2017  6  Stream: 0, Start HTTP stream to 172.31.14.154:44512
 Fri Apr 28 11:59:36.2057 2017  6  Stream: 0, PAT - PID 0: sectionLength: 181 tableDataSize: 188
 Fri Apr 28 11:59:36.2080 2017  6  Stream: 0, PAT: Section Length: 181 TID: 1111 Version: 217 secNr: 0 lastSecNr: 0 CRC: 1E363F47
 Fri Apr 28 11:59:36.2081 2017  6  Stream: 0, PAT: Prog NR: 0 NIT PID: 0016
 Fri Apr 28 11:59:36.2081 2017  6  Stream: 0, PAT: Prog NR: 13 PMT PID: 0032
 Fri Apr 28 11:59:36.2081 2017  6  Stream: 0, PAT: Prog NR: 7201 PMT PID: 7201
 Fri Apr 28 11:59:36.2081 2017  6  Stream: 0, PAT: Prog NR: 7203 PMT PID: 7203
 Fri Apr 28 11:59:36.2081 2017  6  Stream: 0, PAT: Prog NR: 7212 PMT PID: 7212
 Fri Apr 28 11:59:36.2081 2017  6  Stream: 0, PAT: Prog NR: 7216 PMT PID: 7216
 Fri Apr 28 11:59:36.2081 2017  6  Stream: 0, PAT: Prog NR: 7230 PMT PID: 7230
 Fri Apr 28 11:59:36.2081 2017  6  Stream: 0, PAT: Prog NR: 7232 PMT PID: 7232
 Fri Apr 28 11:59:36.2082 2017  6  Stream: 0, PAT: Prog NR: 7243 PMT PID: 7243
 Fri Apr 28 11:59:36.2082 2017  6  Stream: 0, PAT: Prog NR: 7248 PMT PID: 7248
 Fri Apr 28 11:59:36.2082 2017  6  Stream: 0, PAT: Prog NR: 7252 PMT PID: 7252
 Fri Apr 28 11:59:36.2082 2017  6  Stream: 0, PAT: Prog NR: 7260 PMT PID: 7260
 Fri Apr 28 11:59:36.2082 2017  6  Stream: 0, PAT: Prog NR: 7263 PMT PID: 7263
[... 41 lines cut ...]
Fri Apr 28 11:58:17.8218 2017  3  Found no Stream/Client of interest!
Fri Apr 28 11:58:17.8227 2017  6  Stream: 0, Close StreamClient[0] with SessionID -1

Fri Apr 28 12:08:34.8954 2017  6  Stream: 1, Not capable of handling msys=dvbs
Fri Apr 28 12:08:34.8954 2017  3  Found no Stream/Client of interest!
Fri Apr 28 12:08:34.8956 2017  6  Found StreamID x - SessionID x - Creating new SessionID: 0256267502
Fri Apr 28 12:08:34.8956 2017  6  Stream: 1, Not capable of handling msys=dvbs
Fri Apr 28 12:08:34.8956 2017  3  Found no Stream/Client of interest!
Fri Apr 28 12:08:34.8956 2017  6  Stream: 0, Close StreamClient[0] with SessionID -1
Fri Apr 28 12:08:34.8963 2017  6  RTSP Connection from 172.31.14.154 Port 38014 (fd: 9)
Fri Apr 28 12:08:34.8980 2017  6  Stream: 0, Close StreamClient[0] with SessionID -1
Fri Apr 28 12:08:34.9082 2017  6  RTSP Connection from 172.31.14.154 Port 38018 (fd: 9)
[... cut ...]
 Fri Apr 28 12:08:34.9131 2017  6  Stream: 0, Close StreamClient[0] with SessionID -1
 Fri Apr 28 12:08:34.9138 2017  6  RTSP Connection from 172.31.14.154 Port 38026 (fd: 9)
 Fri Apr 28 12:08:34.9157 2017  6  Stream: 0, Close StreamClient[0] with SessionID -1
 Fri Apr 28 12:08:34.9191 2017  6  RTSP Connection from 172.31.14.154 Port 38028 (fd: 9)
 Fri Apr 28 12:08:34.9192 2017  6  Found StreamID x - SessionID x - Creating new SessionID: 0258257498

Could you please help me, what am i doing wrong?

papachappa, 2017-04-28:
Hello, Marc!
Thank you for your reply. About first question - move mpeg ts file to satpi bin directory and execute "vlc http://ipoftheserver:8875/?msys=file&uri=Astra1_DVB-S1_12604MHz_hh_22000.ts". Server sees the stream on frontend 0

Frontend Info
Stream Index	0
Enable	
Attached	yes
Name	TS Reader
Path	Astra1_DVB-S1_12604MHz_hh_22000.ts
Stream Info
Owner	172.31.14.154
Session ID	0279714759
RTP packet count	259181
RTP streamed (MB)	325.281
Stream Configuration
RTCP Signal Update Freq	
1

About second question - yes i am trying to get dbv-s channel from that file Astra1_DVB-S1_12604MHz_hh_22000.ts

I execute "vlc rtsp://ipoftheserver:554/?src=0&freq=12604&pol=h&ro=0.35&msys=dvbs&mtype=qpsk&plts=off&sr=22000&fec=56&pids=all"
Error occured 

Then i execute "http://172.31.14.154:8875/?src=0&freq=12604&pol=h&ro=0.35&msys=dvbs&mtype=qpsk&plts=off&sr=22000&fec=56&pids=all"
Same error

Logs:
Fri Apr 28 11:59:16.0280 2017  6  --- starting SatPI version: 1.5.191~gf82493a ---
Fri Apr 28 11:59:16.0281 2017  6  Number of processors online: 4
Fri Apr 28 11:59:16.0282 2017  6  Default network buffer size: 104 KBytes
Fri Apr 28 11:59:16.0283 2017  6  enp4s0: 172.31.14.154 [d8:cb:8a:ef:b8:f9]
Fri Apr 28 11:59:16.0284 2017  6  Current DVB_API_VERSION: 5.10
Fri Apr 28 11:59:16.0286 2017  6  Detecting frontends in: /dev/dvb
Fri Apr 28 11:59:16.0286 2017  6  Frontends found: 0
Fri Apr 28 11:59:16.0286 2017  6  Setting up TS Reader using path: .
Fri Apr 28 11:59:16.0288 2017  6  Setting up TS Streamer
Fri Apr 28 11:59:16.0289 2017  6  Setting SSDP annouce interval to: 60 Sec
 Fri Apr 28 11:59:16.0292 2017  6  Setting up SSDP server with BOOTID: 14
 Fri Apr 28 11:59:16.0292 2017  6  Setting up RTSP server
 Fri Apr 28 11:59:16.0292 2017  6  Setting up HTTP server
 Fri Apr 28 11:59:16.2263 2017  6  HTTP Connection from 172.31.14.154 Port 43484 (fd: 8)
 Fri Apr 28 11:59:16.5536 2017  6  HTTP Connection from 172.31.14.138 Port 58843 (fd: 9)
 Fri Apr 28 11:59:16.5714 2017  6  HTTP Connection from 172.31.14.143 Port 43984 (fd: 10)
 Fri Apr 28 11:59:16.5964 2017  6  HTTP Connection from 172.31.14.156 Port 43411 (fd: 11)
 Fri Apr 28 11:59:16.6019 2017  6  HTTP Connection from 172.31.14.150 Port 41529 (fd: 12)
 Fri Apr 28 11:59:18.9743 2017  6  HTTP Connection from 172.31.14.152 Port 51701 (fd: 9)
 Fri Apr 28 11:59:19.6992 2017  6  HTTP Connection from 172.31.14.154 Port 43610 (fd: 10)
 Fri Apr 28 11:59:19.6995 2017  6  HTTP Connection from 172.31.14.154 Port 43612 (fd: 11)
 Fri Apr 28 11:59:19.6995 2017  6  HTTP Connection from 172.31.14.154 Port 43614 (fd: 12)
 Fri Apr 28 11:59:19.6996 2017  6  HTTP Connection from 172.31.14.154 Port 43616 (fd: 13)
 Fri Apr 28 11:59:19.6998 2017  6  HTTP Connection from 172.31.14.154 Port 43618 (fd: 14)
 Fri Apr 28 11:59:35.9698 2017  6  HTTP Connection from 172.31.14.154 Port 44512 (fd: 15)
 Fri Apr 28 11:59:35.9714 2017  6  Found StreamID x - SessionID x - Creating new SessionID: 0279714759
 Fri Apr 28 11:59:35.9714 2017  6  Stream: 0, StreamClient[0] with SessionID 0279714759 for file
 Fri Apr 28 11:59:35.9715 2017  6  Stream: 0, TS Reader using path: Astra1_DVB-S1_12604MHz_hh_22000.ts
 Fri Apr 28 11:59:35.9715 2017  6  Stream: 0, HTTP set network buffer size: 25650 KBytes
 Fri Apr 28 11:59:35.9716 2017  6  Stream: 0, Start HTTP stream to 172.31.14.154:44512
 Fri Apr 28 11:59:36.2057 2017  6  Stream: 0, PAT - PID 0: sectionLength: 181 tableDataSize: 188
 Fri Apr 28 11:59:36.2080 2017  6  Stream: 0, PAT: Section Length: 181 TID: 1111 Version: 217 secNr: 0 lastSecNr: 0 CRC: 1E363F47
 Fri Apr 28 11:59:36.2081 2017  6  Stream: 0, PAT: Prog NR: 0 NIT PID: 0016
 Fri Apr 28 11:59:36.2081 2017  6  Stream: 0, PAT: Prog NR: 13 PMT PID: 0032
 Fri Apr 28 11:59:36.2081 2017  6  Stream: 0, PAT: Prog NR: 7201 PMT PID: 7201
 Fri Apr 28 11:59:36.2081 2017  6  Stream: 0, PAT: Prog NR: 7203 PMT PID: 7203
 Fri Apr 28 11:59:36.2081 2017  6  Stream: 0, PAT: Prog NR: 7212 PMT PID: 7212
 Fri Apr 28 11:59:36.2081 2017  6  Stream: 0, PAT: Prog NR: 7216 PMT PID: 7216
 Fri Apr 28 11:59:36.2081 2017  6  Stream: 0, PAT: Prog NR: 7230 PMT PID: 7230
 Fri Apr 28 11:59:36.2081 2017  6  Stream: 0, PAT: Prog NR: 7232 PMT PID: 7232
 Fri Apr 28 11:59:36.2082 2017  6  Stream: 0, PAT: Prog NR: 7243 PMT PID: 7243
 Fri Apr 28 11:59:36.2082 2017  6  Stream: 0, PAT: Prog NR: 7248 PMT PID: 7248
 Fri Apr 28 11:59:36.2082 2017  6  Stream: 0, PAT: Prog NR: 7252 PMT PID: 7252
[... 44 lines cut ...]
Fri Apr 28 11:58:17.8218 2017  3  Found no Stream/Client of interest!
Fri Apr 28 11:58:17.8227 2017  6  Stream: 0, Close StreamClient[0] with SessionID -1

Fri Apr 28 12:08:34.8954 2017  6  Stream: 1, Not capable of handling msys=dvbs
Fri Apr 28 12:08:34.8954 2017  3  Found no Stream/Client of interest!
Fri Apr 28 12:08:34.8956 2017  6  Found StreamID x - SessionID x - Creating new SessionID: 0256267502
Fri Apr 28 12:08:34.8956 2017  6  Stream: 1, Not capable of handling msys=dvbs
Fri Apr 28 12:08:34.8956 2017  3  Found no Stream/Client of interest!
Fri Apr 28 12:08:34.8956 2017  6  Stream: 0, Close StreamClient[0] with SessionID -1
Fri Apr 28 12:08:34.8963 2017  6  RTSP Connection from 172.31.14.154 Port 38014 (fd: 9)
Fri Apr 28 12:08:34.8980 2017  6  Stream: 0, Close StreamClient[0] with SessionID -1
Fri Apr 28 12:08:34.9082 2017  6  RTSP Connection from 172.31.14.154 Port 38018 (fd: 9)
Fri Apr 28 12:08:34.9112 2017  6  Found StreamID x - SessionID x - Creating new SessionID: 0280160960
[... cut ...]
 Fri Apr 28 12:08:34.9131 2017  6  Stream: 0, Close StreamClient[0] with SessionID -1
 Fri Apr 28 12:08:34.9138 2017  6  RTSP Connection from 172.31.14.154 Port 38026 (fd: 9)
 Fri Apr 28 12:08:34.9157 2017  6  Stream: 0, Close StreamClient[0] with SessionID -1
 Fri Apr 28 12:08:34.9191 2017  6  RTSP Connection from 172.31.14.154 Port 38028 (fd: 9)
 Fri Apr 28 12:08:34.9192 2017  6  Found StreamID x - SessionID x - Creating new SessionID: 0258257498

Could you please help me, what am i doing wrong?

Barracuda09, 2017-04-28:
Hi,

I think I know were the problem is.
You stream the TS file and then you have to 'select' an program in VLC.
The example I added was for if you have DVB-S tuners in you server.

But I have not done this, I only 'record' one program and view it.

papachappa, 2017-04-28:
So, is it possible to input ts file(with different programs) with vlc and get or play stream(ts file) from satpi server with also vlc?

Barracuda09, 2017-04-28:
I am not sure about that (never tried it myself)

It is possible to let SatPI 'stream' an TS file (With one program) to VLC

papachappa, 2017-04-28:
But, satpi can input ts file as TSReader, therefore we can play via vlc somehow?

Barracuda09, 2017-04-28:
Yes it should be possible with this example:
``http://ip.of.your.box:8875/?msys=file&uri=test.ts``

This will show the program in VLC (if you open it as an network stream)

papachappa, 2017-04-28:
In other words, can we emulate situation? On the input satPI takes some ts file(dvb-s), transforms it to dbv-s standard and vlc as a client can read it like http://satipserver:8875/?src=0&freq=12604&pol=h&ro=0.35&msys=dvbs&mtype=qpsk&plts=off&sr=22000&fec=56&pids=0,5100,5101,5102,5103,5104,5105,5106

Barracuda09, 2017-04-28:
I don't think it is very useful to do it like this.
maybe add only 'pids=0,2,18...' to this request:
``http://ip.of.your.box:8875/?msys=file&uri=test.ts``
to 'select' the program. But within VLC it should be able to do that as well.

papachappa, 2017-04-28:
Am i correct that if we have some test.ts file with one program, then send to satpi server via vlc like http://ip.of.your.box:8875/?msys=file&uri=test.ts, then we can get the stream from satpi server and play him like http://ip.of.your.box:8875/?src=0&freq=12604&pol=h&ro=0.35&msys=dvbs&mtype=qpsk&plts=off&sr=22000&fec=56&pids=0,5100,5101,5102,5103,5104,5105,5106
??

Barracuda09, 2017-04-29:
No it works like this, on the SatPI server is the test.ts (which you record) this can be send/stream to VLC
with ``http://ip.of.your.box:8875/?msys=file&uri=test.ts``

lars18th, 2017-06-30:
Hi,

I'm not sure what like to do @papachappa ...

But I feel hi wants to "remap" the call with "&msys=dvbs&freq=12604&pol=h&ro=0.35&mtype=qpsk&plts=off&sr=22000&fec=56" 
to "msys=file&uri=test.ts"
So when the first commes then it's processed as the second.

I'm right?
Perhaps it's not very complicated to do it (internal map of input files to some frequencies).

Barracuda09, 2017-10-12:
Hi @papachappa 

I think @lars18th  has it correct, I misunderstood your question.

The Translation of this request is not implemented  at the moment for TSReader.
It is for DVB-S to DVB-C (Or at least in the funtion: bool Frontend::capableToTransform(const std::string &msg, const std::string &method) const )

This can be done for TSReader as well.

lars18th, 2017-10-13:
Hi @Barracuda09 ,

> This can be done for TSReader as well.

You say that the frequency Translation will be enhanced to other sources (like files)? Great!

Thank you for this great project!

lars18th, 2019-11-20:
Hi @Barracuda09 ,

I feel this issue is already fixed. So please can you close it?
Thank you!

Barracuda09, 2019-11-20:
As requested by @lars18th

closed 2019-11-20 by Barracuda09