Barracuda09/SATPI #54: Changing LNB settings - not sticking

Benoire, 2017-07-31:
Hi

I'm trying to change the LNB settings to align with the New Zealand Sky supplied LNBs frequency of L:10750,H:10750,S:0 but every time I go in to the settings of LNB0 on the first tuner it accepts the changes but quickly resets to something else if I click elsewhere on the screen or change tab.  This issue is repeated across all the tuners in the system.

As I can't find a save button for LNB config, am I doing something wrong?

Thanks,

Chris

Barracuda09, 2017-07-31:
Hi,

Thanks for trying SatPi.

Well you should press "Enter" after changing the frequency.
Yes a save button should be an better way.

Hope this helps you out.

Kind regards,

Marc

Benoire, 2017-07-31:
Wonderful thank you!  I didn't see a requirement for pressing Enter so didn't try it!

Barracuda09, 2017-07-31:
Yes it is not intuitive. A save button would make it easier.

Barracuda09, 2019-03-15:
Made a new issue for the 'Save' button see: #65

closed 2019-03-15 by Barracuda09