Barracuda09/SATPI #55: Log tab - reverse order of log display

Benoire, 2017-08-06:
Hi

It appears that the main log window in the webgui is updated frequently and it adds to the bottom, while the display shows the first parts of the log.  I would like to propose that you reverse the log display and show the latest logs at the top rather than having to scroll to the bottom?  If this is not possible, is there anyway to sort out the auto refresh to avoid it jumping back to the beginning on every refresh?

Thanks,

Chris

Barracuda09, 2017-08-07:
Hi,

Yes this is possible to reverse the log, I will look into this.
Stop or pause of the auto refresh is also not possible at the moment.

I will look into these issue as soon as possible, or if anyone feels up for a challenge?

Kind regards,

Marc

Barracuda09, 2020-12-07:
This is fixed by @issue

For testing, please update web directory on your device.

Barracuda09, 2022-09-23:
Already implemented.

closed 2022-09-23 by Barracuda09