Barracuda09/SATPI #56: DVB-T USB tuner is advertise as dvb-s

Ntemis, 2017-08-12:
When you try to use it you get error server is full capacity.
Enigma 2 Vuduo2 SAT>IP plugin

Ntemis, 2017-08-12:
0  Sat Aug 12 23:37:14.4807 2017  6  --- starting SatPI version: 1.5.197~g6971cca ---
1  Sat Aug 12 23:37:14.4816 2017  6  Number of processors online: 4
2  Sat Aug 12 23:37:14.4821 2017  6  Default network buffer size: 80 KBytes
3  Sat Aug 12 23:37:14.4828 2017  6  eth0: 192.168.1.212 [02:81:7f:84:c2:66]
4  Sat Aug 12 23:37:14.4830 2017  6  Current DVB_API_VERSION: 5.10
5  Sat Aug 12 23:37:14.4838 2017  6  Detecting frontends in: /dev/dvb
6  Sat Aug 12 23:37:14.4909 2017  6  Frontend Name: Realtek RTL2832 (DVB-T)
7  Sat Aug 12 23:37:14.4911 2017  6  Frontend Type: Terrestrial (DVB-T)
8  Sat Aug 12 23:37:14.4912 2017  6  Frontend Freq: 174000000 Hz to 862000000 Hz
9  Sat Aug 12 23:37:14.4913 2017  6  Frontend srat: 0 symbols/s to 0 symbols/s
10  Sat Aug 12 23:37:14.4915 2017  6  Frontends found: 1
11  Sat Aug 12 23:37:14.4916 2017  6  Setting up TS Reader using path: .
12  Sat Aug 12 23:37:14.4918 2017  6  Setting up TS Streamer
13  Sat Aug 12 23:37:14.4965 2017  6  Setting up HTTP server
14  Sat Aug 12 23:37:14.4980 2017  6  Setting up RTSP server
15  Sat Aug 12 23:37:14.4985 2017  6  Setting up SSDP server with BOOTID: 3
16  Sat Aug 12 23:37:29.4336 2017  6  HTTP Connection from 192.168.1.241 Port 35872 (fd: 8)
17  Sat Aug 12 23:37:29.5651 2017  6  HTTP Connection from 192.168.1.241 Port 35874 (fd: 9)
18  Sat Aug 12 23:37:29.5659 2017  6  HTTP Connection from 192.168.1.241 Port 35876 (fd: 10)
19  Sat Aug 12 23:37:29.5714 2017  6  HTTP Connection from 192.168.1.241 Port 35878 (fd: 11)
20  Sat Aug 12 23:37:29.5752 2017  6  HTTP Connection from 192.168.1.241 Port 35880 (fd: 12)
21  Sat Aug 12 23:37:29.5782 2017  6  HTTP Connection from 192.168.1.241 Port 35882 (fd: 13)
22  Sat Aug 12 23:38:34.4662 2017  6  HTTP Connection from 192.168.1.241 Port 35886 (fd: 8)
23  Sat Aug 12 23:40:01.7583 2017  6  SAT>IP Client 192.168.1.181 : tries to discover the network, sending reply back
24  Sat Aug 12 23:40:01.7764 2017  6  HTTP Connection from 192.168.1.181 Port 35329 (fd: 14)
25  Sat Aug 12 23:40:21.7483 2017  6  SAT>IP Client 192.168.1.181 : tries to discover the network, sending reply back
26  Sat Aug 12 23:40:21.7529 2017  6  HTTP Connection from 192.168.1.181 Port 35330 (fd: 14)
27  Sat Aug 12 23:41:13.2717 2017  6  SAT>IP Client 192.168.1.181 : tries to discover the network, sending reply back
28  Sat Aug 12 23:41:13.2756 2017  6  HTTP Connection from 192.168.1.181 Port 35331 (fd: 8)
29  Sat Aug 12 23:41:16.1184 2017  6  SAT>IP Client 192.168.1.181 : tries to discover the network, sending reply back
30  Sat Aug 12 23:41:16.1217 2017  6  HTTP Connection from 192.168.1.181 Port 35332 (fd: 8)
31  Sat Aug 12 23:41:23.8269 2017  6  SAT>IP Client 192.168.1.181 : tries to discover the network, sending reply back
32  Sat Aug 12 23:41:23.8318 2017  6  HTTP Connection from 192.168.1.181 Port 35333 (fd: 8)
33  Sat Aug 12 23:41:26.6001 2017  6  SAT>IP Client 192.168.1.181 : tries to discover the network, sending reply back
34  Sat Aug 12 23:41:26.6034 2017  6  HTTP Connection from 192.168.1.181 Port 35334 (fd: 8)

Barracuda09, 2017-08-14:
Hello,

Could you give an log were the error is showing. This is showing it detects the dvb-t tuner correct.
And could you also do an "make debug"

Thanks for giving this a try

Regards,
Marc

Ntemis, 2017-08-14:
Here you go

[enigma2-crash-logs.zip](https://github.com/Barracuda09/SATPI/files/1222789/enigma2-crash-logs.zip)

Ntemis, 2017-08-15:
Log with debug version:
0  Tue Aug 15 18:57:38.0127 2017  6  [                                src/Satpi.cpp:286] --- starting SatPI version: 1.5.197~g6971cca ---
1  Tue Aug 15 18:57:38.0139 2017  6  [                                src/Satpi.cpp:287] Number of processors online: 4
2  Tue Aug 15 18:57:38.0142 2017  6  [                                src/Satpi.cpp:288] Default network buffer size: 80 KBytes
3  Tue Aug 15 18:57:38.0151 2017  6  [                        src/InterfaceAttr.cpp:119] eth0: 192.168.1.212 [02:81:7f:84:c2:66]
4  Tue Aug 15 18:57:38.0153 2017  6  [                        src/StreamManager.cpp:072] Current DVB_API_VERSION: 5.10
5  Tue Aug 15 18:57:38.0162 2017  6  [                   src/input/dvb/Frontend.cpp:168] Detecting frontends in: /dev/dvb
6  Tue Aug 15 18:57:38.0304 2017  6  [                   src/input/dvb/Frontend.cpp:418] Frontend Name: Realtek RTL2832 (DVB-T)
7  Tue Aug 15 18:57:38.0308 2017  6  [                   src/input/dvb/Frontend.cpp:499] Frontend Type: Terrestrial (DVB-T)
8  Tue Aug 15 18:57:38.0309 2017  6  [                   src/input/dvb/Frontend.cpp:537] Frontend Freq: 174000000 Hz to 862000000 Hz
9  Tue Aug 15 18:57:38.0310 2017  6  [                   src/input/dvb/Frontend.cpp:538] Frontend srat: 0 symbols/s to 0 symbols/s
10  Tue Aug 15 18:57:38.0436 2017  6  [                   src/input/dvb/Frontend.cpp:418] Frontend Name: Realtek RTL2832 (DVB-T)
11  Tue Aug 15 18:57:38.0439 2017  6  [                   src/input/dvb/Frontend.cpp:499] Frontend Type: Terrestrial (DVB-T)
12  Tue Aug 15 18:57:38.0440 2017  6  [                   src/input/dvb/Frontend.cpp:537] Frontend Freq: 174000000 Hz to 862000000 Hz
13  Tue Aug 15 18:57:38.0441 2017  6  [                   src/input/dvb/Frontend.cpp:538] Frontend srat: 0 symbols/s to 0 symbols/s
14  Tue Aug 15 18:57:38.0444 2017  6  [                   src/input/dvb/Frontend.cpp:171] Frontends found: 2
15  Tue Aug 15 18:57:38.0446 2017  6  [                  src/input/file/TSReader.cpp:044] Setting up TS Reader using path: .
16  Tue Aug 15 18:57:38.0448 2017  6  [                src/input/stream/Streamer.cpp:051] Setting up TS Streamer
17  Tue Aug 15 18:57:38.0653 2017  6  [                           src/HttpServer.cpp:056] Setting up HTTP server
18  Tue Aug 15 18:57:38.0668 2017  6  [                           src/RtspServer.cpp:047] Setting up RTSP server
19  Tue Aug 15 18:57:38.0681 2017  6  [                     src/upnp/ssdp/Server.cpp:063] Setting up SSDP server with BOOTID: 5
20  Tue Aug 15 18:57:39.0750 2017  6  [                    src/socket/SocketAttr.cpp:130] HTTP Connection from 192.168.1.204 Port 39183 (fd: 8)
21  Tue Aug 15 18:57:39.0777 2017  7  [                           src/HttpServer.cpp:187] Client: 192.168.1.204 requesed desc.xml
22  Tue Aug 15 18:58:34.0737 2017  7  [                     src/socket/TcpSocket.cpp:090] HTTP Client 192.168.1.204:39183 Connection closed with fd: 8
23  Tue Aug 15 18:58:34.0740 2017  7  [                        src/StreamManager.cpp:155] Found message outside session
24  Tue Aug 15 18:59:10.8451 2017  6  [                    src/socket/SocketAttr.cpp:130] HTTP Connection from 192.168.1.241 Port 49730 (fd: 8)
25  Tue Aug 15 18:59:11.1607 2017  6  [                    src/socket/SocketAttr.cpp:130] HTTP Connection from 192.168.1.241 Port 49734 (fd: 9)
26  Tue Aug 15 18:59:11.1734 2017  6  [                    src/socket/SocketAttr.cpp:130] HTTP Connection from 192.168.1.241 Port 49736 (fd: 10)
27  Tue Aug 15 18:59:14.1493 2017  6  [                    src/socket/SocketAttr.cpp:130] HTTP Connection from 192.168.1.241 Port 49738 (fd: 11)
28  Tue Aug 15 18:59:14.1590 2017  6  [                    src/socket/SocketAttr.cpp:130] HTTP Connection from 192.168.1.241 Port 49740 (fd: 12)
29  Tue Aug 15 18:59:14.1614 2017  6  [                    src/socket/SocketAttr.cpp:130] HTTP Connection from 192.168.1.241 Port 49742 (fd: 13)
30  Tue Aug 15 19:00:48.9013 2017  3  [                           src/HttpServer.cpp:068] readFile ./web/dvbapi.xml
31  Tue Aug 15 19:00:48.9016 2017  3  [                           src/HttpServer.cpp:069] File not found: No such file or directory (code 2)
32  Tue Aug 15 19:00:49.4485 2017  3  [                           src/HttpServer.cpp:068] readFile ./web/dvbapi.xml
33  Tue Aug 15 19:00:49.4487 2017  3  [                           src/HttpServer.cpp:069] File not found: No such file or directory (code 2)
34  Tue Aug 15 19:03:16.0759 2017  7  [                     src/socket/TcpSocket.cpp:090] HTTP Client 192.168.1.241:49734 Connection closed with fd: 9
35  Tue Aug 15 19:03:16.0761 2017  7  [                        src/StreamManager.cpp:155] Found message outside session
36  Tue Aug 15 19:03:16.0764 2017  7  [                     src/socket/TcpSocket.cpp:090] HTTP Client 192.168.1.241:49736 Connection closed with fd: 10
37  Tue Aug 15 19:03:16.0767 2017  7  [                        src/StreamManager.cpp:155] Found message outside session
38  Tue Aug 15 19:03:16.0770 2017  7  [                     src/socket/TcpSocket.cpp:090] HTTP Client 192.168.1.241:49738 Connection closed with fd: 11
39  Tue Aug 15 19:03:16.0773 2017  7  [                        src/StreamManager.cpp:155] Found message outside session
40  Tue Aug 15 19:03:16.0775 2017  7  [                     src/socket/TcpSocket.cpp:090] HTTP Client 192.168.1.241:49740 Connection closed with fd: 12
[... cut ...]
45  Tue Aug 15 19:10:26.8855 2017  6  [                    src/socket/SocketAttr.cpp:130] HTTP Connection from 192.168.1.182 Port 55855 (fd: 9)
46  Tue Aug 15 19:10:26.8861 2017  7  [                           src/HttpServer.cpp:187] Client: 192.168.1.182 requesed desc.xml
47  Tue Aug 15 19:10:26.8989 2017  7  [                     src/socket/TcpSocket.cpp:090] HTTP Client 192.168.1.182:55855 Connection closed with fd: 9
48  Tue Aug 15 19:10:26.8991 2017  7  [                        src/StreamManager.cpp:155] Found message outside session

Ntemis, 2017-08-15:
Note that SAT>IP client in Enigma 2 sees tuners as DVB-S

Barracuda09, 2017-08-15:
What if you request "desc.xml" like in the readme.
What is in there?

Ntemis, 2017-08-15:
Here you go:
<?xml version="1.0" encoding="UTF-8"?>
<root xmlns="urn:schemas-upnp-org:device-1-0" configId="0">
	<specVersion>
		<major>1</major>
		<minor>1</minor>
	</specVersion>
	<device>
		<deviceType>urn:ses-com:device:SatIPServer:1</deviceType>
		<friendlyName>SatPI Server</friendlyName>
		<manufacturer>Marc Postema</manufacturer>
		<manufacturerURL>https://github.com/Barracuda09/SATPI</manufacturerURL>
		<modelDescription>SatPI is a SAT&gt;IP Server for Linux</modelDescription>
		<modelName>SatPI Server</modelName>
		
		<modelNumber>1.5.197~g6971cca</modelNumber>
		
		<modelURL>https://github.com/Barracuda09/SATPI</modelURL>
		<serialNumber>1S81A31231000007</serialNumber>
		
		<UDN>uuid:50c958a8-e839-4b96-b7ae-02817f84c266</UDN>
		
		<UPC>123456789012</UPC>
		<iconList>
			<icon>
				<mimetype>image/png</mimetype>
				<width>48</width>
				<height>48</height>
				<depth>24</depth>
				<url>/assets/images/icons/sm.png</url>
			</icon>
			<icon>
				<mimetype>image/png</mimetype>
				<width>120</width>
				<height>120</height>
				<depth>24</depth>
				<url>/assets/images/icons/lr.png</url>
			</icon>
			<icon>
				<mimetype>image/jpeg</mimetype>
				<width>48</width>
				<height>48</height>
				<depth>24</depth>
				<url>/assets/images/icons/sm.jpg</url>
			</icon>
			<icon>
				<mimetype>image/jpeg</mimetype>
				<width>120</width>
				<height>120</height>
				<depth>24</depth>
				<url>/assets/images/icons/lr.jpg</url>
			</icon>
		</iconList>
		<presentationURL>/</presentationURL>
		
		<satip:X_SATIPCAP xmlns:satip="urn:ses-com:satip">DVBS2-0,DVBT-2,DVBT2-0,DVBC-0,DVBC2-0</satip:X_SATIPCAP>
		
		
		<satip:X_SATIPM3U xmlns:satip="urn:ses-com:satip">channellist.m3u</satip:X_SATIPM3U>
		
	</device>
</root>

Ntemis, 2017-08-15:
Got it i think
this DVBT-2 needs to be DVBT-0 :)

Ntemis, 2017-08-15:
So looks like it's recognizing 2 dvb-t tuners
DVBS2-0,DVBT-2,DVBT2-0,DVBC-0,DVBC2-0
But doesn't advertise them correctly to enigma 2 client.

Barracuda09, 2017-08-15:
So it does not have DVBS2 only DVBT in the desc.xml.
So the client can probably only handle dvb-s ?

Ntemis, 2017-08-15:
No, client can talk to any protocol 
https://github.com/oe-alliance/satip-client

Barracuda09, 2017-08-15:
Well the client is requesting the desc.xml which has the correct description string (as you have seen).
So the client is maybe parsing it wrong?

Ntemis, 2017-08-15:
[image]
[image]
[image]
No its not , it is working just fine with minisatip

Barracuda09, 2017-08-15:
Does that show dvb-t as tuner type?

Ntemis, 2017-08-15:
This is how minisatip sees the tuners
[image]
[image]

Ntemis, 2017-08-15:
As you can see SATPI capability is empty

Barracuda09, 2017-08-16:
Yes so it seems. I can not see at the moment the problem.. as I can not see in the zip files to see what is wrong because I'm not at home for the moment.
I will investigate this problem as soon as I can.
Or somebody can point me to the problem.

Barracuda09, 2017-08-17:
Hi,

Just a shoot in the ballpark.
Could you remove in "desc.xml" the "<presentationURL>/</presentationURL>" the "/" (the slash)

Thanks for trying

Barracuda09, 2017-09-25:
Hi @Ntemis 

I have changed the 'presentationURL' could you try the current version if this is working now?

Many thanks,
Marc

Ntemis, 2017-09-25:
Sure. Will test soon, thanks.

Barracuda09, 2017-09-29:
Hi @Ntemis 

You should also update the web directory. The file desc.xml also changed.

Ntemis, 2017-09-29:
thanks for the update and sorry is taking this long to test.

Barracuda09, 2017-12-12:
Hi @Ntemis 

I think it is fixed now, thanks to @SeyedGH for pointing to another issue (or real issue).

Thanks for giving this an try.

Ntemis, 2017-12-12:
It is thanks

closed 2017-12-12 by Ntemis