Barracuda09/SATPI #57: engima2 support?

kartu, 2017-10-20:
Greetings,

thanks for the great project.
Is there a reason why this can't be run on enigma2 boxes (such as gigablue quad)? (I'm no expert in crosscompiling, but could give it a try with some help :))

How long does channel switching take on a device like PI and fast client?
(couldn't find that in wiki/FAQ and also couldn't find better place to ask)

Ultimately, I'm looking for ways to speed up streaming it via network. Enigma has built in "stream over http" support, but unfortunately it is not particularly fast (TV stumbles for 5+ secs while accepting the stream, HD or not).

Thanks,
kartu

Barracuda09, 2017-10-20:
Hello,

Well I have added, in the wiki, an things you could try for cross compiling.

How fast it is switching I could not say, but at least not longer then 5 sec.

Marc

kartu, 2017-11-05:
Heyho,

compilation did work, I have used the following recipy file (which experienced people would probably laugh at):

`SUMMARY = "SatPI"
HOMEPAGE = "https://github.com/Barracuda09/SATPI"
LICENSE = "GPLv2"
LIC_FILES_CHKSUM = "file://LICENSE;md5=8264535c0c4e9c6c335635c4026a8022"
PV = "1.4+git${SRCPV}"

SRCREV = "f19f3d44141cefe29628237f2623356b9eeabb87"
SRC_URI = "git://github.com/Barracuda09/satpi.git"
SRC_URI[md5sum] = "9856d24abd9eddcabb209a92b2cc125d"
SRC_URI[sha256sum] = "5c10572b6d7956e0121c2116ab53ea3855f9060ea9fcdc5e14a8f20eb4748fd0"

PACKAGES_${PN} = "${PN}"

DEPENDS = "libatomic-ops"

S = "${WORKDIR}/git"

inherit pkgconfig 

EXTRA_OEMAKE = "'CXX=${CXX}' 'RANLIB=${RANLIB}' 'AR=${AR}' 'CFLAGS=${CFLAGS} -I${S}/src -DWITHOUT_XATTR' \
	'BUILDDIR=${S}' LDFLAGS='${LDFLAGS} -latomic -lpthread'"

PACKAGEDIR="/usr/opt/satpi"

do_install() {
	install -d ${D}${bindir}
	install -m 0755 ${PN} ${D}${bindir}
	install -d ${D}${PACKAGEDIR}
	cp -r  ${S}/web ${D}${PACKAGEDIR}
	echo "Package files are ${FILES_${PN}}"
}

FILES_${PN} += " ${S}/web/*"
FILES_${PN} += "${PACKAGEDIR}/web/*"
FILES_${PN} += " ${S}/web"
FILES_${PN} += "${PACKAGEDIR}/web"
`

Just installed an SAT>IP client on a PC, that is still scanning 13E (but reported to have found channels, so, yay, works).

I have killed enigma2 process, as it seems to lock the tuner (and so does satip, I guess). 

Which brings me to the next questions:
1) Is there a way to limit number of tuners that satpi would grab?
2) what about diseqc support, how do I configure it? I have 8 tuners and this is how enigma talks to them:

`config.Nims.0.dvbs.advanced.lnb.1.commandOrder=cut
config.Nims.0.dvbs.advanced.lnb.1.commitedDiseqcCommand=AA
config.Nims.0.dvbs.advanced.lnb.1.diseqcMode=1_1
config.Nims.0.dvbs.advanced.lnb.1.fastDiseqc=true
config.Nims.0.dvbs.advanced.lnb.1.uncommittedDiseqcCommand=1
config.Nims.0.dvbs.advanced.lnb.2.commandOrder=cut
config.Nims.0.dvbs.advanced.lnb.2.commitedDiseqcCommand=AB
config.Nims.0.dvbs.advanced.lnb.2.diseqcMode=1_1
config.Nims.0.dvbs.advanced.lnb.2.fastDiseqc=true
config.Nims.0.dvbs.advanced.lnb.2.uncommittedDiseqcCommand=1
config.Nims.0.dvbs.advanced.lnb.3.commandOrder=cut
config.Nims.0.dvbs.advanced.lnb.3.commitedDiseqcCommand=BA
config.Nims.0.dvbs.advanced.lnb.3.diseqcMode=1_1
config.Nims.0.dvbs.advanced.lnb.3.fastDiseqc=true
config.Nims.0.dvbs.advanced.lnb.3.uncommittedDiseqcCommand=1
config.Nims.0.dvbs.advanced.lnb.4.commandOrder=cut
config.Nims.0.dvbs.advanced.lnb.4.commitedDiseqcCommand=BB
config.Nims.0.dvbs.advanced.lnb.4.diseqcMode=1_1
config.Nims.0.dvbs.advanced.lnb.4.fastDiseqc=true
config.Nims.0.dvbs.advanced.lnb.4.uncommittedDiseqcCommand=1
config.Nims.0.dvbs.advanced.lnb.5.commandOrder=cut
config.Nims.0.dvbs.advanced.lnb.5.commitedDiseqcCommand=AA
config.Nims.0.dvbs.advanced.lnb.5.diseqcMode=1_1
config.Nims.0.dvbs.advanced.lnb.5.fastDiseqc=true
config.Nims.0.dvbs.advanced.lnb.5.uncommittedDiseqcCommand=2
config.Nims.0.dvbs.advanced.lnb.6.commandOrder=cut
config.Nims.0.dvbs.advanced.lnb.6.commitedDiseqcCommand=AB
config.Nims.0.dvbs.advanced.lnb.6.diseqcMode=1_1
config.Nims.0.dvbs.advanced.lnb.6.fastDiseqc=true
config.Nims.0.dvbs.advanced.lnb.6.uncommittedDiseqcCommand=2
config.Nims.0.dvbs.advanced.lnb.7.commandOrder=cut
config.Nims.0.dvbs.advanced.lnb.7.commitedDiseqcCommand=BA
config.Nims.0.dvbs.advanced.lnb.7.diseqcMode=1_1
config.Nims.0.dvbs.advanced.lnb.7.fastDiseqc=true
config.Nims.0.dvbs.advanced.lnb.7.uncommittedDiseqcCommand=2
config.Nims.0.dvbs.advanced.lnb.8.commandOrder=cut
config.Nims.0.dvbs.advanced.lnb.8.commitedDiseqcCommand=BB
config.Nims.0.dvbs.advanced.lnb.8.diseqcMode=1_1
config.Nims.0.dvbs.advanced.lnb.8.fastDiseqc=true
config.Nims.0.dvbs.advanced.lnb.8.uncommittedDiseqcCommand=2
`

xiaole51, 2017-11-05:
Hi kartu
Can share the compiled satpi files, used in E2 system
I tried compiling, but it was too difficult for me.
I am using DM800se,
Thank you very much

mail: hero.gong@foxmail.com

kartu, 2017-11-05:
Hi xiaole,

I don't think it would work on DM800se, but here you go
[satpi_1.4+git0+f19f3d4414-r0_mips32el.ipk.zip](https://github.com/Barracuda09/SATPI/files/1444017/satpi_1.4.git0.f19f3d4414-r0_mips32el.ipk.zip)

xiaole51, 2017-11-05:
Hi kartu.
Thank you for your help.
Yes, it can't work on my E2 system.
Lack of libatomic.So.1 file.

/usr/bin/satpi: error while loading Shared libraries: libatomic. So. 1: cannot open Shared object file: No such file or directory

Barracuda09, 2017-11-05:
Hello @kartu,

Question 1:
You can disable in the web interface the frontends

Question 2:
In the web interface you can configure the diseqc per frontends

Thanks for giving this a try, it is much appreciated!

Marc

kartu, 2017-11-05:
@xiaole51 
I installed libatomic using "opkg install". It wasn't part of my compilation.

@Barracuda09 
Q2:
Could you provide more info please? 
I actually have 8 LNBs (yikes), are only 4 supported in SAT>IP?
The only diseqc option that I see as changeable is diseqc version.
To pick up LNB1 my receiver normally sends:
commited - AA (first command)
uncommited - 1 (second command)
(diseqc mode is version 1.1)

Is there a way to do that with satpi?

Barracuda09, 2017-11-05:
@kartu 

Well I only have a 4 way diseqc switch. And an JESS LNB.
So I'm not sure if this will work at the moment, I'm sorry.

What about the frontends, it should detect all. But could you
Attach a debug log here from the start of SatPI.

Advertise as DVB-S2: this is something if you use DVB-C then you can make a client
Believe that it has S2 of course you need the translation mapping.

kartu, 2017-11-05:
@Barracuda09 
Log (both tuners are actually found just fine, it's LNB stuff that I'm struggling with. 
Perhaps easiest way to cover all possible configs would be to allow entering DISEQC command per LNB.

`0  Sun Nov  5 16:12:16.5989 2017  6  [                                src/Satpi.cpp:286] --- starting SatPI version: 1.5.202~gf19f3d4 ---
1  Sun Nov  5 16:12:16.5996 2017  6  [                                src/Satpi.cpp:287] Number of processors online: 2
2  Sun Nov  5 16:12:16.5999 2017  6  [                                src/Satpi.cpp:288] Default network buffer size: 96 KBytes
3  Sun Nov  5 16:12:16.6007 2017  6  [                        src/InterfaceAttr.cpp:119] eth0: 192.168.0.157 [ac:db:ef:30:64:11]
4  Sun Nov  5 16:12:16.6008 2017  6  [                        src/StreamManager.cpp:068] Current DVB_API_VERSION: 5.10
5  Sun Nov  5 16:12:16.6012 2017  3  [                       src/base/M3UParser.cpp:086] Error: could not open file: mapping.m3u
6  Sun Nov  5 16:12:16.6015 2017  6  [                   src/input/dvb/Frontend.cpp:168] Detecting frontends in: /dev/dvb
7  Sun Nov  5 16:12:16.6039 2017  3  [                       src/base/M3UParser.cpp:086] Error: could not open file: mapping.m3u
8  Sun Nov  5 16:12:16.6044 2017  6  [                   src/input/dvb/Frontend.cpp:461] Frontend Name: BCM7346 DVB-S2 NIM (internal)
9  Sun Nov  5 16:12:16.6046 2017  6  [                   src/input/dvb/Frontend.cpp:533] Frontend Type: Satellite (DVB-S)
10  Sun Nov  5 16:12:16.6047 2017  6  [                   src/input/dvb/Frontend.cpp:538] Frontend Type: Satellite (DVB-S2)
11  Sun Nov  5 16:12:16.6047 2017  6  [                   src/input/dvb/Frontend.cpp:580] Frontend Freq: 950000 Hz to 2150000 Hz
12  Sun Nov  5 16:12:16.6048 2017  6  [                   src/input/dvb/Frontend.cpp:581] Frontend srat: 1000000 symbols/s to 45000000 symbols/s
13  Sun Nov  5 16:12:16.6065 2017  3  [                       src/base/M3UParser.cpp:086] Error: could not open file: mapping.m3u
14  Sun Nov  5 16:12:16.6069 2017  6  [                   src/input/dvb/Frontend.cpp:461] Frontend Name: BCM7346 DVB-S2 NIM (internal)
15  Sun Nov  5 16:12:16.6071 2017  6  [                   src/input/dvb/Frontend.cpp:533] Frontend Type: Satellite (DVB-S)
16  Sun Nov  5 16:12:16.6072 2017  6  [                   src/input/dvb/Frontend.cpp:538] Frontend Type: Satellite (DVB-S2)
17  Sun Nov  5 16:12:16.6073 2017  6  [                   src/input/dvb/Frontend.cpp:580] Frontend Freq: 950000 Hz to 2150000 Hz
18  Sun Nov  5 16:12:16.6074 2017  6  [                   src/input/dvb/Frontend.cpp:581] Frontend srat: 1000000 symbols/s to 45000000 symbols/s
19  Sun Nov  5 16:12:16.6076 2017  6  [                   src/input/dvb/Frontend.cpp:171] Frontends found: 2
20  Sun Nov  5 16:12:16.6077 2017  6  [                  src/input/file/TSReader.cpp:045] Setting up TS Reader using path: /usr/opt/satpi
21  Sun Nov  5 16:12:16.6079 2017  3  [                       src/base/M3UParser.cpp:086] Error: could not open file: mapping.m3u
22  Sun Nov  5 16:12:16.6081 2017  6  [                src/input/stream/Streamer.cpp:051] Setting up TS Streamer
23  Sun Nov  5 16:12:16.6139 2017  3  [                       src/base/M3UParser.cpp:086] Error: could not open file: mapping.m3u
24  Sun Nov  5 16:12:16.6184 2017  3  [                       src/base/M3UParser.cpp:086] Error: could not open file: mapping.m3u
25  Sun Nov  5 16:12:16.6195 2017  3  [                       src/base/M3UParser.cpp:086] Error: could not open file: mapping.m3u
26  Sun Nov  5 16:12:16.6210 2017  6  [                           src/HttpServer.cpp:056] Setting up HTTP server
27  Sun Nov  5 16:12:16.6217 2017  6  [                     src/upnp/ssdp/Server.cpp:063] Setting up SSDP server with BOOTID: 5
28  Sun Nov  5 16:12:16.6225 2017  6  [                           src/RtspServer.cpp:047] Setting up RTSP server
29  Sun Nov  5 16:12:16.6847 2017  6  [                    src/socket/SocketAttr.cpp:134] HTTP Connection from 192.168.0.109 Port 63434 (fd: 8)
30  Sun Nov  5 16:12:16.7076 2017  6  [                    src/socket/SocketAttr.cpp:134] HTTP Connection from 192.168.0.242 Port 63429 (fd: 9)
31  Sun Nov  5 16:12:16.9250 2017  6  [                    src/socket/SocketAttr.cpp:134] HTTP Connection from 192.168.0.109 Port 63435 (fd: 10)
32  Sun Nov  5 16:12:16.9360 2017  6  [                    src/socket/SocketAttr.cpp:134] HTTP Connection from 192.168.0.109 Port 63433 (fd: 11)
33  Sun Nov  5 16:12:28.4230 2017  7  [                     src/socket/TcpSocket.cpp:090] HTTP Client 192.168.0.109:63433 Connection closed with fd: 11
34  Sun Nov  5 16:12:28.4231 2017  7  [                        src/StreamManager.cpp:151] Found message outside session
35  Sun Nov  5 16:12:33.0116 2017  6  [                    src/socket/SocketAttr.cpp:134] HTTP Connection from 192.168.0.109 Port 63437 (fd: 11)
36  Sun Nov  5 16:12:33.0120 2017  6  [                    src/socket/SocketAttr.cpp:134] HTTP Connection from 192.168.0.109 Port 63438 (fd: 12)
37  Sun Nov  5 16:12:33.0125 2017  6  [                    src/socket/SocketAttr.cpp:134] HTTP Connection from 192.168.0.109 Port 63439 (fd: 13)
38  Sun Nov  5 16:12:33.0130 2017  6  [                    src/socket/SocketAttr.cpp:134] HTTP Connection from 192.168.0.109 Port 63440 (fd: 14)
`

Barracuda09, 2017-11-05:
@kartu

Yes that would probably be the next best solution.
How is this done in enigma2?

kartu, 2017-11-05:
@Barracuda09 
Well, enigma allows you to state:

- diseqc version
- commited or uncommitted first
- committed command
- uncommitted command

some additional params:

- fast or not
- higher than usual voltage or not
- whether to repeat command

Which is quite a bit of messing with UI.

Barracuda09, 2017-11-05:
@kartu

So there should come a different diseqc option, were you can specify which commands to send etc.

It should be possible to do.

kartu, 2017-11-05:
@Barracuda09 

Yes, I just thought if it can be expressed as a sequence of HEX numbers, plain text input box would be the most flexible way.

kartu, 2017-11-05:
By the way, my SAT>IP client experience was rather turbulent. 
Android client didn't find much.
Windows client could find a lot, but couldn't play it.

Then I tried scanning for particular frequency, it found several channels, but could only play one.

Anyhow, these issues are rather secondary at this point. :)

kartu, 2017-11-05:
@Barracuda09 
By the way, here is the solution by similar project, command line params:

https://github.com/catalinii/minisatip

-d --diseqc ADAPTER1:COMMITTED1-UNCOMMITTED1[,ADAPTER2:COMMITTED2-UNCOMMITTED2[,...]

The first argument is the adapter number, second is the number of committed packets to send to a Diseqc 1.0 switch, third the number of uncommitted commands to sent to a Diseqc 1.1 switch The higher number between the committed and uncommitted will be sent first.
eg: -d 0:1-0 (which is the default for each adapter).
note: * as adapter means apply to all adapters
note: * before committed number enables fast-switch (only voltage/tone)

Barracuda09, 2017-11-05:
Yes it could be a text input with hex numbers.
Which clients did you use, so I can try them myself (again)

kartu, 2017-11-05:
I tried second one in the list:
http://www.satip.info/products/SAT>IP%20Clients

And top 4 Sat>IP results on google store. 

It might be some diseqc initialization issue so don't invest too much time into it, might a non issue.

rayholland, 2018-10-30:
Hi,
Before going to cross-compile (arm), i would like to know if FBC tuners are supported?
At the moment i'm using minisatip on my VU+ Uno 4k (dvb-c), but it is not stable (having troubles with recording multiple simultaneous streams).
Thank you

Barracuda09, 2018-10-31:
Hi @rayholland 

Many thanks for your interest in my project.

I am using also the VU+ Uno 4k (DVB-C) so it is working, but a few problems remain:
- There seems to be a problem with scanning channels (the stream seems so first fill
   a buffer before sending the data and therefore a time-out may occur in the scanning application TvHeadend. This is because of "slow" pids used for scanning. I have not had time to check it out)
- I am not stressing it that much but if you can test it, it would be nice

But I can give you pre compiled binaries if you like to test it.

Regards,

Marc

rayholland, 2018-10-31:
Hi Marc,

That would be great. 
Please send the binaries to raymond at servertje dot org and i will do some stress testing.

Best regards and many thanks,
Raymond

MadsBakholt, 2020-10-29:
This is a great piece of software! :)

Having struggled quite alot with "minisatip" on my VU+ Solo SE v2(no SSDP, no tuning, no nothing), I tried out this application and had a SAT>IP running in minutes - thank you.

I would be grateful if you could provide newer binaries of SatPI as the one from this thread - can you help out? :)

Barracuda09, 2020-10-29:
Hi @MadsBakholt 

Thanks for giving this a try. Here is a debug binary [satpi.tar.gz](https://github.com/Barracuda09/SATPI/files/5460914/satpi.tar.gz)

Give it a try.

MadsBakholt, 2020-10-30:
Sorry, exec format error - VU+ Solo SE V2 is a MIPS32EL based STB! :)

Barracuda09, 2020-10-30:
Hi @MadsBakholt 

Sorry this was a ARM (for Vu+ Uno 4K).

Here is an MIPS32 binary:
[satpi_mips32.tar.gz](https://github.com/Barracuda09/SATPI/files/5467187/satpi_mips32.tar.gz)

MadsBakholt, 2020-11-02:
Well, gotta say that SatPI simply works on a VU+ Solo SE V2 - no problems.

This new version(debug?) is blessed with a few goodies:

1) Log output directly in terminal
2) Much neater web interface

There are, however, 2 uncertainties I cannot resolve:

1) The SSDP datagram is not send out from my STB, that being connected to LAN on Wi-Fi og tethered
2) I'm not sure if adapters are a shared ressource or can be a shared ressource

Thing is: some dudes over @ openATV claims that they can run 'minisatip' on Enigma2 devices and have'em both serve streams at the same time.

I would really like to help out on the SSDP issue, but have not been able to find any clues why this SSDP package is not being broadcasted.

I tested 'minisatip' on a Raspberry Pi with a Xbox One TV adapter connected and it broadcasted its services right away to e.g. Elgato SAT>IP on my Android phone.

Barracuda09, 2020-11-02:
Hi @MadsBakholt 

Thanks for you feedback and tests.

- Is the STB connected via LAN to WiFi and you try to receive the SSDP there? (Broadcast are usually not routed between LAN and WiFi)
 - What do you mean with shared? (Between Enigma and SatPI? so you can use Enigma and at the same time SatPI)

MadsBakholt, 2020-11-03:
Hi @Barracuda09,

Wow... didn't know about broadcasts not passing LAN <-> WLAN, despite having Googled about the subject bigtime the past weeks!
I will hook up the STB through LAN one of the coming evenings and see if that made a change - the funny thing is that I hooked up the STB to wired LAN a few weeks ago when I got frustrated over 'minisatip' missing out on broadcasting as well - that didn't help at all!

Sharing... yes, openATV users claims that starting 'minisatip' up with no DVB device detection makes it run in shared mode! Truth is I suspect its not exactly shared among 'minisatip' and the underlying Linux, but more like taking advantage of the built-in RSTP in openATV!

But honestly... instructions and descriptions are sparse over at openATV and my non-native-english capabilities may not have helped me understand what "they" are trying to accomplish and how they do it!

Barracuda09, 2020-11-03:
Hi @MadsBakholt 

Maybe you can try to add on STB to route SSDP (Maybe its blocked there):
`ip route add 239.255.255.250 dev eth0`

Well broadcasts not passing LAN <-> WLAN is a bit tricky. I am not a 'guru' but in the past I had simulare issue's.

Thanks for you tests and feedback.