Barracuda09/SATPI #58: Build with OSCam DVBAPI is error!!

xiaole51, 2017-11-05:
Can't find the dvbcsa/dvbcsa.h file
Normal make is pass

root@localhost:/opt/satpi# make debug LIBDVBCSA=yes
make "BUILD=debug"
make[1]: Entering directory '/opt/satpi'
g++ -c -I ./src -std=c++11 -Wall -Wextra -Winit-self -pthread  -O0 -g3 -DDEBUG -fstack-protector-all -Wswitch-default -DLIBDVBCSA src/decrypt/dvbapi/Client.cpp -o obj/decrypt/dvbapi/Client.o
src/decrypt/dvbapi/Client.cpp:42:28: fatal error: dvbcsa/dvbcsa.h: No such file or directory
compilation terminated.
Makefile:123: recipe for target 'obj/decrypt/dvbapi/Client.o' failed
make[1]: *** [obj/decrypt/dvbapi/Client.o] Error 1
make[1]: Leaving directory '/opt/satpi'
Makefile:134: recipe for target 'debug' failed
make: *** [debug] Error 2

Barracuda09, 2017-11-05:
Hello,

You must have the libdvbcsa installed on your system

kartu, 2017-11-05:
This doesn't even look like cross platform compile to me, xiaole, check how it works for DM 800 SE here:

http://wiki.blue-panel.com/index.php/Build_images_and_compile_programs_with_OE_2.2

xiaole51, 2017-11-06:
hi Barracuda09 
Libdvbcsa installed and compiled successfully
But what protocol does OScam prot use?
Unable to connect to oscam

hi kartu
I'm learning to cross-compile with Buildroot.
But don't know how to compile ipk files.
There is very little information on the Internet. Do you have any link?

thank you

kartu, 2017-11-06:
@xiaole51 

It's just an archive, I think in zip format. 
You don't really need it,, "installing" IPK is basically unpacking its content. You can simply copy files around manually.

Barracuda09, 2017-11-06:
@xiaole51 

SatPI is using libdvbcsa to decrypt the stream with a valid subscription and oscam.
So SatPI is talking to oscam with dvbapi protocol and receiving keys from your card in oscam.
Using these keys to decrypt the stream with libdvbcsa.

Hope this clears things up.

Barracuda09, 2022-09-23:
See Wiki for more deatails:
https://github.com/Barracuda09/SATPI/wiki/Setup-OSCam-DVBAPI-Protocol

closed 2022-09-23 by Barracuda09