Barracuda09/SATPI #59: How to edit cross-compilation options?

xiaole51, 2017-11-06:
I always can't succeed.
Can you give an example?

export INCLUDES=--sysroot=dir        (  What does this do?   )
export CXXPREFIX=mipsel-unknown-linux-gnu-gcc      (       is ok?       )
export CXXSUFFIX=mipsel-unknown-linux-gnu-gcc      (       is ok?       )               
make debug                            

PS:  make clean Can't remove export value

kartu, 2017-11-06:
@xiaole51 
You shouldn't be creating new issues to ask quesitons that you have already asked in other issue.

Cross-compiling is compiling on one architecture (normally x86) and even platform (whatever version of linux) for another architecture (another CPU, e.g. MIPS) and likely another version of unix.

You can't use normal compilers to achieve that, you need something called "toolchain" that is created exactly for the platform you are targeting.

That isn't easy, but some nice people prepared how-to's and builc packages that help with it.
Check "build instructions" here: 
https://github.com/openatv/enigma2/

On my machine building took several hours and took more than 40GB of HDD space.

Barracuda09, 2017-11-06:
@xiaole51 

The export vars you can't clean with make, it's just an environment var of bash/sh etc.
sysroot is something you let gcc know we're the targets libs and include files are (usually the file
system of the target) and it uses this to cross compile the program.

It is an method I have used for the Raspberry PI, for cross compiling programs.

Barracuda09, 2017-11-06:
Suffix is the stuff after gcc
Prefix is the stuff in front of gcc

So in your example: mipsel-unknown-linux-gnu-gcc
The prefix is: mipsel-unknown-linux-gnu-
The suffix is not needed

Barracuda09, 2021-02-22:
Moved to discussions #114

closed 2021-02-22 by Barracuda09