Barracuda09/SATPI #60: DVB-S tuner does not get exported.

mthqwork, 2018-11-25:
I have 3 adapters, two DVB-S (not S2):
/dev/dvb/adapter0/frontend0
/dev/dvb/adapter1/frontend0 
and one with two frontends: DVB-S and DVB-T:
/dev/dvb/adapter2/frontend0
/dev/dvb/adapter2/frontend1

I tried to understand how SATPI works (I have working minisatip now, but i would like to try SATPI also).

By default, all frontends showing up, and enabled, however the desc.xml only contains line like this:

<satip:X_SATIPCAP xmlns:satip="urn:ses-com:satip">DVBT2-1</satip:X_SATIPCAP>

If I disable the adapter2/frontend1 (the DVB-T one) i just get this:

<satip:X_SATIPCAP xmlns:satip="urn:ses-com:satip"/>

What could be the problem?
Am I missed something?
Thanks.

Barracuda09, 2018-11-25:
Well it seems that I do not count the DVB-S adapters at all.

In Frontend.cpp  around line 510 - 515 is shows..

Barracuda09, 2018-11-25:
You could try for me to also count DVB-S there and see if that works?

mthqwork, 2018-11-25:
I have already tried to use it in tvheadend, manually set the configuration to DVBS2-3, but I had lines like this in SATPI:

Stream: 0, Not capable of handling msys=dvbs.

Im happy to help you and we can figure out what's going on here :)

Barracuda09, 2018-11-25:
Could you try to change in Frontend.cpp around line 510:

				case SYS_DVBS:
					SI_LOG_INFO("Frontend Type: Satellite (DVB-S)");
					++_dvbs2;  // <-- Add this line
					break;
				case SYS_DVBS2:
					// we only count DVB-S2
					++_dvbs2;
					SI_LOG_INFO("Frontend Type: Satellite (DVB-S2)");
					break;

mthqwork, 2018-11-25:
Of course, i will get back soon with the results!

mthqwork, 2018-11-25:
Looks like it exports now fine, and running with tvheadend / TCP mode. 
Thank You :)

Barracuda09, 2018-11-25:
No thanks.. Thank You for trying.

mthqwork, 2018-11-28:
I promised a feedback. It's been running for 3 days without any issues.

Barracuda09, 2018-11-28:
Ok, thanks for your feedback, I appreciate that. So the fix is working out
for DVB-S.

closed 2019-03-08 by Barracuda09