Barracuda09/SATPI #62: support for panasonic tv multicast channel

holli73, 2019-03-06:
i would like to import a m3u file to my panasonic tv's so i can have on all tv's the same channel order/config.

the file i need to put on the tv must look like this:

```
#EXTM3U
#EXTINF:0,1. ZDF
rtp://192.168.249.111:5554?stype=1&onid=1&tsid=1079&svcid=28006
#EXTINF:0,2. Pro Sieben
rtp://192.168.249.111:5554?stype=1&onid=1&tsid=1107&svcid=17501
```

is there some way in satpi i can map this rtp call to the correct dvb-s frequency?
f.ex. in vlc this is working fine

```
#EXTINF:0,6. Pro Sieben
rtsp://192.168.249.111:5554/?src=1&freq=12545&pol=h&ro=0.35&msys=dvbs&mtype=qpsk&plts=off&sr=22000&fec=56&pids=0,17,18,97,511,512,33
```

i tried this content but the tv will not work with this parameters - i would like to avoid to run channel searches on all tv's and then sort them as needed (even with chansort software - it is not easy...

any idea if something like this would be possible?

thanks
holli
.

Barracuda09, 2019-03-07:
Hi,

Is this a Panasonic TV without SAT>IP capability?

holli73, 2019-03-07:
hi,

it has satip build in - but i have a unicable setup with 4 satellites and it takes more than one hour to make a channel scan - because the build in satip client can use multicast as well and the multicast can be supplied with a m3u file - it would be much simpler and save lots of time

i have 4 of this tv's and all of them are from the same series - but the smaller ones channel db causes error on the bigger screens - so i have to use chansort software multiple times to get all going and on the same ordered lists...

if i could just maintain a single m3u file and place it on the tv's - i would save lots of time ...

thanks
holli

Barracuda09, 2019-03-07:
Can you explain to me what is not working, because if it is a compliant SAT>IP client (Which Panasonic is I belief) it should work with:

``
#EXTINF:0,6. Pro Sieben
rtsp://192.168.249.111:5554/?src=1&freq=12545&pol=h&ro=0.35&msys=dvbs&mtype=qpsk&plts=off&sr=22000&fec=56&pids=0,17,18,97,511,512,33
``

because this is a correct SAT>IP request, OR is it trying to open a multicast stream then? and could you give then the log of SatPI.

9000h, 2019-03-07:
Hi,
I think it working a bit different have a look here 
http://download.digital-devices.eu/download/octopus_net_mc/ONET-MC_Manual_DE.pdf
CU
9000h

9000h, 2019-03-07:
to make it work correct  at least for the Panasonic TV's a file on a usb stick is needed

satip_multicast.txt with contents
```
#EXTM3U_DOWNLOAD
#EXTUPDATE: On
#EXTUSE_NIT_LCN: Off
#EXTUSE_SDT_NAME: Off
http://satip.server.de/html/satip.m3u
```
satip.m3u contents looks like
```
#EXTM3U
#EXTINF:0,1. Das Erste HD
rtp://239.190.40.61:5004?stype=25&onid=1&tsid=18002&svcid=10301

#EXTINF:0,2. ZDF HD
rtp://239.190.43.102:5004?stype=25&onid=1&tsid=18005&svcid=11110

#EXTINF:0,3. WDR HD Münster
rtp://239.190.110.170:5004?stype=25&onid=1&tsid=18630&svcid=28330

#EXTINF:0,4. RTL Television
rtp://239.190.46.227:5004?stype=1&onid=1&tsid=18014&svcid=12003

#EXTINF:0,5. SAT.1 NRW
rtp://239.190.68.100:5004?stype=1&onid=1&tsid=19508&svcid=17508

#EXTINF:0,6. ProSieben
rtp://239.190.68.93:5004?stype=1&onid=1&tsid=19501&svcid=17501

#EXTINF:0,7. RTL2
rtp://239.190.46.244:5004?stype=1&onid=1&tsid=18015&svcid=12020

#EXTINF:0,8. VOX
rtp://239.190.47.28:5004?stype=1&onid=1&tsid=18019&svcid=12060

#EXTINF:0,9. kabel eins
rtp://239.190.68.94:5004?stype=1&onid=1&tsid=19502&svcid=17502

#EXTINF:0,10. 3sat HD
rtp://239.190.43.142:5004?stype=25&onid=1&tsid=18086&svcid=11150

#EXTINF:0,11. arte HD
rtp://239.190.40.62:5004?stype=25&onid=1&tsid=18001&svcid=10302
```

Barracuda09, 2019-03-10:
Thanks @9000h  for you info

9000h, 2020-01-08:
for reference and real world use
https://www.stadtwerke-rhede.de/privatkunden/internet-co/downloads

holli73, 2020-01-09:
> for reference and real world use
> https://www.stadtwerke-rhede.de/privatkunden/internet-co/downloads

that is cool - so what kind of software must i place in front of minisatip to make multicast working ? (dvblast, mumudvb,...)

any idea/help?

thanks
holli