Barracuda09/SATPI #64: Can I bind to an IP/Interface

brimur, 2019-03-10:
First off thanks for the software, it works great. My server has 3 network cards. Only one has a default gateway but satPI is binding to the wrong interface. Is it possible to bind it to a specific IP or interface? Thanks

Barracuda09, 2019-03-10:
Hi,

Thanks for you interest in my project.

Sorry, unfortunately it is not possible to bind to a specific "eth" at the moment

Barracuda09, 2019-03-17:
Hi,

This should be fixed with: df1bc1831a228600e367054b560ed31f69a6b414

command option to use: ``--iface-name eth0``

Thanks for your issue report

brimur, 2019-09-03:
I tried adding this to the systemd file and the service would not start. I'm running it in a VM so in my case it was --iface-name ens160

The service will start without the the parameter

````
MediaServer /opt/satpi: git rev-parse HEAD
3539cf5a68240449469996d8a29a727065ca14c4

● satPI.service - SatPI Daemon
   Loaded: loaded (/etc/systemd/system/satPI.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2019-09-03 13:20:45 IST; 43s ago
  Process: 5293 ExecStart=/opt/satpi/satpi --no-daemon --http-path /opt/satpi/web --iface-name ens160 (code=exited, status=1/FAILURE)
 Main PID: 5293 (code=exited, status=1/FAILURE)

Sep 03 13:20:45 MediaServer satpi[5293]:         --http-port      set http port default 8875 (1024 - 65535)
Sep 03 13:20:45 MediaServer satpi[5293]:         --rtsp-port      set rtsp port default 554  ( 554 - 65535)
Sep 03 13:20:45 MediaServer satpi[5293]:         --no-daemon      do NOT daemonize
Sep 03 13:20:45 MediaServer satpi[5293]:         --no-ssdp        do NOT advertise server
Sep 03 13:20:45 MediaServer systemd[1]: satPI.service: Service hold-off time over, scheduling restart.
Sep 03 13:20:45 MediaServer systemd[1]: satPI.service: Scheduled restart job, restart counter is at 5.
Sep 03 13:20:45 MediaServer systemd[1]: Stopped SatPI Daemon.
Sep 03 13:20:45 MediaServer systemd[1]: satPI.service: Start request repeated too quickly.
Sep 03 13:20:45 MediaServer systemd[1]: satPI.service: Failed with result 'exit-code'.
Sep 03 13:20:45 MediaServer systemd[1]: Failed to start SatPI Daemon.

````

Barracuda09, 2019-09-03:
What is the output if you do not start as service and without `--no-daemon`

brimur, 2019-09-03:
Same result if I leave out --no-daemon
````
● satPI.service - SatPI Daemon
   Loaded: loaded (/etc/systemd/system/satPI.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2019-09-03 13:48:45 IST; 1s ago
  Process: 4170 ExecStart=/opt/satpi/satpi --http-path /opt/satpi/web --iface-name ens160 (code=exited, status=1/FAILURE)
 Main PID: 4170 (code=exited, status=1/FAILURE)

Sep 03 13:48:45 MediaServer systemd[1]: satPI.service: Service hold-off time over, scheduling restart.
Sep 03 13:48:45 MediaServer systemd[1]: satPI.service: Scheduled restart job, restart counter is at 5.
Sep 03 13:48:45 MediaServer systemd[1]: Stopped SatPI Daemon.
Sep 03 13:48:45 MediaServer systemd[1]: satPI.service: Start request repeated too quickly.
Sep 03 13:48:45 MediaServer systemd[1]: satPI.service: Failed with result 'exit-code'.
Sep 03 13:48:45 MediaServer systemd[1]: Failed to start SatPI Daemon.

````

brimur, 2019-09-03:
Hah, actually this is what I see if I run it manually...

````
MediaServer /opt/satpi: /opt/satpi/satpi --http-path /opt/satpi/web --iface-name ens160
Usage /opt/satpi/satpi [OPTION]

Options:
        --help           show this help and exit
        --version        show the version number
        --user xx        run as user
        --dvb-path       set path were to find dvb devices default /dev/dvb
        --app-data-path  set path for application state data eg. xml files etc
        --http-path      set root path of web/http pages
        --http-port      set http port default 8875 (1024 - 65535)
        --rtsp-port      set rtsp port default 554  ( 554 - 65535)
        --no-daemon      do NOT daemonize
        --no-ssdp        do NOT advertise server
````

Barracuda09, 2019-09-03:
Well in the latest master you should also see this:
  ` "\t--iface-name     set the network interface to bind to (eg. eth0)\r\n"  `

but your output is not showing this, which is a bit strange I would think.

Could you please check the vesion once more?

brimur, 2019-09-03:
Is this the latest?
MediaServer /opt/satpi: git rev-parse HEAD
3539cf5a68240449469996d8a29a727065ca14c4

I did a git pull earlier today before I started playing with it again.

Barracuda09, 2019-09-03:
Yes it is, but if you run:
`/opt/satpi/satpi --version`

What does the output show?

brimur, 2019-09-03:
MediaServer /opt/satpi: /opt/satpi/satpi --version
SatPI version: 1.5.258~g4c3aacc

Barracuda09, 2019-09-03:
That is an old build 4c3aaccafe9a7b5438bcc8a1dd2e419f648ffbae

You should do an `make clean` and the `make debug` I think.

brimur, 2019-09-03:
Doh! I did the git pull but forgot the make. Sorry

brimur, 2019-09-03:
All good. Sorry for the time wasting!

````
● satPI.service - SatPI Daemon
   Loaded: loaded (/etc/systemd/system/satPI.service; disabled; vendor preset: enabled)
   Active: active (running) since Tue 2019-09-03 14:59:08 IST; 3s ago
 Main PID: 8871 (satpi)
    Tasks: 4 (limit: 2238)
   CGroup: /system.slice/satPI.service
           └─8871 /opt/satpi/satpi --no-daemon --http-path /opt/satpi/web --iface-name ens160

Sep 03 14:59:08 MediaServer SatPI[8871]: Setting up HTTP server
Sep 03 14:59:08 MediaServer SatPI[8871]: Setting up RTSP server
Sep 03 14:59:08 MediaServer SatPI[8871]: Setting up SSDP server with BOOTID: 4  annouce interval: 60 Sec
Sep 03 14:59:10 MediaServer SatPI[8871]: Root Device Client 192.168.200.40 : tries to discover the network, sending reply back
Sep 03 14:59:10 MediaServer SatPI[8871]: Root Device Client 192.168.200.40 : tries to discover the network, sending reply back
Sep 03 14:59:10 MediaServer SatPI[8871]: Root Device Client 192.168.200.40 : tries to discover the network, sending reply back
Sep 03 14:59:10 MediaServer SatPI[8871]: Root Device Client 192.168.200.40 : tries to discover the network, sending reply back
Sep 03 14:59:10 MediaServer SatPI[8871]: HTTP Connection from 192.168.200.40 Port 55611 with fd: 8
Sep 03 14:59:10 MediaServer SatPI[8871]: HTTP Client 192.168.200.40:55611 Connection closed with fd: 8
Sep 03 14:59:11 MediaServer SatPI[8871]: Root Device Client 192.168.200.9 : tries to discover the network, sending reply back
````

Barracuda09, 2019-09-03:
No problem.

closed 2019-09-03 by brimur